crankyoldgit/IRremoteESP8266 #1480: HAIER_AC_YRW02 remote doubt

beckmx, 2021-05-19:
### Version/revision of the library used
2.7.19
### Describe the bug
I have a minisplit, which with this library uses the HAIER_AC_YRW02 codec, I was trying to scan the remote control just to see if I could sync the code, but realized that for some reason, the dump is not able to recognize the control, it is able to recognize other codecs but this in particular looks like is not picking it up

#### To Reproduce
_What steps did you do, and what did or didn't actually happen? How can we reproduce the issue?_
 the dump message is the next one:

ABFAD961
Protocol  : UNKNOWN
Code      : 0xABFAD961 (179 Bits)

F18F7905
Protocol  : UNKNOWN
Code      : 0xF18F7905 (179 Bits)

#### Example code used
The setup details for the readings:

```
IRrecv irrecv(kRecvPin, 1024,15,true);// I also try with 50ms
```

In the loop
```
String description = IRAcUtils::resultAcToString(&results);
if (description.length()) Serial.println("MY_DESC" ": " + description);
irrecv.resume();  // Receive the next value
```

#### Expected behaviour
Using the dump code, 

#### Output of raw data from [IRrecvDumpV2.ino]IRrecvDump is now running and waiting for IR input on Pin 5
Timestamp : 000010.962
Library   : v2.7.18

Protocol  : UNKNOWN
Code      : 0xABFAD961 (179 Bits)
uint16_t rawData[357] = {3096, 2948,  3048, 4388,  588, 1610,  614, 498,  586, 1612,  612, 500,  612, 500,  586, 1610,  588, 1612,  612, 502,  586, 1612,  612, 500,  612, 500,  614, 500,  612, 498,  586, 1610,  586, 1612,  612, 502,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 504,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 502,  614, 498,  586, 1612,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 502,  586, 1612,  612, 500,  586, 1610,  612, 500,  612, 498,  612, 500,  614, 478,  634, 502,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 498,  614, 504,  612, 500,  614, 500,  586, 1612,  612, 500,  612, 500,  612, 500,  612, 500,  612, 502,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 504,  614, 500,  612, 500,  612, 498,  614, 500,  612, 500,  612, 500,  612, 500,  612, 482,  632, 500,  612, 502,  610, 500,  614, 500,  612, 500,  612, 500,  612, 480,  632, 504,  612, 480,  632, 500,  612, 500,  612, 480,  632, 500,  612, 500,  612, 500,  612, 502,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  586, 1612,  612, 500,  586, 1616,  612, 500,  612, 500,  586, 1610,  588, 1612,  612, 502,  612, 500,  614, 498,  586, 1614,  586, 1612,  612, 500,  586, 1610,  586, 1592,  632, 498,  586, 1610,  588, 1610,  586, 1614,  614, 500,  612, 480,  632, 500,  612, 500,  612, 500,  612, 500,  614, 498,  612, 500,  614, 500,  614, 500,  612, 500,  612, 500,  614, 498,  614, 498,  614, 500,  612, 504,  612, 500,  612, 500,  612, 500,  612, 498,  612, 502,  612, 500,  614, 498,  612, 502,  612, 500,  612, 498,  614, 500,  612, 500,  612, 500,  612, 500,  612, 500,  614, 502,  612, 500,  614, 478,  634, 498,  614, 500,  612, 500,  612, 500,  612, 500,  612, 482,  634, 500,  612, 500,  612, 500,  612, 500,  614, 498,  614, 500,  612, 480,  632, 502,  586, 1610,  614, 478,  608, 1610,  588, 1610,  612, 498,  586, 1610,  588, 1610,  586, 1606,  612};  // UNKNOWN ABFAD961

### What brand/model IR demodulator are you using?
Using the sender the model is HAIER_AC_YRW02

### Has this library/code previously worked as expected for you?
Not yet

crankyoldgit, 2021-05-19:
_**Have you got the Brand & Model number for the A/C unit and the remote you're using please?**_

A quick look the `rawData[]` it tells me the capture you have is NOT a `HAIER_AC_YRW02`.

using the [`auto_analyse_raw_data.py`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py) tool, a _real_ `HAIER_AC_YRWO2` message looks like:
```
Found 229 timing entries.
Potential Mark Candidates:
[2998, 640]
Potential Space Candidates:
[4460, 3086, 1692, 614]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 2998
kHdrSpace  = 3086
kBitMark   = 556
kOneSpace  = 1614
kZeroSpace = 543
kSpaceGap = 4460

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+kHdrMark+UNEXPECTED->GAP(4460)kBitMark(UNEXPECTED)1010011000010010000000000000001001000000001000000000000000100000000000000000000000000000000000000000010100111111
  Bits: 112
  Hex:  0xA61200024020002000000000053F (MSB first)
        0xFCA0000000000400040240004865 (LSB first)
  Dec:  3368306103852107447402401108067647 (MSB first)
        5123843726122503529084458218047589 (LSB first)
  Bin:  0b1010011000010010000000000000001001000000001000000000000000100000000000000000000000000000000000000000010100111111 (MSB first)
        0b1111110010100000000000000000000000000000000000000000010000000000000001000000001001000000000000000100100001100101 (LSB first)

Total Nr. of suspected bits: 112
```
i.e. A **112** bit / **14** byte message.

Where as your supplied message is:
```
Found 357 timing entries.
Potential Mark Candidates:
[3096, 634]
Potential Space Candidates:
[4388, 2948, 1616, 504]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3072
kHdrSpace  = 2948
kBitMark   = 609
kOneSpace  = 1610
kZeroSpace = 498
kSpaceGap = 4388

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+kHdrMark+UNEXPECTED->GAP(4388)kBitMark(UNEXPECTED)10100110100001100000000000000000010000001010000000000000001000000000000000000000000000000000000000000101001100011011011100000000000000000000000000000000000000000000000010110111
  Bits: 176
  Hex:  0xA686000040A00020000000000531B7000000000000B7 (MSB first)
        0xED000000000000ED8CA0000000000400050200006165 (LSB first)
  Dec:  62303814805046023047516589831372227655673129067348151 (MSB first)
        88672227340140543086680535493829987771895833784508773 (LSB first)
  Bin:  0b10100110100001100000000000000000010000001010000000000000001000000000000000000000000000000000000000000101001100011011011100000000000000000000000000000000000000000000000010110111 (MSB first)
        0b11101101000000000000000000000000000000000000000000000000111011011000110010100000000000000000000000000000000000000000010000000000000001010000001000000000000000000110000101100101 (LSB first)

Total Nr. of suspected bits: 176
```
i.e. A **176** bit /  **22** byte message.

It's a fair bit bigger, thus it isn't the same protocol.

Looking at it's structure, it is very similar to `HAIER_AC_YRW02` except it is longer. If `HAIER_AC_YRW02` is working for you sending, then it's complete luck.

Creating a simple/basic sender/decoder (i.e. to an array of bytes) will be trivial. See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol

However, fine/detailed control of the settings will require significant leg work from you! e.g. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

crankyoldgit, 2021-05-19:
@beckmx Can you please download and test the branch `Issue1480` (https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1480) / PR #1481 and let me know how it goes for you?
I still need the model data for the AC & the remote.

Based on the data you supplied, the library should now identify, decode, & send your messages in a _basic_ form.

i.e. you're now up to: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#create-a-spreadsheet

beckmx, 2021-05-19:
The model is MMI18HDBWCA6MI8, the brand is mabe, and the remote control doesnt have any marks, only inside the battery has this:V12843 HJ200223, here is a picture https://cdn1.coppel.com/images/catalog/pm/6448383-5.jpg. I will give it a try to the branch, give me a couple of minutes

beckmx, 2021-05-19:
I extracted this behaviour, I can see now some pattern, I am actively reading your documentation as well, I dont know if it should decode the temperature, but at least is giving me the protocol:

```
turn on cool 24
2000A040000086A6
Protocol  : HAIER_AC176
Code      : 0xA686000040A00020000000000531B7000000000000B7 (176 Bits)

cool to 25
2000A040000096A6
Protocol  : HAIER_AC176
Code      : 0xA696000040A0002000000000003CB7000000000000B7 (176 Bits)

turn off at 25
2000A000000090A6
Protocol  : HAIER_AC176
Code      : 0xA690000000A000200000000005FBB7000000000000B7 (176 Bits)

turn on at 25
2000A040000096A6
Protocol  : HAIER_AC176
Code      : 0xA696000040A00020000000000541B7000000000000B7 (176 Bits)

cool to 24
2000A040000086A6
Protocol  : HAIER_AC176
Code      : 0xA686000040A0002000000000012DB7000000000000B7 (176 Bits)

turn off at 24
2000A000000080A6
Protocol  : HAIER_AC176
Code      : 0xA680000000A000200000000005EBB7000000000000B7 (176 Bits)

turn on at 24
2000A040000086A6
Protocol  : HAIER_AC176
Code      : 0xA686000040A00020000000000531B7000000000000B7 (176 Bits)

```

Do you happen to have an example of extracting the temperature from a captured message? I havent found one yet

NiKiZe, 2021-05-19:
You might want to read the documentation for what is needed to get this correct.
Please use IRrecvDumpV2+ (so we know that nothing else interfere with the data collection)
Post any raw dumps for unknown (without them there will be low chance of fixing it 😉)
Collect a set of captures from lowest to highest temp, this allows us to ensure that the data is in expected order. If not the data might flip when we get to know it.

After that the documentation in the wiki has details for what needs to be done to figure out which parts of the message is for what function etc.

beckmx, 2021-05-19:
@NiKiZe ok, following your suggestion, I am attaching captures from cool, heat and off
```
turn on cool at 16
Protocol  : HAIER_AC176
Code      : 0xA606000040A000200000000005B1B7000000000000B7 (176 Bits)
uint16_t rawData[357] = {3076, 2948,  3048, 4388,  586, 1612,  612, 502,  584, 1612,  612, 500,  612, 500,  584, 1612,  586, 1612,  612, 504,  612, 502,  610, 502,  610, 502,  612, 500,  612, 500,  586, 1612,  586, 1614,  612, 504,  612, 500,  612, 502,  612, 500,  612, 500,  612, 500,  612, 502,  612, 500,  612, 504,  610, 502,  610, 502,  610, 502,  612, 502,  610, 502,  612, 500,  612, 500,  612, 504,  612, 498,  586, 1612,  612, 502,  612, 500,  612, 500,  612, 502,  610, 502,  610, 504,  584, 1612,  612, 500,  586, 1612,  612, 502,  612, 502,  610, 500,  612, 500,  612, 504,  612, 500,  612, 502,  610, 500,  612, 502,  610, 502,  610, 500,  612, 502,  612, 504,  612, 500,  612, 500,  586, 1612,  612, 500,  612, 500,  612, 502,  612, 502,  610, 504,  612, 502,  610, 502,  612, 502,  610, 502,  612, 502,  610, 502,  612, 500,  612, 504,  610, 502,  612, 500,  612, 502,  612, 502,  612, 500,  612, 502,  610, 502,  610, 504,  612, 500,  612, 500,  612, 502,  610, 502,  612, 500,  612, 500,  612, 500,  612, 504,  610, 502,  612, 502,  610, 500,  612, 502,  612, 502,  612, 500,  612, 500,  612, 504,  612, 502,  612, 500,  612, 502,  612, 502,  610, 500,  586, 1612,  612, 500,  586, 1614,  586, 1612,  612, 500,  586, 1612,  586, 1614,  612, 500,  612, 500,  612, 500,  586, 1616,  584, 1614,  612, 500,  586, 1612,  586, 1612,  612, 500,  584, 1612,  586, 1610,  586, 1616,  612, 500,  612, 502,  610, 500,  612, 502,  610, 502,  610, 502,  612, 502,  612, 504,  612, 500,  612, 500,  612, 500,  612, 502,  610, 502,  612, 502,  612, 500,  612, 504,  612, 500,  612, 500,  612, 500,  612, 500,  612, 500,  612, 502,  610, 500,  614, 504,  610, 502,  612, 500,  612, 502,  610, 502,  612, 502,  610, 500,  612, 502,  610, 502,  612, 500,  612, 502,  610, 502,  610, 500,  612, 502,  612, 500,  612, 500,  612, 504,  612, 502,  610, 502,  612, 500,  610, 502,  612, 500,  612, 500,  612, 500,  612, 504,  586, 1612,  612, 500,  586, 1612,  586, 1612,  612, 500,  586, 1612,  586, 1612,  586, 1608,  610};  // HAIER_AC176
uint8_t state[22] = {0xA6, 0x06, 0x00, 0x00, 0x40, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x05, 0xB1, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};

cool to 17
Timestamp : 000874.319
Library   : v2.7.18

Protocol  : HAIER_AC176
Code      : 0xA616000040A000200000000000BCB7000000000000B7 (176 Bits)
uint16_t rawData[357] = {3078, 2950,  3048, 4390,  586, 1614,  610, 500,  586, 1612,  612, 500,  612, 500,  586, 1612,  586, 1612,  612, 504,  610, 502,  610, 502,  612, 500,  586, 1612,  612, 500,  586, 1612,  584, 1614,  610, 506,  610, 502,  612, 500,  612, 500,  612, 500,  612, 500,  612, 502,  612, 500,  612, 504,  612, 500,  612, 502,  612, 500,  612, 502,  612, 500,  612, 502,  610, 500,  612, 504,  612, 500,  584, 1612,  612, 500,  612, 502,  610, 500,  612, 500,  612, 502,  610, 504,  586, 1612,  612, 500,  586, 1612,  612, 500,  612, 502,  612, 500,  610, 502,  612, 504,  612, 500,  612, 502,  612, 500,  612, 500,  612, 500,  612, 500,  614, 500,  612, 504,  612, 500,  612, 500,  584, 1612,  612, 502,  610, 502,  610, 502,  610, 502,  612, 504,  612, 502,  612, 502,  612, 500,  612, 502,  612, 502,  612, 500,  612, 502,  612, 504,  612, 502,  610, 502,  610, 502,  612, 502,  612, 500,  612, 502,  612, 500,  612, 504,  612, 500,  612, 502,  610, 502,  612, 502,  610, 502,  612, 500,  612, 502,  612, 504,  612, 502,  610, 502,  610, 502,  612, 502,  610, 502,  612, 502,  612, 500,  612, 504,  612, 502,  610, 502,  612, 500,  612, 502,  610, 502,  610, 502,  612, 500,  612, 504,  584, 1612,  612, 500,  584, 1612,  586, 1612,  586, 1610,  586, 1612,  612, 502,  610, 504,  586, 1614,  612, 500,  586, 1612,  586, 1614,  612, 500,  586, 1612,  586, 1612,  586, 1616,  612, 500,  612, 502,  612, 500,  612, 502,  612, 502,  612, 502,  610, 502,  610, 504,  612, 502,  612, 502,  610, 502,  612, 500,  610, 502,  612, 500,  612, 502,  610, 504,  612, 502,  612, 502,  612, 502,  610, 502,  612, 502,  612, 502,  610, 502,  610, 506,  610, 502,  612, 500,  612, 502,  612, 500,  612, 502,  612, 502,  610, 502,  612, 504,  612, 502,  610, 502,  610, 482,  632, 500,  612, 502,  610, 502,  612, 502,  612, 504,  612, 502,  610, 502,  612, 500,  612, 502,  612, 502,  610, 502,  610, 502,  612, 504,  584, 1614,  610, 500,  586, 1612,  586, 1614,  610, 500,  586, 1612,  586, 1612,  586, 1606,  612};  // HAIER_AC176
uint8_t state[22] = {0xA6, 0x16, 0x00, 0x00, 0x40, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0xBC, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};

cool to 18
Timestamp : 000892.933
Library   : v2.7.18

Protocol  : HAIER_AC176
Code      : 0xA626000040A000200000000000CCB7000000000000B7 (176 Bits)
uint16_t rawData[357] = {3078, 2948,  3048, 4388,  586, 1612,  612, 500,  584, 1612,  612, 502,  610, 500,  586, 1612,  586, 1612,  612, 504,  610, 502,  610, 500,  586, 1612,  612, 500,  612, 500,  584, 1612,  586, 1612,  612, 504,  610, 502,  612, 500,  612, 502,  610, 500,  612, 502,  612, 500,  612, 500,  612, 504,  612, 500,  612, 502,  610, 502,  612, 500,  612, 500,  612, 500,  610, 502,  612, 504,  610, 502,  584, 1612,  612, 502,  610, 502,  612, 502,  610, 502,  610, 502,  612, 504,  584, 1612,  612, 500,  586, 1612,  612, 500,  612, 500,  612, 504,  610, 500,  610, 504,  612, 500,  612, 502,  610 [... cut ...] 500,  610, 502,  610, 500,  610, 500,  610, 504,  610, 502,  610, 502,  610, 502,  610, 500,  612, 500,  610, 500,  610, 502,  610, 504,  612, 500,  612, 500,  612, 500,  610, 502,  610, 500,  610, 502,  610, 502,  610, 502,  586, 1612,  610, 500,  584, 1610,  584, 1612,  610, 500,  586, 1610,  584, 1610,  584, 1608,  610};  // HAIER_AC176
uint8_t state[22] = {0xA6, 0x40, 0x00, 0x00, 0x00, 0xA0, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x05, 0x0B, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};

```

NiKiZe, 2021-05-19:
Thank you, raw data is needed if you get unknown protocol. Otherwise it's not.
The numbers looks good, if you look for what changes when temperature changes you can probably get that data without to much trouble, much more work is needed for the next step.

What we also want to know is if you can use this data to re-send, and have the AC react?

beckmx, 2021-05-19:
Ok, I tried with the https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino example and both data arrays the raw and the:
`irsend.sendHaierAC176(mystate);`

Worked, but my guess is that the decoding phase is not running, because I tried to use:
`if (irrecv.decodeHaierAC176(&results)) {`
And it wont let me compile, probably some of these defines:
`DECODE_HAIER_AC176`
is avoiding it?

Greetings!

crankyoldgit, 2021-05-20:
> Ok, I tried with the https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino example and both data arrays the raw and the state can be sent, the state was sent with this:
> `irsend.sendHaierAC176(mystate);`

Thanks for confirming it works.

> 
> And worked, but my guess is that the decoding phase is not running, because I tried to use:
> `if (irrecv.decodeHaierAC176(&results)) {`
> And it wont let me compile, probably some of these defines:
> `DECODE_HAIER_AC176`
> is avoiding it?

You can't call `irrecv.decodeHaierAC176()` directly. That function/method is only for internal (_private_) use. You need to use the _public_ `irrecv.decode()` function.

beckmx, 2021-05-20:
Hello, yes i realized that afterwards, but i did notice the decode phase is
not running, i tried also to see if the state could be read but wasnt lucky
in the decode side --
*Web Architect  | Neoris*
Innovation over knowledge

crankyoldgit, 2021-05-20:
> @NiKiZe ok, following your suggestion, I am attaching captures from cool, heat and off
> 
> turn on cool at 16
> Protocol  : HAIER_AC176
> uint8_t state[22] = {0xA6, 0x**0**6, 0x00, 0x00, 0x40, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x05, 0xB1, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};
> 
> cool to 17
> Timestamp : 000874.319
> Protocol  : HAIER_AC176
> uint8_t state[22] = {0xA6, 0x**1**6, 0x00, 0x00, 0x40, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0xBC, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};
> 
> cool to 18
> Timestamp : 000892.933
> Protocol  : HAIER_AC176
> uint8_t state[22] = {0xA6, 0x**2**6, 0x00, 0x00, 0x40, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0xCC, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};
>
> [SNIP]
> 
> cool to 25
> Timestamp : 001093.322 
> Protocol  : HAIER_AC176
> uint8_t state[22] = {0xA6, 0x**9**6, 0x00, 0x00, 0x40, 0xA0, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3C, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};

If you read/follow the documentation (See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#example-analysis) you'll easily find the temperature. It's in `state[1]`.
That is the first half (nibble) of `state[1]` (2nd byte) is the temperature offset by `16`. I've highlighted it in bold in your reply. i.e. `0` + 16 = 16C, `1` + 16 = 17C, `2` + 16 = 18C, ... `9` + 16 = 25C

So we have the bit ordering correct, and _you've_ now identified the temperature. **_You_** now need to work out (per wiki) what the rest of the bits do etc and be able to explain any changes in the any of the bytes of the `state`.

When **_you_** have that ready, we can code up _detailed_ support for your A/C.
(Ref: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#The_libraryexample_code_doesnt_understand_my_Air_Conditioner_remote_)

crankyoldgit, 2021-05-20:
> Hello, yes i realized that afterwards, but i did notice the decode phase is not running, i tried also to see if the state could be read but wasnt lucky in the decode side -- *Web Architect | Neoris* Innovation over knowledge

You are probably not using it correctly. Try following the _recv_ examples in https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples

crankyoldgit, 2021-05-20:
> @crankyoldgit hello, thanx for the suggestion, I am still actively reading the docs and examples, so, it's just a matter of time to see some results, btw these were the decode procedures I tried to use:

The reason that _code_ didn't work was because we don't have **detailed** support for _this_ protocol yet.

Re: your PR.
There is far more to it than that for correctly supporting this protocol.

beckmx, 2021-05-20:
from the tests I have made and following ur docs, it seems to be compatible I will continue to see if additional bytes do something else, I am closing the thread, if you agree

crankyoldgit, 2021-05-20:
It probably is very similar, however as the checksum is located at the very end of the 22 bytes, the class you're re-using only handles 14 bytes, and calculates/expected the checksum at the end of those 14 bytes.
So it may sort-of work, it won't properly with.

crankyoldgit, 2021-07-06:
FYI, the committed & merged changes thus far have been included in the newly released version of the library. i.e. v2.7.19

crankyoldgit, 2021-07-10:
@beckmx Ping. We haven't heard from you in over a month. How are you going with decoding/reverse engineering the protocol? Do you have document/spreadsheet with your analysis available?

crankyoldgit, 2021-07-14:
Haven't heard from you in over a month. This seems stale. Marking it closed. Let me know if I should re-open.

instalator, 2021-07-22:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 11">
<link id=Main-File rel=Main-File
href="file:///C:\Users\INSTAL~1\AppData\Local\Temp\msohtml1\01\clip.htm">
<link rel=Edit-Time-Data
href="file:///C:\Users\INSTAL~1\AppData\Local\Temp\msohtml1\01\clip_editdata.mso">
<link rel=OLE-Object-Data
href="file:///C:\Users\INSTAL~1\AppData\Local\Temp\msohtml1\01\clip_oledata.mso">
<style>

</style>
</head>

<body link=blue vlink=purple>

  |   | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | 17 | 18 | 19 | 20 | 21
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Button | Indicator | Prefix | Temp/Button |   |   | Power |   | Turbo | Mode | Sleep |   |   |   | Button | CRC (0-12) | Prefix |   | Fan |   |   |   |   | CRC (14-20)
on |   | A6 | 82 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 05 | 2D | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
cool | cool, 24 C, fan auto | A6 | 82 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 06 | 2E | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
dry | dry, 23 C, fan auto | A6 | 72 | 00 | 00 | 40 | A0 | 00 | 40 | 00 | 00 | 00 | 00 | 06 | 3E | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
heat | heat, 24 C, fan auto | A6 | 8A | 00 | 00 | 40 | A0 | 00 | 80 | 00 | 00 | 00 | 00 | 06 | 96 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
fan | cool, 24 C, fan min | A6 | 82 | 00 | 00 | 40 | 60 | 00 | 20 | 00 | 00 | 00 | 00 | 04 | EC | B7 | 00 | C0 | 00 | 00 | 00 | 00 | 77
fan | cool, 24 C, fan mid | A6 | 82 | 00 | 00 | 40 | 40 | 00 | 20 | 00 | 00 | 00 | 00 | 04 | CC | B7 | 00 | 80 | 00 | 00 | 00 | 00 | 37
fan | cool, 24 C, fan max | A6 | 82 | 00 | 00 | 40 | 20 | 00 | 20 | 00 | 00 | 00 | 00 | 04 | AC | B7 | 00 | 40 | 00 | 00 | 00 | 00 | F7
fan | cool, 24 C, fan auto | A6 | 82 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 04 | 2C | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
fan | heat, 24 C, fan min | A6 | 0A | 00 | 00 | 40 | 60 | 00 | 80 | 00 | 00 | 00 | 00 | 04 | D4 | B7 | 00 | C0 | 00 | 00 | 00 | 00 | 77
fan | heat, 24 C, fan mid | A6 | 0A | 00 | 00 | 40 | 40 | 00 | 80 | 00 | 00 | 00 | 00 | 04 | B4 | B7 | 00 | 80 | 00 | 00 | 00 | 00 | 37
fan | heat, 24 C, fan max | A6 | 0A | 00 | 00 | 40 | 20 | 00 | 80 | 00 | 00 | 00 | 00 | 04 | 94 | B7 | 00 | 40 | 00 | 00 | 00 | 00 | F7
fan | heat, 24 C, fan auto | A6 | 0A | 00 | 00 | 40 | A0 | 00 | 80 | 00 | 00 | 00 | 00 | 04 | 14 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
fan | dry, 24 C, fan min | A6 | 72 | 00 | 00 | 40 | 60 | 00 | 40 | 00 | 00 | 00 | 00 | 04 | FC | B7 | 00 | C0 | 00 | 00 | 00 | 00 | 77
fan | dry, 24 C, fan mid | A6 | 72 | 00 | 00 | 40 | 40 | 00 | 40 | 00 | 00 | 00 | 00 | 04 | DC | B7 | 00 | 80 | 00 | 00 | 00 | 00 | 37
fan | dry, 24 C, fan max | A6 | 72 | 00 | 00 | 40 | 20 | 00 | 40 | 00 | 00 | 00 | 00 | 04 | BC | B7 | 00 | 40 | 00 | 00 | 00 | 00 | F7
fan | dry, 24 C, fan auto | A6 | 72 | 00 | 00 | 40 | A0 | 00 | 40 | 00 | 00 | 00 | 00 | 04 | 3C | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
smart_on | smart режим | A6 | 82 | 00 | 00 | 40 | A0 | 00 | 00 | 00 | 00 | 00 | 00 | 06 | 0E | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
smart_off | off | A6 | 80 | 00 | 00 | 00 | A0 | 00 | 00 | 00 | 00 | 00 | 00 | 05 | CB | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
Swing_on | cool, 21 C, fan auto | A6 | 5C | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 02 | 04 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
Swing_off | cool, 21 C, fan auto | A6 | 50 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 02 | F8 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
sleep_on | cool, 21 C, fan auto | A6 | 50 | 00 | 00 | 40 | A8 | 00 | 20 | 80 | 00 | 00 | 00 | 0B | 89 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
sleep_off | cool, 21 C, fan auto | A6 | 50 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 0B | 01 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
turbo/quiet_turbo | cool, 21 C, fan auto | A6 | 50 | 00 | 00 | 40 | A0 | 40 | 20 | 00 | 00 | 00 | 00 | 08 | 3E | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
turbo/quiet_quiet | cool, 21 C, fan auto | A6 | 50 | 00 | 00 | 40 | A0 | 80 | 20 | 00 | 00 | 00 | 00 | 08 | 7E | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
turbo/quiet_norm(off) | cool, 21 C, fan auto | A6 | 50 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 08 | FE | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
extra жалюзи вверх | cool, 21 C, fan auto | A6 | 51 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 1F | 16 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7
extra жалюзи вниз | cool, 21 C, fan auto | A6 | 53 | 00 | 00 | 40 | A0 | 00 | 20 | 00 | 00 | 00 | 00 | 1F | 18 | B7 | 00 | 00 | 00 | 00 | 00 | 00 | B7

</body>

</html>

crankyoldgit, 2021-07-22:
In your table, `state[5]` changes with no explanation. You need to work that out first.

beckmx, 2021-07-22:
@crankyoldgit I missed the notifications, from my tests, I noticed all the bytes documented for haier were compatible at least with my remote, they seem to fit in place

crankyoldgit, 2021-07-23:
> In your table, `state[5]` changes with no explanation. You need to work that out first.

This comment was directed at @instalator

crankyoldgit, 2021-07-23:
@instalator When looking at the structure of `HAIER_AC_YRW02` it seems `state[5]` is also used by `fan`.
i.e.
https://github.com/crankyoldgit/IRremoteESP8266/blob/9a473c94cbe199844cd3ae032f56575c08d45f84/src/ir_Haier.h#L155-L157

Does that look correct to you for this longer `HAIER_AC176` protocol/variant?

instalator, 2021-07-23:
HAIER_AC176 works correctly with 14 bytes of the HAIER_AC_YRW02 protocol. I just provided more information and maybe it will be useful to someone.

crankyoldgit, 2021-07-23:
@instalator I think I've got the `HAIER_AC_YRW02` & `HAIER_AC176` support mostly merged.
When decoding a known example (https://github.com/crankyoldgit/IRremoteESP8266/issues/1480#issuecomment-844369177 _"turn on cool at 16"_), I noticed that the `swing` is being reported as `Swing: 6 (UNKNOWN)`
`state[1]` contains both the temp & the swing value (not temp & button per your table) in `HAIER_AC_YRW02`. For `HAIER_AC176` are you sure it is button??? As the lower nibble (4 bits) doesn't seem to track with the button you said you've pressed.

Either of @instalator or @beckmx, if it is the swing byte/nibble, what does a value of `6` mean?

Now, on to @instalator 's data regarding `state[1]`, your table says it contains the temp. But the temp value doesn't seem to match the description etc. Also, you report you're pressing the `fan` button, but your description of what the remote display says indicates that the `mode` is also changing. What is up with that?

crankyoldgit, 2021-07-23:
Can you both please download/test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Haier176_detailed and let me know how it goes? e.g. Some of the issues I've listed above need to be clarified/tested/checked etc.

That branch should add detailed support for `HAIER_AC176`.

crankyoldgit, 2021-07-28:
Ping @beckmx & @instalator

crankyoldgit, 2021-07-30:
Have either of you tried that branch and have some feedback?

instalator, 2021-07-30:
I have no way of checking. I no longer have a device.

crankyoldgit, 2021-07-31:
@instalator No worries. Thanks for getting back to us,
I guess we will have to hope @beckmx gets back to us.

Andrey-m, 2021-08-08:
Hi. 
I have that device and try to reverse IR and UART protocol.

i create a little spreadsheet with captured IR data

https://docs.google.com/spreadsheets/d/1wdOVS08wgK2pEP7hTZLYMmrQ9FZVmLpZF2HjNQaVxlU/edit?usp=sharing

crankyoldgit, 2021-08-08:
@Andrey-m Have you tried downloading and using the branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Haier176_detailed

It should have some detailed support branch.

Andrey-m, 2021-08-08:
@crankyoldgit Yes.
Any timers not detected as timer.

//Timer for 30 min (ON)
Timestamp : 000686.742
Library   : v2.7.19

Protocol  : HAIER_AC176
Code      : 0xA682004000A000001E0000001036B7000000000000B7 (176 Bits)
Mesg Desc.: Power: Off, Button: 0 (Temp Up), Mode: 0 (Auto), Temp: 24C, Fan: 5 (Auto), Turbo: 0 (Off), Swing: 2 (Middle), Sleep: Off, Health: Off
uint16_t rawData[357] = {3092, 3024,  3094, 4430,  604, 1636,  604, 516,  598, 1642,  598, 520,  604, 514,  600, 1642,  600, 1642,  598, 520,  626, 1614,  636, 482,  632, 488,  604, 514,  600, 518,  606, 514,  602, 1640,  600, 518,  606, 512,  602, 518,  630, 490,  636, 484,  630, 490,  636, 482,  632, 488,  604, 514,  600, 518,  606, 1634,  606, 514,  602, 518,  606, 512,  602, 518,  598, 522,  604, 516,  598, 520,  604, 514,  600, 520,  606, 514,  600, 518,  606, 514,  602, 518,  608, 512,  602, 1638,  602, 518,  596, 1644,  598, 522,  602, 516,  598, 524,  600, 516,  600, 520,  604, 516,  600, 520,  606, 514,  602, 518,  606, 512,  602, 518,  608, 512,  602, 516,  598, 522,  604, 516,  598, 520,  604, 516,  600, 520,  606, 514,  632, 486,  606, 512,  602, 518,  606, 512,  602, 518,  596, 1644,  598, 1644,  598, 1642,  598, 1644,  598, 522,  604, 516,  598, 522,  604, 514,  600, 520,  606, 514,  600, 518,  606, 512,  602, 518,  596, 522,  602, 516,  598, 520,  604, 516,  598, 520,  604, 514,  600, 518,  606, 514,  600, 518,  608, 512,  602, 518,  598, 522,  604, 516,  598, 520,  604, 516,  600, 520,  606, 514,  600, 518,  608, 512,  602, 1638,  602, 518,  596, 522,  602, 516,  598, 522,  636, 484,  630, 490,  604, 1636,  604, 1636,  604, 514,  600, 1640,  600, 1640,  600, 520,  606, 1634,  606, 514,  602, 1638,  600, 1640,  600, 518,  606, 1634,  606, 1634,  606, 1634,  606, 512,  602, 516,  598, 522,  602, 516,  598, 520,  604, 516,  598, 520,  606, 514,  600, 520,  606, 514,  600, 518,  606, 512,  602, 516,  598, 522,  602, 516,  598, 522,  604, 516,  600, 520,  606, 514,  600, 520,  606, 514,  602, 518,  606, 512,  602, 518,  598, 522,  602, 516,  598, 522,  604, 516,  598, 522,  604, 516,  600, 520,  606, 514,  600, 518,  606, 512,  602, 518,  608, 512,  602, 516,  598, 522,  604, 516,  598, 520,  604, 514,  600, 518,  606, 514,  600, 518,  606, 512,  602, 518,  596, 522,  634, 486,  598, 1642,  598, 520,  604, 1636,  604, 1638,  604, 516,  600, 1640,  632, 1610,  598, 1642,  598};  // HAIER_AC176
uint8_t state[22] = {0xA6, 0x82, 0x00, 0x40, 0x00, 0xA0, 0x00, 0x00, 0x1E, 0x00, 0x00, 0x00, 0x10, 0x36, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};

//Disable Timer
Timestamp : 000723.570
Library   : v2.7.19

Protocol  : HAIER_AC176
Code      : 0xA682000040A00000000000001018B7000000000000B7 (176 Bits)
Mesg Desc.: Power: On, Button: 0 (Temp Up), Mode: 0 (Auto), Temp: 24C, Fan: 5 (Auto), Turbo: 0 (Off), Swing: 2 (Middle), Sleep: Off, Health: Off
uint16_t rawData[357] = {3128, 2986,  3122, 4402,  632, 1608,  632, 486,  628, 1614,  628, 492,  634, 488,  626, 1612,  628, 1612,  628, 492,  634, 1608,  632, 486,  628, 490,  634, 486,  628, 490,  636, 484,  630, 1610,  630, 490,  624, 494,  630, 488,  626, 492,  632, 488,  626, 492,  634, 486,  628, 492,  634, 486,  630, 490,  634, 484,  630, 488,  636, 484,  630, 488,  626, 492,  632, 488,  626, 492,  632, 486,  628, 1612,  628, 492,  634, 486,  628, 490,  634, 484,  630, 490,  634, 484,  630, 1610,  630, 488,  626, 1616,  636, 484,  632, 488,  626, 494,  632, 488,  628, 492,  632, 486,  638, 482,  634, 486,  628, 490,  634, 484,  630, 488,  626, 494,  632, 488,  626, 492,  632, 486,  628, 492,  632, 486,  628, 490,  634, 484,  630, 490,  636, 484,  630, 488,  626, 492,  632, 486,  628, 492,  632, 486,  628, 490,  634, 486,  630, 490,  636, 484,  630, 490,  636, 484,  630, 488,  626, 494,  632, 488,  628, 492,  634, 486,  628, 492,  634, 484,  630, 490,  636, 484,  632, 488,  626, 494,  632, 488,  628, 492,  634, 486,  628, 492,  634, 486,  630, 490,  636, 484,  630, 488,  626, 494,  632, 488,  628, 492,  632, 486,  628, 492,  634, 1608,  634, 486,  628, 490,  636, 484,  630, 488,  636, 484,  632, 488,  626, 492,  632, 1608,  634, 1608,  632, 486,  628, 490,  634, 486,  628, 1612,  630, 490,  636, 1606,  634, 1606,  634, 484,  630, 1612,  630, 1610,  630, 1610,  630, 490,  636, 484,  632, 488,  626, 494,  632, 488,  638, 482,  634, 486,  628, 490,  634, 484,  630, 490,  636, 484,  632, 488,  626, 492,  632, 488,  628, 492,  634, 486,  628, 490,  634, 484,  630, 490,  636, 484,  632, 488,  626, 494,  632, 488,  628, 492,  634, 486,  628, 492,  634, 486,  630, 490,  636, 484,  632, 488,  626, 492,  632, 488,  628, 492,  634, 486,  628, 492,  634, 486,  630, 490,  636, 484,  630, 488,  626, 492,  632, 488,  626, 492,  634, 486,  628, 490,  634, 484,  630, 490,  636, 484,  632, 488,  626, 1614,  626, 494,  632, 1608,  632, 1608,  632, 488,  626, 1614,  628, 1614,  626, 1614,  628};  // HAIER_AC176
uint8_t state[22] = {0xA6, 0x82, 0x00, 0x00, 0x40, 0xA0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x18, 0xB7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xB7};
`

crankyoldgit, 2021-08-08:
> Any timers not detected as timer.

Correct. It's early stages at the moment. We are trying to confirm what is there is currently working.

The timer details have not yet been fully analysed yet, as they are a much lower priority. If you have them fully worked out, I can add support for them.

crankyoldgit, 2021-08-21:
> @crankyoldgit Yes.
> Any timers not detected as timer.

The latest update to the [branch](https://github.com/crankyoldgit/IRremoteESP8266/compare/Haier176_detailed) should add preliminary Timer detection/setting etc. Please download and test it, and let me know how it goes.

crankyoldgit, 2021-08-23:
The latest update to the branch should add Timer & Timer Mode detection/setting etc. Please download and test it, and let me know how it goes.

Andrey-m, 2021-08-23:
> The latest update to the branch should add Timer & Timer Mode detection/setting etc. Please download and test it, and let me know how it goes.

I quickly checked. All OK!

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

commit 610ad366 (2021-08-21): HAIER_AC176: Add experimental detailed support
* `IRHaierACYRW02` inherits from the new `IRHaierAC176`.
  - Except for a few functions that need to be special.
* Add it to the Common A/C API (`IRac`)
* Expand & update the unit tests.

Not fully happy with the support yet.

Fixes #1480
commit 2d24333a (2021-08-21): HAIER_AC176: Add experimental detailed support
* `IRHaierACYRW02` inherits from the new `IRHaierAC176`.
  - Except for a few functions that need to be special.
* Add it to the Common A/C API (`IRac`)
* Expand & update the unit tests.

Not fully happy with the support yet.

Fixes #1480
commit 23b66445 (2021-08-28): HAIER_AC176: Add experimental detailed support (#1571)

* `IRHaierACYRW02` inherits from the new `IRHaierAC176`.
  - Except for a few functions that need to be special.
* Add `[sg]et(On|Off)Timer()` & `[sg]etTimerMode()` class methods.
* Update `toString()`.
* Add timer specific unit tests.
* Update existing Unit Tests.
* Add it to the Common A/C API (`IRac`)
* Expand & update the unit tests.

Fixes #1480