crankyoldgit/IRremoteESP8266 #1483: ESP32 task problem

huexpub, 2021-05-22:
**Version/revision of the library used**
IRremoteESP8266@2.7.18

platform
arduino esp32 

when use xTaskCreatePinnedToCore create dual task and more core 0 or core1 , IR send codes Destroyed, i think freertos xtask scheduling with  cpu delay cause this error, and i found every reboot esp32 frist send is right and then is wrong ....any Suggestions？ thanks everybody!

crankyoldgit, 2021-05-22:
Can you please be more detailed about what the problem is? The way you've described it isn't very clear.
Note: We don't officially support use outside of the Arduino Framework, so getting a "fix" may take a while.

You haven't supplied any simplified code that reproduces the problem, which makes debugging what could be going wrong extremely difficult.

>  found every reboot esp32 frist send is right and then is wrong

Sending is pretty rock-solid and simple. If it works once, it should work a second time. I'd check your code flow. Have you changed the GPIO settings in between transmissions. Have you captured using IRrecvDumpV2 or V3, the "wrong" code produced?

TL;DR: You need to give us a lot more info and explain it better if you want anyone to able to realistically help you.

huexpub, 2021-05-24:
task vdelay change to 50 then ok,,thanks!

closed 2021-05-24 by huexpub