crankyoldgit/IRremoteESP8266 #1487: irsend.cpp: how to turn off ir leds in idle state?

dekuort, 2021-05-29:
### Version/revision of the library used
2.7.18

### Describe the bug
IR Emitter is causing the heat from the LED constant current circuit being on  when not being used.

#### Sample code
_I saw patch in esphome: [apply patch to remote_transmitter_esp32.cpp to turn off ir leds in idle state](https://github.com/AdamKGoddard/XiaomiIR_ESPHome/commit/bc353d373dedf3a89c28b44706f77382246b7220)._

`// Start building idle configuration
  this->config_rmt(c);
  c.rmt_mode = RMT_MODE_TX;
  c.gpio_num = gpio_num_t(this->pin_->get_pin());
  c.tx_config.loop_en = false;

  c.tx_config.carrier_en = false;
  c.tx_config.idle_output_en = true;
  // Force inverted logic in idle state
  c.tx_config.carrier_level = RMT_CARRIER_LEVEL_LOW;
  c.tx_config.idle_level = RMT_IDLE_LEVEL_HIGH;

  // Update configuration
  esp_err_t error = rmt_config(&c);
  if (error != ESP_OK) {
    this->error_code_ = error;
    this->mark_failed();
    return;`

### What brand/model IR demodulator are you using?
_Xiaomi Mija universal remote controller_

### Circuit diagram and hardware used (if applicable)
_ESP32-S0WD ESP32-Solo._

https://cloud.mail.ru/public/36kc/Jp2urE72y

### Other useful information
_IR emitter is very high power and more useful than the cheaper alternatives on the market. Also I can't identify this elements. Maybe it's npn transistors
https://cloud.mail.ru/public/jTLi/oZthD5ynv
https://cloud.mail.ru/public/HKDU/t8ktx8pzp_

NiKiZe, 2021-05-29:
In exactly which code pathos the LED left on by IRRemoteESP8266? Can you create a simple example that only uses this library, and not external code?

crankyoldgit, 2021-05-29:
This library does not use the RMT functions of the ESP32. So this is not applicable. When

closed 2021-05-29 by crankyoldgit

dekuort, 2021-05-30:
> In exactly which code pathos the LED left on by IRRemoteESP8266? Can you create a simple example that only uses this library, and not external code?

unfortunately I do not have enough experience to deal with this, so I opened the issue and provided as much information as possible so that the developer could find a solution.

> This library does not use the RMT functions of the ESP32. So this is not applicable.

I understand that this is  code from RMT library esphome, but i presented it for example.

 > When idle, the gpio controlling the ir led held low. Thus no power/current draw.

Unfortunately you are wrong. On this device, it works the other way around. I attached photos with thermal camera.
Here's  board working without selected gpio irsend:
https://prnt.sc/13ks52q web settings
https://cloud.mail.ru/public/SrNw/oWqTCKCjJ board

Here's board with selected gpio irsend. Ir leds and transistors gets hot immediatly after power on:
https://prnt.sc/13ks8pv web
https://cloud.mail.ru/public/mVoJ/S4TGtBLB9 board

If this information is not enough, please write about it

crankyoldgit, 2021-05-30:
Without know what the code is (You haven't told us what software you're running) **and** what the circuit is doing, we can _only_ make guesses.

Looking at what you've provided, I'm going to take a few guesses:

1. Your/The code isn't making the [`IRsend::begin()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a386f026bf739b0718efde4cffa6ce129) call that sets up the pin correctly.
or
2. The IR circuit "transmits" (IR LEDs are on) when the GPIO is pulled `LOW` rather than the usual `HIGH`. i.e. The opposite of what is typical.

If `2.`, then you can try setting the [`inverted` flag](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a792780b7de996c90c86dd7b700eaf271) to `true` when you make the [`IRsend::IRsend()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#a792780b7de996c90c86dd7b700eaf271) instantiation.
e.g.
```
IRsend irsend(26, true);  // Transmit on GPIO26 and invert the signal. i.e. HIGH is off, LOW is on
```