crankyoldgit/IRremoteESP8266 #1495: Deprecation warning on ESP8266 core v3.0.0

sblantipodi, 2021-06-07:
### Version/revision of the library used
"2.7.18"

### Describe the bug
Every time I use this lib I thanks the author (in my mind) for its great work here.

I recently switched to the new Arduino Core ESP8266 V3.0.0 and now when I compile using this lib I get some depracation warning, the lib works awesome as always in my case using a D1 Mini and a Samsung AC.

Is this something that can be fixed before the final deprecation?

```
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:346:62: warning: 'void read_timeout(void*)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  346 |   os_timer_setfn(&timer, reinterpret_cast<os_timer_func_t *>(read_timeout),
      |                                                              ^~~~~~~~~~~~
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:144:27: note: declared here
  144 | static void USE_IRAM_ATTR read_timeout(void *arg __attribute__((unused))) {
      |                           ^~~~~~~~~~~~
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:346:74: warning: 'void read_timeout(void*)' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  346 |   os_timer_setfn(&timer, reinterpret_cast<os_timer_func_t *>(read_timeout),
      |                                                                          ^
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:144:27: note: declared here
  144 | static void USE_IRAM_ATTR read_timeout(void *arg __attribute__((unused))) {
      |                           ^~~~~~~~~~~~
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:350:35: warning: 'void gpio_intr()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  350 |   attachInterrupt(params.recvpin, gpio_intr, CHANGE);
      |                                   ^~~~~~~~~
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:166:27: note: declared here
  166 | static void USE_IRAM_ATTR gpio_intr() {
      |                           ^~~~~~~~~
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:350:52: warning: 'void gpio_intr()' is deprecated: Use IRAM_ATTR in place of ICACHE_RAM_ATTR to move functions into IRAM [-Wdeprecated-declarations]
  350 |   attachInterrupt(params.recvpin, gpio_intr, CHANGE);
      |                                                    ^
.pio\libdeps\smartostat\IRremoteESP8266\src\IRrecv.cpp:166:27: note: declared here
  166 | static void USE_IRAM_ATTR gpio_intr() {
      |                           ^~~~~~~~~
```

Thank you for your awesome work here.

NiKiZe, 2021-06-07:
I think this was fixed in c49084947 could you download and test current master?

sblantipodi, 2021-06-07:
thank you for the answer @NiKiZe @crankyoldgit, it works awesome and the warning vanished.
waiting for the official release. thanks!

closed 2021-06-07 by sblantipodi