crankyoldgit/IRremoteESP8266 #1496: Question: why use this library over the original IRremote library?

gannonprudhomme, 2021-06-09:
Apologies if this a dumb question, but the code that I wrote to test out the IR Receiver using the [Arduino-IRRemote](https://github.com/Arduino-IRremote/Arduino-IRremote) for my Arduino Uno also seems to work completely fine on my ESP8266 (specifically [this one](https://www.amazon.com/dp/B081PX9YFV)). As such, I was wondering what this library achieves over the original and why I should use one or the other? This and the original library are obviously well-used and active, which is why this has me scratching my head

crankyoldgit, 2021-06-09:
It's only recently that the original library became active again, it had not been changed for years, it also only recently supported the ESP8266.
The reason the projects split was due to inactivity and no support for the esp8266.
We've tried to keep this library compatible as much as we can with the original library to ease transition.

This library supports more A/Cs and protocols with more features, thus is bigger, and won't fit on tiny AVR/Arduino chips. It's also uses a different approach to capture.

It also the case that this library fixed a whole load of protocol mistakes and had a far better test suite and tools.

In short, why use this library over the original. If you want to control A/C's, this one supports lots and lots, most with detailed control. If you only need to do simple IR stuff (like TVs) and you need to fit in a small amount of memory, then the original one works fine/better.

They diverged so much it's too much work to merge them.

gannonprudhomme, 2021-06-09:
Makes sense, thank you for the detailed response!

closed 2021-06-09 by crankyoldgit