crankyoldgit/IRremoteESP8266 #1498: Connect via MDNS hostname.local fails

touchgadget, 2021-06-11:
### Version/revision of the library used
master at b81945d145a4e423c5aa58de1d8c4018447a1698 June 6, 2021

### Describe the bug
Firefox and Chromium on Ubuntu fail to connect to ESP8266 (d1 mini) running the IRMQTTServer and IRServer examples using MDNS.

#### To Reproduce
Load IRMQTTServer example.
Connect to http://ir_server.local/ using Firefox or Chromium.
The hostname never resolves to an IP address. Connecting to the IP address works fine. Connecting to
other devices using MDNS work fine.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Not applicable.

### Has this library/code previously worked as expected for you?
No.

### Other useful information
I will post a PR that fixes the examples.

crankyoldgit, 2021-06-11:
Thanks for the report & PR. Agreed, this looks like an oversight on my part. Comments etc in the PR

closed 2021-06-12 by crankyoldgit

crankyoldgit, 2021-07-06:
FYI, the committed & merged changes have been included in the newly released version of the library. i.e. v2.7.19

PR #1499 (2021-06-12): Fix MDNS in IRServer and IRMQTTServer
Fix MDNS service in IRServer and IRMQTTServer examples

In the examples IRServer and IRMQTTServer, Chromium and Firefox running on Ubuntu fail to connect to the ESP8266 using the MDNS hostname. For example, esp8266.local and ir_server.local do not resolve.

Based on the MDNS examples included with the ESP8266 board package, adding calls to mdns.addService() and mdns.update() fixes the problems.

Fixes #1498
commit b86aa36d (2021-06-12): Fix MDNS in IRServer and IRMQTTServer (#1499)

Fix MDNS service in IRServer and IRMQTTServer examples

In the examples IRServer and IRMQTTServer, Chromium and Firefox running on Ubuntu fail to connect to the ESP8266 using the MDNS hostname. For example, esp8266.local and ir_server.local do not resolve.

Based on the MDNS examples included with the ESP8266 board package, adding calls to mdns.addService() and mdns.update() fixes the problems.

Fixes #1498