crankyoldgit/IRremoteESP8266 #1501: Best if order of sending signal in "COOLIX" protocol is alternate

MarkEvens, 2021-06-14:
Right now what I am facing issue is In "COOLIX" for turbo mode it is first send turbo toggle and after it send all the other settings. So because of this if my AC is OFF and I directly send turbo frame with Turbo ON is useless, but instead you just change the order and alternate by first send all the other settings after it send turbo toggle than it is worked even If my AC is OFF.

crankyoldgit, 2021-06-16:
Hey @MarkEvens Can you please download/try out PR #1502 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1501 and let me know if this fixes the issue for you?

My tests indicate it should, but nothing beats testing against a real device.

i.e. It should now send in order: the normal/on message, then the turbo message.

crankyoldgit, 2021-06-21:
Ping @MarkEvens

MarkEvens, 2021-06-22:
Sorry for late response, I tested it and it's working
```
 ac.next.protocol = decode_type_t::COOLIX;  // Set a protocol to use.
  ac.next.model = 1;  // Some A/Cs have different models. Try just the first.
  ac.next.mode = stdAc::opmode_t::kCool;  // Run in cool mode initially.
  ac.next.celsius = true;  // Use Celsius for temp units. False = Fahrenheit
  ac.next.degrees = 25;  // 25 degrees.
  ac.next.fanspeed = stdAc::fanspeed_t::kMedium;  // Start the fan at medium.
  ac.next.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
  ac.next.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
  ac.next.light = false;  // Turn off any LED/Lights/Display that we can.
  ac.next.beep = false;  // Turn off any beep from the A/C if we can.
  ac.next.econo = false;  // Turn off any economy modes if we can.
  ac.next.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
  ac.next.turbo = true;  // Don't use any turbo/powerful/etc modes.
  ac.next.quiet = false;  // Don't use any quiet/silent/etc modes.
  ac.next.sleep = -1;  // Don't set any sleep time or modes.
  ac.next.clean = false;  // Turn off any Cleaning options if we can.
  ac.next.clock = -1;  // Don't set any current time if we can avoid it.
  ac.next.power = true;  // Initially start with the unit off.
```
In my setup I fired this frame when my AC is OFF and It turns ON and start in turbo mode

crankyoldgit, 2021-06-23:
Thanks for confirming it works

closed 2021-06-23 by crankyoldgit

crankyoldgit, 2021-07-06:
FYI, the committed & merged changes thus far have been included in the newly released version of the library. i.e. v2.7.19

PR #1502 (2021-06-23): IRac: Change Coolix to send special messages after a normal message.
Sending a special message after a normal message ensures the unit is on.
Reports are that if a state with special setting (e.g. Turbo) and turning the unit on, the special message is ignored because the unit must be powered on before it will accept it.
So, change the order, send the normal state first, then the special message/settings, except for "Off" of course.

Fixes #1501
commit 609abce2 (2021-06-23): IRac: Change Coolix to send special messages after a normal message. (#1502)

Sending a special message after a normal message ensures the unit is on.
Reports are that if a state with special setting (e.g. Turbo) and turning the unit on, the special message is ignored because the unit must be powered on before it will accept it.
So, change the order, send the normal state first, then the special message/settings, except for "Off" of course.

Fixes #1501