crankyoldgit/IRremoteESP8266 #1503: Decode SANYO Air conditioner remote 270bits

dotanger, 2021-06-19:
### Version/revision of the library used
2.7.18 
running the IRRecvDumpV3

### Describe the bug
Can't Decode a remote control that is SANYO according to google search. (mine is marked Tornado - probably OEM).
Getting protocol: UNKNOWN

#### Expected behaviour
find protocol: SANYO and the codes received

#### Output of raw data from [IRrecvDumpV3.ino]

For On/23C/Auto/cool/no swing I get different code each times, here are two on/off and on/off:

```
ON1
Timestamp : 003963.999
Library   : v2.7.18

Protocol  : UNKNOWN
Code      : 0x426182FB (270 Bits)
Raw Timing[539]:
   +  5486, -  1874,    +   580, -   700,    +   576, -  1432,    +   578, -   676, 
   +   576, -  1432,    +   578, -   676,    +   576, -  1432,    +   580, -   702, 
   +   572, -  1438,    +   552, -   698,    +   574, -   676,    +   574, -   678, 
   +   574, -   678,    +   572, -  1438,    +   572, -   708,    +   574, -  1438, 
   +   556, -   724,    +   554, -   698,    +   548, -   700,    +   552, -   706, 
   +   566, -   680,    +   606, -   704,    +   548, -  1462,    +   512, -   786, 
   +   492, -  1516,    +   496, -  1516,    +   494, -  1516,    +   494, -  1518, 
   +   544, -   734,    +   546, -  1466,    +   506, -   774,    +   506, -   748, 
   +   504, -   774,    +   506, -  1522,    +   494, -  1514,    +   494, -   758, 
   +   494, -   784,    +   494, -   758,    +   494, -   758,    +   494, -   740, 
   +   510, -   742,    +   510, -  1500,    +   512, -   740,    +   534, -   762, 
   +   544, -  1466,    +   544, -  1466,    +   546, -   734,    +   546, -   706, 
   +   572, -   682,    +   572, -   680,    +   574, -   704,    +   576, -   702, 
   +   578, -   676,    +   704, -   704,    +   580, -   700,    +   580, -   654, 
   +   672, -   676,    +   582, -   698,    +   560, -   724,    +   654, -   696, 
   +   582, -   652,    +   596, -   658,    +   568, -   684,    +   570, -   788, 
   +   574, -   696,    +   556, -   696,    +   556, -   826,    +   546, -   760, 
   +   520, -   732,    +   560, -   772,    +   520, -   734,    +   568, -   684, 
   +   570, -   708,    +   572, -   682,    +   572, -   704,    +   602, -   652, 
   +   604, -   650,    +   604, -   674,    +   580, -   672,    +   580, -   698, 
   +   580, -   698,    +   518, -   734,    +   518, -   734,    +   518, -   734, 
   +   518, -   760,    +   570, -   736,    +   516, -   764,    +   514, -   766, 
   +   618, -  1394,    +   562, -  3660,    +  5398, -  1942,    +   578, -   702, 
   +   576, -  1450,    +   578, -   734,    +   596, -  1432,    +   576, -   674, 
   +   682, -  1432,    +   576, -   702,    +   578, -  1432,    +   578, -   676, 
   +   576, -   676,    +   578, -   700,    +   574, -   706,    +   574, -  1436, 
   +   572, -   680,    +   598, -  1412,    +   574, -   706,    +   572, -   680, 
   +   562, -   718,    +   592, -   760,    +   518, -   716,    +   536, -   760, 
   +   596, -  1468,    +   516, -   736,    +   516, -  1496,    +   514, -  1498, 
   +   556, -  1454,    +   476, -  1552,    +   518, -   716,    +   506, -  1520, 
   +   520, -   734,    +   518, -   760,    +   518, -   718,    +   534, -  1492, 
   +   552, -  1458,    +   552, -   702,    +   552, -   700,    +   552, -   702, 
   +   552, -   700,    +   550, -   788,    +   570, -   704,    +   550, -  1464, 
   +   512, -   736,    +   550, -   702,    +   550, -  1462,    +   490, -  1526, 
   +   560, -   718,    +   560, -   692,    +   456, -   822,    +   456, -   798, 
   +   454, -   824,    +   592, -   730,    +   548, -   732,    +   546, -   706, 
   +   624, -   708,    +   548, -   738,    +   538, -   708,    +   544, -   710, 
   +   488, -   864,    +   518, -   786,    +   492, -   760,    +   492, -   838, 
   +   492, -   788,    +   490, -   760,    +   490, -   892,    +   518, -   760, 
   +   492, -   760,    +   492, -   760,    +   490, -   742,    +   510, -   788, 
   +   492, -   786,    +   492, -   762,    +   492, -   742,    +   510, -   786, 
   +   492, -   762,    +   490, -   764,    +   486, -   748,    +   504, -   774, 
   +   504, -   748,    +   474, -   776,    +   616, -   762,    +   518, -   760, 
   +   518, -   736,    +   616, -   748,    +   536, -   718,    +   500, -   752, 
   +   502, -   796,    +   484, -   820,    +   544, -  1414,    +   544, -  3688, 
   +  5528, -  1970,    +   476, -   776,    +   506, -  1524,    +   568, -   764, 
   +   488, -  1522,    +   488, -   762,    +   490, -  1522,    +   484, -   798, 
[... 254 lines cut ...]
   +   430, -   822,    +   432, -   820,    +   458, -   744,    +   510, -   848, 
   +   536, -   768,    +   510, -   770,    +   508, -   872,    +   430, -   822, 
   +   432, -   794,    +   484, -   744,    +   508, -   744,    +   508, -   744, 
   +   508, -   744,    +   508, -   744,    +   510, -   744,    +   508, -   736, 
   +   516, -   770,    +   508, -   770,    +   508, -   770,    +   508, -   770, 
   +   508, -   772,    +   506, -   744,    +   560, -   768,    +   484, -   744, 
   +   508, -   688,    +   486

uint16_t rawData[539] = {5472, 1876,  572, 750,  560, 1448,  562, 718,  638, 1430,  564, 688,  564, 1450,  572, 706,  602, 1424,  586, 666,  588, 692,  586, 666,  584, 652,  594, 1486,  584, 666,  586, 1422,  586, 696,  582, 668,  602, 678,  582, 670,  582, 666,  584, 668,  586, 1428,  580, 720,  600, 688,  592, 1418,  626, 1426,  582, 1428,  582, 670,  582, 1428,  582, 704,  578, 672,  578, 670,  582, 670,  584, 694,  638, 674,  554, 694,  582, 1432,  562, 686,  686, 694,  584, 668,  584, 1426,  584, 670,  662, 1404,  570, 1442,  598, 1430,  560, 676,  576, 720,  560, 676,  562, 690,  564, 688,  600, 680,  644, 704,  570, 684,  572, 706,  650, 686,  604, 646,  606, 644,  608, 672,  606, 722,  632, 672,  608, 646,  606, 724,  610, 644,  608, 670,  610, 644,  608, 644,  608, 644,  610, 642,  610, 644,  608, 670,  610, 668,  610, 644,  608, 646,  608, 670,  606, 672,  606, 672,  606, 672,  632, 648,  606, 676,  602, 648,  602, 650,  600, 754,  604, 674,  572, 666,  556, 690,  602, 650,  560, 720,  576, 590,  572, 3578,  5402, 1944,  564, 758,  520, 1488,  524, 756,  522, 1490,  520, 730,  522, 1488,  522, 762,  516, 1496,  562, 686,  564, 716,  514, 738,  514, 734,  518, 1496,  558, 720,  568, 1458,  556, 724,  554, 698,  552, 686,  596, 724,  554, 700,  554, 682,  568, 1458,  604, 700,  552, 700,  552, 1458,  552, 1462,  548, 1466,  460, 822,  456, 1554,  566, 728,  550, 686,  508, 744,  512, 784,  492, 742,  480, 772,  458, 792,  490, 1526,  508, 786,  520, 758,  494, 756,  494, 1516,  496, 810,  494, 1516,  494, 1516,  492, 1524,  482, 770,  484, 794,  480, 802,  502, 748,  618, 730,  522, 732,  520, 732,  522, 732,  516, 718,  538, 740,  536, 720,  502, 748,  456, 794,  458, 794,  458, 820,  458, 824,  476, 774,  480, 800,  508, 772,  532, 722,  500, 750,  588, 760,  518, 734,  518, 734,  492, 760,  490, 840,  492, 760,  492, 760,  492, 760,  492, 760,  492, 788,  492, 760,  492, 744,  506, 748,  508, 788,  490, 762,  486, 748,  476, 774,  538, 786,  492, 786,  492, 760,  570, 762,  490, 734,  492, 3762,  5478, 1990,  518, 762,  492, 1518,  606, 764,  502, 1520,  492, 788,  492, 1518,  490, 762,  490, 1518,  492, 762,  516, 762,  516, 738,  540, 738,  516, 1494,  516, 764,  514, 1496,  486, 768,  512, 738,  516, 738,  514, 738,  514, 764,  514, 842,  540, 1496,  514, 764,  566, 738,  514, 1498,  482, 1578,  454, 1508,  530, 770,  456, 1524,  512, 730,  494, 758,  466, 824,  428, 824,  536, 766,  512, 766,  512, 740,  590, 1472,  512, 744,  508, 742,  512, 740,  512, 1498,  536, 822,  430, 1582,  482, 1498,  512, 1498,  512, 766,  512, 742,  512, 740,  512, 768,  510, 768,  512, 768,  512, 766,  512, 766,  590, 758,  520, 770,  510, 740,  590, 742,  510, 768,  510, 820,  562, 768,  510, 770,  508, 770,  558, 824,  430, 822,  432, 820,  458, 744,  510, 848,  536, 768,  510, 770,  508, 872,  430, 822,  432, 794,  484, 744,  508, 744,  508, 744,  508, 744,  508, 744,  510, 744,  508, 736,  516, 770,  508, 770,  508, 770,  508, 770,  508, 772,  506, 744,  560, 768,  484, 744,  508, 688,  486};  // UNKNOWN 225422BE
```

### Circuit diagram and hardware used (if applicable)
Used an esp8266 gpio 4 using a TSOP4838 IR receiver.

### I have followed the steps in the [Troubleshooting Guide]
Yes. tried auto_analyse_raw_data.py to the best of my ability.

### Has this library/code previously worked as expected for you?
Yes, including this last version for the same remotes I've tested before.

Any suggestions would be appreciated.
remote control picture attached as well for reference.

[image]
[image]

crankyoldgit, 2021-06-21:
It looks like this is an unsupported protocol at the moment. How far thru https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol did you get?

crankyoldgit, 2021-06-21:
Looking at your raw data via our tools, it looks like you've got a poor quality capture. That could be for a number of reasons.
See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#receiving-an-ir-signal

I had to bump up the _range_ matching on our tools to make heads or tails of your data. i.e. `-r 300`. That is a strong indicator something is not good with your capture setup/hardware etc.

It appears to be a 88 bit message with two identical repeats. i.e. 3 x 88 bit messages.

```
$ ./tools/auto_analyse_raw_data.py -f /tmp/off2 -r 300
Found 539 timing entries.
Potential Mark Candidates:
[5478, 686]
Potential Space Candidates:
[3762, 1990, 1582, 872]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 5450
kHdrSpace  = 1936
kBitMark   = 538
kOneSpace  = 1480
kZeroSpace = 732
kSpaceGap = 3670

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+0101010100001010000001001110100000001000101110000000000000000000000000000000000000000000GAP(3578)
  Bits: 88
  Hex:  0x550A04E808B80000000000 (MSB first)
        0x00000000001D10172050AA (LSB first)
  Dec:  102806008838864397892845568 (MSB first)
        31954944676010 (LSB first)
  Bin:  0b0101010100001010000001001110100000001000101110000000000000000000000000000000000000000000 (MSB first)
        0b0000000000000000000000000000000000000000000111010001000000010111001000000101000010101010 (LSB first)
kHdrMark+kHdrSpace+0101010100001010000001001110100000001000101110000000000000000000000000000000000000000000GAP(3762)
  Bits: 88
  Hex:  0x550A04E808B80000000000 (MSB first)
        0x00000000001D10172050AA (LSB first)
  Dec:  102806008838864397892845568 (MSB first)
        31954944676010 (LSB first)
  Bin:  0b0101010100001010000001001110100000001000101110000000000000000000000000000000000000000000 (MSB first)
        0b0000000000000000000000000000000000000000000111010001000000010111001000000101000010101010 (LSB first)
kHdrMark+kHdrSpace+0101010100001010000001001110100000001000101110000000000000000000000000000000000000000000
  Bits: 88
  Hex:  0x550A04E808B80000000000 (MSB first)
        0x00000000001D10172050AA (LSB first)
  Dec:  102806008838864397892845568 (MSB first)
        31954944676010 (LSB first)
  Bin:  0b0101010100001010000001001110100000001000101110000000000000000000000000000000000000000000 (MSB first)
        0b0000000000000000000000000000000000000000000111010001000000010111001000000101000010101010 (LSB first)

Total Nr. of suspected bits: 264
```

We can potentially work with the data you've collected but I am reluctant to do so till you can get cleaner captures. That is, the respective Mark and Space values are within an acceptable/repeatable range for their reference. i.e. The default range for a value is 200 usecs (+/-100 usecs), which most protocol values are typical _well_ within. Most captures have their respective values within +/-50 usecs. Your captures required +/-150 usecs to get _any_ semblance of decoding.

dotanger, 2021-06-21:
HI,
thanks a lot for getting back so quickly.
I didn't go too far with the Adding-support-for-a-new-IR-protocol because I couldn't make heads or tails of my data especially due to the fact that the codes are not repeating at all.
I read the manual you've shared on mitigating bad capture and I will try capturing the signal over 30mm away, I think I did violate that.

I have two tv remotes which work fine and another coolix AC remote that also works fine with out a glitch using this setup.

could it be that the transmission beam BW is wide enough to be received by the 38kHz receiver and actually I have the wrong frequency receiver for this remote?

until I get to do that this is my setup schematics:
[image]
this is how it looks:
[image]

crankyoldgit, 2021-06-21:
> I read the manual you've shared on mitigating bad capture and I will try capturing the signal over 30mm away, I tried capturing from over 10cm away but it's the same.

10cm is possibly even too close. 30cm to 100cm is typically optimal in my experience. ie. Remotes are not designed with Point Blank operation in mind.

> 
> I have two tv remotes which work fine and another coolix AC remote that also works fine with out a glitch using this setup.

Good to know.

> 
> could it be that the transmission beam BW is wide enough to be received by the 38kHz receiver and actually I have the wrong frequency receiver for this remote?

That is entirely possible.  A 36, 40, or 56 kHz signal could present like this to a 38kHz receiver. 
An oscilloscope is the definitive tool required on the remote to be sure of the frequency in operation.
 
> maybe parameters for the capture program?

Not really. In your situation, the raw data from the receiver looks fine. The timing issues are up stream from the ESP. i.e. receiver/remote/environment etc.

The library is going to recognise it as `UNKNOWN` regardless as it doesn't (yet) support this 88 bit protocol.

> I didn't go too far with the Adding-support-for-a-new-IR-protocol because I couldn't make heads or tails of my data especially due to the fact that the codes are not repeating at all.

In the meantime, try adding/using the `-r 300` parameter argument to [`tools/auto_analyse_raw_data.py`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py). It will help you see what is going on.

dotanger, 2021-06-25:
Hi,
Can you please look at the following data again? I played with it now for a while and distance helps to give me confidence - I put the transmitter well away and the readings seem more stable. using -r 200 still doesn't work though.
btw, it's hard to source IR receivers with frequencies other than 38kHz. ideas?
Another point to note, is that this remote has two IR transmitters it seems. I tried to block one of them for some of the readings each time and the result was the same. don't know what that's about.

I guess my question is if you still think the problem is with the setup? is the timing still bad? or can it be maybe something else?
I tried 2 other remotes I have, 3 AC in total and 2 tv and 1 audio receiver and they work great...

a scope is harder to locate (though may be possible)

ON:
`Protocol  : UNKNOWN
Code      : 0xABF4C698 (270 Bits)
Raw Timing[539]:
   +  5374, -  1972,    +   512, -   762,    +   522, -  1510,    +   498, -   780, 
   +   578, -  1486,    +   462, -   790,    +   462, -  1550,    +   544, -   708, 
   +   510, -  1500,    +   544, -  1470,    +   516, -   762,    +   482, -   770, 
   +   516, -  1496,    +   518, -  1510,    +   498, -   780,    +   468, -  1544, 
   +   466, -   786,    +   466, -   786,    +   466, -   786,    +   466, -   786, 
   +   486, -   748,    +   526, -   756,    +   506, -  1520,    +   516, -   764, 
   +   516, -  1492,    +   518, -   762,    +   544, -   708,    +   548, -   704, 
   +   602, -  1434,    +   556, -  1456,    +   524, -   726,    +   552, -   700, 
   +   554, -   698,    +   648, -  1438,    +   548, -   684,    +   536, -  1478, 
   +   568, -   682,    +   568, -   682,    +   578, -   702,    +   568, -   684, 
   +   570, -   712,    +   624, -   724,    +   528, -  1482,    +   528, -   726, 
   +   528, -  1482,    +   526, -   724,    +   546, -  1524,    +   568, -   682, 
   +   540, -   710,    +   544, -   734,    +   542, -  1516,    +   492, -   762, 
   +   488, -   744,    +   546, -  1482,    +   492, -   788,    +   544, -   734, 
   +   494, -   758,    +   492, -   760,    +   490, -   744,    +   548, -   828, 
   +   494, -   760,    +   494, -   758,    +   494, -   758,    +   494, -   786, 
   +   492, -   760,    +   494, -   758,    +   494, -   742,    +   508, -   746, 
   +   508, -   788,    +   492, -   760,    +   494, -   786,    +   494, -   786, 
   +   494, -   742,    +   508, -   744,    +   508, -   788,    +   570, -   760, 
   +   494, -   784,    +   494, -   788,    +   568, -   786,    +   492, -   744, 
   +   508, -   772,    +   608, -   768,    +   538, -   714,    +   490, -   786, 
   +   628, -   728,    +   494, -   786,    +   494, -   758,    +   494, -   742, 
   +   508, -  1390,    +   494, -  3692,    +  5394, -  1960,    +   494, -   784, 
   +   494, -  1516,    +   494, -   760,    +   494, -  1516,    +   494, -   760, 
   +   492, -  1516,    +   494, -   760,    +   492, -  1518,    +   524, -  1486, 
   +   494, -   790,    +   486, -   764,    +   492, -  1524,    +   538, -  1472, 
   +   540, -   740,    +   538, -  1546,    +   546, -   728,    +   524, -   726, 
   +   526, -   728,    +   524, -   728,    +   526, -   728,    +   524, -   728, 
   +   524, -  1486,    +   526, -   726,    +   546, -  1464,    +   550, -   702, 
   +   550, -   728,    +   552, -   702,    +   550, -  1460,    +   550, -  1460, 
   +   550, -   702,    +   552, -   728,    +   550, -   702,    +   550, -  1460, 
   +   550, -   728,    +   550, -  1464,    +   548, -   704,    +   546, -   704, 
   +   548, -   732,    +   546, -   704,    +   548, -   704,    +   548, -   730, 
   +   548, -  1462,    +   548, -   706,    +   518, -  1492,    +   546, -   706, 
   +   518, -  1490,    +   546, -   706,    +   520, -   732,    +   520, -   732, 
   +   522, -  1492,    +   518, -   760,    +   518, -   732,    +   546, -  1468, 
   +   516, -   736,    +   516, -   732,    +   520, -   732,    +   520, -   758, 
   +   520, -   818,    +   478, -   772,    +   480, -   802,    +   454, -   894, 
   +   520, -   734,    +   518, -   734,    +   518, -   760,    +   518, -   734, 
   +   518, -   734,    +   518, -   718,    +   506, -   746,    +   480, -   818, 
   +   488, -   790,    +   490, -   744,    +   508, -   790,    +   490, -   744, 
   +   510, -   770,    +   508, -   788,    +   490, -   762,    +   544, -   788, 
   +   490, -   788,    +   462, -   818,    +   566, -   788,    +   490, -   744, 
   +   480, -   770,    +   456, -   794,    +   458, -   894,    +   492, -   762, 
   +   492, -   760,    +   490, -   744,    +   508, -  1390,    +   492, -  3664, 
   +  5398, -  1960,    +   490, -   746,    +   476, -  1552,    +   492, -   742, 
   +   508, -  1504,    +   508, -   746,    +   504, -  1522,    +   494, -   786, 
   +   492, -  1516,    +   494, -  1516,    +   494, -   760,    +   492, -   786, 
   +   494, -  1520,    +   490, -  1520,    +   458, -   822,    +   458, -  1554, 
   +   510, -   788,    +   494, -   784,    +   494, -   760,    +   492, -   742, 
   +   508, -   770,    +   512, -   786,    +   492, -  1516,    +   520, -   760, 
   +   520, -  1488,    +   520, -   734,    +   520, -   760,    +   518, -   734, 
   +   520, -  1490,    +   520, -  1494,    +   516, -   734,    +   518, -   732, 
   +   520, -   762,    +   516, -   762,    +   488, -  1522,    +   458, -  1554, 
[... cut ...] 524,  598, 1494,  516, 734,  520, 734,  518, 734,  566, 1500,  458, 796,  456, 820,  480, 1532,  508, 772,  506, 746,  504, 748,  476, 820,  540, 816,  518, 734,  518, 760,  540, 798,  508, 772,  506, 744,  510, 772,  506, 822,  538, 736,  518, 734,  518, 736,  516, 736,  490, 746,  508, 790,  516, 762,  516, 736,  516, 716,  508, 744,  504, 748,  478, 800,  478, 772,  480, 772,  480, 800,  586, 794,  516, 736,  514, 766,  538, 796,  456, 796,  478, 800,  510, 774,  506, 1392,  514};'
`

Thanks!

NiKiZe, 2021-06-25:
The best way to know is to open the remote and put an oscilloscope over the LEDs.
If there is multiple LEDs they are most often just in series.

crankyoldgit, 2021-06-26:
I'll try to take another look at the data soon.
In the meantime, please include all the model info of both the remote and the AC unit you can. Try searching for a circuit diagram for the AC unit. You might get lucky and it could list the part number for the ir receiver. Maybe call their service dept, and they may be able to tell you the part &/or carrier frequency.

Have you tried fresh batteries? (Long shot)

Their are a number of electronic component stores that should carry ir demodulators tuned to various carrier frequencies. E.g. element14, rs-online, eBay, Amazon, AliExpress, etc.

A logic analyser may work too instead of an oscilloscope. No promises.

crankyoldgit, 2021-06-29:
@dotanger I've added some basic experimental support for this protocol. You'll need to test it (sending and receiving) against the real thing. It may work, it may not. It's just a stab in the dark. Download the branch via: https://github.com/crankyoldgit/IRremoteESP8266/tree/SanyoAc88
 
Note: Don't count or rely on the `state[]` values reported by it at present, this may change as per: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#determining-the-bit-order-of-the-protocol

We need the model/brand info for the A/C unit and the remote too. Also, how sure are you that this really is a Sanyo device or protocol? So far, it doesn't look very Sanyo-like.

dotanger, 2021-06-29:
Thanks a lot!
I had a quick check for the branch.
it does get the commands as the protocol. I didn't try to send yet.
I will do further testing during the weekend including changing different states to try and see what part is what.

I will have to do a lot of checks I think because it wasn't 100 percent stable. most of the encoding shows a 'header' of AA50 and then suddenly I had a streak of AA4F.

Regarding the remote, I have a problem at the moment because there is absolutely no indication on the plastic/stickers/PCB - absolutely nothing. I did find remotes from other stores that claim it's a substitute but the model seems private for the store. I found the same looking remote on aliexpress after photographing the remote and applying google lens on it. most to all references are for a Sanyo AC.
I will try to investigate more during the weekend.

```
Timestamp : 000726.364
Library   : v2.7.18

Protocol  : SANYO_AC88
Code      : 0xAA50 A0 16 07 170 200000080 (88 Bits)
uint16_t rawData[539] = {5438, 1900,  554, 726,  552, 1456,  554, 698,  554, 1454,  554, 698,  554, 1456,  554, 698,  554, 1538,  576, 678,  568, 684,  566, 684,  564, 686,  534, 1478,  568, 684,  566, 1446,  574, 726,  556, 722,  558, 696,  552, 682,  572, 724,  556, 724,  554, 1454,  556, 698,  554, 1454,  556, 696,  556, 1454,  556, 1452,  558, 696,  556, 1454,  530, 752,  530, 720,  554, 698,  556, 1456,  562, 1452,  566, 1462,  550, 814,  592, 712,  566, 686,  568, 812,  526, 726,  526, 1482,  632, 1482,  492, 1518,  494, 760,  492, 1518,  494, 758,  494, 758,  490, 762,  536, 698,  550, 1478,  542, 738,  594, 710,  544, 734,  570, 682,  676, 690,  588, 664,  558, 692,  590, 692,  584, 762,  578, 702,  578, 676,  578, 778,  600, 704,  578, 658,  590, 764,  598, 674,  578, 674,  578, 782,  594, 660,  584, 668,  590, 688,  592, 690,  592, 702,  576, 660,  588, 664,  590, 704,  576, 658,  594, 688,  592, 702,  578, 658,  590, 660,  586, 664,  560, 694,  638, 668,  594, 676,  570, 662,  558, 694,  560, 1340,  578, 3576,  5476, 1868,  554, 696,  552, 1458,  556, 694,  550, 1462,  558, 694,  556, 1454,  590, 706,  572, 1438,  572, 680,  546, 732,  546, 734,  544, 708,  544, 1464,  544, 708,  544, 1466,  544, 708,  544, 708,  544, 710,  542, 710,  594, 710,  540, 738,  540, 1476,  538, 708,  542, 1502,  456, 794,  544, 1466,  508, 1522,  488, 744,  506, 1522,  492, 760,  490, 744,  510, 768,  508, 1518,  492, 1516,  492, 1516,  464, 816,  492, 786,  492, 760,  544, 738,  490, 762,  490, 1522,  488, 1526,  506, 1504,  508, 788,  492, 1518,  492, 760,  492, 788,  492, 760,  492, 786,  492, 1518,  492, 760,  522, 756,  524, 756,  522, 730,  522, 756,  522, 756,  546, 738,  516, 732,  522, 730,  522, 758,  520, 786,  538, 742,  534, 716,  458, 794,  458, 794,  460, 820,  458, 822,  530, 720,  532, 746,  534, 744,  532, 718,  534, 746,  536, 762,  518, 716,  540, 758,  520, 716,  534, 744,  536, 716,  534, 762,  516, 816,  520, 758,  520, 732,  542, 796,  478, 774,  480, 800,  480, 774,  456, 896,  544, 1438,  520, 3692,  5492, 1974,  456, 794,  458, 1556,  476, 776,  454, 1572,  486, 820,  516, 1492,  518, 736,  540, 1524,  460, 822,  456, 796,  458, 822,  564, 762,  492, 1518,  492, 788,  492, 1518,  492, 762,  492, 760,  492, 762,  490, 762,  570, 764,  490, 760,  492, 1520,  490, 760,  596, 1518,  492, 760,  492, 1518,  492, 1516,  492, 764,  510, 1496,  518, 764,  516, 736,  518, 734,  516, 1494,  516, 1494,  490, 1522,  508, 774,  506, 772,  506, 746,  478, 772,  508, 772,  506, 1506,  532, 1494,  518, 1492,  518, 734,  518, 1492,  518, 734,  518, 736,  518, 736,  516, 760,  518, 1492,  518, 736,  518, 734,  518, 736,  542, 762,  518, 762,  516, 736,  640, 722,  506, 792,  516, 716,  614, 736,  516, 736,  516, 736,  516, 762,  516, 846,  478, 774,  508, 772,  504, 848,  516, 762,  516, 762,  516, 814,  510, 746,  478, 774,  478, 902,  516, 764,  514, 764,  514, 842,  540, 764,  514, 764,  514, 818,  458, 796,  456, 822,  456, 896,  516, 738,  512, 740,  484, 746,  508, 772,  532, 720,  504, 1396,  512};  // SANYO_AC88
uint8_t state[11] = {0xAA, 0x50, 0xA0, 0x16, 0x07, 0x17, 0x02, 0x00, 0x00, 0x00, 0x80};

Timestamp : 000731.596
Library   : v2.7.18

Protocol  : SANYO_AC88
Code      : 0xAA50 20 16 0C 170 200000080 (88 Bits)
uint16_t rawData[539] = {5454, 1902,  548, 836,  554, 1460,  566, 712,  612, 1470,  554, 698,  554, 1454,  556, 698,  554, 1456,  552, 698,  554, 724,  554, 726,  554, 724,  606, 1438,  566, 682,  564, 1468,  554, 724,  554, 682,  570, 726,  554, 724,  554, 726,  602, 732,  522, 1488,  564, 716,  564, 688,  598, 728,  550, 1460,  550, 1460,  550, 728,  550, 1462,  550, 702,  552, 702,  552, 728,  552, 700,  586, 720,  568, 1444,  566, 1448,  568, 702,  548, 714,  572, 722,  558, 696,  550, 1460,  556, 1454,  556, 1454,  556, 696,  556, 1454,  554, 700,  552, 724,  528, 726,  526, 726,  526, 1482,  528, 726,  494, 760,  652, 682,  546, 734,  480, 798,  566, 760,  492, 786,  540, 740,  646, 718,  560, 738,  544, 734,  598, 708,  570, 708,  574, 680,  574, 704,  576, 756,  568, 690,  562, 720,  586, 762,  576, 704,  576, 676,  576, 676,  576, 676,  576, 676,  576, 702,  576, 702,  578, 676,  576, 702,  578, 676,  576, 676,  576, 702,  576, 676,  576, 702,  576, 676,  630, 676,  576, 676,  576, 676,  576, 1306,  574, 3578,  5478, 1878,  574, 678,  574, 1436,  574, 680,  572, 1436,  572, 680,  572, 1438,  574, 704,  546, 1470,  546, 732,  458, 822,  458, 794,  458, 794,  548, 1462,  458, 796,  544, 1468,  478, 774,  478, [... cut ...]   518, 718,  588, 762,  518, 734,  518, 734,  518, 734,  518, 812,  518, 736,  516, 764,  514, 740,  486, 786,  490, 792,  484, 768,  488, 792,  484, 766,  514, 738,  516, 766,  512, 738,  516, 736,  516, 736,  516, 738,  516, 736,  594, 736,  488, 794,  458, 792,  514, 740,  512, 842,  540, 764,  514, 738,  514, 816,  486, 794,  512, 740,  514, 738,  514, 766,  512, 740,  514, 1372,  504};  // SANYO_AC88
uint8_t state[11] = {0xAA, 0x50, 0x20, 0x16, 0x30, 0x1B, 0x02, 0x00, 0x00, 0x00, 0x80};
```

crankyoldgit, 2021-06-30:
> Regarding the remote, I have a problem at the moment because there is absolutely no indication on the plastic/stickers/PCB - absolutely nothing.

But what about the A/C unit itself?

Re: Your collected data.

If you don't tell us what the corresponding settings are for each `state[]` code, no one can help you.
See/read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#a-note-on-collecting-data
All of the data you've included at present doesn't tell us anything other than the library is now correctly detecting your protocol. Thanks for the confirmation btw.

The **next** step is to collect a temperature range so we can check if we have the _bit ordering_ correct or not.
See/read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#determining-the-bit-order-of-the-protocol

dotanger, 2021-07-02:
Hi @crankyoldgit,

> But what about the A/C unit itself?
It is a mini central or whatever it's called so I don't really have anything to go with. it is inside a compartment in the ceiling.
I did have a go with the remote control again and took apart the display and found underneath a "Rolbit" chip part number 86C820F-3DC7. it seems like TOSHIBA has something similar with 86C820F. I guess it needs programming anyway so that is probably not helpful...

I found remotes which are very similar in how they look, most to all are considered Sanyo replacement.
many hits by searching "sanyo air conditioner remote control" in google.
example from amazon: "HCDZ Replacement Remote Control for Sanyo KS2462R RCS-5S1E KS2432A 26KS72R KHS0952 RCS-BH80UA-WL RCS-6FHPMS2C-D Multi-Split AC Air Conditioner." https://www.amazon.com/HCDZ-Replacement-RCS-BH80UA-WL-RCS-6FHPMS2C-D-Multi-Split/dp/B00UV3TZQ6
example from joybuy "Air conditioner remote control suitable for sanyo air conditioning 2GHR1 replace RCS-5WS1C RCS-7WN1C". https://www.joybuy.com/product/650567987.html

I did further testing and am happy to report that the codes are sent well (on/off with 22C, Cool, auto) and accepted by the AC receiver.

I will continue in trying to map bits for the modes/temp/etc.

Thanks!

crankyoldgit, 2021-07-03:
> It is a mini central or whatever it's called so I don't really have anything to go with. it is inside a compartment in the ceiling.

Well, that's a bugger. :-(

> I found remotes which are very similar in how they look, most to all are considered Sanyo replacement.
many hits by searching "sanyo air conditioner remote control" in google.
example from amazon: "HCDZ Replacement Remote Control for Sanyo KS2462R RCS-5S1E KS2432A 26KS72R KHS0952 RCS-BH80UA-WL RCS-6FHPMS2C-D Multi-Split AC Air Conditioner." https://www.amazon.com/HCDZ-Replacement-RCS-BH80UA-WL-RCS-6FHPMS2C-D-Multi-Split/dp/B00UV3TZQ6
example from joybuy "Air conditioner remote control suitable for sanyo air conditioning 2GHR1 replace RCS-5WS1C RCS-7WN1C". https://www.joybuy.com/product/650567987.html

I'm really reluctant to add "possible" supported brands/models etc. When we say something is supported, we like to be fairly sure.
Oh well.

> I did further testing and am happy to report that the codes are sent well (on/off with 22C, Cool, auto) and accepted by the AC receiver.

Excellent! Thanks for confirming that bit seems to be working. I'll note it in the code when we have the bit order confirmed.

> I will continue in trying to map bits for the modes/temp/etc.

Thanks in advance. Looking forward to the results.

dotanger, 2021-07-03:
This is what I have so far, I'd say it's the important parts.
The rest I would guess are relevant to timer of STOP and START and also just clock.

The majority is already covered so input would be welcome at this stage.

https://docs.google.com/spreadsheets/d/1weUmGAsEpfX38gg5rlDN69Uchnbr6gQl9FqHffLBIRk/edit?usp=sharing
how to read:
columns are the description of the field.
rows are the value of the fields.
the matrix is colored to description with values.
There is a use of bits to on/off features in this protocol. so more specific then nibbles and bytes.

Thanks

crankyoldgit, 2021-07-04:
Please make the document/spreadsheet publically readable.

dotanger, 2021-07-04:
done

dotanger, 2021-07-09:
I have been playing with ElectraAC coding (physically near my 'SanyoAc88)' as well for one of my ACs and I had 2 observations,
1. Sometimes my supposed 'Sanyo88Ac' beeped (despite not doing anything other than beep) - does that make any sense?
2. couldn't find a minRepeats for the ElectraAC, is that on purpose? (I know it's out of scope for this thread)

thanks

crankyoldgit, 2021-07-10:
Okay, that's weird.

I've got a few other issues on the go at the moment, and I'll get back to this issue/task when they are done. i.e. I haven't forgotten. Just busy & distracted.

crankyoldgit, 2021-07-10:
> 2\. couldn't find a minRepeats for the ElectraAC, is that on purpose? (I know it's out of scope for this thread)

Yes. ElectraAC doesn't have required repeats to work. You can make it repeat, but no captures for it ever showed repeats.

dotanger, 2021-07-16:
Hey,
Thanks for the clarification of ElectraAC, it makes sense.

updated the spreadsheet to include the bit for filter.
The remote has timers and they do have effect on the code obviously  - I did not cover those...

crankyoldgit, 2021-08-01:
@dotanger I've added some experimental _detailed_ support for this protocol. You'll need to test it (sending and receiving) against the real thing. It may work, it may not. It's based on your spreadsheet, which has a LOT of incomplete data & analysis.

Download the branch via: https://github.com/crankyoldgit/IRremoteESP8266/tree/SanyoAc88

dotanger, 2021-08-01:
Hi,
Thanks. I will test.
Regarding the missing info, my observation is that it's clock and timers.
I did find that I covered information you'd normally parse for other protocols.

Also, answered your comments in the spreadsheet.

crankyoldgit, 2021-08-10:
@dotanger I've added Clock support in a new commit to that branch. Have you tried it out yet? Any feedback?

dotanger, 2021-08-13:
Hey,
first test looks very much correct.
After transmitting I checked what the remote thinks about the time and it said 23.4 which is correct since it does not show the minutes and seconds.

I'll be trying to understand what the second byte is next.

Protocol  : SANYO_AC88
Code      : 0xAA5BA015102E1700000080 (88 Bits)
Mesg Desc.: Power: On, Mode: 2 (Cool), Temp: 21C, Fan: 0 (Auto), Swing(V): Off, Turbo: Off, Sleep: Off, Clock: 23:46

uint16_t rawData[539] = {5406, 1926,  556, 696,  556, 1454,  556, 696,  556, 1456,  552, 698,  556, 1456,  530, 750,  566, 1518,  556, 1454,  556, 1454,  556, 698,  632, 1438,  560, 1446,  574, 676,  568, 1462,  554, 680,  572, 726,  552, 682,  568, 682,  560, 692,  652, 700,  550, 1460,  550, 702,  550, 1460,  550, 1462,  550, 730,  464, 1548,  528, 724,  462, 1548,  534, 746,  532, 718,  530, 724,  526, 724,  524, 728,  650, 704,  548, 706,  546, 1464,  552, 726,  608, 726,  524, 754,  552, 702,  656, 1454,  556, 1454,  556, 1454,  556, 696,  558, 1454,  538, 744,  526, 724,  554, 1458,  532, 1478,  568, 1462,  550, 682,  568, 1462,  554, 724,  528, 724,  528, 706,  544, 708,  544, 708,  568, 682,  546, 734,  546, 704,  572, 682,  542, 710,  570, 710,  644, 710,  544, 692,  556, 696,  560, 720,  560, 766,  596, 680,  574, 704,  574, 792,  590, 660,  592, 706,  576, 754,  576, 676,  576, 704,  576, 676,  576, 704,  576, 676,  576, 678,  576, 702,  576, 676,  576, 702,  576, 702,  576, 702,  578, 706,  572, 1330,  576, 3574,  5442, 1898,  576, 676,  578, 1436,  552, 774,  600, 1434,  574, 704,  600, 1412,  570, 680,  554, 1454,  546, 1466,  574, 1440,  458, 792,  460, 1552,  560, 1468,  542, 694,  560, 1466,  544, 690,  562, 734,  544, 710,  490, 746,  508, 786,  494, 742,  512, 1516,  494, 740,  508, 1520,  496, 1514,  494, 758,  494, 1516,  494, 758,  494, 1514,  496, 758,  494, 758,  494, 784,  494, 784,  494, 784,  494, 784,  600, 712,  542, 1486,  462, 772,  540, 710,  566, 784,  494, 758,  494, 1516,  494, 1514,  496, 1514,  494, 758,  494, 1518,  492, 758,  494, 758,  494, 1516,  494, 1520,  490, 1522,  540, 738,  540, 1488,  522, 730,  522, 712,  540, 712,  542, 738,  540, 714,  538, 740,  544, 736,  542, 754,  526, 710,  622, 726,  526, 726,  526, 728,  524, 728,  526, 728,  524, 754,  550, 702,  550, 684,  538, 712,  536, 716,  458, 792,  460, 790,  520, 730,  522, 756,  546, 706,  546, 704,  548, 704,  524, 730,  522, 834,  542, 716,  460, 792,  460, 790,  462, 818,  460, 792,  518, 732,  518, 1384,  522, 3630,  5350, 1990,  518, 732,  520, 1494,  516, 736,  518, 1494,  460, 818,  460, 1570,  490, 744,  458, 1572,  518, 1492,  518, 1490,  518, 734,  518, 1490,  520, 1492,  516, 734,  492, 1518,  518, 764,  516, 734,  518, 764,  488, 762,  492, 760,  492, 786,  494, 1520,  460, 792,  490, 1520,  460, 1552,  482, 816,  488, 1522,  492, 786,  492, 1516,  494, 760,  494, 758,  494, 784,  494, 760,  494, 758,  494, 758,  492, 740,  480, 1550,  490, 744,  510, 786,  492, 760,  488, 746,  476, 1550,  492, 1518,  494, 1514,  516, 736,  518, 1490,  520, 734,  518, 734,  520, 1490,  520, 1490,  520, 1492,  490, 790,  460, 1554,  456, 792,  460, 822,  458, 792,  458, 792,  516, 734,  520, 732,  520, 732,  520, 732,  520, 760,  542, 766,  490, 762,  518, 734,  518, 734,  520, 814,  512, 790,  490, 742,  458, 794,  458, 792,  460, 794,  516, 734,  518, 732,  520, 762,  518, 734,  518, 734,  518, 734,  520, 734,  490, 816,  518, 734,  518, 760,  568, 768,  506, 746,  508, 772,  508, 792,  516, 788,  542, 1440,  518};  // SANYO_AC88
uint8_t state[11] = {0xAA, 0x5B, 0xA0, 0x15, 0x10, 0x2E, 0x17, 0x00, 0x00, 0x00, 0x80};

crankyoldgit, 2021-08-16:
So, it appears to be decoding the remote correctly. Is it able to control the A/C correctly?

dotanger, 2021-08-16:
It controls it perfectly

crankyoldgit, 2021-08-16:
> It controls it perfectly

Thanks for confirming.

closed 2021-08-16 by crankyoldgit

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

dotanger, 2021-09-02:
Thank a lot!
has been a pleasure.

PR #1568 (2021-08-16): Detailed support for Tornado/Sanyo 88-bit A/C protocol
* Settings supported:
  - Power
  - Mode
  - Temp
  - Fan Speed
  - Sleep
  - Turbo
  - Filter
  - Clock
* Add support in `IRac` class.
* Unit tests added & updated.
* Other minor code cleanups.

Fixes #1503
commit 89d039af (2021-08-16): Detailed support for Tornado/Sanyo 88-bit A/C protocol (#1568)

* Settings supported:
  - Power
  - Mode
  - Temp
  - Fan Speed
  - Sleep
  - Turbo
  - Filter
  - Clock
* Add support in `IRac` class.
* Unit tests added & updated.
* Other minor code cleanups.

Fixes #1503