crankyoldgit/IRremoteESP8266 #1508: Cooper&Hunter Protocol based on Gree

k-korn, 2021-06-26:
### Version/revision of the library used
2.7.18

### Intro
The hardware is as follows:
- esp-01+TSSOP+IR Leds device to send/receive IR signals, managed over MQTT.
- Couple of other home appliances perfectly accepting commands from IRremoteESP8266
- Saleae Logic Analyzer to see what happens after IR receiver
- Cooper&Hunter AC, not currently controllable with this library.

### Previous Works
As mentioned in #602, some other software had success in controlling C&H with Gree protocol, so I have tried comparing signals sent by this library on Gree protocol with what is actually sent by C&H remote, and binary protocol indeed matches.

Timings, however, are slightly different - inter-message space is 19980ms instead of 19000ms for C&H, for example.
So "as is" Gree protocol does not work

### Raw Data
I've attached a dump of actual C&H remote signals in Saleae Logic format (viewable with https://www.saleae.com/downloads/, no actual hardware needed to view):
[c_h-examples.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/6719692/c_h-examples.zip)

### The Suggestion
Cooper&Hunter AC can be controlled if we replace constants in  https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Gree.cpp#L22 with the following:
```
const uint16_t kGreeHdrMark = 9000;
const uint16_t kGreeHdrSpace = 4500;
const uint16_t kGreeBitMark = 620;
const uint16_t kGreeOneSpace = 1670;
const uint16_t kGreeZeroSpace = 570;
const uint16_t kGreeMsgSpace = 19980;
```
(most significant change is kGreeMsgSpace)

### The Issue
However, I am unsure how to properly add new AC protocol to the library with proper detection, etc - so this is not a pull request, but an issue instead.
Gree protocol with the changes outlined above works perfectly for me controlling C&H air conditioner - so it may be helpful to others with similar hardware.

NiKiZe, 2021-06-26:
We really like to have the "rawData" that you get with IRrecvDumpV2+.
Also verify what receiver you actually have, as stated in FAQ and issue template.

Please give full brand and model of both unit and remote.

k-korn, 2021-06-26:
> We really like to have the "rawData" that you get with IRrecvDumpV2+.
My hardware (esp-01) makes it difficult to have both serial console and IR receiver to be available simultaneously, but I'll try to obtain that data.
Here are some decoded binary messages (32bit first half + spacer + 32bit second half): [ch-dumps.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/6719742/ch-dumps.txt)

> Also verify what receiver you actually have, as stated in FAQ and issue template.
> 
> Please give full brand and model of both unit and remote.
Remote: YB1F2 (Google search even shows it's Gree branded, but my has C&H logo on it)
Unit: CH-S09FTXG

crankyoldgit, 2021-06-26:
FYI, try the Kelvinator protocol, it is very similar to the Gree protocol, and has the 20ms gap as you're describing.

k-korn, 2021-06-26:
> FYI, try the Kelvinator protocol, it is very similar to the Gree protocol, and has the 20ms gap as you're describing.

It is indeed similar in timings, and even works on basic level (on/off, set temps, etc).
However, actual features (number of fan modes, overall message length of 8 bytes and not 16, etc),
Cooper&Hunter looks more similar to
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Gree.h#L36
than to
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Kelvinator.h#L37

crankyoldgit, 2021-06-26:
> Remote: YB1F2 (Google search even shows it's Gree branded, but my has C&H logo on it)
> Unit: CH-S09FTXG

Thanks for that. I've added it to the supported list.

> Timings, however, are slightly different - inter-message space is 19980ms instead of 19000ms for C&H, for example.
> So "as is" Gree protocol does not work

I've reviewed previous data we have, and your word on your data, and my personal experience with the Kelvinator protocol which overlaps a lot with the Gree Protocol, and it lines up. It looks like the value should be closer to 20ms (19980**us**) so I've created a branch with it updated: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1508 / PR #1509

Can you please download it, try it out, & let us know how it goes?

k-korn, 2021-06-26:
> I've reviewed previous data we have, and your word on your data, and my personal experience with the Kelvinator protocol which overlaps a lot with the Gree Protocol, and it lines up. It looks like [...]

Yes, that works perfectly.
So the one actual change needed is indeed kGreeMsgSpace=19000 to 19980

crankyoldgit, 2021-06-26:
Thanks for confirming.

closed 2021-06-26 by crankyoldgit

crankyoldgit, 2021-07-06:
FYI, the committed & merged changes thus far have been included in the newly released version of the library. i.e. v2.7.19

PR #1509 (2021-06-26): [GREE] Update inter-message gap timing
* Based on ground-truth data & similarity to the Kelvinator protocol, I'm changing the `kGreeMsgSpace` to approx 20ms instead of 19ms.

Ref: 
  - https://github.com/crankyoldgit/IRremoteESP8266/blob/e70cd5b9a46b7a4e14f7f21e439f1b70568d50cf/test/ir_Gree_test.cpp#L609
  - https://github.com/crankyoldgit/IRremoteESP8266/blob/e70cd5b9a46b7a4e14f7f21e439f1b70568d50cf/src/ir_Kelvinator.cpp#L42-L43

* Update supported model info.

Fixed #1508
commit 371b82e2 (2021-06-26): [GREE] Update inter-message gap timing (#1509)

* Based on ground-truth data & similarity to the Kelvinator protocol, I'm changing the `kGreeMsgSpace` to approx 20ms instead of 19ms.

Ref: 
  - https://github.com/crankyoldgit/IRremoteESP8266/blob/e70cd5b9a46b7a4e14f7f21e439f1b70568d50cf/test/ir_Gree_test.cpp#L609
  - https://github.com/crankyoldgit/IRremoteESP8266/blob/e70cd5b9a46b7a4e14f7f21e439f1b70568d50cf/src/ir_Kelvinator.cpp#L42-L43

* Update supported model info.

Fixed #1508