crankyoldgit/IRremoteESP8266 #1513: Support for LG AKB74955603 remote

czagatti, 2021-07-02:
Hi @crankyoldgit 

I'm following the steps on the Wiki to share with you the details to support this remote - although the current LG2 works to turn it on and off and change modes, the fan speeds and vertical swing sets are not OK.

I really hope I'm taking it right, but I apologize in advance if something is not as I'm not a developer and I'm doing my best to help.

Photos of the (original) remote and manual, just for reference:

[image]
[image]

Created a Google Sheet with the details from IRrecvDumpV2 available at https://docs.google.com/spreadsheets/d/1zF0FI2ENvbLdk4zaWBY9ZYVM3MB_4oxro9wCM7ETX4Y/edit?usp=sharing with  comments enabled.

Thank you for your support!

NiKiZe, 2021-07-02:
Could you write the model number of both AC and remote (we want to have it in text, and. It parse from pictures, but the pictures are great as verification)

czagatti, 2021-07-02:
> Could you write the model number of both AC and remote (we want to have it in text, and. It parse from pictures, but the pictures are great as verification)

Sure, that was on the sheet but I forgot to mention here:

AC External unit: A4UW30GFA2
AC Internal unit: AMNW09GSJA0
Remote: AKB74955603

Thank you!

crankyoldgit, 2021-07-03:
Okay looking at the spreadsheet, some of this is easy. e.g. Light sends the same code, so it's a toggle. That's an easy-ish add.

As this is an already supported protocol, just a different model the procedure will be slightly different.
Can you please include what the text description is from IRrecvDumpV2 or V3 and highlight (pref with a colour) when it is _wrong_ etc.?

If we are lucky, we can find something that can tell us the difference between the existing `AKB75215403` LG2 model and your new one, so we can automatically detect the difference etc.

> the fan speeds and vertical swing sets are not OK.

Are they the only settings that don't work correctly and are not supported? e.g. Light button isn't currently supported.

czagatti, 2021-07-03:
Thank you all, I wasn't expecting to see these replies so fast.

@crankyoldgit I created a new tab on the sheet (named "Fan and Swing errors") and highlighted in red the errors comparing the description from IRrecvDumpV2 with the actual aircon behavior.

It's clear the main issue with this remote is with the vertical swing commands, fan only has 1 issue.

I really appreciate your support!

crankyoldgit, 2021-07-04:
Turns out _fan speed_ had multiple issues. I think I've got it .. not sure.

Please download and test out branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/AKB74955603 Let me know how it goes please.

It has required a new `LG`/`LG2` model of `AKB74955603`. You may need to set that to get any transmission of generated codes to work as you expect.

Now, on to things like `light`, it seems it is a specialised "Off" message. Not sure. We will need to do more testing, data collection, & analysis to be sure.

The `swing` settings on the other hand are a weird kettle of fish. **You** are going to have to do some data analysis to get support for them.

If you look here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/7006b9f39cada52936ea3643c1ed1866697cf243/src/ir_LG.h#L34-L46
You'll see our existing mapping of message bits to functions etc. i.e. `data` code left to right: Signature (8 bits), then Power (2 bits), then UNKNOWN (3 bits), then Mode (3 bits), then Temperature (4 bits), then Fan speed (4 bits), then lastly the Checksum (4 bits).

It seems the swing overlaps with existing setting bits & uses some unknown ones too.
e.g.
Fan speed is controlled by the second last nibble/octet/digit of [0x880A3**4**1](https://docs.google.com/spreadsheets/d/1zF0FI2ENvbLdk4zaWBY9ZYVM3MB_4oxro9wCM7ETX4Y/edit#gid=797409634&range=C6). i.e. `4` which is the max fan speed.

However, if you look at this [range of swing values](https://docs.google.com/spreadsheets/d/1zF0FI2ENvbLdk4zaWBY9ZYVM3MB_4oxro9wCM7ETX4Y/edit#gid=797409634&range=C9:C10) you'll see that value also changes. So I think it has dual purpose on your model/protocol.
e.g. 0x88130**4**8 & 0x88130**5**9

I think the answer lies with the 3rd & 4th nibble/octet/digit. i.e. 0x88**13**048 & 0x88**13**059 i.e. `0x13` or `0b00010011`
Of those digits (2 nibbles; 8 bits), we expect the first two bits (0b**xx**010011) is the power. (`0b00` is power on, `0b11` is power off), the next three bits `0b00xxx000` we have no clue about their use. **_You need to work this out, what they do, what they affect, how it changes the use of the fan speed bits, etc etc._** The last 3 bits `0b00000xxx` are the operating mode of the A/C. In those data values, that equates to `0b011` or `3` which is `kLgAcAuto`.

So, of that `0x13` value, those mystery bits are `0b010`. You need to explain **any** result that is not `0b000` for those bits, and how anything else that changes is linked to them.
e.g. Is it that `0b010` in those three bits, plus a `fan` of `4` means it's the swing `middle` position? e.g. [0x88**13**0**4**8](https://docs.google.com/spreadsheets/d/1zF0FI2ENvbLdk4zaWBY9ZYVM3MB_4oxro9wCM7ETX4Y/edit#gid=797409634&range=C9)

When you got that worked out on your spreadsheet/documented etc. Let me know.

czagatti, 2021-07-06:
It's been great pushing my limits trying to understand the more technical side of this, thanks for the opportunity.

I understood the approach with ir_LG.h and as you can see on the sheet (created a new tab named Fan Speed debug (branch code)), it is showing UNKNOWN when it reads "9" but the other modes are named OK - maybe need to declare the kLgAcFanLowAlt variable elsewhere? I didn't get that well.

For the swing, I must confess I tried to understand but I'll really need to dig deeper on that - I'm a fast learner and I'll do my best next weekend.

crankyoldgit, 2021-07-06:
> It's been great pushing my limits trying to understand the more technical side of this, thanks for the opportunity.

Anytime. ;-)

> 
> I understood the approach with ir_LG.h and as you can see on the sheet (created a new tab named Fan Speed debug (branch code)), it is showing UNKNOWN when it reads "9" but the other modes are named OK - maybe need to declare the kLgAcFanLowAlt variable elsewhere? I didn't get that well.

Nope, that's my fault. I didn't handle the output well. Fixed in a new commit to that branch. Please download/build/test with the latest in that existing branch.

> For the swing, I must confess I tried to understand but I'll really need to dig deeper on that - I'm a fast learner and I'll do my best next weekend.

Cool. Take your time. Yeah, it's not simple or easy, but it is doable. It comes slightly easier for me as I've done so many of them, but they all take _considerable_ time, data, & testing to reverse-engineer.  I've only got so much spare time & energy. ;-) Hence: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Can_you_reverse_engineer_my_AirCon_protocol_for_me

czagatti, 2021-07-07:
> Nope, that's my fault. I didn't handle the output well. Fixed in a new commit to that branch. Please download/build/test with the latest in that existing branch.

It is working perfectly, tried with all fan speeds and the data for the 2 modes that you highlighted on the sheet is updated.
Next weekend I'll work on the swing debug!

czagatti, 2021-07-08:
@crankyoldgit 

> Nope, that's my fault. I didn't handle the output well. Fixed in a new commit to that branch. Please download/build/test with the latest in that existing branch.

Fan speed seems to be working perfectly, tried with all fan speeds and the data for the 2 modes that you highlighted on the sheet is updated. Tried only the "receive" mode with IRrecvDumpV2, but I'm wondering how'd I send the proper command with IRsend on Tasmota (if/when this is merged 🤪)...

One thing I couldn't find on the FAQ, how can I send you a beer/coffee? I really appreciate your work and would like to collaborate :-)

Now about the swing, I created a new tab (Swing Tests) and did a lot with the codes (uint64_t data and uint16_t rawData[59]) I got from IRrecvDumpV2. At first my conclusion was that the codes (uint64_t data) were fixed (you can see it doesn't matter the mode/temperature they're always the same, even on power on they're resolved the same). But if that was true, turning the A/C on with the swing at a specific point wouldn't work and the A/C would always turn on at a default position - I tried that with the actual A/C unit and "there's more between the heaven and land", as it turns at the position displayed on the remote. Maybe need to dig the raw data? Not sure how to do that, I tried to compare with the sheet examples but they show hex/binary codes that I can't get with IRrecvDumpV2. Would you give me some guidance on how to do it?

Thanks in advance!

crankyoldgit, 2021-07-09:
> One thing I couldn't find on the FAQ, how can I send you a beer/coffee? I really appreciate your work and would like to collaborate :-)

So far, I've only accepted _in person_ beer offers. Ask @sheppy99 🍺 
My intention is to never ask for or accept money etc for this project. It's _free_ Open Source. Just my way of paying back to the Open Source movement etc.

> Now about the swing, I created a new tab (Swing Tests) and did a lot with the codes (uint64_t data and uint16_t rawData[59]) I got from IRrecvDumpV2. At first my conclusion was that the codes [...]

Having _fixed_ messages for some settings is fairly common. Especially on _short_ bit length protocols. From you analysis, it seems that is the case here.

Regarding powering on etc.: It's likely the case the A/C unit remembers that last swing position, and it isn't sent again. If it was sent at the same time as two back-to-back messages, I'd expect the raw data to be longer than `59` entries.

A way to test that theory would be to set the position to the highest, then turn off the a/c. Then cover the remote (or take it to another room etc) and change the position to the lowest on the remote, Then uncover the remote, and turn the a/c back on etc.

If the a/c doesn't change position, then it should be that the a/c just remembers the last position it saw. (Most likely scenario)
If it does change to the lowest, then there is something in the signal or message we are not seeing/working out etc.

FYI, unless the raw data changes in length, it's _highly unlikely_ there is any hidden data in it that the `uint64_t data` isn't representing.

kzagatti, 2021-07-09:
@crankyoldgit 

> So far, I've only accepted _in person_ beer offers. Ask @sheppy99 🍺
> My intention is to never ask for or accept money etc for this project. It's _free_ Open Source. Just my way of paying back to the Open Source movement etc.

I understand and I really appreciate your work!
 
> Having _fixed_ messages for some settings is fairly common. Especially on _short_ bit length protocols. From you analysis, it seems that is the case here.

And you got it, again! Tested turning the A/C unit on, set a specific swing position, turned it off, covered the remote to turn it on + set another swing position + turned off and finally turned the A/C unit on. It went on with the first swing position, confirming this setting is fixed.

So now we have the fan speed, swing positions and AC light worked out. Do you want me to debug other remote options (Jet Mode, Timer...)?

crankyoldgit, 2021-07-09:
Speaking of light, can you please test the "light" button with other modes/settings, to see if it changes at all from the code in the sheet? If it doesn't, you don't need to record it. Only need to if it _changes_.

kzagatti, 2021-07-09:
No changes. Every time I press any button (mode, temperature, swing) the light comes back on and you have to press Light Off again.

crankyoldgit, 2021-07-10:
> No changes. Every time I press any button (mode, temperature, swing) the light comes back on and you have to press Light Off again.

Interesting. So, any "normal" message will reset "light" to `on`. Correct?

kzagatti, 2021-07-10:
> Interesting. So, any "normal" message will reset "light" to `on`. Correct?
> i.e. If we want it to be off, we need to send a "light" toggle message after every "normal" command. Correct?

Exactly! (quite annoying, but that's how this A/C behaves)

crankyoldgit, 2021-07-11:
Does the "light" toggle/button work when the power is off?. i.e. Can you change the light/led/display/whatever when the A/C is off, and it it actually changes the "light" on the A/C without it powering the A/C back up?

crankyoldgit, 2021-07-11:
And can you please test the latest update to that branch, it has experimental support for the light setting/toggle etc.

kzagatti, 2021-07-11:
> Does the "light" toggle/button work when the power is off?. i.e. Can you change the light/led/display/whatever when the A/C is off, and it it actually changes the "light" on the A/C without it powering the A/C back up?
> And of course, does the hex value change?

The A/C unit "beeps" when I press the light button with the power is off, but nothing actually happens (no light is turned on). And the hex is always the same as we have on the sheet, 0x88C00A6.

> And can you please test the latest update to that branch, it has experimental support for the light setting/toggle etc.

Added to the sheet (tab Light Tests) just to have it documented, but it's OK. Just a suggestion to the text, if it was only "Light Toggle" or "Light Off" that'd represent better because the name on the remote if "Light Off" and every other command you send turns the light on, so you basically press this button to turn the light off.

And just to confirm, this version did not bring anything related to the swing positions, right?

crankyoldgit, 2021-07-11:
> The A/C unit "beeps" when I press the light button with the power is off, but nothing actually happens (no light is turned on). And the hex is always the same as we have on the sheet, 0x88C00A6.

Noted. Thanks for that.

> Added to the sheet (tab Light Tests) just to have it documented, but it's OK. Just a suggestion to the text, if it was only "Light Toggle" or "Light Off" that'd represent better because the name on the remote is "Light Off" and every other command you send turns the light on, so you basically press this button to turn the light off.

Excellent. I was going with Toggle, as you spreadsheet data indicated that multiple consecutive presses changed the state each time. Is that not the case?

> And just to confirm, this version did not bring anything related to the swing positions, right?

Correct. But I'm almost finished ...

crankyoldgit, 2021-07-11:
Now ... can you please download and test the branch again? Hopefully Vertical Swing should work. 🤞

kzagatti, 2021-07-11:
> Excellent. I was going with Toggle, as you spreadsheet data indicated that multiple consecutive presses changed the state each time. Is that not the case?

Toggle is fine! Button name on the remote is "Light Off", but you press it to toggle on/off so Toggle is more accurate than the original remote 🤪

> Now ... can you please download and test the branch again? Hopefully Vertical Swing should work. 🤞

Confirmed, everything is working correctly with IRrecvDumpV2!

kzagatti, 2021-07-11:
And to test both reading and sending commands, I compiled this branch with Tasmota and I'm trying sending some combinations to make sure everything works.

I'm saving all info on the same sheet, new tab called Tasmota. It's working almost OK, but I'm not sure if these kind of test would be reported here or on the Tasmota github.

Just so you to know, until now I noticed only two errors, lines 35 and 49 on the sheet, the swing reads "middle" where it should be "upper middle" (value from IRrecvDumpV2) and the "Off" swing position that does not respond, but that's not a major issue since I can set any other position...

sheppy99, 2021-07-11:
> So far, I've only accepted _in person_ beer offers. Ask @sheppy99 🍺

That was a great afternoon - will have to do it again sometime if we can ever travel freely from Prison Ship New Zealand again

crankyoldgit, 2021-07-12:
> That was a great afternoon - will have to do it again sometime if we can ever travel freely from Prison Ship New Zealand again

_Ahem!_ If anything is a Prison Ship, it's Australia. We were _the_ Penal Colony after all. ;-)

crankyoldgit, 2021-07-12:
> Just so you to know, until now I noticed only two errors, lines 35 and 49 on the sheet, the swing reads "middle" where it should be "upper middle" (value from IRrecvDumpV2) and the "Off" swing position that does not respond, but that's not a major issue since I can set any other position...

@kzagatti 

Re: line 35 & "middle", that issue arises because the "Common A/C" api (aka. `IRac`) doesn't have an "Upper Middle" setting.
The native positions has one more than the _common_ model supports, thus two get squished together.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/dd301061d60f8ae92484bd8ca8e2e78713545486/src/IRsend.h#L69-L80

Re: "49" i.e. SwingV Off not stopping the swing via Tasmota. I'll take a look to see if it is something on my side. Thanks for the heads up.

Also, thanks for confirming things are working. That's good news!

crankyoldgit, 2021-07-12:
> Re: "49" i.e. SwingV Off not stopping the swing via Tasmota. I'll take a look to see if it is something on my side. Thanks for the heads up.

Hopefully I've fixed in the latest update to the branch. Please download & test again .. yadda yadda yadda.

Fingers crossed, that's it.

kzagatti, 2021-07-12:
> > Re: "49" i.e. SwingV Off not stopping the swing via Tasmota. I'll take a look to see if it is something on my side. Thanks for the heads up.
> 
> Hopefully I've fixed in the latest update to the branch. Please download & test again .. yadda yadda yadda.
> 
> Fingers crossed, that's it.

I was really hoping to give you the great news today, but swingv: off is still not recognized (the A/C beeps "ok" but it doesn't stop). Setting a specific position works OK - included these tests in the file too (tab Tasmota lines 51-54). And Tasmota reads the off position when I press on the remote... Don't understand why it doesn't send it correctly.

I want to help building the best for the community, but if you want to stop here I'm totally ok with what we have now.

Just to share with you, I'm using Home Assistant + Tasmota and the climate entity just recognizes 3 swing modes (vertical auto, horizontal auto or both, but not specific swing positions), so this is better than what HA can handle. To work with HA I'm using the Tasmota-IRHVAC integration and I changed it's code to send "lowest" instead of "off".

But if you want me to do any more testing, just let me know! I love this feeling of "almost there!"

crankyoldgit, 2021-07-13:
> I was really hoping to give you the great news today, but swingv: off is still not recognized (the A/C beeps "ok" but it doesn't stop). Setting a specific position works OK - included these tests in [...]

FYI, in Google Sheets, you can select a range of cells, then right click, which gives you contextual menu which can give you a link for those cells. e.g. https://docs.google.com/spreadsheets/d/1zF0FI2ENvbLdk4zaWBY9ZYVM3MB_4oxro9wCM7ETX4Y/edit#gid=348220307&range=A51:B54

As for why. I'm not sure either. Did you make sure you updated the library in tasmota and completely rebuilt?

I've simulated as best I can, with some unit tests, and as far as I can work out it should be working after the previous update.
It was broken before [this commit](https://github.com/crankyoldgit/IRremoteESP8266/commit/1e5ddf6dbfe84b858a811ba16e14b31c22547fb4): 
Tests:
https://github.com/crankyoldgit/IRremoteESP8266/blob/3f7af6bc9d6ed243da563b8551d165bda290ab24/test/IRac_test.cpp#L1074-L1131

> 
> I want to help building the best for the community, but if you want to stop here I'm totally ok with what we have now.

We may have to until we get more of an idea what is going on in Tasmota. Thanks for confirming _everything else_ is working.

> Just to share with you, I'm using Home Assistant + Tasmota and the climate entity just recognizes 3 swing modes (vertical auto, horizontal auto or both, but not specific swing positions), so this is [...]

Not sure what to ask for, other than the above.

FYI, you can use [**IRMQTTServer**](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) with HA's climate as well, as an alternative to Tasmota.

See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/dd301061d60f8ae92484bd8ca8e2e78713545486/examples/IRMQTTServer/IRMQTTServer.ino#L218-L267

I too, use HA, Tasmota, and IRMQTTServer. I just don't use Tasmota for IR stuff, I know some people have had issues with their implementation. Plus, my thing (IRMQTTServer) predated Tasmota's support for AirCons. ;-)

P.S. I'm going to merge that PR #1530 which will automatically close this Issue. Feel free to add to this issue, I'll still see it. But if you find anything new that's a _bug_, I'd prefer a new issue to keep things cleaner/separate.

closed 2021-07-13 by crankyoldgit

crankyoldgit, 2021-07-13:
FYI, you are probably best off downloading the `master` branch now, as it's merged.

kzagatti, 2021-07-13:
Hi @crankyoldgit 

I'm using gitpod to compile Tasmota and I thought this could be something related to cache, so I cloned the repo again (now from master) and checked before compiling:

```
gitpod /workspace/Tasmota/lib/lib_basic/IRremoteESP8266/.git/logs $ cat HEAD
0000000000000000000000000000000000000000 178940c404e9c2eaf6e2cb3f948df3512957d80c name <email@gmail.com> 1626193949 +0000    clone: from https://github.com/crankyoldgit/IRremoteESP8266
gitpod /workspace/Tasmota/lib/lib_basic/IRremoteESP8266/.git/logs $ 
```

Tested the swingv off and it still doesn't work, so I mounted the sending device in front of my IR receiver to get the reading (as Tasmota does not shows the exact command it is sending):

```
IRhvac {"Vendor":"LG2", "Model": 3, "Power": "On", "Mode": "fan_only", "FanSpeed": "min", "SwingV": "Auto", "Light": "On"}
```

This is what I get at IRrecvDumpV2:

```
Timestamp : 000907.684
Library   : v2.7.19

Protocol  : LG2
Code      : 0x8802608 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 2 (Fan), Temp: 21C, Fan: 0 (Quiet)
uint16_t rawData[59] = {3282, 9870,  550, 1546,  554, 504,  532, 506,  534, 506,  532, 1536,  554, 504,  532, 508,  532, 508,  536, 504,  532, 508,  534, 506,  534, 504,  532, 508,  532, 508,  532, 1540,  556, 500,  534, 506,  536, 1536,  552, 1536,  554, 504,  532, 506,  532, 508,  534, 504,  534, 506,  534, 1538,  552, 506,  532, 508,  532, 506,  532};  // LG2 8802608
uint32_t address = 0x88;
uint32_t command = 0x260;
uint64_t data = 0x8802608;

Timestamp : 000907.795
Library   : v2.7.19

Protocol  : LG2
Code      : 0x8813149 (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Swing(V): 20 (Swing)
uint16_t rawData[59] = {3264, 9838,  558, 1538,  552, 504,  532, 506,  532, 506,  532, 1538,  554, 502,  534, 506,  532, 508,  534, 506,  532, 506,  532, 508,  534, 1538,  554, 502,  532, 508,  534, 1538,  550, 1554,  536, 504,  532, 506,  532, 508,  536, 1534,  554, 506,  530, 1542,  552, 506,  530, 508,  534, 1538,  552, 506,  532, 508,  538, 1532,  556};  // LG2 8813149
uint32_t address = 0x88;
uint32_t command = 0x1314;
uint64_t data = 0x8813149;
```

2 commands are sent, different then what we get when sending the command with swingv off:

```
IRhvac {"Vendor":"LG2", "Model": 3, "Power": "On", "Mode": "fan_only", "FanSpeed": "min", "SwingV": "Off", "Light": "On"}
```

IRrecvDumpV2 reads:

```
Timestamp : 000986.351
Library   : v2.7.19

Protocol  : LG2
Code      : 0x8802608 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 2 (Fan), Temp: 21C, Fan: 0 (Quiet)
uint16_t rawData[59] = {3262, 9874,  546, 1546,  554, 504,  532, 506,  532, 506,  532, 1540,  554, 502,  532, 506,  534, 506,  532, 506,  534, 506,  534, 504,  536, 504,  532, 506,  532, 506,  534, 1536,  556, 502,  532, 508,  534, 1536,  552, 1538,  550, 506,  532, 506,  534, 506,  534, 506,  532, 508,  532, 1538,  552, 506,  532, 508,  532, 510,  530};  // LG2 8802608
uint32_t address = 0x88;
uint32_t command = 0x260;
uint64_t data = 0x8802608;
```

Just to make sure, I tested different combinations and when sending swingv off, it does not send the second command. To me, this sounds like that Tasmota is not sending it when it's "off"...

Thanks for merging the code into master, good to know it will soon be part "of the family". And in this spirit of contributing, I have another LG A/C unit (remote model AKB73757604) which has a wifi module I'm using to control it via HA, but I'd be happy to share and help those who doesn't have the wifi module. I'll open a new issue for it, ok?

And I'll surely check IRMQTTServer, thanks for sharing this!

crankyoldgit, 2021-07-14:
> To me, this sounds like that Tasmota is not sending it when it's "off"...

I think you are right. I _think_ I know what they are doing _"wrong"_ (i.e. not optimally). I _think_ they are not storing/saving the _previous_ state, and not passing that to the `IRac` class as well as the _desired_ state.

I'll try one more _hack_ to see if can work around their code. i.e. Always send the `swingv` setting following the first message. However, if the A/C beeps with just the `swingv` message, it may double beep **every** time Tasmota sends an A/C command.

So questions for you before I implement it. a) Does it beep with just the swingv msg? b) If it does, can you live with two beeps every time (via tasmota)?

crankyoldgit, 2021-07-14:
Oh, btw, if you can reproduce the same problem with `IRMQTTServer` then it probably is something wrong with the library. At least I know the code in `IRMQTTServer` pretty well. 😉

kzagatti, 2021-07-14:
> I'll try one more _hack_ to see if can work around their code. i.e. Always send the `swingv` setting following the first message. However, if the A/C beeps with just the `swingv` message, it may double beep **every** time Tasmota sends an A/C command.

Turning the A/C on via the climate card on HA triggers the following command on Tasmota:

 ```
19:06:10.051 MQT: stat/tasmota_6B9B6E/RESULT = {"IRHVAC":{"Vendor":"LG2","Model":3,"Mode":"fan_only","Power":"On","Celsius":"On","Temp":21,"FanSpeed":"Min","SwingV":"Lowest","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
 ```

And using IRrecDumpV2 I see that 3 signals are sent already (and I hear their beeps). Tried to get them decoded (even changed the tolerance to 30%):

```
Timestamp : 000434.918
Library   : v2.7.19

Tolerance : 30%
Protocol  : LG2
Code      : 0x8802608 (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 2 (Fan), Temp: 21C, Fan: 0 (Quiet)
uint16_t rawData[59] = {3276, 9870,  552, 1546,  552, 504,  532, 508,  532, 508,  532, 1536,  556, 504,  532, 506,  532, 508,  534, 506,  532, 506,  532, 508,  532, 506,  532, 506,  536, 504,  532, 1540,  550, 504,  532, 508,  532, 1540,  552, 1536,  552, 506,  532, 506,  532, 506,  534, 506,  532, 508,  530, 1540,  554, 504,  532, 508,  532, 508,  534};  // LG2 8802608
uint32_t address = 0x88;
uint32_t command = 0x260;
uint64_t data = 0x8802608;

Timestamp : 000435.027
Library   : v2.7.19

Tolerance : 30%
Protocol  : UNKNOWN
Code      : 0xB4B05D88 (32 Bits)
uint16_t rawData[63] = {244, 5044,  288, 3014,  3256, 9818,  578, 1540,  554, 504,  532, 508,  532, 508,  536, 1538,  550, 506,  532, 506,  532, 508,  532, 508,  534, 504,  530, 508,  532, 1538,  554, 504,  532, 508,  532, 1540,  556, 1534,  554, 502,  532, 506,  532, 506,  532, 508,  534, 506,  532, 1538,  556, 502,  532, 506,  530, 1542,  552, 504,  534, 506,  530, 508,  534};  // UNKNOWN B4B05D88

Timestamp : 000435.137
Library   : v2.7.19

Tolerance : 30%
Protocol  : UNKNOWN
Code      : 0x8F0208A1 (32 Bits)
uint16_t rawData[63] = {250, 1850,  406, 4972,  3252, 9858,  542, 1534,  552, 504,  532, 508,  532, 508,  530, 1540,  552, 504,  532, 506,  534, 504,  532, 1538,  552, 1540,  548, 506,  530, 508,  534, 504,  532, 508,  532, 508,  532, 508,  532, 506,  534, 506,  534, 504,  532, 508,  532, 1540,  550, 506,  532, 1538,  556, 502,  532, 508,  532, 1540,  554, 1534,  554, 504,  532};  // UNKNOWN 8F0208A1

Although the codes are not properly decoded (now I understand why you ask people to use only the original remotes), I think I understand what's happening from the A/C behavior:

- The first blast is clear (got decoded), turns the unit on with fan_only and fanspeed at 0;
- Second blast sends the lowest swingv position; and
- The last one turns the light off.

> So questions for you before I implement it. a) Does it beep with just the swingv msg? b) If it does, can you live with two beeps every time (via tasmota)?

I'm totally fine with it. It is actually doing exactly what I'd do with the remote, but in batch :-)

kzagatti, 2021-07-14:
> Oh, btw, if you can reproduce the same problem with `IRMQTTServer` then it probably is something wrong with the library. At least I know the code in `IRMQTTServer` pretty well. 😉

Sure, I just have to find a way to compile and get it deployed to this Tasmota device. Would you happen to have the bin file for it so I can deploy via OTA? Otherwise next weekend I can get it wired.

Thanks!

crankyoldgit, 2021-07-15:
> Sure, I just have to find a way to compile and get it deployed to this Tasmota device. Would you happen to have the bin file for it so I can deploy via OTA? Otherwise next weekend I can get it wired.

You can compile it, and just upload the `firmware.bin` file (or what ever you're IDE produces) via tasmota's OTA.
e.g.:  https://randomnerdtutorials.com/bin-binary-files-sketch-arduino-ide/

If you want to revert to Tasmota via OTA, you need to ensure you set a password during setup of the initial setup of IRMQTTServer. This is done at first boot via the setup menu. You can practice it on a spare ESP if you have one. ;-)

Relevant Warning:
https://github.com/crankyoldgit/IRremoteESP8266/blob/2d4659cd174ee5b555b2ef70e674e54053f282c5/examples/IRMQTTServer/IRMQTTServer.h#L107-L114
and instructions:
https://github.com/crankyoldgit/IRremoteESP8266/blob/2d4659cd174ee5b555b2ef70e674e54053f282c5/examples/IRMQTTServer/IRMQTTServer.ino#L22-L56

crankyoldgit, 2021-07-15:
> And using IRrecDumpV2 I see that 3 signals are sent (and I hear their beeps). Tried to get them decoded (even changed the tolerance to 30%):

Looking at the raw data, Tolerance isn't the main issue. Each subsequent message is being preceeded by some IR noise/garbage. The LG message _starts_ with a Mark signal of approx `3200`, and then a Space signal of approx `9800`.
The 2nd and 3rd messages have some extra entries before the start of the message, thus the decoder doesn't recognise it.
e.g.
Good:
uint16_t rawData[59] = **{3276, 9870,**  552, 1546,  552, 504, ...
Bad:
uint16_t rawData[63] = {_250, 1850,  406, 4972,_  **3252, 9858,** ...

You can get the library to try to brute force software filter that garbage by adjusting this line & adding a [`max_skip`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#aeaa5c07a8b46f8fbb982f996cc1f9f4b) parameter:
https://github.com/crankyoldgit/IRremoteESP8266/blob/2d4659cd174ee5b555b2ef70e674e54053f282c5/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L149

e.g.
```c++
 if (irrecv.decode(&results, NULL, 2)) {  // max_skip of 2 pairs. i.e. Skip up to 4 bad Raw entries before a message.
```

> ```
> Timestamp : 000434.918
> Library   : v2.7.19
>
> Tolerance : 30%
> Protocol  : LG2
> Code      : 0x8802608 (28 Bits)
> Mesg Desc.: Model: 2 (AKB75215403), Power: On, Mode: 2 (Fan), Temp: 21C, Fan: 0 [...]

Correct. There are three messages, batched.

And now for the **_HACK_** I alluded to earlier. If you change this line:
https://github.com/crankyoldgit/IRremoteESP8266/blob/2d4659cd174ee5b555b2ef70e674e54053f282c5/src/ir_LG.cpp#L228
to:
```c++
  _swingv_prev = 0;
```

It should force the library to send the swing message with the first message. That should cause Tasmota to _ALWAYS_ send the swing message.

Hope that helps. Let me know how it goes.

kzagatti, 2021-07-20:
Hi @crankyoldgit 
I'm out of home for a couple of weeks and I'll get back to this as soon as I get back home next week, ok?

crankyoldgit, 2021-07-20:
No worries.

kzagatti, 2021-07-29:
Hi @crankyoldgit 

IRMQTTServer is awesome, I've been playing/testing it before moving away from Tasmota and, for now, there is one issue (that I'd probably post on IRMQTTServer git):

- Flashed a NodeMCU with IRMQTT directly from the Arduino IDE, checked the flash size to make sure we still have room for OTA;
- It boots up ok, I'm able to configure wifi and etc... it comes to life!
- I go to the admin page and try to upload tasmota-minimal via OTA (to test the plan B).... Nothing happens

After hitting "update" it never comes back up. I see the default AP again but when I connect to it, I can't access 192.168.4.1.
The only way to get it back to life is flashing it again from the Arduino IDE (it comes back with the previous configs).

Turned the debug on and this is what I get from the serial monitor:

```
0000340.506: Update:
0000340.507: tasmota-minimal (1).bin.gz
0000347.663: Update Success:
0000347.664: 284410
0000347.664: Rebooting...
0000347.665: Attempting firmware update & reboot
0000348.669: Post firmware reboot.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0008d6e0
@cp:0
ld
```

Well, this the plan B, but I'm doing some other tests too. Would you like me to keep posting that here?

Thanks!

crankyoldgit, 2021-07-29:
> 0000340.507: tasmota-minimal (1).bin.gz

There is probably your problem. IRMQTTServer's OTA is expecting a `.bin` file, not a `.bin.gz` file.
Tasmota added the ability to handle compressed (`.gz`) binaries. IRMQTTServer doesn't have that. Thus you need to decompress it before OTA flashing it.

crankyoldgit, 2021-07-29:
> Edit: tried with both .bin and .bin.gz files, results are the same. And there is free memory (Free Sketch Space: 368k).

Hmm. Not sure. IRMQTTServer is using the standard/original OTA upload method. How have you built the tasmota binary? And what ESP8266 variant are you uploading to? e.g. How much flash? What memory format/partitioning/FS size did you chose?

kzagatti, 2021-07-29:
> Hmm. Not sure. IRMQTTServer is using the standard/original OTA upload method. How have you built the tasmota binary? And what ESP8266 variant are you uploading to? e.g. How much flash? What memory format/partitioning/FS size did you chose?

Just downloaded the files from http://ota.tasmota.com/tasmota/release/
Even tried with an older (smaller) version...

It's a standard NodeMCU v3, 1MB and I'm using flash size 1MB FS:64kb

kzagatti, 2021-07-29:
Ah, also tried with a larger file just to confirm the debug message and when it fails (due to space) it shows up an "update failed, not enough space" on the serial screen...

crankyoldgit, 2021-07-29:
> It's a standard NodeMCU v3, 1MB and I'm using flash size 1MB FS:64kb

Odd. I thought all NodeMCU boards were 4Mb of flash.

The standard builds of Tasmota are made for 1Mb devices. (Which in theory should just work). It may be that the flash needs to be fully erased.

So, what is the serial output of you loading (via IRMQTTServer) a Tasmota `.bin` file. (i.e. Not a .bin.gz file)?

crankyoldgit, 2021-07-29:
ie. serial of it loading it and the subsequent reboots etc.

kzagatti, 2021-07-29:
It was the same. But I tried again with the .bin file and got a not enough space message - so this is probably due to the size of the memory. I confirmed this NodeMCU (v3 lol1n) has only 1M but it usually works with Tasmota (I'm using tasmota-ir.bin so when I want to change it via OTA I have to upload the minimal version to free up the space).

```
0000192.159: Update:
0000192.162: tasmota-minimal-8.5.1.bin
ERROR[4]: Not Enough Space
ERROR[4]: Not Enough Space
ERROR[4]: Not Enough Space
ERROR[4]: Not Enough Space
ERROR[4]: Not Enough Space
ERROR[4]: Not Enough Space
0000202.176: Attempting firmware update & reboot
0000203.180: Post firmware reboot.
```

Well, I'm going all-in with IRMQTTServer and if I have to come back I'll just hook up the pins :-)
Thanks for your quick reply @crankyoldgit, I'll let you know about the other tests!

crankyoldgit, 2021-07-29:
Yeah, I'd say there isn't enough space on 1M to do the OTA. Tasmota jumps through a lot of hoops to try to still allow it on a 1M platform.

How big is the Tasmota `minimal.bin` file?

Options are:
* Compile a smaller version of IRMQTTServer (by disabling protocols etc), upload that, and it might give you space for the tasmota minimal.
  - This is probably a wise thing to do as your board is 1M, and if you want to OTA upgrade it later, you need to get it (IRMQTTServer's `.bin` file well below 512kb) See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9a473c94cbe199844cd3ae032f56575c08d45f84/src/IRremoteESP8266.h#L63-L105
* Upload/use a very small intermediate OTA program. e.g. https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino
Load this between using IRMQTTServer and Tasmota
* Move to a bigger ESP8266 Flash. ;-)
* Resort to serial uploading :-(

kzagatti, 2021-07-29:
> How big is the Tasmota `minimal.bin` file?

368kb. But don't worry, if I ever need to come back I'll flash it via serial.

As for the tests with IRMQTTServer, I set the Protocol for LG2 and Model 3 on the Air Conditioner Control page and I'm able to control the A/C properly, the only command that doesn't work is power on/off (but it works when I select it from the Mode input). With Home Assistant the setup was straightforward, as per the description on the .ino file. But when I turn the A/C on (fan_only, for example), I see that two commands are actually sent to the A/C:

```
Message 1 received on ir_server_quarto/ac/cmnd/mode at 11:59 PM:
fan_only
QoS: 0 - Retain: false
Message 0 received on ir_server_quarto/ac/cmnd/power at 11:59 PM:
ON
QoS: 0 - Retain: false
```

The first one, power, is exactly what's not working via Air Conditioning Control.... And the beep I hear from the A/C is as if the unit was being turned off - maybe something with the codes or model? Can't debug exactly what's being sent in this case...

Anyway, I just changed the power topic to something that will not trigger anything and so far... so good! Moving the other IR devices (TV, etc) to IRMQTTServer too :-)

If you want me to keep debugging, just let me know! But thanks for your hard work.

kzagatti, 2021-07-29:
Well, last for today - tried to send a MQTT Discovery message but from the log listener I'm getting the following message:

```
Message 0 received on ir_server_quarto/log at 12:16 AM:
MQTT climate discovery FAILED to send.
QoS: 0 - Retain: false
```

crankyoldgit, 2021-07-29:
I've just added a minimal OTA build option (via PlatformIO) at 380496 bytes (371k) vs a "full" version at 549032 bytes.
See branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/IRMQTTServer_minimal

That's as small as I can easily make it without going to a lot of effort.

crankyoldgit, 2021-07-29:
> Well, last for today - tried to send a MQTT Discovery message but from the log listener I'm getting the following message:
> 
> ```
> Message 0 received on ir_server_quarto/log at 12:16 AM:
> MQTT climate discovery FAILED to send.
> QoS: 0 - Retain: false
> ```

I honestly haven't used this feature in ages. I'll try to take a look to see what I can see.

crankyoldgit, 2021-07-29:
Oh and yeah, this is now kind of not the same original issue. Should probably create a new one(s) for these.

crankyoldgit, 2021-07-29:
> > How big is the Tasmota `minimal.bin` file?
> 
> 368kb. But don't worry, if I ever need to come back I'll flash it via serial.
> 
> As for the tests with IRMQTTServer, I set the Protocol for LG2 and Model 3 on the Air Conditioner Control page and I'm able to control the A/C properly, the only command that doesn't work is power on/off (but it works when I select it from the Mode input). With Home Assistant the setup was straightforward, as per the description on the .ino file. But when I turn the A/C on (fan_only, for example), I see that two commands are actually sent to the A/C:
> 
> ```
> Message 1 received on ir_server_quarto/ac/cmnd/mode at 11:59 PM:
> fan_only
> QoS: 0 - Retain: false
> Message 0 received on ir_server_quarto/ac/cmnd/power at 11:59 PM:
> ON
> QoS: 0 - Retain: false
> ```
> 
> The first one, power, is exactly what's not working via Air Conditioning Control.... And the beep I hear from the A/C is as if the unit was being turned off - maybe something with the codes or model? Can't debug exactly what's being sent in this case...
> 
> Anyway, I just changed the power topic to something that will not trigger anything and so far... so good! Moving the other IR devices (TV, etc) to IRMQTTServer too :-)
> 
> If you want me to keep debugging, just let me know! But thanks for your hard work.
> 
> EDIT: If it helps, this is the configuration I'm using:
> 
> ```
> climate:
>   - platform: mqtt
>     name: Ar Quarto MQTT
>     modes:
>       - "off"
>       - "cool"
>       - "heat"
>       - "fan_only"
>     fan_modes:
>       - "Auto"
>       - "Min"
>       - "Low"
>       - "Medium"
>       - "High"
>       - "Max"
>     swing_modes:
>       - "Off"
>       - "Auto"
>       - "Highest"
>       - "High"
>       - "Middle"
>       - "Low"
>       - "Lowest"
>     power_command_topic: "not_used/ac/cmnd/power"
>     mode_command_topic: "ir_server_quarto/ac/cmnd/mode"
>     mode_state_topic: "ir_server_quarto/ac/stat/mode"
>     temperature_command_topic: "ir_server_quarto/ac/cmnd/temp"
>     temperature_state_topic: "ir_server_quarto/ac/stat/temp"
>     fan_mode_command_topic: "ir_server_quarto/ac/cmnd/fanspeed"
>     fan_mode_state_topic: "ir_server_quarto/ac/stat/fanspeed"
>     swing_mode_command_topic: "ir_server_quarto/ac/cmnd/swingv"
>     swing_mode_state_topic: "ir_server_quarto/ac/stat/swingv"
>     min_temp: 18
>     max_temp: 30
>     temp_step: 1
>     retain: false
> ```

I'll also need to know the "state" of the MQTT topics too to debug this. e.g. Is the `model` topic set etc? and what the "web page" thinks things are set too.

kzagatti, 2021-07-29:
Moving to #1542

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

commit daf35d40 (2021-07-11): Support for SwingV & Light
* Add support for SwingV settings. (A large mess.)
* Add missing `IRac` support for light.
* Update/Add unit tests.

Fixes #1513
commit 1e5ddf6d (2021-07-12): Improve handling of SwingV in IRac.
* Cover an edge case where we are changing from a SwingV position back to the default position.
* Add Unit tests to confirm the fix(es) work.

Fixes #1513
PR #1530 (2021-07-13): [LG2] Add fan speed, Swing, & Light support for new `AKB74955603` model
* Create new `AKB74955603` LG/LG2 model.
* Add support for different fan speed values for the `AKB74955603` model.
* Tweak `IRLgAc::setRaw()` to accept an optional protocol parameter.
* Improve model detection code & unit tests.
* Improve fan speed text output handling.
* Add support for SwingV & Light settings.
  - Because a normal message turns the light on, we can treat it not as a toggle in most cases.
* Misc cleanup and code improvements.
* Unit tests coverage and improvement.
* Update supported devices info.

Fixes # [... cut]
commit 178940c4 (2021-07-13): [LG2] Add Fan speed, Swing, & Light support for new `AKB74955603` model (#1530)

* Create new `AKB74955603` LG/LG2 model.
* Add support for different fan speed values for the `AKB74955603` model.
* Tweak `IRLgAc::setRaw()` to accept an optional protocol parameter.
* Improve model detection code & unit tests.
* Improve fan speed text output handling.
* Add support for SwingV & Light settings.
  - Because a normal message turns the light on, we can treat it not as a toggle in most cases.
* Misc cleanup and code improvements.
* Unit tests coverage and improvement.
* Update supported devices info. [... cut]