crankyoldgit/IRremoteESP8266 #1514: Need support for Hitachi Inverter AC

shivaniumredkar1102, 2021-07-03:
looking library support to decode IR code for hitachi inverter AC model no - R410A and
Remote model name - Hitachi AC220

Attaching link here to get appropriate details of remote. please check.
https://www.flipkart.com/lipiworld-220-ac-remote-compatible-inverter-hitachi-controller/p/itm5ee346d81032a?pid=REMFMMQH3PWKFFXN&lid=LSTREMFMMQH3PWKFFXNCLWGOW&marketplace=FLIPKART&cmpid=content_remote-controller_11333386289_u_8965229628_gmc_pla&tgi=sem,1,G,11214002,u,,,472069863535,,,,c,,,,,,,&ef_id=CjwKCAjwlYCHBhAQEiwA4K21mwKxllx20aINNYFh8yQ42OTMlQHOX6d9sAiGRPOUEC2uKu77hTfdfhoCywAQAvD_BwE:G:s&s_kwcid=AL!739!3!472069863535!!!u!296287989246!&gclid=CjwKCAjwlYCHBhAQEiwA4K21mwKxllx20aINNYFh8yQ42OTMlQHOX6d9sAiGRPOUEC2uKu77hTfdfhoCywAQAvD_BwE

NiKiZe, 2021-07-03:
Then you should follow the documentation that is listed in both the issue template that you should have filled in and also in the FAQ.

Dumping a request like this without doing even minimal effort to try and understand what is involved, is to me unacceptable.

crankyoldgit, 2021-07-03:
Please read and follow the instructions in https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

crankyoldgit, 2021-07-10:
Ping.

shivaniumredkar1102, 2021-07-10:
sure, thank you for the active response. we started to capturing raw codes with the provided format. will update soon.

crankyoldgit, 2021-07-30:
Haven't heard anything for weeks. If we don't hear something soon, we will consider this stale and close the issue.

crankyoldgit, 2021-08-02:
Marking this issue stale/closed.

closed 2021-08-02 by crankyoldgit