crankyoldgit/IRremoteESP8266 #1528: TCL mute mode support

thanxx, 2021-07-08:
Hi, and first of all, thanks for supporting this project

My TCL has a `mute` button, but the library doesn't catch the change of the state.
In both cases (mute on / off) the state is the same
```
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x00, 0x00, 0x00, 0x00, 0x82, 0x05};
```

Can you please give me some clue on how to catch that command?

thanxx, 2021-07-08:
here's a full dump of mute(quiet) on / off taken via irrecvdumpv2

Mute ON
```
Timestamp : 000074.617
Library   : v2.7.18

Protocol  : UNKNOWN
Code      : 0x2757C6B2 (114 Bits)
uint16_t rawData[227] = {3040, 1632,  500, 1084,  502, 1084,  500, 318,  474, 344,  474, 344,  472, 1110,  474, 344,  474, 344,  472, 1110,  474, 1110,  474, 344,  472, 1112,  474, 344,  472, 346,  470, 1112,  472, 1112,  472, 346,  470, 1114,  470, 1114,  470, 348,  468, 348,  468, 1116,  470, 348,  468, 350,  464, 354,  424, 1158,  426, 392,  424, 394,  424, 394,  424, 392,  424, 392,  424, 394,  424, 392,  424, 392,  424, 394,  424, 392,  424, 392,  424, 394,  424, 392,  460, 358,  460, 358,  458, 358,  458, 358,  460, 358,  460, 358,  460, 1124,  460, 1124,  460, 358,  460, 358,  458, 358,  458, 360,  482, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  486, 332,  484, 334,  484, 332,  484, 334,  484, 332,  484, 334,  484, 332,  484, 332,  484, 332,  486, 332,  484, 332,  484, 334,  484, 334,  484, 332,  484, 334,  484, 334,  484, 332,  484, 332,  486, 332,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  482, 334,  482, 334,  482, 336,  482, 336,  482, 336,  482, 336,  482, 336,  482, 336,  482, 336,  482, 336,  480, 336,  480, 338,  480, 338,  480, 338,  480, 336,  480, 338,  480, 338,  480, 338,  480, 338,  478, 1104,  478, 340,  478, 1104,  480, 338,  478, 340,  478, 340,  476, 340,  476, 1106,  478};  // UNKNOWN 2757C6B2

Timestamp : 000074.789
Library   : v2.7.18

Protocol  : TCL112AC
Code      : 0x23CB260100640307000000008205 (112 Bits)
Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Econo: Off, Health: Off, Light: Off, Turbo: Off, Swing(H): Off, Swing(V): Off
uint16_t rawData[227] = {3016, 1656,  484, 1100,  484, 1102,  482, 336,  482, 336,  480, 336,  480, 1102,  482, 336,  480, 338,  480, 1104,  480, 1104,  480, 338,  480, 1104,  480, 338,  478, 338,  478, 1104,  478, 1106,  478, 340,  476, 1106,  476, 1110,  476, 342,  474, 342,  474, 1110,  476, 342,  474, 344,  472, 1112,  474, 344,  448, 370,  448, 370,  448, 370,  448, 370,  448, 370,  446, 370,  446, 372,  446, 372,  446, 370,  446, 372,  444, 372,  446, 372,  446, 372,  444, 372,  444, 374,  442, 374,  420, 1162,  422, 396,  420, 398,  420, 1164,  420, 1164,  422, 396,  420, 1164,  420, 1164,  420, 398,  418, 400,  398, 420,  402, 416,  398, 420,  398, 420,  398, 1186,  398, 1186,  398, 1186,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  396, 420,  396, 422,  396, 420,  
396, 1184,  400, 420,  396, 422,  396, 420,  396, 420,  396, 420,  398, 1184,  398, 1186,  398, 422,  396, 1186,  398, 420,  396, 422,  396, 420,  396, 422,  396, 422,  396};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x00, 0x00, 0x00, 0x00, 0x82, 0x05};
```

Mute OFF
```

Timestamp : 000080.515
Library   : v2.7.18

Protocol  : UNKNOWN
Code      : 0x662A72B1 (114 Bits)
uint16_t rawData[227] = {3016, 1658,  486, 1100,  484, 1100,  484, 334,  484, 334,  484, 332,  484, 1100,  486, 332,  484, 334,  484, 1098,  484, 1100,  486, 332,  484, 1100,  484, 334,  484, 334,  484, 1098,  484, 1100,  484, 334,  484, 1098,  486, 1100,  484, 334,  484, 334,  484, 1098,  486, 332,  484, 334,  484, 332,  484, 1098,  486, 334,  484, 332,  484, 334,  484, 332,  484, 334,  484, 332,  484, 334,  484, 334,  484, 332,  484, 334,  484, 334,  484, 332,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 334,  484, 1100,  484, 334,  484, 334,  484, 334,  482, 334,  482, 336,  482, 334,  484, 334,  482, 334,  484, 334,  482, 336,  482, 336,  482, 336,  482, 336,  480, 336,  480, 336,  480, 338,  480, 338,  480, 338,  480, 338,  478, 340,  478, 340,  476, 340,  476, 342,  476, 342,  474, 342,  474, 342,  448, 370,  448, 370,  446, 370,  446, 370,  446, 370,  446, 372,  444, 372,  444, 374,  422, 396,  420, 396,  420, 398,  398, 420,  398, 420,  398, 420,  398, 420,  398, 418,  400, 418,  398, 418,  398, 418,  398, 420,  396, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  398, 1184,  430, 388,  430, 1154,  430, 388,  430, 388,  430, 1154,  430, 1154,  430, 388,  430};  // UNKNOWN 662A72B1

Timestamp : 000080.687
Library   : v2.7.18

Protocol  : TCL112AC
Code      : 0x23CB260100640307000000008205 (112 Bits)
Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Econo: Off, Health: Off, Light: Off, Turbo: Off, Swing(H): Off, Swing(V): Off
uint16_t rawData[227] = {3048, 1624,  430, 1154,  432, 1152,  432, 388,  430, 386,  430, 388,  430, 1152,  432, 386,  430, 388,  430, 1152,  432, 1152,  432, 388,  430, 1154,  430, 388,  430, 388,  430, 1152,  430, 1154,  432, 388,  430, 1154,  430, 1156,  428, 388,  430, 388,  430, 1152,  432, 388,  430, 388,  430, 1152,  432, 388,  454, 362,  454, 364,  454, 362,  430, 388,  430, 388,  430, 388,  430, 388,  430, 388,  430, 388,  430, 388,  430, 388,  430, 388,  430, 388,  452, 364,  430, 388,  430, 388,  430, 1154,  430, 388,  430, 388,  430, 1154,  430, 1154,  430, 388,  430, 1154,  430, 1154,  430, 388,  430, 388,  430,  [... cut ...] ,  428, 390,  428, 390,  428, 390,  452, 366,  450, 366,  452, 366,  452, 366,  452, 366,  452, 364,  452, 364,  452, 366,  452, 364,  454, 364,  452, 366,  452, 364,  452, 364,  454, 364,  452, 364,  
454, 1128,  454, 366,  452, 364,  454, 364,  454, 364,  452, 364,  454, 1128,  456, 1130,  454, 364,  452, 1130,  456, 364,  452, 364,  454, 364,  454, 364,  452, 364,  454};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x00, 0x00, 0x00, 0x00, 0x82, 0x05};

```

crankyoldgit, 2021-07-08:
Does Mute equate to turning off the Beep when it gets a new message via IR?

thanxx, 2021-07-09:
> Does Mute equate to turning off the Beep when it gets a new message via IR?

No, it turns down the fan speed to a minimum and makes the AC sound very quiet and suitable for sleeping. However, one can't reach that minimum via the fan speed command.

crankyoldgit, 2021-07-09:
Looking at the data you have presented, I can see there is an `UNKNOWN` message just prior to the `TCL112AC` message(s).

The reason the library is not detecting that message is because it is using an unexpected checksum algorithm for that protocol.

I'll see if I can work up some proper support this variant but in the meantime, you _should_ be able to emulate it by using:
```c++

const uint8_t mute_on[kTcl112AcStateLength] = {
    0x23, 0xCB, 0x26, 0x02, 0x00, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x85};
const uint8_t mute_off[kTcl112AcStateLength] = {
    0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x65};

irsend.sendTcl112Ac(mute_on);
delay(2000);
irsend.sendTcl112Ac(mute_off);
delay(2000);
```

My guess is that your A/C uses this special message/protocol to do something outside of it's "normal" operation.

thanxx, 2021-07-09:
Thank you, will try it soon

crankyoldgit, 2021-07-09:
> Thank you, will try it soon

Excellent. Looking forward to the results.

crankyoldgit, 2021-07-09:
BTW, can you please include the Brand/Model numbers of both the Remote **&** the A/C unit?

crankyoldgit, 2021-07-09:
> I'll see if I can work up some proper support this variant but in the meantime, you _should_ be able to emulate it by using:
>
> ```c++
> const uint8_t mute_on[kTcl112AcStateLength] = {
>     0x23, 0xCB, [...]

Just for future reference, I calculated those `state[]` values using [`tools/auto_analyse_raw_data.py`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py) on the `UNKNOWN`'s rawdata.
e.g.
```
$ ../tools/auto_analyse_raw_data.py -f /tmp/mute_off
Found 227 timing entries.
Potential Mark Candidates:
[3016, 486]
Potential Space Candidates:
[1658, 1184, 420]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3016
kHdrSpace  = 1658
kBitMark   = 459
kOneSpace  = 1113
kZeroSpace = 359

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+1100010011010011011001000100000000000000000000100000000000000000000000000000000000000000000000000000000010100110
  Bits: 112
  Hex:  0xC4D36440000200000000000000A6 (MSB first)
        0x650000000000000040000226CB23 (LSB first)
  Dec:  3992100450478508661657186103984294 (MSB first)
        2048523369968818712889041160162083 (LSB first)
  Bin:  0b1100010011010011011001000100000000000000000000100000000000000000000000000000000000000000000000000000000010100110 (MSB first)
        0b0110010100000000000000000000000000000000000000000000000000000000010000000000000000000010001001101100101100100011 (LSB first)

Total Nr. of suspected bits: 112
```
and converting `0x650000000000000040000226CB23 (LSB first)` to `{0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x65};` i.e. Last byte becomes first byte, 2nd last byte becomes 2nd byte, ...

crankyoldgit, 2021-07-09:
@thanxx Oh, can you please also  capture what happens when the unit is "off" and you press the "mute" button a few times.

crankyoldgit, 2021-07-09:
Also with the light/display on the A/C on & off, plus Econo mode too if it has it.

thanxx, 2021-07-09:
@crankyoldgit 

the model is `TCL TAC-09CHSD/XA31I`
the remote doesn't have any markings except `0802 KT` on the PCB.

link to picure:
https://tcl-ukraine.com.ua/image/cache/catalog/tcl/kondicioner/xa31/tac-09chsd-xa31i-inverter-r32-wi-fi-ready/52enew-500x500-1000x1000.jpg

the states you posted didn't work for me.
the AC doesn't even "beep" when I send those states, however, they are received by the IRRecvdumpv2 (but not decoded)
on , off
```

Timestamp : 000970.991
Library   : v2.7.18

Protocol  : UNKNOWN
Code      : 0x2757C6B2 (114 Bits)
uint16_t rawData[227] = {2936, 1740,  440, 1126,  438, 1120,  440, 394,  438, 396,  438, 394,  438, 1118,  468, 368,  438, 400,  438, 1118,  440, 1120,  438, 396,  438, 1118,  442, 394,  438, 396,  438, 1120,  438, 1120,  440, 394,  438, 1118,  440, 1118,  440, 394,  438, 396,  438, 1118,  440, 396,  438, 396,  438, 396,  438, 1120,  440, 394,  438, 396,  438, 396,  440, 394,  438, 394,  438, 396,  438, 396,  438, 396,  438, 396,  438, 394,  440, 394,  438, 400,  434, 396,  438, 396,  438, 394,  438, 396,  438, 396,  438, 396,  438, 396,  438, 1118,  440, 1118,  440, 396,  438, 396,  438, 396,  438, 394,  438, 396,  438, 394,  438, 396,  438, 394,  438, 396,  438, 394,  440, 396,  438, 396,  438, 396,  438, 396,  438, 394,  438, 396,  438, 396,  440, 396,  438, 396,  438, 396,  438, 396,  438, 396,  438, 394,  438, 396,  438, 396,  440, 394,  438, 394,  438, 394,  438, 396,  438, 394,  440, 396,  438, 394,  438, 396,  438, 396,  438, 396,  440, 394,  438, 394,  440, 394,  440, 394,  438, 394,  438, 394,  440, 394,  438, 396,  438, 396,  436, 396,  438, 396,  438, 396,  438, 394,  438, 396,  438, 394,  438, 394,  440, 394,  438, 394,  438, 396,  438, 396,  438, 394,  440, 396,  438, 1118,  440, 394,  438, 1118,  440, 394,  440, 394,  438, 394,  438, 394,  438, 1118,  440};  // UNKNOWN 2757C6B2

Timestamp : 000976.203
Library   : v2.7.18

Protocol  : UNKNOWN
Code      : 0x662A72B1 (114 Bits)
uint16_t rawData[227] = {2936, 1742,  438, 1128,  440, 1118,  442, 394,  464, 370,  438, 396,  438, 1116,  442, 394,  440, 400,  438, 1118,  440, 1118,  468, 368,  438, 1118,  440, 394,  438, 394,  440, 1116,  442, 1118,  440, 396,  438, 1120,  466, 1092,  440, 396,  438, 396,  438, 1120,  438, 396,  438, 396,  438, 396,  440, 1118,  440, 396,  438, 394,  438, 396,  438, 394,  438, 396,  438, 396,  438, 396,  438, 394,  438, 396,  438, 394,  438, 396,  438, 394,  438, 396,  438, 396,  438, 394,  436, 396,  438, 394,  438, 396,  438, 396,  438, 396,  438, 1118,  440, 396,  438, 396,  438, 396,  438, 396,  438, 396,  440, 394,  438, 396,  438, 396,  438, 396,  438, 396,  438, 394,  438, 394,  440, 396,  438, 396,  438, 394,  438, 396,  438, 396,  438, 394,  438, 396,  438, 396,  438, 394,  438, 394,  438, 394,  438, 394,  440, 396,  438, 394,  438, 396,  438, 396,  438, 394,  438, 394,  438, 394,  440, 396,  438, 394,  438, 394,  438, 396,  438, 396,  436, 396,  438, 394,  440, 394,  440, 394,  438, 396,  438, 394,  440, 394,  438, 396,  438, 396,  440, 394,  438, 394,  440, 394,  438, 396,  438, 396,  438, 394,  438, 394,  440, 394,  438, 394,  438, 396,  438, 394,  438, 396,  438, 1120,  440, 394,  440, 1118,  440, 396,  438, 394,  440, 1118,  440, 1118,  440, 394,  438};  // UNKNOWN 662A72B1
```

>Just for future reference, I calculated those state[] values using tools/auto_analyse_raw_data.py on the UNKNOWN's rawdata.
e.g.

oh, cool, thanks for sharing that

Ok, here are the dumps that you requested:

When the unit is off (and the remote is in the "off" mode) the remote doesn't emit anything.
The only buttons that work are Health and Clean
Did I get you right?

If the unit and remote are ON:
I posted the dumps of mute on/off in the 1st comment

**light / display off / on**
```
Timestamp : 002342.651
Library   : v2.7.18

Protocol  : UNKNOWN
Code      : 0x662A72B1 (114 Bits)
uint16_t rawData[227] = {3046, 1628,  458, 1126,  460, 1126,  458, 360,  458, 360,  458, 360,  458, 1126,  460, 358,  458, 360,  458, 1126,  460, 1126,  458, 360,  456, 1126,  458, 360,  458, 360,  458, 1126,  458, 1126,  458, 360,  458, 1126,  458, 1126,  458, 362,  456, 362,  456, 1126,  458, 362,  456, 362,  458, 360,  456, 1126,  458, 360,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 360,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 1126,  458, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  454, 362,  456, 362,  454, 362,  456, 362,  456, 362,  456, 362,  454, 362,  456, 362,  454, 362,  456, 362,  454, 364,  454, 362,  456, 362,  456, 362,  454, 362,  456, 362,  454, 364,  454, 364,  454, 362,  454, 364,  454, 362,  456, 362,  454, 364,  454, 364,  454, 362,  454, 364,  454, 364,  454, 364,  454, 362,  454, 364,  454, 362,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 362,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 362,  454, 364,  454, 364,  454, 364,  454, 364,  454, 1128,  456, 364,  454, 1128,  456, 364,  454, 364,  454, 1130,  454, 1130,  454, 364,  454};  // UNKNOWN 662A72B1

Timestamp : 002342.805
Library   : v2.7.18

Protocol  : TCL112AC
Code      : 0x23CB260100640308000000008004 (112 Bits)
Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 23C, Fan: 0 (Auto), Econo: Off, Health: Off, Light: Off, Turbo: Off, Swing(H): Off, Swing(V): Off
[... cut ...] 68,  450, 366,  452, 366,  450, 368,  450, 366,  452, 366,  450, 368,  450, 366,  450, 368,  450, 368,  450, 368,  450, 368,  450, 368,  450, 368,  450, 368,  450, 368,  450, 366,  450, 368,  450, 368,  450, 368,  450, 368,  450, 1134,  452, 368,  450, 368,  450, 1134,  450, 368,  450, 368,  450, 368,  448, 1134,  450, 1134,  450};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x03, 0x08, 0x00, 0x00, 0x00, 0x00, 0x80, 0xC4};

```

Please let me know if any captures needed

crankyoldgit, 2021-07-09:
> the states you posted didn't work for me.
> the AC doesn't even "beep" when I send those states, however, they are received by the IRRecvdumpv2 (but not decoded)
> on , off

Okay, then you may need to try this instead:
```c++

const uint8_t mute_on[kTcl112AcStateLength] = {
    0x23, 0xCB, 0x26, 0x02, 0x00, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x85};
const uint8_t mute_off[kTcl112AcStateLength] = {
    0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x65};

// Current settings.
const uint8_t state[kTcl112AcStateLength] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x00, 0x00, 0x00, 0x00, 0x82, 0x05};

// Turn mute/quiet on.
irsend.sendTcl112Ac(mute_on);
irsend.sendTcl112Ac(state);
delay(20000);  // Wait 20 seconds.
// Turn mute/quiet off.
irsend.sendTcl112Ac(mute_off);
irsend.sendTcl112Ac(state);
```

i.e. That special message may need to be followed up by a "normal" message. This should emulate your original captures more closely. There might be some gap timings to be finessed though.

crankyoldgit, 2021-07-09:
Also, can you please download and try out this Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1528

I think I've coded up some support for your Mute/Quiet mode.

Also, if you rebuild `IRrecvDumpV2` etc using that branch, it should now detect those `UNKNOWN` messages better.

Let me know how it goes please.

thanxx, 2021-07-09:
@crankyoldgit it started working after your latest suggestion!

Also, I rebuilt the library with new branch and now it's fully decoded 
the output for mute on / off is below

I Will try out `setQuiet()` soon as well
Again, huge thanks for the reply and your awesome work!

```
Timestamp : 000087.734
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB260200600000000000000085 (112 Bits)
Mesg Desc.: Type: 2, Quiet: On, Light: Off
uint16_t rawData[227] = {3046, 1626,  482, 1102,  482, 1102,  480, 338,  480, 338,  480, 338,  480, 1102,  482, 338,  480, 338,  480, 1104,  480, 1104,  480, 338,  480, 1104,  480, 338,  480, 338,  480, 1104,  480, 1104,  482, 338,  480, 1104,  482, 1104,  480, 338,  480, 338,  478, 1106,  478, 340,  478, 338,  478, 340,  478, 1106,  478, 340,  478, 340,  478, 340,  478, 340,  476, 340,  478, 342,  476, 340,  476, 342,  476, 342,  474, 342,  474, 342,  476, 342,  476, 342,  474, 342,  474, 344,  474, 344,  474, 344,  472, 344,  472, 344,  448, 1134,  474, 1110,  474, 344,  448, 370,  448, 370,  448, 370,  446, 370,  448, 370,  448, 370,  446, 370,  446, 372,  446, 370,  446, 370,  446, 372,  446, 372,  446, 372,  444, 374,  444, 372,  444, 372,  444, 374,  444, 374,  438, 380,  422, 396,  420, 396,  420, 396,  420, 398,  420, 398,  420, 398,  420, 398,  420, 398,  418, 398,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  400, 418,  398, 420,  398, 418,  398, 420,  398, 420,  400, 418,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 1186,  398, 420,  398, 1186,  398, 420,  398, 418,  398, 420,  398, 418,  398, 1186,  398};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x02, 0x00, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x85};

Timestamp : 000087.916
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB2601002403080000000080C4 (112 Bits)
Mesg Desc.: Type: 1, Power: On, Mode: 3 (Cool), Temp: 23C, Fan: 0 (Auto), Econo: Off, Health: Off, Turbo: Off, Swing(H): Off, Swing(V): Off, Light: On
uint16_t rawData[227] = {3046, 1628,  448, 1136,  448, 1136,  448, 370,  446, 372,  446, 372,  446, 1138,  446, 372,  446, 372,  448, 1136,  446, 1140,  446, 370,  444, 1140,  444, 376,  442, 374,  420, 1162,  422, 1162,  422, 396,  420, 1164,  420, 1164,  422, 396,  420, 398,  420, 1164,  420, 398,  420, 398,  398, 1184,  400, 420,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  398, 1184,  398, 420,  398, 420,  398, 1184,  400, 418,  398, 420,  398, 1184,  400, 1184,  400, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 
420,  398, 420,  398, 420,  398, 1184,  400, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  396, 420,  398, 420,  396, 420,  396, 420,  398, 420,  396, 420,  396, 420,  398, 420,  396, 420,  396, 422,  396, 1184,  398, 422,  396, 420,  396, 1186,  398, 422,  396, 420,  396, 422,  396, 1186,  398, 1186,  398};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x03, 0x08, 0x00, 0x00, 0x00, 0x00, 0x80, 0xC4};

Timestamp : 000101.624
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB260200400000000000000065 (112 Bits)
Mesg Desc.: Type: 2, Quiet: Off, Light: Off
uint16_t rawData[227] = {3046, 1626,  478, 1106,  478, 1106,  478, 340,  478, 340,  476, 342,  474, 1108,  476, 342,  476, 342,  476, 1108,  476, 1108,  476, 342,  476, 1108,  476, 342,  474, 344,  474, 1108,  476, 1108,  476, 344,  474, 1110,  474, 1110,  476, 344,  472, 344,  448, 1136,  448, 370,  448, 370,  448, 370,  448, 1136,  450, 370,  446, 370,  448, 370,  446, 372,  446, 372,  446, 372,  446, 372,  446, 372,  446, 372,  446, 372,  446, 372,  444, 374,  444, 374,  444, 372,  444, 374,  442, 374,  420, 396,  422, 396,  420, 396,  420, 398,  420, 1164,  420, 398,  420, 398,  420, 398,  420, 398,  420, 398,  420, 398,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  400, 418,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 418,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  398, 1186,  398, 420,  398, 1184,  398, 420,  398, 420,  398, 1184,  400, 1186,  398, 420,  398};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x65};

Timestamp : 000101.808
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB2601002403080000000088CC (112 Bits)
Mesg Desc.: Type: 1, Power: On, Mode: 3 (Cool), Temp: 23C, Fan: 0 (Auto), Econo: Off, Health: Off, Turbo: Off, Swing(H): On, Swing(V): Off, Light: On
[... cut ...] ,  398, 420,  398, 420,  398, 420,  398, 420,  398, 420,  396, 420,  398, 420,  398, 420,  396, 420,  398, 420,  396, 420,  398, 420,  396, 420,  396, 422,  396, 420,  396, 420,  398, 420,  396, 420,  396, 422,  396, 420,  396, 1186,  398, 422,  396, 420,  396, 422,  396, 1186,  398, 420,  396, 422,  396, 1186,  398, 1186,  398, 422,  396, 420,  396, 1186,  426, 1158,  428};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x03, 0x08, 0x00, 0x00, 0x00, 0x00, 0x88, 0xCC};
```

crankyoldgit, 2021-07-10:
> I Will try out `setQuiet()` soon as well

Excellent. Looking forward to the results.

The data you just collected indicates that I may have the "Light" setting incorrect for the `Type: 2` messages, so I'll have to make changes there I think.

thanxx, 2021-07-10:
> The data you just collected indicates that I may have the "Light" setting incorrect for the `Type: 2` messages, so I'll have to make changes there I think.

yes, the light in Type2 is always off
switching the light on / off
```
Protocol  : TCL112AC
Code      : 0x23CB260200400000000000000065 (112 Bits)
Mesg Desc.: Type: 2, Quiet: Off, Light: Off
uint16_t rawData[227] = {3048, 1624,  430, 1154,  430, 1154,  432, 386,  430, 386,  432, 386,  430, 1154,  430, 386,  432, 386,  430, 1152,  432, 1152,  430, 388,  430, 1152,  432, 388,  430, 386,  430, 1152,  432, 1154,  430, 388,  430, 1152,  432, 1152,  432, 388,  430, 388,  430, 1152,  432, 386,  430, 388,  430, 388,  430, 1152,  432, 386,  430, 388,  430, 386,  430, 388,  430, 386,  430, 388,  430, 388,  430, 386,  430, 388,  430, 388,  430, 386,  430, 386,  430, 388,  430, 386,  430, 388,  430, 388,  430, 388,  430, 386,  430, 388,  430, 388,  430, 1152,  430, 388,  454, 362,  456, 362,  454, 364,  454, 362,  454, 362,  456, 362,  456, 362,  456, 362,  456, 362,  454, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 360,  456, 362,  456, 362,  456, 362,  456, 360,  456, 362,  456, 362,  456, 360,  456, 362,  456, 360,  456, 362,  456, 362,  456, 362,  456, 360,  456, 362,  456, 362,  454, 362,  456, 362,  456, 362,  456, 362,  456, 362,  456, 362,  454, 362,  456, 362,  456, 362,  456, 362,  456, 362,  454, 362,  454, 364,  430, 388,  430, 386,  430, 388,  430, 388,  428, 388,  430, 388,  430, 388,  430, 1154,  430, 388,  430, 1154,  430, 388,  430, 388,  428, 1154,  430, 1154,  430, 388,  428};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x65};

Timestamp : 000127.110
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB260100640307000000008003 (112 Bits)
Mesg Desc.: Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Econo: Off, Health: Off, Turbo: Off, Swing(H): Off, Swing(V): Off, Light: Off
uint16_t rawData[227] = {3044, 1628,  456, 1128,  456, 1126,  458, 362,  456, 360,  458, 360,  456, 1126,  458, 360,  456, 360,  458, 1126,  458, 1126,  458, 360,  458, 1126,  458, 360,  460, 358,  456, 1126,  458, 1126,  458, 360,  458, 1124,  458, 1126,  458, 362,  456, 360,  456, 1126,  458, 360,  456, 360,  458, 1126,  456, 362,  456, 360,  456, 362,  456, 362,  456, 360,  456, 360,  456, 362,  456, 362,  456, 362,  454, 362,  454, 362,  456, 362,  456, 362,  456, 362,  454, 362,  454, 362,  454, 362,  430, 1152,  430, 388,  430, 388,  428, 1154,  430, 1154,  430, 388,  428, 1154,  430, 1154,  430, 388,  428, 388,  428, 388,  430, 388,  428, 388,  428, 388,  428, 1154,  430, 1154,  430, 1154,  430, 390,  428, 388,  428, 390,  428, 390,  428, 388,  428, 390,  428, 390,  428, 390,  452, 366,  452, 366,  452, 364,  452, 364,  452, 364,  454, 364,  452, 364,  452, 364,  454, 364,  454, 364,  454, 364,  452, 364,  452, 364,  454, 364,  454, 364,  452, 364,  454, 364,  454, 364,  452, 364,  452, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  
454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 1128,  456, 1128,  454, 1128,  456, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454, 364,  454};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x00, 0x00, 0x00, 0x00, 0x80, 0x03};

Timestamp : 000154.934
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB260200400000000000000065 (112 Bits)
Mesg Desc.: Type: 2, Quiet: Off, Light: Off
uint16_t rawData[227] = {3044, 1628,  460, 1124,  460, 1124,  460, 358,  460, 358,  460, 358,  460, 1124,  460, 358,  458, 358,  460, 1124,  460, 1124,  460, 360,  458, 1124,  460, 358,  458, 360,  458, 1124,  460, 1124,  460, 358,  458, 1124,  460, 1124,  460, 360,  458, 358,  458, 1124,  460, 358,  460, 360,  458, 358,  458, 1124,  460, 360,  458, 358,  458, 360,  458, 358,  458, 358,  458, 360,  458, 358,  458, 358,  460, 358,  458, 360,  458, 360,  458, 360,  458, 358,  458, 360,  458, 360,  458, 360,  458, 360,  458, 358,  458, 360,  458, 360,  458, 1124,  458, 360,  458, 358,  458, 358,  458, 360,  458, 358,  458, 360,  458, 360,  458, 358,  458, 358,  458, 360,  458, 360,  458, 358,  460, 358,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 360,  458, 358,  458, 360,  456, 360,  456, 360,  458, 360,  456, 360,  458, 360,  456, 360,  456, 362,  456, 362,  456, 360,  456, 362,  456, 362,  456, 362,  456, 360,  456, 362,  456, 362,  456, 362,  454, 362,  454, 362,  456, 362,  454, 364,  454, 362,  454, 364,  454, 364,  454, 362,  454, 362,  454, 1128,  454, 364,  454, 1128,  456, 364,  454, 362,  454, 1128,  456, 1128,  456, 364,  454};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x65};

Timestamp : 000155.119
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB2601002403070000000080C3 (112 Bits)
Mesg Desc.: Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Econo: Off, Health: Off, Turbo: Off, Swing(H): Off, Swing(V): Off, Light: On
uint16_t rawData[227] = {3044, 1626,  460, 1124,  460, 1126,  458, 360,  458, 358,  458, 360,  458, 1126,  458, 360,  458, 362,  456, 1124,  460, 1124,  458, 360,  458, 1124,  460, 360,  458, 360,  458 [... cut ...] 2,  454, 362,  456, 362,  454, 362,  454, 362,  454, 364,  454, 364,  454, 364,  454, 362,  454, 362,  454, 364,  454, 364,  454, 362,  454, 362,  454, 362,  454, 364,  454, 364,  454, 362,  454, 364,  454, 364,  454, 364,  454, 362,  454, 362,  454, 364,  454, 364,  454, 1128,  456, 1128,  456, 1128,  456, 364,  454, 364,  454, 364,  454, 364,  454, 1128,  454, 1128,  456};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x03, 0x07, 0x00, 0x00, 0x00, 0x00, 0x80, 0xC3};
```

crankyoldgit, 2021-07-10:
I've removed that _Light_ setting from the `quiet` messages in the latest update to that branch. Please download & retest it if you can.

thanxx, 2021-07-10:
> I've removed that _Light_ setting from the `quiet` messages in the latest update to that branch. Please download & retest it if you can.

confirm, it's removed now
```
Mesg Desc.: Type: 2, Quiet: On
```

crankyoldgit, 2021-07-10:
Now we just need you to test transmission

thanxx, 2021-07-10:
not sure I am doing it right
I need to turn the AC on with HSwing on for 14 seconds to get the flaps left. Then turn off the HSwing and turn on Mute finally.
```
  ac.on();
  ac.setTemp(24);
  ac.setLight(false);
  ac.setSwingHorizontal(true);
  ac.send();
  delay(14000);
  ac.setSwingHorizontal(false);
  ac.send();
  delay(5000);
  ac.setQuiet(true);
  ac.send();
```

Result: no type2 message
```
Timestamp : 000129.798
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB2601006403074000000008CB (112 Bits)
Mesg Desc.: Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Econo: Off, Health: Off, Turbo: Off, Swing(H): On, Swing(V): Off, Light: Off
uint16_t rawData[227] = {2896, 1768,  496, 1068,  494, 1064,  494, 340,  492, 342,  492, 342,  492, 1066,  494, 342,  464, 370,  466, 1090,  466, 1094,  464, 368,  492, 1066,  494, 342,  492, 342,  492, 1064,  494, 1066,  466, 370,  464, 1094,  464, 1094,  464, 370,  464, 370,  462, 1094,  492, 342,  490, 346,  462, 1094,  464, 370,  462, 370,  462, 372,  462, 372,  462, 372,  460, 372,  460, 372,  464, 370,  464, 370,  464, 372,  462, 370,  464, 370,  464, 370,  460, 372,  460, 374,  460, 372,  460, 374,  460, 1098,  466, 370,  462, 370,  462, 1094,  464, 1094,  464, 372,  458, 1098,  460, 1098,  414, 420,  412, 420,  412, 420,  412, 420,  414, 420,  460, 374,  412, 1144,  414, 1144,  414, 1142,  416, 420,  412, 420,  412, 420,  414, 420,  412, 420,  458, 376,  414, 422,  412, 420,  412, 420,  412, 422,  412, 420,  412, 1142,  416, 422,  414, 420,  414, 420,  414, 420,  414, 420,  414, 
420,  414, 420,  414, 420,  414, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 420,  414, 420,  414, 422,  414, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 422,  412, 422,  414, 420, 
 414, 420,  414, 420,  414, 1142,  416, 420,  414, 422,  412, 420,  414, 422,  412, 1142,  416, 1142,  416, 420,  412, 1144,  414, 422,  412, 422,  414, 1144,  414, 1148,  414};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x40, 0x00, 0x00, 0x00, 0x08, 0xCB};

Timestamp : 000144.015
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB2601006403074000000000C3 (112 Bits)
Mesg Desc.: Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Econo: Off, Health: Off, Turbo: Off, Swing(H): Off, Swing(V): Off, Light: Off
uint16_t rawData[227] = {2914, 1754,  492, 1070,  492, 1068,  490, 342,  490, 342,  494, 340,  494, 1062,  496, 340,  494, 342,  464, 1092,  466, 1092,  466, 368,  464, 1090,  466, 370,  464, 368,  494, 1064,  494, 1068,  464, 368,  464, 1092,  466, 1092,  466, 370,  464, 370,  464, 1092,  464, 370,  492, 344,  490, 1066,  466, 370,  462, 370,  462, 370,  462, 370,  462, 370,  464, 370,  462, 372,  490, 344,  490, 344,  490, 344,  488, 346,  486, 346,  464, 370,  464, 370,  464, 370,  460, 372,  460, 372,  460, 1096,  462, 374,  460, 374,  462, 1094,  464, 1094,  464, 372,  460, 1096,  462, 1098,  460, 376,  458, 376,  458, 376,  458, 374,  454, 378,  456, 378,  414, 1142,  414, 1144,  414, 1144,  414, 420,  412, 420,  412, 420,  412, 420,  414, 420,  460, 374,  460, 374,  460, 374,  414, 420,  412, 420,  412, 422,  412, 1144,  414, 420,  414, 420,  414, 420,  414, 420,  414, 420,  414, 
420,  414, 420,  414, 420,  414, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 422,  414, 420,  412, 420,  412, 420,  412, 420,  414, 420,  412, 420,  412, 420,  412, 422,  414, 420, 
 414, 420,  414, 420,  414, 420,  414, 420,  414, 420,  414, 420,  414, 420,  412, 1142,  416, 1142,  414, 420,  412, 420,  412, 420,  414, 420,  414, 1142,  416, 1146,  416};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x40, 0x00, 0x00, 0x00, 0x00, 0xC3};

Timestamp : 000149.232
Library   : v2.7.19

Protocol  : TCL112AC
Code      : 0x23CB2601006403074000000000C3 (112 Bits)
Mesg Desc.: Type: 1, Power: On, Mode: 3 (Cool), Temp: 24C, Fan: 0 (Auto), Econo: Off, Health: Off, Turbo: Off, Swing(H): Off, Swing(V): Off, Light: Off
uint16_t rawData[227] = {2890, 1780,  466, 1098,  466, 1092,  466, 370,  464, 370,  460, 374,  460, 1096,  462, 374,  460, 374,  458, 1098,  464, 1094,  466, 374,  460, 1096,  464, 372,  462, 372,  462, 1094,  462, 1098,  460, 374,  458, 1098,  460, 1098,  460, 374,  462, 372,  462, 1096,  464, 372,  462, 372,  458, 1098,  416, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 420,  412, 422,  460, 374,  414, 420,  414, 420,  412, 420,  412, 420,  412, 420,  412, 420,  414, 420,  458, 376,  414, 420,  414, 1142,  416, 422,  412, 420,  414, 1144,  416, 1142,  416, 422,  412, 1144,  414, 1142,  416, 420,  414, 420,  414, 420,  414, 420,  414, 420,  414, 420,  414, 1142,  416, 1142,  414, 1144,  414, 420,  412, 420,  414, 420,  412, 420,  414, 420,  414, 420,  414, 420,  414, 420,  414, 420,  412, 420,  412, 420,  412, 1144,  414, 422,  414, 422,  414, 420,  414, 420,  414, 420,  414, 
420,  414, 420,  414, 420,  414, 420,  412, 422,  412, 420,  412, 420,  412, 422,  412, 420,  412, 420,  414, 420,  414, 422,  412, 420,  412, 422,  412, 422,  412, 422,  412, 420,  412, 420,  412, 420,  412, 422,  414, 420, 
 414, 420,  414, 420,  414, 420,  414, 422,  412, 420,  412, 422,  412, 422,  412, 1144,  414, 1142,  416, 420,  414, 422,  414, 420,  414, 420,  414, 1142,  416, 1146,  416};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x40, 0x00, 0x00, 0x00, 0x00, 0xC3};
```

[... cut ...]
394,  440, 394,  440, 392,  440, 394,  440, 392,  440, 392,  464, 370,  440, 394,  440, 394,  440, 394,  440, 392,  442, 392,  440, 394,  440, 394,  440, 394,  440, 394,  440, 394,  440, 394,  466, 368,  440, 394,  440, 392, 
 440, 394,  440, 394,  440, 394,  440, 394,  466, 368,  440, 392,  440, 394,  440, 1116,  444, 1116,  442, 394,  440, 394,  440, 394,  440, 394,  440, 1116,  442, 1116,  442};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x64, 0x03, 0x07, 0x40, 0x00, 0x00, 0x00, 0x00, 0xC3};

```

thanxx, 2021-07-10:
a clarification regarding my last comment:
the detection is working fine with native remote
it fails with ac.send() on latest code

crankyoldgit, 2021-07-13:
Hmm. I rebased this branch with the `master` branch prior to the last commit etc. It maybe that the newly added `TEKNOPOINT` protocol overlaps with or is the same as this protocol. I'll have to take a look-see.

crankyoldgit, 2021-07-13:
Looking at this some more, it looks like `TEKNOPOINT`, `MITSUBISHI112`, & `TCL112AC` are all the same protocol, just with subtle timing differences in some cases, and with possible errors introduced with/by crappy IR demodulators. To work with those crappy devices, we bumped up the tolerances to get it to match, hence false posatives. :-(
See #1486

crankyoldgit, 2021-07-13:
> not sure I am doing it right
> I need to turn the AC on with HSwing on for 14 seconds to get the flaps left. Then turn off the HSwing and turn on Mute finally.
>
> ```
>   ac.on();
>   ac.setTemp(24); [...]

Your code snippet should work. In fact, I'm pretty darn sure it does. See a [unit test](https://github.com/crankyoldgit/IRremoteESP8266/commit/6ada9303c7346d7290dc094ca0ef54e4ccbf7680) I just wrote to verify it for confirmation.

I think the _issue_ may be elsewhere in your code. i.e. Are you sure `quiet` is not `true` before your above code executes? If it is, it won't send the `Type: 2, Quiet: On` message you are expecting. i.e. It only sends a `quiet` message when it thinks it needs too.
e.g. If that code snippet was looping, nothing is setting `quiet` back to `false`.
Or to explain it with code ...
```c++
  ac.setQuiet(true);
  ac.send();  // Might send a "Quiet: on" if it was off before hand, **and** will send a "normal" settings msg.
  delay(5000);
  ac.setQuiet(true);
  ac.send();  // This will only ever send a single "normal" settings msg, as the _quiet_ setting hasn't changed since it was last sent.
```

Try:
```c++
  ac.setQuiet(false);  // Or ac.stateReset();
  ac.on();
  ac.setTemp(24);
  ac.setLight(false);
  ac.setSwingHorizontal(true);
  ac.send();
  delay(14000);
  ac.setSwingHorizontal(false);
  ac.send();
  delay(5000);
  ac.setQuiet(true);
  ac.send();
```

thanxx, 2021-07-13:
@crankyoldgit thanks a lot for explanations, will investigate and will get back here with results

thanxx, 2021-07-15:
@crankyoldgit 
can confirm that all works
your latest suggestion did the trick!
thank you once again

crankyoldgit, 2021-07-15:
Thanks for confirming.

closed 2021-07-15 by crankyoldgit

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

stepanselyuk, 2021-12-05:
I hope I found this earlier because I did not find Daewoo (DSB-F0934ELH-V) among supported models in this repository and did all analysis and code in python on RPi and then I was really interested in how to mute the buzzer. I tried to search over the remote type (which I disassembled) `GYKQ-52E` and saw something about TCL a/c and finally went back to this repo and checked TCL, and the protocol seemed the same :-)

crankyoldgit, 2021-12-06:
> Daewoo DSB-F0934ELH-V / GYKQ-52E

Thanks for the info. It will be added to the Supported Devices list.

stepanselyuk, 2021-12-23:
I did not find a way to mute my AC by the way, and the remote does not contain the mute button as on TCL models/remotes.

thanxx, 2021-12-23:
the Mute here should be considered a Quiet mode when AC works on minimum
power

PR #1529 (2021-07-15): TCL112AC: Add support for quiet/mute setting.
* Changes required to handle quiet setting, which is sent in a separate special message.
* Modify checksum alg. for these special messages.
* Add `IRac` support for quiet for TCL112
* Add & update unit tests.
* Update supported models
* General code style cleanups.

Fixes #1528
commit dbc8e86c (2021-07-15): TCL112AC: Add support for quiet/mute setting. (#1529)

* Changes required to handle quiet setting, which is sent in a separate special message.
* Modify checksum alg. for these special messages.
* Add `IRac` support for quiet for TCL112
* Add & update unit tests.
* Update supported models
* General code style cleanups.

Fixes #1528