crankyoldgit/IRremoteESP8266 #1531: Support for LG AKB73757604 remote (new model with vertical and horizontal swing)

kzagatti, 2021-07-13:
Hi @crankyoldgit 

Helping you grow the support for LG remotes: 

External A/C: A4UW30GFA2
Internal A/C: AMNW24GTPA1
Remote model: AKB73757604

Photos of the original remote:

[image] [image]

I have the wifi module for this unit so most of the time I'm using either HA or the LG ThinQ app to manage it. But to help others, I created the sheet with the codes I collected:

https://docs.google.com/spreadsheets/d/17C_Ay7OjsYNSAxxj8uXbh0Vi2jrqyrncwzIyUOGSuNo/edit?usp=sharing

Most of the normal operations are recognized correctly considering the development for AKB75215403 (Model: 2) and AKB74955603 (Model: 3), but this A/C unit has horizontal swing and you can also set the Vent Angle for the vertical swing positions.

And as we did for Model: 3, I tested adjusting the swing to a specific position, "hide" the remote and change the position and turn it off/on - the A/C unit "remembers" its last position and does not assume what's on the remote - the codes are fixed and there is a logic you can notice on the sheet.

So I think this might be simple to implement, but let me know if you need any other tests and I'll be happy to help!

crankyoldgit, 2021-07-14:
I'll try to take a longer look later. I really should address some other folks issues for a bit. ;-)

> the codes are fixed and there is a logic you can notice on the sheet.

If _"there is a logic you can notice"_, especially in the bit patterns. Please spell it out.

kzagatti, 2021-07-14:
Here's a photo of the A/C unit:

[image]

It has 4 vanes (4 independent vertical swings) and you can set their position using the remote.

> If "there is a logic you can notice", especially in the bit patterns. Please spell it out.

Sure, they're fixed codes, just as in Model 3. With Model 3, IRrecvDumpV2 identifies the swing positions from 4 (lowest) to 9 (highest) along with Auto (20) and Off (21). This "Model 4" implements some new codes, 22 and 23 for horizontal swing (auto and off respectively) and 33 to 62 for each vane / position - these are all on the sheet.

I should say that I have this A/C unit for a year but I've never changed the vane angle position either on the remote or on the app - I'm barely using the auto vertical (which is already supported). It's great to work on this and discover "new" features.

crankyoldgit, 2021-08-03:
Hey @kzagatti can you please try out branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/AKB73757604 hopefully it adds the support you've pointed out.

kzagatti, 2021-08-03:
@crankyoldgit 

Almost there! Tested and horizontal swing and vanes 0 and 2 are OK, but vanes 1 and 3 are showing up as UNKNOWN. Probably a little adjust...

Vane 1 results: https://docs.google.com/spreadsheets/d/17C_Ay7OjsYNSAxxj8uXbh0Vi2jrqyrncwzIyUOGSuNo/edit#gid=0&range=A47:G52
Vane 3 results: https://docs.google.com/spreadsheets/d/17C_Ay7OjsYNSAxxj8uXbh0Vi2jrqyrncwzIyUOGSuNo/edit#gid=0&range=A59:G64

Tks

crankyoldgit, 2021-08-03:
Ah yep, I know what I did wrong. Will fix it soon. Thanks for the prompt & detailed feedback. That sort of thing really helps.

crankyoldgit, 2021-08-04:
> Almost there! Tested and horizontal swing and vanes 0 and 2 are OK, but vanes 1 and 3 are showing up as UNKNOWN. Probably a little adjust...

Now fixed in the latest commit to that branch. Please verify and let me know if everything is all good.

kzagatti, 2021-08-04:
> Ah yep, I know what I did wrong. Will fix it soon. Thanks for the prompt & detailed feedback. That sort of thing really helps.

Happy to help!

> Now fixed in the latest commit to that branch. Please verify and let me know if everything is all good.

Tested everything, all commands again and I confirm, it's all good!

closed 2021-08-04 by crankyoldgit

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

PR #1545 (2021-08-04): [LG] Add support for `AKB73757604` model
* Support for Vertical Swing on multiple vanes.
* Support for Horizontal Swing.
* Model detection.
* Unit tests added & updated.

Fixes #1531
commit 1c9a75b8 (2021-08-04): [LG] Add support for `AKB73757604` model (#1545)

* Support for Vertical Swing on multiple vanes.
* Support for Horizontal Swing.
* Model detection.
* Unit tests added & updated.

Fixes #1531