crankyoldgit/IRremoteESP8266 #1533: Interfere with other libraries

TavakoliAli, 2021-07-15:
### Version/revision of the library used
2.7.19

### Describe the bug
Actually it is not a Bug. I compiled and tested IRrecvDumpV2 example on my hardware successfully but when merge it to my program don't work any way. However, I copied sketch of IRrecvDumpV2 to my sketch (no merge, now IRrecvDumpV2 and my program execute In parallel) ‌but don't work too.

I want to know there is any well known interfere with other libraries or core features of Arduino?
I am using following libraries :
	dancol90/ESP8266Ping@^1.0
	knolleary/PubSubClient@^2.8
	bblanchon/ArduinoJson@^6.18.0
And using following features (Internal libraries) :
    Ticker

Thanks

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes

TavakoliAli, 2021-07-15:
Well, I tested it myself, and as I guessed, this library interfered with the Ticker library. I think esp8266 has 1 Timer and both of libraries use timer and timer interrupt.
Now i am thinking this convert to Bug. Is there any way to solve this interference?

NiKiZe, 2021-07-15:
[Second item in the FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me)

Try to add one at a time to the basic example, once you figure out what conflicts, and get a good description of the symptoms, please let us know.

crankyoldgit, 2021-07-15:
The Ticker library and `IRrecv` class both use the sole timer on the ESP8266. (Note: The ESP32 has four.) Thus, you can't have `IRrecv` and `Ticker` _enabled_ at the exact same time. You'd have to juggle enabling/disabling each appropriately yourself to share the single hardware resource.
If your project can't accommodate that level of resource sharing, then these libraries and the ESP8266 are not for you. I strongly suggest you upgrade to the ESP32 which has multiple timers. This library supports them.

closed 2021-07-15 by crankyoldgit

crankyoldgit, 2021-07-16:
FYI, this is now in the FAQ: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#What_libraries_are_known_to_be_incompatible_with_this_library

Before anyone asks:
**Q**: Can you modify the library to use `Ticker` so it works?
**A**: It is technically possible, but this library controls the timer from inside an Interrupt handler. Thus all the code needs to be in IRAM (a _VERY_ limited resource). The ticker library is not designed to be run from inside an Interrupt, and would use up to much IRAM, hence the answer is **No**, it won't be implemented.