crankyoldgit/IRremoteESP8266 #1534: Interference of pins

TavakoliAli, 2021-07-17:
### Version/revision of the library used
2.7.19

### Board
ESP-07 (ESP8266)

### Describe the bug
I compiled and tested "IRrecvDumpV2" example successfully (only i changed kRecvPin to 4) but when i set GPIO5 as output(GPIO5 is connected to indicator LED), it don't work anymore.

#### To Reproduce
1. Open IRrecvDumpV2 example.
2. Add pinMode(5, OUTPUT) anywhere you want in root of "setup" or "loop" functions.

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
Nothing

### What brand/model IR demodulator are you using?
VS838

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Other useful information
1. I tested it with other pins and all was Ok.
2. I know that GPIO5 is default clock pin and GPIO4 is default data pin of I2C, but i didn't use  I2C any way.
3. I guess when GPIO5 is setting as output, I2C will enabled automatically. However i added "Wire" library and tried to redirect pins of I2C, but it didn't work too.

crankyoldgit, 2021-07-17:
I can't think of any reason why only adding `pinMode(5, OUTPUT);` would cause our example code to fail in any way. Do you have anything connected to that pin?

Are you still getting the startup text? e.g. `"IRrecvDump is now running and waiting for IR input on Pin %d"`

TavakoliAli, 2021-07-19:
> I can't think of any reason why only adding `pinMode(5, OUTPUT);` would cause our example code to fail in any way. Do you have anything connected to that pin?

As i mentioned in first comment, GPIO5 is connected to a normal led (which i called it "indicator led").

> Are you still getting the startup text? e.g. `"IRrecvDump is now running and waiting for IR input on Pin %d"`

Yes, all are Ok, except receiving of IR signal.
I think when the mode of GPIO5 is changed the function of GPIO4 and 5 are changed.

crankyoldgit, 2021-07-19:
> As i mentioned in first comment, GPIO5 is connected to a normal led (which i called it "indicator led").

Can I suggest you remove anything connected to that pin, to ensure there isn't a hardware/electrical issue.

> 
> > Are you still getting the startup text? e.g. `"IRrecvDump is now running and waiting for IR input on Pin %d"`
> 
> Yes, all are Ok, except receiving of IR signal.

Thanks for confirming that. 

> I think when the mode of GPIO5 is changed the function of GPIO4 and 5 are changed.

I'll try to duplicate this when I get time. Assuming all you're running is just the example code, with that extra line, I really can't see why that addition would stop capture.

Can you please document the circuit and hardware configuration you are using so we can try to reproduce this? And perhaps a photo.

As the pins are adjacent, have you checked for a short circuit or bridge between the two? If they are cross connected it could explain what you are seeing.

TavakoliAli, 2021-07-19:
> Can I suggest you remove anything connected to that pin, to ensure there isn't a hardware/electrical issue.

I remove LED, but no thing happen.
 
> Can you please document the circuit and hardware configuration you are using so we can try to reproduce this? And perhaps a photo.

[Schematic](https://drive.google.com/file/d/12lXtfbcldAumUguy7_uGeY_xKtEHWYRM/view?usp=sharing)

> As the pins are adjacent, have you checked for a short circuit or bridge between the two? If they are cross connected it could explain what you are seeing.

I made a PCB for my hardware. It is not a prototype. I ensure that there are not any hardware problem. I checked all track and all was Ok. However i test it on multiple PCB.

NiKiZe, 2021-07-19:
Maybe not relevant to your current issue, but I find it weird that,
According to the schematic you have a coil (L2) on the IR receiver pin? is that correct?

TavakoliAli, 2021-07-19:
> Maybe not relevant to your current issue, but I find it weird that,
> According to the schematic you have a coil (L2) on the IR receiver pin? is that correct?

Yes, actually it is a Ferrite bead.

crankyoldgit, 2021-07-21:
@TavakoliAli I've just tested this with a ESP8266 **NodeMCU** board, with an IR demodulator _directly_ connected to `GND`, `3V3`, & `D2` (`GPIO4`), _Nothing_ connected to `D1`/`GPIO5` and a git clone of the master branch (as of https://github.com/crankyoldgit/IRremoteESP8266/commit/dbc8e86c0c8dce0ee3379de6edd77c6dd31e04b6)
_Without_ `pinMode(5, OUTPUT);` it ran fine and as expected. i.e. Decoded a Samsung remote correctly.
_With_ `pinMode(5, OUTPUT);` in `setup()`, it ran fine and as expected. i.e. Decoded a Samsung remote correctly.

Here is the diff of the changes I made to the example code per your suggestions:
```
diff --git a/examples/IRrecvDumpV2/IRrecvDumpV2.ino b/examples/IRrecvDumpV2/IRrecvDumpV2.ino
index ae7e21da..d02bfca3 100644
--- a/examples/IRrecvDumpV2/IRrecvDumpV2.ino
+++ b/examples/IRrecvDumpV2/IRrecvDumpV2.ino
@@ -38,7 +38,7 @@
 // An IR detector/demodulator is connected to GPIO pin 14
 // e.g. D5 on a NodeMCU board.
 // Note: GPIO 16 won't work on the ESP8266 as it does not have interrupts.
-const uint16_t kRecvPin = 14;
+const uint16_t kRecvPin = 4;
 
 // The Serial connection baud rate.
 // i.e. Status message will be sent to the PC at this baud rate.
@@ -141,6 +141,7 @@ void setup() {
 #endif  // DECODE_HASH
   irrecv.setTolerance(kTolerancePercentage);  // Override the default tolerance.
   irrecv.enableIRIn();  // Start the receiver
+  pinMode(5, OUTPUT);
 }
 
 // The repeating section of the code
```

**If** that is the only change you made to the example code, then you have a problem with your circuit/hardware, or your build/testing environment.

As you've said it works with other GPIOs on the board, then your demodulator is probably fine. Perhaps replace the ESP8266, or try this on a breadboard and slowly build up your circuit till you find your issue.

crankyoldgit, 2021-07-21:
FWIW, nothing in the library should have changed from `v2.7.19` to the current `master` branch that would affect this.

crankyoldgit, 2021-07-30:
Ping @TavakoliAli
Any update?

crankyoldgit, 2021-08-01:
Closing due to inactivity. If you get back to us on this, we will consider re-opening it.

closed 2021-08-01 by crankyoldgit