crankyoldgit/IRremoteESP8266 #1535: "Humidify" mode for Daikin AC

francescor93, 2021-07-17:
I own a Daikin air conditioner model _FTXZ25NV1B_ with a remote control model _ARC477A1_, which is recognized in this library as _DAIKIN2_ protocol.
I wrote my own code to control my air conditioner using this library on a NodeMCU board, which receives commands via MQTT, and everything works well, except I don't know how to activate humidification mode, because in the code it seems to be unsupported.
My remote control is like the one in the picture: 
[image]
In it there is the command _HUMIDIFY_, so it is allowed by the air conditioner, but if I use the example sketch _IRrecvDumpV2_, the operating mode returned is always _Mode: 4 (Heat)_.

To test this I connected only one IR receiver to the NodeMCU board on which the IRrecvDumpV2 sketch was running and sent two commands from my remote control: the first for the activation of the Humidify mode, the second for the activation of the Heat mode:
```
23:12:46.701 -> Timestamp : 000198.068
23:12:46.701 -> Library   : v2.7.19
23:12:46.701 -> 
23:12:46.701 -> Protocol  : DAIKIN2
23:12:46.701 -> Code      : 0x11DA2700010E4050600C8004B0162400008BC3D911DA27000049C032A0000006600000C1906004 (312 Bits)
23:12:46.747 -> Mesg Desc.: Power: On, Mode: 4 (Heat), Temp: 96C, Fan: 10 (Auto), Swing(V): 3 (Upper Middle), Swing(H): 139 (UNKNOWN), Clock: 00:14, On Timer: Off, Off Timer: Off, Sleep Timer: Off, Beep: 1 (Quiet), Light: 1 (High), Mould: Off, Clean: On, Fresh: Off, Eye: Off, Eye Auto: Off, Quiet: Off, Powerful: Off, Purify: On, Econo: Off
23:12:46.792 -> uint16_t rawData[633] = {9960, 25128,  3448, 1756,  386, 1344,  408, 458,  408, 456,  408, 460,  406, 1324,  386, 482,  408, 460,  408, 456,  386, 482,  408, 1324,  408, 458,  406, 1324,  384, 1348,  408, 456,  386, 1348,  410, 1320,  384, 1346,  386, 1348,  406, 1324,  386, 482,  386, 482,  408, 1324,  408, 458,  408, 458,  386, 480,  408, 460,  386, 480,  408, 460,  408, 456,  408, 460,  386, 478,  386, 482,  408, 1322,  408, 460,  410, 458,  408, 456,  408, 460,  406, 460,  408, 458,  388, 480,  408, 458,  406, 1324,  408, 1326,  386, 1346,  408, 460,  384, 482,  408, 456,  408, 456,  408, 456,  386, 482,  408, 456,  408, 458,  408, 460,  408, 458,  386, 1346,  384, 480,  408, 458,  408, 458,  408, 458,  408, 460,  384, 1348,  386, 478,  386, 1348,  408, 456,  386, 480,  408, 458,  408, 460,  408, 458,  410, 456,  406, 1328,  408, 1324,  384, 482,  408, 458,  406, 462,  406, 1324,  386, 1348,  386, 482,  408, 456,  408, 460,  386, 482,  406, 458,  386, 482,  386, 480,  408, 458,  408, 458,  386, 482,  408, 456,  410, 1320,  386, 482,  406, 460,  408, 1324,  410, 456,  408, 458,  408, 460,  386, 482,  386, 480,  386, 482,  408, 458,  406, 458,  408, 460,  408, 1324,  406, 1326,  406, 458,  408, 1322,  408, 460,  430, 1300,  410, 1322,  408, 458,  408, 1320,  408, 458,  408, 460,  408, 456,  386, 482,  408, 458,  406, 1326,  386, 480,  408, 456,  386, 1346,  408, 456,  386, 478,  386, 480,  408, 460,  406, 460,  386, 480,  408, 458,  408, 456,  386, 478,  408, 458,  388, 480,  386, 480,  430, 436,  408, 460,  386, 480,  408, 460,  386, 480,  386, 480,  384, 1346,  386, 1348,  408, 458,  386, 1348,  408, 456,  410, 456,  386, 482,  406, 1322,  386, 1346,  408, 1324,  410, 456,  408, 458,  408, 460,  384, 482,  408, 1326,  408, 1322,  384, 1346,  408, 458,  386, 480,  408, 1324,  406, 1322,  408, 460,  408, 1322,  408, 1322,  386, 35098,  3426, 1778,  386, 1346,  408, 456,  408, 456,  408, 460,  408, 1322,  408, 458,  408, 460,  408, 458,  408, 460,  408, 1322,  386, 480,  408, 1324,  386, 1350,  384, 480,  406, 1324,  410, 1322,  408, 1324,  408, 1324,  408, 1324,  408, 458,  406, 458,  408, 1326,  408, 456,  408, 458,  386, 482,  408, 458,  408, 456,  386, 482,  406, 462,  384, 480,  408, 458,  384, 480,  410, 458,  386, 480,  408, 460,  408, 458,  406, 458,  408, 460,  408, 458,  386, 480,  384, 1346,  386, 478,  386, 480,  408, 1324,  386, 480,  384, 480,  386, 1346,  386, 478,  406, 458,  408, 456,  386, 482,  408, 456,  408, 458,  410, 456,  408, 1322,  408, 1324,  406, 462,  406, 1326,  386, 480,  384, 480,  408, 1324,  384, 1348,  406, 460,  408, 458,  408, 460,  408, 460,  408, 456,  410, 458,  408, 456,  386, 1344,  408, 458,  408, 1322,  408, 458,  408, 458,  406, 458,  388, 480,  408, 462,  406, 458,  408, 458,  408, 458,  386, 480,  408, 458,  406, 460,  386, 480,  408, 458,  408, 460,  386, 480,  408, 460,  408, 460,  408, 1326,  406, 1326,  408, 458,  406, 456,  408, 458,  408, 460,  408, 458,  406, 460,  408, 458,  386, 480,  406, 460,  408, 460,  386, 1348,  406, 1322,  408, 458,  406, 460,  408, 456,  410, 458,  386, 482,  408, 456,  408, 458,  386, 478,  408, 458,  408, 458,  386, 480,  408, 458,  410, 458,  408, 456,  406, 460,  386, 482,  408, 458,  406, 1324,  384, 480,  408, 456,  408, 458,  386, 482,  408, 458,  408, 1326,  384, 1344,  432, 434,  406, 460,  408, 458,  386, 482,  408, 1324,  386, 480,  408, 458,  386, 1346,  408, 458,  384, 480,  408, 458,  406, 460,  386, 482,  410, 1324,  408, 1322,  406, 462,  406, 460,  386, 480,  408, 1324,  406, 458,  408, 460,  384, 480,  408, 458,  408, 460,  386};  // DAIKIN2
23:12:47.111 -> uint8_t state[39] = {0x11, 0xDA, 0x27, 0x00, 0x01, 0x0E, 0x40, 0x50, 0x60, 0x0C, 0x80, 0x04, 0xB0, 0x16, 0x24, 0x00, 0x00, 0x8B, 0xC3, 0xD9, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x49, 0xC0, 0x32, 0xA0, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 0x90, 0x60, 0x04};
```
```
23:12:50.053 -> Timestamp : 000201.431
23:12:50.053 -> Library   : v2.7.19
23:12:50.053 -> 
23:12:50.053 -> Protocol  : DAIKIN2
23:12:50.100 -> Code      : 0x11DA2700010E4050600C8004B0162400008BC3D911DA270000493200A0000006600000C1906044 (312 Bits)
23:12:50.100 -> Mesg Desc.: Power: On, Mode: 4 (Heat), Temp: 25C, Fan: 10 (Auto), Swing(V): 3 (Upper Middle), Swing(H): 139 (UNKNOWN), Clock: 00:14, On Timer: Off, Off Timer: Off, Sleep Timer: Off, Beep: 1 (Quiet), Light: 1 (High), Mould: Off, Clean: On, Fresh: Off, Eye: Off, Eye Auto: Off, Quiet: Off, Powerful: Off, Purify: On, Econo: Off
23:12:50.146 -> uint16_t rawData[633] = {9956, 25132,  3448, 1756,  410, 1322,  408, 456,  410, 456,  408, 458,  406, 1326,  384, 480,  408, 458,  408, 458,  408, 458,  410, 1322,  410, 458,  410, 1322,  408, 1320,  388, 480,  406, 1324,  408, 1324,  388, 1344,  408, 1324,  408, 1324,  386, 480,  408, 458,  406, 1324,  408, 458,  406, 458,  406, 458,  408, 460,  408, 460,  384, 480,  408, 460,  408, 458,  406, 460,  406, 458,  408, 1322,  386, 478,  410, 454,  408, 456,  408, 458,  408, 460,  386, 480,  384, 482,  408, 458,  406, 1326,  386, 1344,  386, 1344,  410, 456,  384, 480,  406, 458,  408, 458,  408, 460,  410, 458,  386, 480,  406, 460,  408, 460,  408, 458,  384, 1346,  408, 456,  406, 458,  408, 458,  408, 458,  386, 482,  408, 1324,  408, 458,  406, 1326,  408, 456,  408, 456,  410, 458,  408, 460,  408, 456,  408, 456,  408, 1322,  406, 1324,  410, 458,  384, 480,  410, 456,  386, 1346,  408, 1324,  386, 482,  408, 456,  410, 458,  386, 482,  406, 458,  386, 482,  406, 460,  408, 456,  408, 458,  386, 482,  408, 460,  384, 1348,  408, 458,  386, 478,  386, 1344,  410, 454,  386, 478,  408, 458,  408, 458,  408, 460,  410, 456,  408, 458,  408, 460,  408, 458,  408, 1322,  408, 1324,  408, 460,  386, 1348,  408, 458,  406, 1328,  406, 1322,  408, 458,  408, 1322,  408, 460,  406, 460,  386, 480,  408, 460,  408, 458,  388, 1346,  386, 480,  386, 480,  384, 1346,  386, 478,  386, 478,  408, 458,  406, 460,  408, 460,  406, 462,  406, 460,  406, 458,  408, 458,  408, 460,  406, 458,  408, 456,  408, 460,  408, 458,  408, 458,  408, 458,  408, 460,  408, 458,  408, 1324,  408, 1324,  408, 458,  408, 1326,  386, 478,  408, 458,  408, 468,  386, 1346,  408, 1322,  386, 1346,  408, 456,  408, 456,  410, 456,  406, 460,  408, 1324,  408, 1322,  408, 1326,  406, 460,  408, 456,  408, 1322,  408, 1326,  384, 482,  408, 1322,  408, 1322,  408, 35082,  3448, 1756,  408, 1320,  408, 458,  408, 458,  386, 480,  406, 1324,  384, 480,  408, 458,  408, 456,  408, 458,  408, 1324,  406, 460,  408, 1324,  406, 1324,  408, 458,  406, 1326,  408, 1322,  406, 1324,  408, 1324,  408, 1324,  386, 480,  386, 480,  410, 1324,  410, 456,  408, 458,  408, 458,  408, 460,  386, 478,  408, 460,  406, 460,  408, 456,  386, 482,  384, 478,  410, 458,  384, 480,  408, 460,  408, 458,  408, 458,  406, 460,  408, 460,  406, 458,  386, 1346,  408, 456,  386, 478,  408, 1324,  408, 458,  406, 460,  408, 1322,  406, 458,  406, 458,  408, 1326,  406, 458,  408, 460,  408, 1324,  386, 1346,  [... cut ...]
23:12:50.464 -> uint8_t state[39] = {0x11, 0xDA, 0x27, 0x00, 0x01, 0x0E, 0x40, 0x50, 0x60, 0x0C, 0x80, 0x04, 0xB0, 0x16, 0x24, 0x00, 0x00, 0x8B, 0xC3, 0xD9, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x49, 0x32, 0x00, 0xA0, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 0x90, 0x60, 0x44};
```

So my request is: is it possible, in a future version of the library, to also add this mode for air conditioners that support it? I think it would be very useful and even more comprehensive.

Looking forward hoping it can be implemented, I remain available in case of need.
Thanks in advance

NiKiZe, 2021-07-17:
Thank you for a pretty well defined report, I think we still need some details tho.

You also have hum% can you go thru all the values for that?
Include mesg desc and state for each value.

And finally off and on command with humidity
And the same with heat.
When doing this please include which value for hum% that was shown in the display.

If you while doing this can try and figure out which bits are changing as described in the FAQ that would be even better, but just getting the data and I could try and take a look.

crankyoldgit, 2021-07-18:
Differences:
0x11DA2700010E4050600C8004B0162400008BC3D911DA27000049**C032**A0000006600000C19060**04** Humidify
0x11DA2700010E4050600C8004B0162400008BC3D911DA27000049**3200**A0000006600000C19060**44** Heat
Last byte is the checksum, so that means it can be ignored safely.

If I'm reading/counting it right, those two bytes are `state[26]` & `state[27]` which we currently have mapped as:
https://github.com/crankyoldgit/IRremoteESP8266/blob/dbc8e86c0c8dce0ee3379de6edd77c6dd31e04b6/src/ir_Daikin.h#L228-L232

Given the Humidity message is reporting with a temperature of `96C` (not normal) it seems we have the correct byte range.
From the code above, we had expected the temperature to by 7 bits long in `state[26]`, this is probably wrong, or something unusual is happening. As for `state[27]` we have no idea what bits in this byte do.

@francescor93 **_You_** are going to need to do the analysis to work out what bits do what with respect to Humidity etc.
Some tips for how you can do this are on this page: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data

On the plus side, most of this protocol is done already, so you only need to explain/document a very small section of it. ;-)

Good luck. Let us know when you think you have it sorted out.

francescor93, 2021-07-18:
Hi @NiKiZe and @crankyoldgit, and thanks for your answers.
I deepened the analysis on my AC using the same installation of the first post (therefore an IR receiver connected to a NodeMCU on which the _IRrecvDumpV2_ sketch is running), verifying the `state[]` changes as the operating modes or humidity values vary; I attach an .xlsx file with the complete results of my checks and some notes.

[Daikin Analysis.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/6836641/Daikin.Analysis.xlsx)

Basically I found two things:

- The value of `state[25]` for both _Heat_ and _Humidify_ modes is the same (0x49), so that's why _Humidify_ mode is recognized as _Heat_ and not, for example, as unknown or default; to correctly detect which mode it is, it might be useful to also refer to `state[27]` (see next point)
- The value of `state[27]` most likely represents the configured humidity setting: in fact this is the only field that changes when, in _Humidify_ mode, I switch through the various humidity values available (_Low_, _Std_, _Hi_, _Cont_, as shown on the remote control); furthermore, only for the _Dry_ and _Humidify_ modes (in which it is possible to set a humidity) this field is valued, while in the other operating modes it is always 0x00.

Again, all the details of my tests are available in the attached file. Please let me know if this information is sufficient or if you need to know more, I will help you as I can.
Thanks again.

NiKiZe, 2021-07-18:
Can you convert that file to a Google sheet or similar and make it publicly readable, it would be great for everyone to be able to read it without needing extra software and steps.

francescor93, 2021-07-18:
Sure, of course, and sorry for the inconvenience.
I uploaded the file here, please let me know if everything is ok now.
https://docs.google.com/spreadsheets/d/1kxHgFqiUB9ETXYEkszAIN5gE-t2ykvnPCnOV-sPUE0A/edit?usp=sharing

crankyoldgit, 2021-07-21:
What does "CONT" mean?

francescor93, 2021-07-21:
I answered you on Google Sheet, but I also answer here for convenience:
> "CONT" stands for Continuous.
> The values that can be shown on the remote control are: LOW, STD, HI, CONT. According to the operation manual:
> LOW: Humidifies or dehumidifies with low power
> STD: Humidifies or dehumidifies with moderate power
> HI: Humidifies or dehumidifies with high power
> CONT: Continues humidifying or dehumidifying

crankyoldgit, 2021-07-22:
Please download and try out branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Daikin_Humid

It should support setting & reporting the humidity etc.

Let me know how it goes please.

crankyoldgit, 2021-07-22:
FYI, you should probably make that spreadsheet view/comment only.

francescor93, 2021-07-22:
Hi @crankyoldgit ,
I tested with my remote by sending to an IR receiver with the _IRrecvDumpV2_ script and I can confirm that the _Humid_ value is correctly detected as set. Here are the results of my tests:
```
13:57:08.140 -> Timestamp : 012268.919
13:57:08.140 -> Library   : v2.7.19
13:57:08.140 -> 
13:57:08.140 -> Protocol  : DAIKIN2
13:57:08.140 -> Code      : 0x11DA270001454350600C8004B01624000081CE1411DA27000049C028B0000006600000C18060FA (312 Bits)
13:57:08.140 -> Mesg Desc.: Power: On, Mode: 4 (Heat), Temp: 32C, Fan: 11 (Quiet), Swing(V): 14 (Off), Swing(H): 129 (UNKNOWN), Clock: 13:57, On Timer: Off, Off Timer: Off, Sleep Timer: Off, Beep: 1 (Quiet), Light: 1 (High), Mould: Off, Clean: On, Fresh: Off, Eye: Off, Eye Auto: Off, Quiet: Off, Powerful: Off, Purify: Off, Econo: Off, Humid: 40%
13:57:08.187 -> uint16_t rawData[633] = {10004, 25090,  3496, 1732,  430, 1300,  430, 438,  430, 436,  430, 436,  430, 1304,  426, 436,  430, 436,  428, 436,  430, 464,  404, 1304,  426, 438,  428, 1302,  428, 1302,  430, 438,  430, 1300,  430, 1302,  430, 1308,  426, 1328,  404, 1304,  426, 438,  428, 462,  402, 1302,  428, 436,  428, 436,  430, 438,  430, 438,  428, 436,  430, 442,  428, 438,  428, 436,  428, 438,  430, 436,  428, 1304,  428, 462,  404, 458,  406, 436,  426, 438,  428, 438,  428, 436,  430, 464,  402, 1300,  430, 438,  430, 1304,  428, 438,  428, 438,  430, 436,  430, 1330,  402, 438,  428, 1302,  428, 1302,  428, 436,  430, 462,  404, 462,  402, 438,  430, 1300,  430, 440,  428, 436,  428, 438,  428, 438,  428, 466,  404, 1302,  430, 436,  430, 1302,  430, 436,  430, 436,  430, 438,  430, 438,  426, 436,  430, 438,  430, 1304,  430, 1300,  430, 440,  430, 436,  428, 438,  428, 1302,  430, 1302,  430, 436,  430, 436,  430, 438,  428, 436,  430, 436,  430, 442,  426, 436,  430, 434,  430, 440,  428, 436,  430, 438,  432, 1302,  430, 434,  432, 438,  430, 1298,  430, 436,  430, 436,  432, 434,  428, 438,  428, 440,  430, 436,  426, 438,  430, 440,  428, 434,  428, 1330,  404, 1302,  430, 466,  406, 1298,  430, 438,  432, 1300,  430, 1304,  428, 436,  426, 1300,  430, 440,  430, 436,  428, 436,  430, 438,  432, 434,  428, 1306,  428, 436,  430, 436,  430, 1302,  428, 436,  428, 436,  428, 438,  430, 434,  428, 436,  430, 436,  432, 436,  428, 438,  428, 464,  402, 436,  430, 440,  430, 432,  428, 438,  430, 438,  428, 436,  430, 442,  426, 438,  430, 436,  432, 1298,  430, 438,  430, 432,  428, 436,  430, 442,  426, 436,  430, 436,  432, 1302,  430, 438,  428, 1304,  428, 1300,  430, 1300,  430, 434,  430, 436,  432, 1300,  432, 1298,  430, 438,  430, 440,  426, 1308,  426, 436,  430, 1304,  428, 436,  430, 438,  432, 434,  428, 35038,  3496, 1734,  430, 1302,  428, 436,  430, 434,  432, 438,  428, 1300,  432, 438,  426, 436,  430, 434,  430, 466,  402, 1300,  430, 438,  428, 1304,  430, 1300,  428, 436,  430, 1300,  430, 1304,  428, 1300,  430, 1304,  428, 1330,  402, 436,  430, 436,  430, 1304,  430, 438,  428, 436,  428, 438,  430, 440,  426, 436,  430, 440,  428, 436,  430, 436,  430, 436,  430, 438,  430, 432,  430, 436,  432, 434,  428, 438,  432, 436,  428, 436,  430, 438,  430, 436,  432, 1304,  428, 436,  430, 436,  430, 1302,  430, 436,  430, 434,  430, 1302,  430, 436,  430, 438,  426, 438,  428, 464,  404, 436,  430, 442,  426, 436,  430, 1302,  430, 1302,  430, 464,  402, 436,  430, 440,  426, 1304,  428, 438,  430, 1300,  430, 438,  430, 436,  430, 436,  432, 436,  430, 464,  402, 436,  430, 1300,  430, 1302,  432, 436,  428, 1302,  430, 436,  428, 434,  430, 440,  426, 438,  430, 440,  428, 434,  430, 438,  430, 436,  430, 436,  430, 438,  428, 436,  430, 440,  426, 438,  430, 436,  430, 434,  430, 436,  430, 442,  424, 1304,  430, 1300,  430, 434,  430, 436,  430, 436,  430, 438,  430, 436,  428, 436,  430, 440,  428, 436,  430, 438,  430, 442,  424, 1304,  430, 1300,  428, 438,  448, 418,  430, 436,  450, 418,  428, 438,  448, 416,  428, 438,  450, 442,  424, 416,  450, 418,  448, 418,  450, 414,  450, 444,  424, 418,  530, 336,  450, 442,  424, 418,  450, 1284,  450, 414,  452, 416,  528, 338,  528, 340,  450, 440,  504, 1204,  526, 1206,  450, 442,  452, 390,  450, 440,  428, 418,  450, 414,  448, 442,  450, 390,  450, 1282,  450, 416,  474, 390,  476, 414,  452, 416,  450, 392,  476, 1256,  478, 1252,  478, 390,  476, 412,  454, 1256,  478, 412,  454, 1256,  476, 1254,  476, 1276,  482, 1226,  516, 1240,  510};  // DAIKIN2
13:57:08.517 -> uint8_t state[39] = {0x11, 0xDA, 0x27, 0x00, 0x01, 0x45, 0x43, 0x50, 0x60, 0x0C, 0x80, 0x04, 0xB0, 0x16, 0x24, 0x00, 0x00, 0x81, 0xCE, 0x14, 0x11, 0xDA, 0x27, 0x00, 0x00, 0x49, 0xC0, 0x28, 0xB0, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 0x80, 0x60, 0xFA};
13:57:08.564 -> 
13:57:08.564 -> 
13:57:10.692 -> Timestamp : 012271.462
13:57:10.692 -> Library   : v2.7.19
13:57:10.692 -> 
13:57:10.692 -> Protocol  : DAIKIN2
13:57:10.692 -> Code      : 0x11DA270001454350600C8004B01624000081CE1411DA27000049C02DB0000006600000C18060FF (312 Bits)
13:57:10.692 -> Mesg Desc.: Power: On, Mode: 4 (Heat), Temp: 32C, Fan: 11 (Quiet), Swing(V): 14 (Off), Swing(H): 129 (UNKNOWN), Clock: 13:57, On Timer: Off, Off Timer: Off, Sleep Timer: Off, Beep: 1 (Quiet), Light: 1 (High), Mould: Off, Clean: On, Fresh: Off, Eye: Off, Eye Auto: Off, Quiet: Off, Powerful: Off, Purify: Off, Econo: Off, Humid: 45%
[... cut ...]
13:57:16.408 -> 
13:57:16.408 -> 
```
I see that the _Mode_ value is always _4 (Heat)_ due to `state[25]` that does not change, so in my sketch I will identify the actual mode (heating only or even humidification) based also on the new _Humid_ value.
Tonight I also try to send some actual humidification commands to my AC (which I don't have here now) to verify that everything is working, but I don't think there will be any problems.
I will give you confirmation as soon as possible.
Thanks for now.

francescor93, 2021-07-22:
As anticipated, I have now also tested by sending commands to my AC with a custom sketch, and I can confirm that the _HUMIDIFY_ mode works correctly.
So I think everything is fine now with this modification.

crankyoldgit, 2021-07-22:
Awesome. Thanks for confirming.

closed 2021-07-22 by crankyoldgit

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

francescor93, 2021-08-31:
Hi @crankyoldgit, thank you, I have already downloaded it and I am using it with satisfaction in my projects. Thanks again for your availability and support for this development.

PR #1540 (2021-07-22): Daikin2: Add support for Humidity setting/operation.
* Allow setting known humidity percentages via `setHumidity()` & `getHumidity()`
* Report the setting correctly in `toString().
* Enforce it being available only in Heat & Dry Modes.
* Change temp to Max when it's in operation.
* Misc code style cleanups.
* Update unit tests, and add coverage for new features.
* Update supported models.

Fixes #1535
commit 9a473c94 (2021-07-22): Daikin2: Add support for Humidity setting/operation. (#1540)

* Allow setting known humidity percentages via `setHumidity()` & `getHumidity()`
* Report the setting correctly in `toString().
* Enforce it being available only in Heat & Dry Modes.
* Change temp to Max when it's in operation.
* Misc code style cleanups.
* Update unit tests, and add coverage for new features.
* Update supported models.

Fixes #1535