crankyoldgit/IRremoteESP8266 #1537: dump v2 and receive  data and saw in raw array

mrdsn3, 2021-07-20:
### Version/revision of the library used
I use last version 

### Describe the bug
I write program that receive  data  and save in raw data 
when we want send data that received data 
my problem is that data dosent work  I try IR smart repeater it working well 
then I  use irdump to dump data and use it for irsenddemo the code dosent work 
why irrepeater do it`s work well reading well sending well but irdump can`t show the right code for sending whith raw data 
thanks for any answers
#### To Reproduce
_What steps did you do, and what did or didn't actually happen? How can we reproduce the issue?_
 I write a new code for reading raw data and change to string 
I think somthinge wrong in  sorting raw data and serial printing int 
_e.g._
_1. Initialise the IRsend class._
_2. `IRsend.sendFoobar(0xdeadbeef);`_
_3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead._

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable in the report](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Expected behaviour
_What steps did you do and what should it have done? A clear and concise description of what you expected to happen._

_e.g._
_1. Initialise the IRsend class._
_2. `IRsend.sendFoobar(0xdeadbeef);`_
_3. Foobar branded BBQ turns on and cooks me some ribs._

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
_Include some serial text of the raw dumps of what the device saw._

_**Note**: Output from Tasmota is not acceptable. We can't easily use their raw format._

### What brand/model IR demodulator are you using?
_Brand/Model code_ or _None_

_If VS1838b: That is likely your problem. Please buy a better one & try again. Reporting a capture or decode issue when you use one of these is effectively wasting our & your time. See the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator)_

### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

mrdsn3, 2021-07-20:
this is main program
`#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <DNSServer.h> 
#include <Arduino.h>
#include <Hash.h> 
#include <Arduino.h>

#include "strtoraw.h"
#include <assert.h>
#include <IRrecv.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRtext.h>
#include <IRutils.h>
#include "constroctor.h"

// ==================== start of TUNEABLE PARAMETERS ====================
// An IR detector/demodulator is connected to GPIO pin 14
// e.g. D5 on a NodeMCU board.
// Note: GPIO 16 won't work on the ESP8266 as it does not have interrupts.
const uint16_t kRecvPin = 14;

// The Serial connection baud rate.
// i.e. Status message will be sent to the PC at this baud rate.
// Try to avoid slow speeds like 9600, as you will miss messages and
// cause other problems. 115200 (or faster) is recommended.
// NOTE: Make sure you set your Serial Monitor to the same speed.
const uint32_t kBaudRate = 115200;

// As this program is a special purpose capture/decoder, let us use a larger
// than normal buffer so we can handle Air Conditioner remote codes.
const uint16_t kCaptureBufferSize = 1024;

// kTimeout is the Nr. of milli-Seconds of no-more-data before we consider a
// message ended.
// This parameter is an interesting trade-off. The longer the timeout, the more
// complex a message it can capture. e.g. Some device protocols will send
// multiple message packets in quick succession, like Air Conditioner remotes.
// Air Coniditioner protocols often have a considerable gap (20-40+ms) between
// packets.
// The downside of a large timeout value is a lot of less complex protocols
// send multiple messages when the remote's button is held down. The gap between
// them is often also around 20+ms. This can result in the raw data be 2-3+
// times larger than needed as it has captured 2-3+ messages in a single
// capture. Setting a low timeout value can resolve this.
// So, choosing the best kTimeout value for your use particular case is
// quite nuanced. Good luck and happy hunting.
// NOTE: Don't exceed kMaxTimeoutMs. Typically 130ms.
#if DECODE_AC
// Some A/C units have gaps in their protocols of ~40ms. e.g. Kelvinator
// A value this large may swallow repeats of some protocols
 uint8_t kTimeout = 50;
//const uint8_t kTimeout = 90;
#else   // DECODE_AC
// Suits most messages, while not swallowing many repeats.
 uint8_t kTimeout = 15;
//uint8_t kTimeout = 95;

#endif  // DECODE_AC
// Alternatives:
//uint8_t kTimeout = 95;
// Suits messages with big gaps like XMP-1 & some aircon units, but can
// accidentally swallow repeated messages in the rawData[] output.
//
// const uint8_t kTimeout = kMaxTimeoutMs;
// This will set it to our currently allowed maximum.
// Values this high are problematic because it is roughly the typical boundary
[... 211 lines cut ...]
    //delay(1);
        pch = strtok (NULL, ",");
                          //sscanf(rawbuffer,"63,%d",&rps);
       

    rps = atoi(pch);

     rawarrayforsend[d-1]=rps;
     Serial.print(String(rps));Serial.print(",");
    d++;
        } 
        Serial.println("");
        d=1;
    pch=NULL;
      
       
        String strpassize=irswitch.size;
        const  char* rawbuffersize=strpassize.c_str();
        
         sscanf(rawbuffersize,"%d",&size); Serial.println(""); Serial.println("size:");
         Serial.print(String(size));
    irsend.sendRaw(rawarrayforsend,size, kFrequency);
      //String objforsendclint=
       server.send(200,"text/plain",rawbufferchar);
      }`

mrdsn3, 2021-07-20:
and this is "strtoraw.h"

#include <assert.h>
#include <IRrecv.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRtext.h>
#include <IRutils.h> 

String resultToSource(const decode_results * const results) {
  String output = "";
  // Reserve some space for the string to reduce heap fragmentation.
  output.reserve(1536);  // 1.5KB should cover most cases.
  // Start declaration
  output += uint64ToString(getCorrectedRawLength(results), 10);
  // array size
  output += F(",");  // Start declaration

  // Dump data
  for (uint16_t i = 1; i < results->rawlen; i++) {
    uint32_t usecs;
    for (usecs = results->rawbuf[i] * kRawTick; usecs > UINT16_MAX;
         usecs -= UINT16_MAX) {
      output += uint64ToString(UINT16_MAX);
      if (i % 2)
        output += F(",0,");
      else
        output += F(",0,");
    }
    output += uint64ToString(usecs, 10);
    if (i < results->rawlen - 1)
      output += ",";            // ',' not needed on the last one
  //  if (i % 2 == 0) output += ' ';  // Extra if it was even.
  }

  // End declaration
 //output += F(" ");

  // Comment
 // output += typeToString(results->decode_type, results->repeat);
  // Only display the value if the decode type doesn't have an A/C state.
  if (!hasACState(results->decode_type))
    //output += ',' + uint64ToString(results->value, 16);
//  output += F("\n");

  // Now dump "known" codes
  if (results->decode_type != UNKNOWN) {
    if (hasACState(results->decode_type)) {
#if DECODE_AC
      uint16_t nbytes = results->bits / 8;
      
      output += uint64ToString(nbytes);
      output += F(",");
      for (uint16_t i = 0; i < nbytes; i++) {
        output += F("0x");
        if (results->state[i] < 0x10) output += '0';
        output += uint64ToString(results->state[i], 16);
        if (i < nbytes - 1) output += kCommaSpaceStr;
      }
      output += F("\n");
#endif  // DECODE_AC
    } else {
      // Simple protocols
      // Some protocols have an address &/or command.
      // NOTE: It will ignore the atypical case when a message has been
      // decoded but the address & the command are both 0.
      if (results->address > 0 || results->command > 0) {
        
        output += uint64ToString(results->address, 16);
        output += uint64ToString(results->command, 16);
        
      }
      // Most protocols have data
       
      output += uint64ToString(results->value, 16);
       
    }
  }
  return output;
}

mrdsn3, 2021-07-20:
this is raw array that it show 

this is raw array [63]::,634,490,634,490,636,490,634,490,636,488,636,490,636,488,612,1634,662,1582,612,1634,634,1610,636,1608,612,1634,610,1636,660,1584,660,462,638,490,634,1610,636,490,634,1608,636,488,636,490,638,488,636,1610,634,1608,636,490,634,1608,636,490,610,1632,612,39602,9056,2164,636

and I think it should be like that
[63]::9056,2164,636,634,490,634,490,636,490,634,490,636,488,636,490,636,488,612,1634,662,1582,612,1634,634,1610,636,1608,612,1634,610,1636,660,1584,660,462,638,490,634,1610,636,490,634,1608,636,488,636,490,638,488,636,1610,634,1608,636,490,634,1608,636,490,610,1632,612,39602

mrdsn3, 2021-07-20:
I test what I think raight 
and it`s working well

mrdsn3, 2021-07-20:
I try again and again and again until it working well 
I think when IR rcv start decoding 3 part of array missed out 
and I don`t knew how can I fix it

mrdsn3, 2021-07-20:
When I press a button twice in a row, the second data obtained is completely correct.
I think something wrong in 
if (irrecv.decode(&results))
what you think? can you help me?

NiKiZe, 2021-07-21:
Not sure what you want with this issue? Do you want us to debug and fix your program/code?

crankyoldgit, 2021-07-21:
Please read:

- https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me
- https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#reporting-an-issue _(Point 4)_

You need to simplify your code to the minimum amount that produces the problem. You code is too complex for us to easily see what is going on and **if** it is the libraries code, or yours that has the issue.

**_TL;DR_**: I'm not going to spend my time trying to debug your _entire_ project. You shouldn't expect anyone to do it for you.

If `IRrecvDumpV2` etc is showing the data correctly, you can pretty much rest assured that the problem is with _your_ code.

FYI, if you look at:
```
[63]::,634,490,634,490,636,490,634,490,636,488,636,490,636,488,612,1634,662,1582,612,1634,634,1610,636,1608,612,1634,610,1636,660,1584,660,462,638,490,634,1610,636,490,634,1608,636,488,636,490,638,488,636,1610,634,1608,636,490,634,1608,636,490,610,1632,612,39602,9056,2164,636
```
You'll see it's data starts with `,634` i.e. the `,` indicates your cod is probably doing something wrong _at least_ there.

mrdsn3, 2021-07-21:
thanks 
when I use IRrecvDumpV2 first three parameter is show in the end of rawarray why ?and how can I fix it?

mrdsn3, 2021-07-21:
I knew if I  define temp raw array I can replace them to correct location but how can I read correctly?

crankyoldgit, 2021-07-21:
> thanks
> when I use IRrecvDumpV2 first three parameter is show in the end of rawarray why ?and how can I fix it?

Can you show us the serial output of an unmodified `IRrecvDumpV2.ino` that you think is reporting the data incorrectly?
e.g. Data please.

crankyoldgit, 2021-07-21:
> I knew if I define temp raw array I can replace them to correct location but how can I read correctly?

I think you have a more fundamental problem, but to answer your question:

https://github.com/crankyoldgit/IRremoteESP8266/blob/dbc8e86c0c8dce0ee3379de6edd77c6dd31e04b6/examples/DumbIRRepeater/DumbIRRepeater.ino#L113-L117
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/dbc8e86c0c8dce0ee3379de6edd77c6dd31e04b6/examples/DumbIRRepeater/DumbIRRepeater.ino#L123-L124

Read:
* https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRutils_8cpp.html#a7b3bbfa1f2bf2dea2fc40a2fefe05a2a
* https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRutils_8cpp.html#aad5f25cf6a2dded8b48f4a6dd16857be

closed 2021-07-21 by mrdsn3