crankyoldgit/IRremoteESP8266 #1542: IRMQTTServer and AKB74955603

kzagatti, 2021-07-29:
Moving from #1513 

@crankyoldgit good morning!

> Upload/use a very small intermediate OTA program. e.g. https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino Load this between using IRMQTTServer and Tasmota

Good news, compiled WebUpdate.ino and managed to go back/forth with Tasmota and IRMQTT - so the issue is indeed related to the memory size. Did not try with IRMQTTServer's minimal though (tried but it failed compiling, probably something my side too). But it's good to share with this community the plan B.

> I honestly haven't used this feature in ages. I'll try to take a look to see what I can see.

Must confess I don't really understand what this was supposed to do... Instead of configuring the entity at configuration.yaml it would show up "magically"?

> I'll also need to know the "state" of the MQTT topics too to debug this. e.g. Is the model topic set etc? and what the "web page" thinks things are set too.

Does this print shows you the info?
[image]

Now for the MQTT debug, I set a listener to the following topics:

- ir_server_quarto/ac/cmnd/#
- ir_server_quarto/log

The first one it's where I'm hearing those command messages from (on + set mode).

You also mentioned the model topic, I couldn't find a way to set the model on the HA side...

crankyoldgit, 2021-07-30:
> Must confess I don't really understand what this was supposed to do... Instead of configuring the entity at configuration.yaml it would show up "magically"?

Kind of. HA's MQTT Discovery is all about devices announcing themselves via MQTT telling HA that it exists, and what attributes/topics etc it has/uses, so that it can be _edited_ via the HA user interface etc. i.e. Not having to edit `configuration.yaml`.

> Sorry, does this print shows you the web page info you mentioned?

Yep.

> I couldn't find a way to set the model on the HA side - am I missing something?

No. It has to be done via MQTT or the IRMQTTServer's Web UI.

Looking at your MQTT log, I notice a number of things:

1. The `.../stat/...` topic messages should have `retain` set on them. i.e. `Retain: true` in your logs. Having `retain` set is important/vital for normal operation after reboots.
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/b6368bd6926eaab15a48410d080979d5cf02f751/examples/IRMQTTServer/IRMQTTServer.ino#L199-L216
2. It took me a while to realise your MQTT log is in reverse order to most _logs_. Most logs are _newest_ entries at the bottom, _oldest_ at the top. Your's are the opposite.
3. Is there any chance you can increase the time resolution of time stamp in your mqtt logging program. i.e. include Seconds, or even milliseconds? That would be helpful to see what actions were done etc.
4. The log seems to show everything happened at `10:36` is that really correct?

If I'm reading it (log) correctly, You told it to reboot. It rebooted successfully. It read the previous `state` from retained MQTT messages. That `state` did **not** include the `power` or `mode` settings for some reason. At this point, depending how it was built, it may have sent the `state` via IR to the a/c. As there was no saved state for `power` & `mode`, IIRC it defaults of `off` for both. It then received a `power` (on) & `mode` (fan) message(s).  Assuming it was compiled with `MQTT_CLIMATE_HA_MODE` set to `true`, which binds `power` `off` and `mode` `off` together. i.e. either turns it off. It should turn "on" when it gets the "fan_only mode message.

I just checked my HA config:
https://github.com/crankyoldgit/Home-AssistantConfig/blob/047dd931ec7d74e7ad00517cea8d6674d3734000/configuration.yaml#L846-L886

I too, clearly have disabled the `power` topic to cope with HA's different way of doing things.

Re: `MQTT climate discovery FAILED to send.`

Looking at the code, the reason that message is likely generated is because the MQTT Discovery message is too long/big for the MQTT library to handle. i.e. Choose a larger value of `MQTT_BUFFER_SIZE`. The default is:
https://github.com/crankyoldgit/IRremoteESP8266/blob/b6368bd6926eaab15a48410d080979d5cf02f751/examples/IRMQTTServer/IRMQTTServer.h#L119-L123

kzagatti, 2021-07-31:
So I think everything is fine and it's all working OK! Let's close the thread then?
Answering your comments:

> 2. It took me a while to realise your MQTT log is in reverse order to most _logs_. Most logs are _newest_ entries at the bottom, _oldest_ at the top. Your's are the opposite.

Haha sorry, this is how HA/mqtt shows it.

> 3. Is there any chance you can increase the time resolution of time stamp in your mqtt logging program. i.e. include Seconds, or even milliseconds? That would be helpful to see what actions were done etc.

Agree, but couldn't find a way to set that but I believe that the message number helps

> 4. The log seems to show everything happened at `10:36` is that really correct?
Yep!

> Looking at the code, the reason that message is likely generated is because the MQTT Discovery message is too long/big for the MQTT library to handle. i.e. Choose a larger value of `MQTT_BUFFER_SIZE`. 

Well, this doesn't actually seems worth it :-D

Again, @crankyoldgit, I appreciate your work and if there's anything I can do to help you or the community, count me in!

Best

crankyoldgit, 2021-07-31:
No worries. Thanks for getting back to us. Closing.

closed 2021-07-31 by crankyoldgit