crankyoldgit/IRremoteESP8266 #1543: Different Models in Daikin Remote BRC4C151

rahul121190, 2021-07-31:
I have 2 units of Daikin AC in my bedroom controlled by different remotes. Both uses the same model (BRC4C151), but I have set the dip switch values to A and B respectively in each of the remotes. The same setting is done at the individual indoor units to map the remote to the respective units. The model BRC4C151 corresponds to DAIKIN176 when I test both the remotes using IRRecvDumpV3, however using the IRDaikin library to send the IR signal works only for one of the units (Remote set to model A). And the other unit (set to Model B) does not respond to the send method.

I have currently stored the raw value for the On/Off for the second remote and using the sendRaw method in IRSend to make it work.

I use Hitachi in my living room and recently saw the setModel method in IRHitachi for HITACHI_AC1 model. And I was able to set the value R_LT0541_HTA_A/R_LT0541_HTA_B and it corresponds to A/B dip switch in the Hitachi Remote. However I could not find setModel method in IRDaikin for DAIKIN176.

Is there any way I can make the Model B Daikin remote to work without setting the dip switch to A (since I want to control each of these units separately)?

NiKiZe, 2021-07-31:
So what is the difference from the remotes sending the same data from them and picking it up with IRRecvDumpV3? (we need you to tell us the difference)

setModel is the model of the device, not the unit id, please again provide the captures for the difference of only changing the dip switch.

You need a unitID setting, not Model.

crankyoldgit, 2021-07-31:
> please again provide the captures for the difference of only changing the dip switch.

@NiKiZe is correct.

We need the full output of the dump program (plus your description of the settings of the remote) with every setting being the same, except only changing the A/B setting. That way it should be possible to workout what changes and what part of the message has that info.

rahul121190, 2021-07-31:
**Power Off in Remote BRC4C151 with 'A' setting**

Timestamp : 000338.702
Library   : v2.7.19

Protocol  : DAIKIN176
Code      : 0x11DA171804001E11DA17180073002000001E36002021 (176 Bits)
Mesg Desc.: **Power: Off,** Mode: 2 (Cool), Temp: 24C, Fan: 3 (High), Swing(H): 6 (Off)
uint16_t rawData[359] = {5102, 2118,  400, 1756,  402, 676,  402, 676,  402, 676,  402, 1756,  402, 678,  402, 676,  402, 676,  402, 676,  402, 1756,  402, 678,  402, 1756,  402, 1756,  400, 676,  402, 1756,  402, 1756,  402, 1756,  402, 1756,  404, 1754,  402, 676,  402, 1756,  402, 678,  400, 678,  402, 676,  400, 676,  402, 676,  400, 678,  402, 1756,  402, 1756,  402, 676,  402, 676,  402, 678,  400, 676,  402, 676,  402, 1756,  402, 678,  400, 676,  404, 676,  402, 676,  402, 678,  400, 678,  402, 676,  402, 678,  400, 676,  402, 678,  402, 676,  402, 676,  402, 676,  402, 678,  402, 1756,  402, 1756,  402, 1756,  402, 1756,  402, 676,  402, 678,  402, 676,  402, 29344,  5102, 2116,  402, 1756,  402, 676,  402, 676,  402, 676,  402, 1756,  402, 676,  402, 676,  402, 676,  400, 678,  402, 1756,  400, 678,  402, 1756,  400, 1758,  402, 676,  402, 1756,  402, 1756,  402, 1756,  400, 1756,  402, 1756,  402, 676,  402, 1756,  402, 676,  402, 676,  404, 676,  402, 676,  402, 676,  402, 676,  402, 1756,  402, 1756,  402, 676,  402, 676,  402, 676,  402, 676,  402, 678,  402, 676,  402, 678,  402, 676,  402, 676,  402, 676,  402, 678,  402, 1756,  402, 1756,  402, 676,  402, 676,  402, 1756,  402, 1756,  400, 1756,  402, 676,  402, 676,  402, 676,  402, 676,  402, 678,  402, 676,  402, 676,  402, 676,  402, 676,  402, 676,  402, 678,  402, 676,  402, 676,  402, 676,  402, 1756,  402, 676,  402, 676,  402, 676,  400, 678,  402, 676,  402, 678,  402, 676,  402, 680,  398, 676,  402, 676,  402, 676,  402, 676,  402, 678,  402, 676,  402, 676,  402, 676,  402, 678,  400, 676,  402, 676,  402, 1756,  402, 1756,  400, 1756,  402, 1756,  402, 676,  402, 678,  402, 674,  402, 676,  402, 1756,  402, 1756,  402, 678,  400, 1758,  400, 1756,  402, 676,  402, 676,  402, 676,  402, 676,  402, 676,  400, 676,  402, 676,  402, 676,  402, 676,  402, 676,  402, 676,  402, 676,  402, 676,  402, 676,  402, 676,  400, 1756,  402, 676,  402, 676,  402, 1756,  402, 678,  402, 676,  402, 676,  402, 678,  400, 1756,  402, 676,  402, 678,  400};  // DAIKIN176
uint8_t state[22] = {0x11, 0xDA, 0x17, 0x18, 0x04, 0x00, 0x1E, 0x11, 0xDA, 0x17, 0x18, 0x00, 0x73, 0x00, 0x20, 0x00, 0x00, 0x1E, 0x36, 0x00, 0x20, 0x21};

**Power Off in Remote BRC4C151 with 'B' setting**

Timestamp : 000343.550
Library   : v2.7.19

Protocol  : DAIKIN176
Code      : 0x11DA171904001F11DA17190073002000001E36002022 (176 Bits)
Mesg Desc.: **Power: Off**, Mode: 2 (Cool), Temp: 24C, Fan: 3 (High), Swing(H): 6 (Off)
uint16_t rawData[359] = {5110, 2120,  402, 1758,  404, 678,  402, 678,  402, 678,  402, 1760,  402, 678,  402, 678,  402, 678,  402, 676,  404, 1758,  402, 678,  402, 1758,  402, 1758,  402, 678,  402, 1758,  404, 1758,  404, 1758,  402, 1758,  404, 1758,  402, 678,  402, 1758,  404, 676,  402, 678,  404, 676,  404, 1758,  402, 678,  402, 678,  402, 1758,  402, 1758,  402, 678,  404, 678,  402, 678,  402, 678,  402, 678,  402, 1758,  402, 678,  404, 676,  402, 678,  402, 676,  404, 678,  402, 676,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  404, 676,  402, 678,  402, 1758,  402, 1758,  402, 1758,  404, 1758,  402, 1758,  402, 678,  402, 678,  402, 678,  402, 29392,  5110, 2118,  402, 1760,  402, 678,  402, 678,  402, 678,  402, 1760,  402, 678,  404, 678,  402, 678,  402, 678,  402, 1758,  402, 678,  402, 1758,  402, 1758,  404, 676,  402, 1758,  402, 1758,  402, 1758,  402, 1758,  402, 1758,  402, 678,  402, 1760,  402, 678,  402, 678,  402, 678,  402, 1758,  404, 676,  404, 678,  402, 1758,  404, 1758,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 1758,  402, 1760,  402, 678,  402, 678,  402, 1758,  402, 1760,  402, 1760,  402, 678,  402, 676,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  404, 678,  402, 678,  402, 678,  404, 1758,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 680,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 676,  402, 1760,  402, 1758,  402, 1760,  400, 1760,  402, 678,  402, 678,  402, 678,  402, 678,  402, 1758,  402, 1758,  402, 678,  402, 1760,  402, 1760,  402, 678,  402, 676,  402, 678,  402, 678,  402, 678,  404, 676,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 1758,  402, 678,  402, 678,  402, 678,  402, 1758,  402, 678,  402, 678,  400, 678,  402, 1758,  402, 678,  404, 678,  402};  // DAIKIN176
uint8_t state[22] = {0x11, 0xDA, 0x17, 0x19, 0x04, 0x00, 0x1F, 0x11, 0xDA, 0x17, 0x19, 0x00, 0x73, 0x00, 0x20, 0x00, 0x00, 0x1E, 0x36, 0x00, 0x20, 0x22};

**Power On in Remote BRC4C151 with 'A' setting**

Timestamp : 000348.150
Library   : v2.7.19

Protocol  : DAIKIN176
Code      : 0x11DA171804001E11DA17180073002100001E36002022 (176 Bits)
Mesg Desc.: **Power: On**, Mode: 2 (Cool), Temp: 24C, Fan: 3 (High), Swing(H): 6 (Off)
uint16_t rawData[359] = {5100, 2116,  402, 1756,  402, 676,  402, 676,  402, 676,  402, 1756,  402, 676,  402, 676,  402, 678,  400, 678,  402, 1756,  402, 676,  402, 1756,  402, 1756,  402, 676,  402, 1756,  402, 1756,  402, 17 [... cut ...]   402, 678,  402, 1760,  400, 1760,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 678,  402, 1758,  402, 680,  402, 678,  402, 1758,  404, 1758,  402, 678,  402, 678,  402, 678,  402, 1760,  402, 678,  402, 678,  402};  // DAIKIN176
uint8_t state[22] = {0x11, 0xDA, 0x17, 0x19, 0x04, 0x00, 0x1F, 0x11, 0xDA, 0x17, 0x19, 0x00, 0x73, 0x00, 0x21, 0x00, 0x00, 0x1E, 0x36, 0x00, 0x20, 0x23};

rahul121190, 2021-07-31:
If I use the below sample program to send IR command, the AC unit configured with remote 'A' setting only turns on. Nothing happens to the AC unit with 'B' setting. Now my question is how to set the remote 'B' using the below code. (Currently I use the stored rawData of remote 'B' and use sendRaw method in IRSend to make it work.)

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Daikin.h>

const uint16_t kIrLed = 16;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRDaikin176 ac(kIrLed);  // Set the GPIO to be used to sending the message

void setup() {
  ac.begin();
  Serial.begin(115200);
}

void loop() {
  Serial.println("Sending...");

  // Set up what we want to send. See ir_Daikin.cpp for all the options.
  ac.on();
  ac.setFan(kDaikinFanMax);
  ac.setMode(kDaikin176Cool);
  ac.setTemp(24);
  ac.setSwingHorizontal(kDaikin176SwingHOff);
  // Display what we are going to send.
  SERIALPLN(ac.toString());
  // Now send the IR signal.
    ac.send();

  // Display what we are going to send.
  Serial.println(ac.toString());

  // Now send the IR signal.
#if SEND_DAIKIN176
  ac.send();
#endif  // SEND_DAIKIN176
}

crankyoldgit, 2021-07-31:
_0x11DA171**8**04001E11DA17180073002000001E36002021_ A
_0x11DA171**9**04001F11DA17190073002000001E36002022_ B
_0x11DA171**8**04001E11DA17180073002100001E36002022_ A
_0x11DA171**9**04001F11DA17190073002100001E36002023_ B

Last byte is the checksum, so the **bold** number is likely the A/B unit designator.

I'll code something up soon.

rahul121190, 2021-07-31:
That's great @crankyoldgit. Thanks for the quick response.

crankyoldgit, 2021-07-31:
Please download and try out branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1543

It should now give you `setId()` and `getId()` for the `IRDaikin176` class.
e.g.
```c++
ac.setId(1);  // Create "B" setting messages.
ac.setId(0);  // Create "A" setting messages.
```

If you recompile the dump program using this branch, it should report the `Id:` in the text description of the message.

Let me know how it goes, but based on the data you've provided, it should work fine.

NiKiZe, 2021-07-31:
Is A and B the only settings, or are there more?

Also when you mention model, is it the model number of the actual unit/remote? Or is it what works in the library? (If we should expand on the supported models list)

NiKiZe, 2021-07-31:
I would also like to see the same test on your Hitachi unit with only A/B changed for verification there as well. (Even if it is a different issue if it is something that we can improve on)

rahul121190, 2021-08-01:
[image]

rahul121190, 2021-08-01:
Thanks @crankyoldgit . That worked perfectly! I was able to set the Id to 0 and 1 and I am now able to control each of the units separately.

closed 2021-08-01 by crankyoldgit

rahul121190, 2021-08-01:
Thanks @NiKiZe. I think they are just remote settings, nothing more. Both the Daikin AC units are in the same room and I want to control them via separate remotes, I set the dip switch value in each of the remote to different positions. (as seen in the above image). As you can see, there are only 2 positions that I can set the dip switch to. There is similar dip switch present in each of the indoor units as well, and I need to set them to A/B so that I can use each of the remote separately for the units.

I have only one unit of Hitachi_AC and it works when I set the model to R_LT0541_HTA_A in the code and dip switch 'A' in the remote.

crankyoldgit, 2021-08-01:
@rahul121190 Thanks for confirming. The PR is now merged in to the `master` branch.

crankyoldgit, 2021-08-01:
> I have only one unit of Hitachi_AC and it works when I set the model to R_LT0541_HTA_A in the code and dip switch 'A' in the remote.

If I remember correctly, we implemented as a model on that protocol as the user needed to access it via Tasmota. So I mapped Model to ID in that particular protocol to help them out.

rahul121190, 2021-08-01:
> > I have only one unit of Hitachi_AC and it works when I set the model to R_LT0541_HTA_A in the code and dip switch 'A' in the remote.
> 
> If I remember correctly, we implemented it as a model on that protocol as the user needed to access it via Tasmota. So I mapped Model to ID in that particular protocol to help them out.

Yes @crankyoldgit. I think the dip switches in Hitachi are for different models of the Hitachi_AC1, but for Daikin176, setting the ID 0 and 1 are merely used when we want to use different remotes for different units in the same room.

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

rahul121190, 2021-08-28:
> FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

Thanks to @crankyoldgit and @NiKiZe for all the help.

PR #1544 (2021-08-01): Daikin176: Add support for Unit Id.
* Add `IRDaikin176:setId()` & `IRDaikin176:getId()` for dealing with Unit Identifiers.
* Add "Id" to `.toString()` output.
* Add unit test coverage for that, and update existing tests.
* Misc code style cleanups.

Fixes #1543
commit 89ead22d (2021-08-01): Daikin176: Add support for Unit Id. (#1544)

* Add `IRDaikin176:setId()` & `IRDaikin176:getId()` for dealing with Unit Identifiers.
* Add "Id:" to `.toString()` output.
* Add unit test coverage for that, and update existing tests.
* Misc code style cleanups.

Fixes #1543