crankyoldgit/IRremoteESP8266 #1552: Daikin AC OFF not working in Daikin Remote ARC484A4

rahul121190, 2021-08-07:
Hi @crankyoldgit . I hope you are doing well.

I have a Daikin AC unit which uses the remote ARC484A4. The model ARC484A4 corresponds to protocol DAIKIN (280 bits/35 bytes), and hence I am using the IRDaikinESP class to send the IR signal and it works when I try to turn ON the AC. However it does not work when I try to turn OFF the AC. I have currently stored the raw value for the Off command for the remote and using the sendRaw method in IRSend to make it work.

I managed to get the IR Dump data of my remote as well as the one generated by the library file. Below are the IR data captured via IRrecvDumpV3. I see a difference of 9 bytes between the code generated by library and the remote.

**AC OFF**
0x11DA2700C500**10E7**11DA27004200005411DA27000038**28**00**A0**000006600000**C180**00**B9** (via Remote Control)
0x11DA2700C500**00D7**11DA27004200005411DA27000038**34**00**30**000006600000**C000**00**D4** (library generated)

**AC ON**
0x11DA2700C500**10E7**11DA27004200005411DA27000039**28**00**A0**000006600000**C180**00**BA** (via Remote Control)
0x11DA2700C500**00D7**11DA27004200005411DA27000039**34**00**30**000006600000**C000**00**D5** (library generated)

Is there any way I can make the AC OFF to work using the library.

Thanks in Advance

rahul121190, 2021-08-07:
[image]

NiKiZe, 2021-08-07:
For reference, could you also give us the model number of the A/C Unit?

What does IRrecvDumpV3 show as message when you use the off command?

We can sure make this work, but might need some more data before we can pin-point the needed changes

crankyoldgit, 2021-08-07:
> **AC OFF**
> 0x11DA2700C500**10E7**11DA27004200005411DA27000038**28**00**A0**000006600000**C180**00**B9** (via Remote Control)
> 0x11DA2700C500**00D7**11DA27004200005411DA27000038**34**00**30**000006600000**C000**00**D4** (library generated)

Okay, breaking this down as I go.
0x11DA2700C500**10E7** ... That's the `comfort` bit. So the _remote_ is set with `comfort` on, and `DaikinESP` class hasn't had that set. That difference can be ignored.

The last byte is a checksum, so that can be ignored.

That leaves the "middle" 4 bytes that are different.

The second last different byte (i.e `80` vs `00`) is the weekly timer. So that can be ignored. i.e. Remote has the weekly timer on, the class has it off.

The 3rd last different byte (`C1` vs `C0`) is unknown to us. That is, we don't know it's function.

The 4th last different byte is vertical swing. It's just in a different position. (e.g. `A0` vs `30`)It can be ignored,

And the 5th last different byte is the temperature, it too can be ignored. (e.g. `28` vs `34`)

So that, leaves `byte[31]` (`C1` vs `C0`) as the culprit. How does this behave with other commands from the remote?
We/you need to find out why your remote has it as `C1` instead of `C0`. e.g is it always `C1` etc.

crankyoldgit, 2021-08-07:
FYI, you don't need to use `sendRaw()`, you can use `sendDaikin()` instead.
e.g.
```c++
const uint8_t remote_off[kDaikinStateLength] = {
    0x11, 0xDA, 0x27, 0x00, 0xC5, 0x00, 0x10, 0xE7, 0x11, 0xDA, 0x27, 0x00, 0x42, 0x00, 0x00, 0x54,
    0x11, 0xDA, 0x27, 0x00, 0x00, 0x38, 0x28, 0x00, 0xA0, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 
    0x80, 0x00, 0xB9}; // (via Remote Control)
irsend.sendDaikin(remote_off, kDaikinStateLength);
```

NiKiZe, 2021-08-07:
Correct me if I'm wrong but it is also possible to set the raw state to toggle the bit/byte and still have it in the class?

rahul121190, 2021-08-07:
@NiKiZe & @crankyoldgit : Thanks for the quick response. I stepped out for some work. Once I get back, I will check if the byte[31] behaves the same way for other commands from the remote.

rahul121190, 2021-08-07:
> FYI, you don't need to use `sendRaw()`, you can use `sendDaikin()` instead.
> e.g.
>
> ```c++
> const uint8_t remote_off[kDaikinStateLength] = {
>     0x11, 0xDA, 0x27, 0x00, 0xC5, 0x00, 0x10, 0xE7, 0x11, [...]

Sure will use sendDaikin going forward.

rahul121190, 2021-08-07:
> For reference, could you also give us the model number of the A/C Unit?
> 
> What does IRrecvDumpV3 show as message when you use the off command?
> 
> We can sure make this work, but might need some more data before we can pin-point the needed changes

I stepped out for some work. Once I get back, I will check  the model of the AC unit, as well as the message appearing in the IRrecvDumpv3 when I press Off

crankyoldgit, 2021-08-07:
> Correct me if I'm wrong but it is also possible to set the raw state to toggle the bit/byte and still have it in the class?

Yep, Correctl That byte, if set via `setRaw()` should persist until another `setRaw()` or `stateReset()`.

I was going to suggest that as a possible workaround, but if that single bit is just some model identifier it will be an easy enhancement. (Famous last words)

NiKiZe, 2021-08-07:
> 
> I was going to suggest that as a possible workaround, but if that single bit is just some model identifier it will be an easy enhancement. (Famous last words)

For that we would like to confirm that that bit is set in everything that the remote sends? (yes that is what you wrote)
Does the remote have any kind of id setting under some cover that can change? (but odd  that everything works except off)

rahul121190, 2021-08-07:
> For that we would like to confirm that that bit is set in everything that the remote sends? (yes that is what you wrote)
> Does the remote have any kind of id setting under some cover that can change? [...]

No. The remote does not have ant dip switch under the cover (unlike my other Daikin Remote BRC4C151 for DAIKIN176 has one, and @crankyoldgit fixed it via setId/getId.)

rahul121190, 2021-08-07:
> For reference, could you also give us the model number of the A/C Unit?
> 
> What does IRrecvDumpV3 show as message when you use the off command?
> 
> We can sure make this work, but might need some more data before we can pin-point the needed changes

[image]

rahul121190, 2021-08-07:
> Okay, breaking this down as I go.
> 0x11DA2700C500**10E7** ... That's the `comfort` bit. So the _remote_ is set with `comfort` on, and `DaikinESP` class hasn't had that set. That difference can be [...]

@crankyoldgit : Looks like the byte[31] does not change no matter what buttons I press on the remote. I tried changing the mode/temperature/fan speed/swing/timer/turbo etc., but still byte[31] remains the same. Attaching the IR Dump data for various buttons pressed in the remote.

rahul121190, 2021-08-07:
15:41:13.371 -> Protocol  : DAIKIN
15:41:13.371 -> Code      : 0x11DA2700C50000D711DA27004200005411DA2700004D2E00400000060F0100**C1**800024 (280 Bits)
15:41:13.371 -> Mesg Desc.: Power: On, **Mode: 4 (Heat),** Temp: 23C, Fan: 2 (UNKNOWN), Powerful: On, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: 04:00, Weekly Timer: Off

15:41:27.686 -> 
15:41:27.686 -> Protocol  : DAIKIN
15:41:27.686 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000069320040000006600100**C1**800095 (280 Bits)
15:41:27.719 -> Mesg Desc.: Power: On, **Mode: 6 (Fan),** Temp: 25C, Fan: 2 (UNKNOWN), Powerful: On, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:41:32.943 -> Protocol  : DAIKIN
15:41:32.943 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000009320040000006600000**C1**800034 (280 Bits)
15:41:32.943 -> Mesg Desc.: Power: On, **Mode: 0 (Auto)**, Temp: 25C, Fan: 2 (UNKNOWN), Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off
15:41:35.371 -> 
15:41:35.371 -> Protocol  : DAIKIN
15:41:35.371 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000029C000A0000006600000**C1**800042 (280 Bits)
15:41:35.371 -> Mesg Desc.: Power: On, **Mode: 2 (Dry),** Temp: 96C, Fan: 10 (Auto), Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:41:38.031 -> Protocol  : DAIKIN
15:41:38.031 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000039320040000006600000**C1**800064 (280 Bits)
15:41:38.031 -> Mesg Desc.: Power: On, **Mode: 3 (Cool)**, Temp: 25C, Fan: 2 (UNKNOWN), Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off
15:41:38.397 -> 
15:41:46.254 -> Protocol  : DAIKIN
15:41:46.254 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000038320040000006600000**C1**800063 (280 Bits)
15:41:46.254 -> Mesg Desc.: **Power: Off**, Mode: 3 (Cool), Temp: 25C, **Fan: 2 (UNKNOWN)**, Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:41:50.478 -> Protocol  : DAIKIN
15:41:50.478 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000038320050000006600000**C1**800073 (280 Bits)
15:41:50.478 -> Mesg Desc.: Power: Off, Mode: 3 (Cool), Temp: 25C, **Fan: 3 (Medium)**, Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:41:54.206 -> Protocol  : DAIKIN
15:41:54.206 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000038320060000006600000**C1**800083 (280 Bits)
15:41:54.206 -> Mesg Desc.: Power: Off, Mode: 3 (Cool), Temp: 25C, **Fan: 4 (UNKNOWN)**, Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off
15:41:57.466 -> 
15:41:57.466 -> Protocol  : DAIKIN
15:41:57.466 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000038320070000006600000**C1**800093 (280 Bits)
15:41:57.466 -> Mesg Desc.: Power: Off, Mode: 3 (Cool), Temp: 25C, **Fan: 5 (High)**, Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:42:00.793 -> Protocol  : DAIKIN
15:42:00.793 -> Code      : 0x11DA2700C50000D711DA27004200005411DA270000383200A0000006600000**C1**8000C3 (280 Bits)
15:42:00.793 -> Mesg Desc.: Power: Off, Mode: 3 (Cool), Temp: 25C, **Fan: 10 (Auto)**, Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:42:04.618 -> Protocol  : DAIKIN
15:42:04.618 -> Code      : 0x11DA2700C50000D711DA27004200005411DA270000383200B0000006600000**C1**8000D3 (280 Bits)
15:42:04.618 -> Mesg Desc.: Power: Off, Mode: 3 (Cool), Temp: 25C, **Fan: 11 (Quiet),** Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:42:07.379 -> Protocol  : DAIKIN
15:42:07.412 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000038320030000006600000**C1**800053 (280 Bits)
15:42:07.412 -> Mesg Desc.: Power: Off, Mode: 3 (Cool), Temp: 25C, **Fan: 1 (Low),** Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: Off, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

15:42:51.867 -> Protocol  : DAIKIN
15:42:51.867 -> Code      : 0x11DA2700C50010E711DA27004200005411DA270000393200A0000006600000**C1**8000C4 (280 Bits)
15:42:51.900 -> Mesg Desc.: **Power: On**, Mode: 3 (Cool), Temp: 25C, **Fan: 10 (Auto),** Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, Comfort: On, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

[... cut ...]
16:02:41.334 -> Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 25C, Fan: 10 (Auto), Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, **Comfort: On,** Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

16:02:57.938 -> Protocol  : DAIKIN
16:02:57.938 -> Code      : 0x11DA2700C50000D711DA27004200005411DA27000039320070000006600000**C1**800094 (280 Bits)
16:02:57.938 -> Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 25C, Fan: 5 (High), Powerful: Off, Quiet: Off, Sensor: Off, Mould: Off, **Comfort: Off**, Swing(H): Off, Swing(V): Off, Clock: 00:00, Day: 0 (UNKNOWN), On Timer: Off, Off Timer: Off, Weekly Timer: Off

crankyoldgit, 2021-08-07:
> Sure will use sendDaikin going forward.

Before I code something up. Can you please confirm the above actually works. i.e. It turns off the AC.

Next, assuming it does work. Try:
```c++
IRDaikinESP ac(your_led);
const uint8_t remote_off[kDaikinStateLength] = {
    0x11, 0xDA, 0x27, 0x00, 0xC5, 0x00, 0x10, 0xE7, 0x11, 0xDA, 0x27, 0x00, 0x42, 0x00, 0x00, 0x54,
    0x11, 0xDA, 0x27, 0x00, 0x00, 0x38, 0x28, 0x00, 0xA0, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0xC1, 
    0x80, 0x00, 0xB9}; // (via Remote Control)
ac.begin();
ac.setRaw(remote_off);
ac.setBlah(what ever controls etc);
ac.on();
ac.send();
delay(50000); // 50 seconds
ac.off();
ac.send();
```

That should simulate what I'm going to do to add this different "model". It should let us know if it works before I code anything up.
```

crankyoldgit, 2021-08-07:
@rahul121190 Can you please download & test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/ARC484A4 (PR #1553)
It should give you `IRDaikinESP::setModel()` which should control that differing bit.
My unit tests demonstrate I can recreate the exact same `state[]` code that your real remote produces with the library, so it should work for you fine.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/04124344b002fc514cc2af28f3ae8e256fe351b0/test/ir_Daikin_test.cpp#L3925-L3951

Let us know how it goes, please.

rahul121190, 2021-08-07:
@crankyoldgit : thanks for the quick turnaround. I used the latest branch and I did some testing. Please find the below details.

1. The basic ON and OFF is working
2. Changing the temperature is working.
3. If I assign setFan to any value other than kdaikinFanAuto(10), then it is NOT working.
4. The same issue happens when I assign setMode to any other value other than kDaikinCool(3).

I am yet to test the other parameters like powerful, Mold, Swing etc.

rahul121190, 2021-08-07:
> @rahul121190 Can you please download & test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/ARC484A4 (PR #1553)
> It should give you `IRDaikinESP::setModel()` which should control that [...]

I initially used the exact values as stated in the above test code and both ON and Off worked.

crankyoldgit, 2021-08-07:
Per the wiki for adding a new AC, you'll need to document and work out the new/different fan speeds and modes.

The data you collected earlier indicated fan high/4 worked.

rahul121190, 2021-08-07:
> Per the wiki for adding a new AC, you'll need to document and work out the new/different fan speeds and modes.
> 
> The data you collected earlier indicated fan high/4 worked.

Ok. Will work out the different fan speeds and modes. However I am not sure if it is generating the same bytes as intended by the AC. Will do a comparison between the remote and the library IR dump for different modes/fan speeds.

crankyoldgit, 2021-08-08:
Looking forward to it.

crankyoldgit, 2021-08-13:
Friendly ping/reminder to get to this this weekend if you can.

rahul121190, 2021-08-13:
Hi @crankyoldgit. Sorry for the delayed response. I was testing numerous things on my side. Please see the below findings.

1. The remote that I was initially using for testing (from which I took the dump data DAIKIN280) is not the Original remote which came with the Ac unit. It is a compatible remote that was bought since the Original Remote did not work.

2. Since many of the features like modes/fan/swing did not work on my AC unit using the test branch as well, I decided to get my Original remote working (It had a display issue).

3. Fortunately, I was able to fix the display, and tried to capture the IRdata using Dump V3. Surprisingly, the protocol used here is DAIKIN216 (for Original Remote - ARC484A4) and not DAIKIN280 (protocol from compatible remote).

4. So I started testing my AC unit with DAIKIN216 functions, and it started working as expected. (Even though the 26th byte in library was C0, and the remote it was C2, but still all actions were working)

5. However there is one issue with the above setup. I have positioned my IR Emitter on the adjacent wall, all functions work properly except for the AC OFF. I tried moving my IR emitter closer to the AC unit, and only after that I am able to switch off the AC Unit. Has anyone reported this issue in the past? (OFF function is working only if the IR Emitter is in close proximity to the AC Unit, whereas mode/temperature/fan/swing functions works from the initial position itself.) Currently I am trying to find a position for my IR Emitter so that all functions including OFF works. I am currently using an IR Led connected to PN222 transistor via the high side and is controlled by Wemos D1 Mini (5v).

Having said that, I think it is safe to do the below things on your end.

1. Delete this branch as the original remote ARC484A4 maps to DAIKIN216 protocol. (Custom DAIKIN280 is no longer required)
2. Add support for ARC484A4 remote using the DAIKIN216 library, so that it will be helpful for others.
3. Any suggestions/pointers to resolve the range issue for OFF function will be much appreciated.

Kudos to your effort and timely support.

NiKiZe, 2021-08-13:
The data that was provided here is not from the ARC484A4 remote, but actually from some other remote?

rahul121190, 2021-08-13:
> The data that was provided here is not from the ARC484A4 remote, but actually from some other remote?

Unfortunately Yes. I thought I bought the same remote again when the original ARC484A4 remote did not work. However it turned out to be a different model (since the protocol is different - DAIKIN280). Since the modified library was not working for most functions except the basic ones, I realized something was wrong and I fixed the Original ARC484A4 remote and it turned out to have DAIKIN216 protocol.

crankyoldgit, 2021-08-15:
> 4\. So I started testing my AC unit with DAIKIN216 functions, and it started working as expected. (Even though the 26th byte in library was C0, and the remote it was C2, but still all actions were working)

Got data/captures to show this?

> However there is one issue with the above setup. I have positioned my IR Emitter on the adjacent wall, all functions work properly except for the AC OFF. I tried moving my IR emitter closer to the AC unit, and only after that I am able to switch off the AC Unit. Has anyone reported this issue in the past? (OFF function is working only if the IR Emitter is in close proximity to the AC Unit, whereas mode/temperature/fan/swing functions works from the initial position itself.) Currently I am trying to find a position for my IR Emitter so that all functions including OFF works. I am using an IR Led connected to PN222 transistor via the high side and is controlled by Wemos D1 Mini (5v).
> 3\. Any suggestions/pointers to resolve the range issue for OFF function will be much appreciated.

Yes, we have seen this sort of thing before.
Have a read of https://github.com/crankyoldgit/IRremoteESP8266/issues/1298#issuecomment-716153694 and the issue in general.

Usually it is an environment or hardware problem.
e.g Not enough current being supplied to the IR led (resistors, the transistor limiting the current, not enough current in the power supply/voltage regulator etc), poor quality IR LED, wrong frequency of IR LED, to wide an angle of the IR LED, etc etc etc.

To boost the signal, try:
1. Aiming better.
2. Narrower beam if IR LED.
3. More current. More voltage.
4. More efficient/powerful IR LED.
5. Use multiple IR LEDs.
6. Improve the environment. eg. cooler and/or darker room. No direct sunlight or heat sources. etc.
7. Tune the resistor value(s)
8. Better transistor. etc etc.

rahul121190, 2021-08-15:
> > 4. So I started testing my AC unit with DAIKIN216 functions, and it started working as expected. (Even though the 26th byte in library was C0, and the remote it was C2, but still all actions were working)
> 
> Got data/captures to show this?
> 
> > However there is one issue with the above setup. I have positioned my IR Emitter on the adjacent wall, all functions work properly except for the AC OFF. I tried moving my IR emitter closer to the AC unit, and only after that I am able to switch off the AC Unit. Has anyone reported this issue in the past? (OFF function is working only if the IR Emitter is in close proximity to the AC Unit, whereas mode/temperature/fan/swing functions works from the initial position itself.) Currently I am trying to find a position for my IR Emitter so that all functions including OFF works. I am using an IR Led connected to PN222 transistor via the high side and is controlled by Wemos D1 Mini (5v).
> > 3. Any suggestions/pointers to resolve the range issue for OFF function will be much appreciated.
> 
> Yes, we have seen this sort of thing before.
> Have a read of [#1298 (comment)](https://github.com/crankyoldgit/IRremoteESP8266/issues/1298#issuecomment-716153694) and the issue in general.
> 
> Usually it is an environment or hardware problem.
> e.g Not enough current being supplied to the IR led (resistors, the transistor limiting the current, not enough current in the power supply/voltage regulator etc), poor quality IR LED, wrong frequency of IR LED, to wide an angle of the IR LED, etc etc etc.
> 
> To boost the signal, try:
> 
> 1. Aiming better.
> 2. Narrower beam if IR LED.
> 3. More current. More voltage.
> 4. More efficient/powerful IR LED.
> 5. Use multiple IR LEDs.
> 6. Improve the environment. eg. cooler and/or darker room. No direct sunlight or heat sources. etc.
> 7. Tune the resistor value(s)
> 8. Better transistor. etc etc.

Thanks @crankyoldgit . We were also trying some of the above steps.
It finally worked when we reduced the resistor in series with the IR led from 22 Ω to 10 Ω. We have also changed the position of the IR Blaster closer to the AC.

closed 2021-08-15 by crankyoldgit

crankyoldgit, 2021-08-28:
FYI, the changes mentioned above have been included in the just released v2.7.20 of the library.

commit 963a8067 (2021-08-15): DAIKIN216: Update supported models. (#1567)

Ref: https://github.com/crankyoldgit/IRremoteESP8266/issues/1552#issuecomment-898348494

Fixes #1552