crankyoldgit/IRremoteESP8266 #1557: [Web-AC-control] Example : File Not Found

641i130, 2021-08-11:
### Version/revision of the library used
v2.7.15

### Describe the bug
After inputting the wifi connection information to the device, when connecting to it via browser on the same network, it isn't able to find any of the files.

#### To Reproduce
1. Install the Example, replace line 78 with `String contentType = mime::getContentType(path);` (or it will not compile)
2. Connect to it, then input the wifi connection info.
3. Do a network scan to find its IP (or enable the serial by uncommenting all the lines in step 1)

#### Example code used
[All of this](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/Web-AC-control)

#### Expected behaviour
It should go to the page in the screenshot.
Instead, it sends me to a page with this:
```
File Not Found

URI: /ui.html
Method: GET
Arguments: 0
```
#### Output of serial
```
mounting LittleFS...
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.61
*WM: freeing allocated params!
```

### Has this library/code previously worked as expected for you?
Yes, I'm able to run the V2 receive example correctly.

### Other useful information
I'm on Arch Linux right now, but I doubt this matters. I think the filesystem library or system in place is broken somehow.

NiKiZe, 2021-08-11:
Exactly which example have you used?

641i130, 2021-08-11:
@NiKiZe https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/Web-AC-control

641i130, 2021-08-11:
Update: Trying to fall back on SPIFFS didn't help either.

closed 2021-08-11 by 641i130

crankyoldgit, 2021-08-11:
I believe the issue of you need to upload the files into the ESP's flash filesystem in addition to just simply flashing the binary. That example code was user submitted.

crankyoldgit, 2021-08-12:
@641i130 Just following up on this, the instructions are in the `README.md` file. It looks like you missed performing a step:
https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/Web-AC-control#instructions

641i130, 2021-08-12:
Ah, you're right, that's exactly what I missed. Thanks for helping!