crankyoldgit/IRremoteESP8266 #1561: Adding support for PANASONIC AC CW Series

PepperBurst, 2021-08-12:
### Version/revision of the library used
v2.7.19

### Specific Model Used
AC - CW-XN620JPH
Remote - ACXA75C17670

### Description

IR Receive Dump detects the protocol and number of bits.
There are only 2 fan speeds, `LOW` and `HIGH`.
Temp range is from `16` ~ `30` degrees Celcius.
Timer range is from `1` ~ `12` hours.
Modes are `FAN`, `COOL`, and `SLEEP`.

Is there any guide on how to implement this inside the library so I can try and do a PR?

### IR Dump

#### OPERATION OFF/ON

Protocol  : PANASONIC_AC
Code      : 0x0220E004000000060220E00480A1335A (128 Bits)

```cpp
uint16_t rawData[263] = {3544, 1680,  476, 408,  478, 1256,  476, 408,  478, 406,  482, 404,  484, 400,  474, 410,  476, 420,  478, 408,  480, 406,  482, 404,  484, 400,  476, 408,  478, 1256,  478, 408,  478, 418,  480, 404,  482, 402,  474, 410,  478, 408,  478, 406,  482, 1252,  482, 1252,  480, 1266,  478, 406,  480, 404,  484, 1250,  482, 402,  474, 410,  476, 408,  478, 406,  480, 416,  482, 404,  484, 402,  476, 410,  478, 408,  480, 406,  482, 402,  484, 400,  476, 422,  476, 408,  478, 406,  482, 404,  484, 402,  474, 412,  476, 408,  478, 408,  480, 416,  480, 404,  452, 432,  474, 410,  456, 430,  448, 438,  448, 436,  450, 434,  452, 444,  454, 430,  446, 1288,  446, 1288,  456, 430,  446, 438,  450, 436,  452, 434,  454, 428,  450, 10504,  3526, 1700,  454, 430,  448, 1286,  448, 438,  450, 434,  452, 434,  454, 430,  446, 440,  448, 450,  448, 436,  452, 434,  454, 432,  454, 430,  446, 438,  450, 1284,  448, 436,  450, 446,  452, 434,  454, 432,  446, 440,  450, 436,  450, 434,  452, 1280,  452, 1280,  452, 1294,  450, 434,  454, 432,  444, 1288,  456, 430,  448, 438,  450, 436,  452, 434,  454, 444,  454, 430,  446, 440,  448, 436,  450, 434,  452, 432,  454, 432,  456, 428,  448, 1298,  446, 1288,  456, 430,  446, 438,  450, 436,  452, 432,  454, 1280,  454, 430,  448, 1298,  446, 1288,  446, 1288,  456, 428,  448, 436,  452, 1282,  452, 1284,  450, 436,  452, 446,  452, 434,  454, 1280,  454, 430,  456, 1278,  456, 1278,  456, 428,  446, 1288,  446, 436,  452};  // PANASONIC_AC
uint8_t state[16] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0xA1, 0x33, 0x5A};
```

#### MODE

Protocol  : PANASONIC_AC
Code      : 0x0220E004000000060220E00480A2335B (128 Bits)

```cpp
uint16_t rawData[263] = {3516, 1706,  450, 432,  454, 1280,  454, 430,  446, 438,  448, 436,  450, 434,  454, 430,  456, 440,  446, 438,  448, 436,  450, 434,  454, 430,  446, 438,  450, 1284,  450, 436,  452, 446,  452, 432,  456, 430,  448, 438,  450, 434,  452, 432,  456, 1278,  456, 1276,  446, 1300,  454, 430,  448, 436,  450, 1284,  450, 436,  452, 434,  454, 430,  446, 438,  448, 448,  450, 436,  452, 432,  454, 430,  446, 438,  448, 436,  450, 434,  454, 432,  456, 440,  446, 438,  450, 436,  452, 434,  454, 430,  446, 438,  448, 436,  452, 434,  454, 444,  454, 430,  458, 428,  448, 436,  452, 434,  454, 430,  446, 438,  448, 436,  450, 446,  452, 432,  456, 1278,  456, 1278,  456, 430,  446, 438,  450, 436,  452, 432,  454, 426,  450, 10500,  3520, 1704,  452, 432,  456, 1278,  456, 430,  446, 438,  450, 434,  452, 432,  454, 430,  446, 450,  448, 438,  450, 436,  452, 432,  454, 430,  448, 436,  450, 1284,  450, 434,  452, 444,  454, 432,  446, 438,  448, 436,  450, 434,  454, 430,  446, 1286,  446, 1286,  448, 1298,  456, 432,  444, 436,  450, 1282,  450, 434,  452, 432,  456, 430,  446, 438,  450, 446,  450, 434,  454, 432,  456, 430,  446, 438,  450, 434,  452, 432,  454, 430,  446, 1298,  456, 428,  448, 1284,  450, 436,  452, 432,  454, 430,  448, 1286,  456, 428,  450, 1294,  450, 1284,  450, 1284,  450, 438,  448, 432,  456, 1278,  446, 1288,  456, 428,  448, 448,  450, 1282,  450, 1284,  450, 434,  452, 1280,  454, 1280,  454, 430,  446, 1288,  456, 426,  450};  // PANASONIC_AC
uint8_t state[16] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0xA2, 0x33, 0x5B};
```

#### TEMP/TIMER DOWN

Protocol  : PANASONIC_AC
Code      : 0x0220E004000000060220E00480A4335D (128 Bits)

```cpp
uint16_t rawData[263] = {3524, 1700,  446, 440,  446, 1284,  450, 434,  452, 432,  454, 430,  446, 438,  450, 436,  452, 446,  452, 432,  456, 430,  448, 438,  450, 436,  452, 432,  456, 1278,  454, 430,  446, 450,  448, 438,  450, 436,  452, 434,  454, 430,  448, 436,  450, 1282,  452, 1282,  450, 1294,  450, 436,  452, 432,  454, 1278,  456, 430,  446, 438,  448, 436,  452, 434,  454, 442,  454, 430,  446, 438,  448, 436,  452, 432,  454, 430,  446, 440,  448, 438,  450, 446,  452, 434,  454, 430,  446, 438,  448, 436,  452, 434,  454, 430,  446, 438,  448, 448,  450, 434,  452, 432,  456, 430,  448, 438,  450, 434,  452, 432,  454, 430,  448, 450,  448, 436,  452, 1280,  452, 1280,  452, 432,  454, 430,  446, 438,  450, 436,  452, 430,  448, 10504,  3516, 1706,  480, 404,  484, 1250,  452, 432,  454, 430,  446, 440,  448, 440,  448, 434,  484, 412,  454, 432,  478, 408,  480, 404,  482, 402,  484, 400,  478, 1256,  476, 408,  480, 418,  480, 404,  482, 402,  476, 410,  478, 406,  480, 404,  484, 1250,  484, 1248,  484, 1262,  482, 402,  486, 398,  478, 1256,  478, 408,  480, 406,  482, 404,  484, 402,  476, 422,  476, 410,  478, 408,  480, 406,  482, 404,  484, 400,  476, 408,  478, 406,  484, 1262,  480, 406,  482, 402,  476, 1258,  484, 400,  476, 406,  480, 1254,  480, 404,  484, 1262,  482, 1250,  484, 1250,  482, 402,  474, 410,  478, 1254,  478, 1254,  478, 406,  480, 414,  484, 1250,  482, 402,  476, 1256,  476, 1256,  476, 1258,  476, 408,  478, 1254,  478, 402,  474};  // PANASONIC_AC
uint8_t state[16] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0xA4, 0x33, 0x5D};
```

#### TEMP/TIMER UP

Protocol  : PANASONIC_AC
Code      : 0x0220E004000000060220E00480A5335E (128 Bits)

```cpp
uint16_t rawData[263] = {3516, 1708,  448, 438,  448, 1284,  450, 436,  452, 434,  454, 430,  456, 428,  448, 438,  450, 448,  450, 436,  452, 432,  454, 430,  456, 428,  448, 436,  452, 1282,  452, 434,  452, 444,  454, 432,  456, 428,  448, 438,  450, 436,  452, 432,  456, 1278,  456, 1278,  456, 1290,  452, 432,  456, 428,  448, 1286,  448, 438,  450, 436,  452, 434,  454, 432,  456, 442,  446, 440,  448, 436,  450, 434,  452, 432,  454, 430,  448, 438,  448, 436,  452, 444,  452, 432,  456, 430,  446, 438,  448, 436,  452, 434,  454, 432,  456, 428,  448, 448,  448, 436,  452, 434,  452, 432,  456, 428,  448, 438,  450, 434,  452, 434,  454, 442,  456, 428,  448, 1286,  448, 1286,  446, 438,  450, 436,  452, 432,  454, 430,  446, 436,  452, 10502,  3550, 1676,  480, 404,  484, 1250,  482, 402,  474, 410,  476, 408,  480, 406,  482, 404,  484, 414,  484, 400,  476, 408,  478, 406,  480, 404,  482, 402,  476, 1258,  484, 400,  476, 420,  476, 408,  480, 406,  482, 404,  484, 402,  476, 408,  478, 1254,  478, 1254,  478, 1268,  476, 408,  478, 406,  482, 1252,  480, 404,  482, 402,  484, 400,  476, 408,  446, 450,  448, 436,  450, 434,  452, 432,  454, 430,  446, 438,  448, 436,  450, 434,  456, 1288,  452, 1280,  452, 432,  488, 1248,  454, 430,  446, 438,  450, 1284,  448, 436,  452, 1294,  450, 1284,  450, 1284,  450, 436,  452, 432,  456, 1278,  456, 1278,  454, 430,  456, 440,  458, 428,  448, 1284,  450, 1284,  450, 1284,  450, 1284,  448, 436,  452, 1282,  450, 430,  446};  // PANASONIC_AC
uint8_t state[16] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0xA5, 0x33, 0x5E};
```

#### TIMER

Protocol  : PANASONIC_AC
Code      : 0x0220E004000000060220E00480A83361 (128 Bits)

```cpp
uint16_t rawData[263] = {3522, 1702,  454, 430,  448, 1286,  446, 438,  448, 436,  450, 434,  452, 432,  454, 430,  446, 450,  448, 438,  450, 434,  452, 432,  454, 430,  446, 438,  450, 1284,  450, 436,  452, 446,  452, 432,  454, 432,  446, 438,  448, 436,  450, 434,  454, 1278,  454, 1278,  454, 1292,  452, 432,  454, 430,  446, 1286,  446, 438,  448, 436,  450, 434,  452, 432,  456, 442,  456, 430,  448, 438,  450, 434,  452, 432,  454, 430,  446, 438,  448, 436,  450, 446,  452, 432,  454, 430,  446, 438,  448, 436,  450, 434,  452, 432,  456, 430,  446, 450,  448, 438,  450, 434,  452, 434,  454, 430,  446, 438,  448, 436,  450, 434,  452, 444,  454, 430,  448, 1286,  448, 1286,  446, 438,  450, 436,  484, 402,  474, 410,  476, 404,  484, 10470,  3548, 1676,  480, 404,  484, 1250,  482, 402,  474, 410,  478, 408,  480, 406,  482, 404,  484, 414,  474, 412,  476, 410,  478, 406,  448, 436,  452,  [... cut ...] 38,  448, 436,  452, 434,  454, 432,  456, 430,  446, 440,  448, 436,  452, 1294,  448, 436,  484, 1250,  484, 1250,  482, 402,  484, 400,  476, 408,  478, 404,  482, 1262,  482, 1252,  480, 404,  484, 1250,  484, 400,  476, 1258,  476, 1258,  476, 410,  478, 418,  480, 1254,  478, 406,  482, 404,  484, 402,  476, 410,  478, 406,  482, 1252,  480, 402,  476};  // PANASONIC_AC
uint8_t state[16] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0x86, 0x35, 0x41};
```

crankyoldgit, 2021-08-12:
Hi @PepperBurst
The only guide we have is here: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol
The `PANASONIC_AC` protocol however, isn't a boring simple implementation, as it supports multiple byte sizes and protocols. So I can't even suggest looking at other example/protocols to show you how to implement it.
Best is to follow the wiki link and collect the data and try to analyse the protocol structure. When you've worked it out, let us know and we can try to code it in with the other Panasonic ACs.

I'm guessing from the lack of display on the remote ... that this is a really "dumb" protocol / AC system. That the messages are just "this button was pressed", not the usual "this is the state/settings of the remote I want the A/C to have" of a typical A/C message. i.e. The messages are always the same for each button press. Like a TV remote would be.

If it is a _Dumb_ protocol/message style, we really can't produce an A/C class like the other protocols to set temps/settings to exactly what you desire, as that relies on the A/C getting a "state" message, rather than a "button" message. Have I explained it enough?

Actually, looking at the data you've provided. It's pretty clear its a dumb protocol.
0x0220E004000000060220E00480**86**35**41** (Powerful)
0x0220E004000000060220E00480**A1**33**5A** (Power)
0x0220E004000000060220E00480**A2**33**5B** (Mode)
0x0220E004000000060220E00480**A3**33**5C** (Fan)
0x0220E004000000060220E00480**A4**33**5D** (Down)
0x0220E004000000060220E00480**A5**33**5E** (Up)
0x0220E004000000060220E00480**A6**33**5F** (Swing)
0x0220E004000000060220E00480**A7**33**60** (Set/Cancel)
0x0220E004000000060220E00480**A8**33**61** (Timer)

The First **bold** byte/value will be the button press. The Last **bold** byte/will be the checksum of the entire message.
e.g. Power is `0x02 + 0x20 + 0xE0 + 0x04 + 0x00 + 0x00 + 0x00 + 0x06 + 0x02 + 0x20 + 0xE0 + 0x04 + 0x80 + 0xA1 + 0x33 + `[`kPanasonicChecksumInit`](https://github.com/crankyoldgit/IRremoteESP8266/blob/0ec49ff4049580d1e8dc71987c9932b5507650d2/src/ir_Panasonic.h#L87) = 0x4**5A**

https://github.com/crankyoldgit/IRremoteESP8266/blob/0ec49ff4049580d1e8dc71987c9932b5507650d2/src/ir_Panasonic.h#L87

PepperBurst, 2021-08-12:
I see, so this is more of a TV-style implementation. I suppose it wouldn't really be recommended to implement these inside the library itself since it would take too much memory in adding each discovered command unless used with build flags. I'll try and see what I can do with this. Thanks for the info.

crankyoldgit, 2021-08-12:
Pretty much. As the messages are commands/buttons, not full states, you really don't need anything smart or fancy to create the needed entire state.
Sadly however, based on the data provided, it means you can't just tell the A/C to go to some arbitrary configuration.
e.g. "Cool, 21C, Swing on, Medium Fan" etc. Each message will just be relative to what ever the A/C is currently configured as.
e.g. "Add 1 degree to the desired temp.", or "Change to the next mode". We can't easily track that in the library.

PepperBurst, 2021-08-12:
So for now I'm forced to use the complete `rawData[263]` as a lookup right?

crankyoldgit, 2021-08-12:
No, you can send using the `state[]` code.
e.g.
```c++
uint8_t state[16] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0x86, 0x35, 0x41};
irsend.sendPanasonicAC(state, 16);
```

PepperBurst, 2021-08-12:
> No, you can send using the `state[]` codes instead of `rawData[]`.
> e.g.
> 
> ```c++
> uint8_t powerful_state[16] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0x86, 0x35, 0x41};
> irsend.sendPanasonicAC(powerful_state, 16);
> ```

Dang, totally overlooked that function parameter that sets the number of bytes to send. This makes things so much easier then. Thanks a lot for the info and input <3.

crankyoldgit, 2021-08-16:
Going to mark this closed, as it seems we've done all that we are going to do.

closed 2021-08-16 by crankyoldgit