crankyoldgit/IRremoteESP8266 #1595: Creating the new protocol for ARRIS V1113 set-top-box

lucky62, 2021-09-02:
Hello, 
I am trying to create a new protocol according guide in the wiki.
I have captured all the codes from remote (file attached).
When trying to analyze I am getting the error:

```
$ cat raw_data.txt | python3 auto_analyse_raw_data.py --stdin
Found 59 timing entries.
Potential Mark Candidates:
[2584, 674, 340]

Potential Space Candidates:
[1896, 612, 312]

Guessing encoding type:
Sorry, it looks like it is Mark encoded. I can't do that yet. Exiting.
```
What it means?

I have also captured signal - seems not complicated:
[image]
Seems like 0 = pulse-space, 1 = space-pulse (or vice versa).
So how to create a protocol files?...

[raw_data.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/7100289/raw_data.txt)

crankyoldgit, 2021-09-03:
Unfortunately `auto_analyse_raw_data.py` only handles the most common of IR protocols. e.g. Space encoded. It's not smart enough to handle Mark encoded, or as it appears on the surface, this encoding which is [Manchester code](https://en.wikipedia.org/wiki/Manchester_code).

I'll look at trying to code it up, but there is no "easy" mode for it. You can see similar protocols that use this in this library if you want to try coding it on your own. e.g.:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Airwell.cpp
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_RC5_RC6.cpp
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Lasertag.cpp

crankyoldgit, 2021-09-03:
Can you please supply the Brand & Model number of the remote too?

lucky62, 2021-09-03:
> Can you please supply the Brand & Model number of the remote too?

It is a IPTV Set-Top-Box ARRIS VIP1113M.
[Datasheet](https://github.com/crankyoldgit/IRremoteESP8266/files/7104177/VIP1113.Compact.IP.Set-Top.Data.Sheet.pdf)
[Product Page](https://www.commscope.com/product-type/broadband-video-devices/video-devices/set-tops/video-devices-for-kreatv/vip1113/)

(**arris.com** is redirecting now to **commscope.com**, so I don't know, if company was sold or renamed or what..)

crankyoldgit, 2021-09-04:
@lucky62 I've added some experimental code that **might** work with your devices & IR protocol.
Please download, test, & give us feedback on the code in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Arris / PR #1598 

When I asked for the brand/model info previously, I was _expressly_ interested if there is any info written on the remote itself. Sometimes it's in the battery compartment etc. Is there anything on the remote at all?

lucky62, 2021-09-04:
@crankyoldgit , thanks for the prompt support. I will test this after the weekend.

crankyoldgit, 2021-09-06:
Great. Looking forward to the results.

lucky62, 2021-09-06:
hi @crankyoldgit
I tested the both - receiving and sending part - both are working ok.
This is requested photo - unfortunately sticker is partially erased:
[image]
[image]

Some more observations:
Remote is sending one extra code when button is released with two bits inverted.
Sample from IrRecvDump - normal code + "release" code:
```
17:43:59.217 -> uint16_t rawData[55] = {2608, 1872,  690, 282,  364, 276,  358, 280,  692, 586,  366, 272,  362, 276,  360, 280,  366, 278,  366, 272,  362, 276,  358, 280,  366, 274,  392, 246,  358, 280,  364, 274,  360, 284,  362, 276,  358, 280,  384, 254,  698, 276,  358, 282,  364, 276,  358, 592,  700, 580,  700, 580,  692, 588,  692};  // ARRIS 10000F55
17:43:59.217 -> uint64_t data = 0x10000F55;
17:43:59.217 -> 
17:43:59.217 -> 10000000000000000111101010101
17:43:59.217 -> 
17:43:59.317 -> uint16_t rawData[55] = {2610, 1868,  694, 278,  356, 282,  364, 274,  698, 582,  360, 278,  358, 280,  364, 274,  698, 586,  356360, 280,  366, 272,  362, 276,  358, 280,  366, 278,  366, 272,  364, 274,  360, 278,  694, 282,  364, 276,  358, 282,  364, 586,  694, 584,  696, 280,  366, 274,  360, 584,  696};  // ARRIS 10800F5D
17:43:59.317 -> uint64_t data = 0x10800F5D;
17:43:59.317 -> 
17:43:59.317 -> 10000100000000000111101011101
```

When button is kept pressed, then code is repeating after each cca. 102ms:

[image]

#1598

crankyoldgit, 2021-09-06:
Thanks for the remote model info. It's been added.

I'll change the inter-message gap for repeats to:
```c++
uint32_t kArrisGapSpace = 102144 - ((8 + 6 + kArrisBits * 2) *  kArrisHalfClockPeriod);  // 77184 uSeconds
```

However, please note that some of the data you collected in [raw_data.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/7100289/raw_data.txt) has a gap of approx 6ms between messages.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/51b6230b8b46764d18774547088e8ecd20c3770d/test/ir_Arris_test.cpp#L44

> Remote is sending one extra code when button is released with two bits inverted.

I've added a routine (`toggleArrisRelease()`) that does as you suggest, but it is something a user of the library will have to invoke as needed.
While inverting the bits you suggests works, I think if you look at the data, you'll find that the last nibble is a checksum of the preceding nibbles.
e.g. 
_0x10000F5**5** (0x1 + 0x0 +  0x0 +  0x0 +  0x0 + 0xF + 0x5 == 0x1**5**)_
_0x10800F5**D** (0x1 + 0x0 +  0x8 +  0x0 +  0x0 + 0xF + 0x5 == 0x1**D**)_
_0x1080695**D** (0x1 + 0x0 +  0x8 +  0x0 +  0x6 + 0x9 + 0x5 == 0x1**D**)_
_0x1000085**E** (0x1 + 0x0 +  0x0 +  0x0 +  0x0 + 0x8 + 0x5 == 0x**E**)_

In short, flipping the bits you suggests works, but I think it is just technically flipping the first bit, and recalcing the checksum.

Can you confirm this (checksum) with all of your captures? If so, I can add a _checksum_ check to the `decodeArris()` function to ensure it's a "good" capture and more confidence it is the correct protocol.

Assuming my _checksum_ assumption is correct, I've modified `decodeArris()` to report on the button release status (via `address`) and the likely `command`. i.e. rest minus the checksum. I've also added a `encodeArris(command, release)` to enable building the raw Arris code too.

Please let me know how the timing changes and the new code goes for you.

lucky62, 2021-09-08:
Hi, thanks for the update. 

> However, please note that some of the data you collected in [raw_data.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/7100289/raw_data.txt) has a gap of approx 6ms between messages.

Shorter gaps are probably between last normal code and the "release" code.
Release code is generated immediately when button is released.

> Can you confirm this (checksum) with all of your captures?

Sure, I will do this.

lucky62, 2021-09-08:
Checksums tested with this small procedure:

```
void loop() {
  if (irrecv.decode(&ircode)) {
    Serial.print(ircode.value,HEX);
    Serial.print(" ");
    uint32_t value = ircode.value;
    byte i = 0; byte j;
    value = value >> 4; j = value & 0xF; Serial.print(j,HEX); i = i + j; 
    value = value >> 4; j = value & 0xF; Serial.print(j,HEX); i = i + j;
    value = value >> 4; j = value & 0xF; Serial.print(j,HEX); i = i + j;
    value = value >> 4; j = value & 0xF; Serial.print(j,HEX); i = i + j;
    value = value >> 4; j = value & 0xF; Serial.print(j,HEX); i = i + j;
    value = value >> 4; j = value & 0xF; Serial.print(j,HEX); i = i + j;
    value = value >> 4; j = value & 0xF; Serial.print(j,HEX); i = i + j;
    Serial.print(" ");
    Serial. print(i & 0xF,HEX);
    if ( (i & 0xF) == (ircode.value & 0xF) ) {
      Serial.println(" OK");
    } else {
      Serial.println(" ERROR");
    }
  }
}
```

Result is as expected. last nibble (4 bits) is a sum of previous 7 nibbles.
Output from test attached.
[Checksum_Test_Result.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/7131570/Checksum_Test_Result.txt)

crankyoldgit, 2021-09-09:
Awesome! Thanks for that. I've added it to the branch/PR. I'll merge it as soon as all the checks have finished.

closed 2021-09-09 by crankyoldgit

lucky62, 2021-09-09:
Just I am wondering why is the latest nibble before the Checksum always 5... 

22:00:57.250 -> 100059**5**4 5950001 4 OK
22:00:57.317 -> 108059**5**C 5950801 C OK
22:00:58.082 -> 100069**5**5 5960001 5 OK
22:00:58.182 -> 100069**5**5 5960001 5 OK
22:00:58.248 -> 108069**5**D 5960801 D OK
22:00:59.047 -> 100079**5**6 5970001 6 OK
22:00:59.147 -> 100079**5**6 5970001 6 OK
22:00:59.213 -> 108079**5**E 5970801 E OK
22:01:00.943 -> 10001C**5**3 5C10001 3 OK
22:01:01.043 -> 10001C**5**3 5C10001 3 OK
22:01:01.143 -> 10001C**5**3 5C10001 3 OK
22:01:01.209 -> 10801C**5**B 5C10801 B OK
22:01:02.142 -> 10002C**5**4 5C20001 4 OK

crankyoldgit, 2021-09-09:
> Just I am wondering why is the latest nibble before the Checksum always 5...

Dunno. I can only speculate. You could contact Arris/Commscope and find out.

commit 3a500449 (2021-09-06): Arris: Update gap & add some encoding tools.
* Change the inter-message gap per observations.
* Add helper routines:
  - `toggleArrisRelease()` for changing mesg status between a press & a release of a button.
  - `encodeArris()` to construct a valid ARRIS mesg from a `command` and a `release` status.
* Update & add Unit Tests accordingly.
* Update status of send & decode as they've been confirmed working.
* Extend decodeArris to provide release status (via `address`) and the likely `command`.

Fixes #1595
commit 92a4684c (2021-09-09): Arris: Add checksum validation to decode.
* Minor code style cleanup & potential optimisation.

Fixes #1595
PR #1598 (2021-09-09): Experimental support for Arris protocol.
* Add `sendArris()` & `decodeArris()`.
* Uses Manchester Encoding.
* Add send & decode unit tests.

Fixes #1595