crankyoldgit/IRremoteESP8266 #1607: Fujitsu ARJW2 Timer/Sleep not working

kARTechnology, 2021-09-19:
### Version/revision of the library used
3.3.0

### Describe the bug
Fujitsu ARJW2 Timer/Sleep not working

#### To Reproduce
    
  ac.begin();
  ac.setModel(ARJW2);
  ac.setCmd(kFujitsuAcCmdTurnOn);
  ac.setSwing(kFujitsuAcSwingOff);
  ac.setMode(kFujitsuAcModeCool);
  ac.setFanSpeed(kFujitsuAcFanAuto);
  ac.setTemp(27);  // 24C
  ac.setCmd(kFujitsuAcCmdTurnOn);
  ac.setSleepTimer(15);
  ac.send();

Model: 4 (ARJW2), Id: 0, Power: On, Mode: 1 (Cool), Temp: 27C, Fan: 0 (Auto), Command: N/A
no mention of timersleep

crankyoldgit, 2021-09-19:
It appears we only support Timers for the `ARRAH2E` model(s). See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/4b7dac7b0fb72025ba2b94dc690b7cb218a3093e/src/ir_Fujitsu.cpp#L912-L913

If you think your A/C model supports it, **you** will need to do some analysis to find out & document how it works for your particular model. See:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

kARTechnology, 2021-09-19:
I could do the research and get the data back to you in a while. thanks.

crankyoldgit, 2021-09-20:
> I could do the research and get the data back to you in a while. thanks.

Looking forward to it.

crankyoldgit, 2021-09-27:
Friendly chase up. How did you go?

kARTechnology, 2021-09-27:
> 
> 
> Friendly chase up. How did you go?
1h sleep

Library   : v2.7.18

Protocol  : FUJITSU_AC
Code      : 0x1463001010FC08303011013C08004A (120 Bits)
Mesg Desc.: Model: 2 (ARDB1), Id: 0, Power: On, Mode: 1 (Cool), Temp: 19C, Fan: 1 (High), Command: N/A
uint16_t rawData[243] = {3362, 1538,  492, 324,  466, 350,  490, 1140,  492, 326,  486, 1146,  490, 326,  466, 300,  516, 350,  466, 1164,  488, 1142,  468, 350,  466, 348,  466, 350,  466, 1164,  468, 1166,  466, 350,  466, 350,  466, 350,  466, 348,  466, 350,  466, 348,  466, 350,  466, 348,  466, 348,  468, 350,  466, 348,  466, 348,  468, 348,  468, 1164,  468, 348,  468, 348,  470, 346,  470, 346,  470, 344,  468, 346,  468, 346,  468, 1162,  468, 348,  468, 348,  468, 348,  466, 348,  468, 348,  466, 1164,  466, 1164,  468, 1164,  466, 1166,  466, 1166,  464, 1166,  466, 352,  464, 352,  464, 352,  464, 1168,  464, 352,  462, 352,  462, 354,  462, 352,  464, 352,  462, 354,  462, 354,  462, 354,  462, 1168,  464, 1168,  436, 380,  434, 380,  436, 380,  434, 382,  454, 360,  434, 380,  434, 1196,  436, 1196,  436, 380,  434, 380,  436, 1196,  436, 380,  434, 380,  434, 382,  434, 1196,  436, 380,  436, 380,  434, 380,  434, 1196,  436, 380,  436, 380,  434, 380,  436, 380,  434, 380,  436, 380,  434, 380,  436, 380,  434, 380,  436, 1196,  436, 1196,  436, 1194,  436, 1196,  436, 380,  436, 380,  434, 380,  436, 380,  434, 380,  434, 1196,  436, 380,  436, 380,  436, 380,  434, 380,  434, 380,  436, 380,  434, 396,  418, 382,  436, 380,  434, 380,  436, 382,  434, 380,  434, 380,  436, 1196,  436, 380,  436, 1196,  436, 380,  434, 380,  434, 1196,  436, 380,  434};  // FUJITSU_AC
uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x30, 0x11, 0x01, 0x3C, 0x08, 0x00, 0x4A};

1h 35m sleep
Timestamp : 000624.791
Library   : v2.7.18

Protocol  : FUJITSU_AC
Code      : 0x1463001010FC08303011015F080027 (120 Bits)
Mesg Desc.: Model: 2 (ARDB1), Id: 0, Power: On, Mode: 1 (Cool), Temp: 19C, Fan: 1 (High), Command: N/A
uint16_t rawData[243] = {3384, 1516,  490, 326,  490, 326,  490, 1140,  490, 326,  490, 1142,  490, 326,  490, 326,  490, 326,  490, 1140,  490, 1140,  490, 326,  490, 326,  490, 326,  490, 1140,  492, 1140,  490, 326,  490, 326,  490, 326,  490, 326,  490, 326,  490, 326,  490, 326,  490, 326,  490, 326,  488, 326,  490, 326,  488, 326,  490, 326,  488, 1142,  490, 326,  490, 326,  488, 326,  488, 326,  490, 326,  488, 326,  490, 326,  490, 1140,  490, 326,  490, 326,  488, 326,  490, 326,  488, 328,  488, 1142,  490, 1142,  490, 1142,  490, 1142,  490, 1142,  490, 1142,  490, 326,  488, 326,  490, 326,  488, 1142,  490, 326,  488, 328,  488, 326,  488, 328,  488, 326,  490, 328,  488, 326,  488, 328,  488, 1142,  490, 1144,  488, 326,  488, 328,  488, 328,  488, 328,  488, 326,  488, 328,  488, 1144,  488, 1142,  488, 328,  486, 328,  488, 1144,  488, 328,  488, 328,  486, 328,  486, 1144,  490, 328,  488, 328,  486, 328,  486, 1144,  488, 328,  486, 328,  486, 328,  488, 328,  486, 330,  486, 328,  486, 328,  486, 1144,  488, 1144,  488, 1146,  486, 1144,  488, 1144,  488, 328,  486, 1146,  486, 330,  486, 330,  486, 330,  486, 330,  486, 1144,  488, 330,  484, 330,  486, 330,  486, 330,  486, 330,  484, 332,  484, 330,  486, 330,  484, 330,  484, 330,  484, 332,  484, 330,  486, 1146,  486, 1146,  484, 1146,  486, 332,  484, 332,  484, 1146,  486, 330,  484, 332,  484};  // FUJITSU_AC
uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x30, 0x11, 0x01, 0x5F, 0x08, 0x00, 0x27};

kARTechnology, 2021-09-27:
reset/cancel timer

Timestamp : 000727.151
Library   : v2.7.18

Protocol  : FUJITSU_AC
Code      : 0x1463001010FC08303001010000009E (120 Bits)
Mesg Desc.: Model: 2 (ARDB1), Id: 0, Power: On, Mode: 1 (Cool), Temp: 19C, Fan: 1 (High), Command: N/A
uint16_t rawData[243] = {3298, 1600,  430, 388,  428, 388,  428, 1202,  430, 386,  428, 1202,  430, 386,  428, 388,  426, 388,  428, 1204,  430, 1202,  430, 386,  428, 388,  426, 388,  428, 1204,  430, 1202,  428, 388,  428, 386,  428, 388,  428, 388,  428, 388,  428, 388,  428, 388,  428, 386,  428, 390,  426, 388,  428, 388,  426, 388,  428, 388,  428, 1204,  428, 388,  428, 384,  436, 388,  420, 390,  428, 388,  428, 388,  426, 388,  430, 1202,  430, 386,  428, 388,  428, 388,  428, 386,  428, 388,  428, 1204,  430, 1202,  428, 1204,  430, 1202,  428, 1202,  430, 1202,  428, 388,  428, 388,  428, 388,  426, 1204,  430, 386,  428, 388,  428, 386,  428, 388,  428, 388,  428, 388,  426, 388,  428, 386,  428, 1204,  430, 1202,  428, 388,  428, 388,  428, 388,  428, 388,  428, 388,  428, 388,  428, 1202,  430, 1202,  430, 386,  428, 388,  428, 1206,  426, 388,  428, 388,  428, 388,  426, 388,  428, 388,  426, 388,  428, 388,  428, 1204,  428, 388,  428, 388,  428, 388,  428, 388,  428, 386,  428, 388,  428, 388,  430, 386,  428, 388,  428, 388,  428, 388,  428, 388,  426, 388,  428, 388,  428, 388,  428, 388,  428, 386,  428, 388,  428, 388,  428, 388,  426, 388,  428, 388,  428, 388,  426, 388,  428, 388,  428, 388,  428, 388,  428, 388,  428, 388,  428, 388,  428, 388,  426, 388,  428, 1204,  428, 1204,  428, 1204,  428, 1204,  428, 388,  428, 388,  428, 1204,  428};  // FUJITSU_AC
uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x30, 0x01, 0x01, 0x00, 0x00, 0x00, 0x9E};

crankyoldgit, 2021-09-28:
uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x30, 0x**1**1, 0x01, 0x**3C**, 0x0**8**, 0x00, 0x4A};  // 1 hour
uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x30, 0x**1**1, 0x01, 0x**5F**, 0x0**8**, 0x00, 0x27};  // 1h35m
uint8_t state[15] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFC, 0x08, 0x30, 0x30, 0x**0**1, 0x01, 0x**00**, 0x0**0**, 0x00, 0x9E};  // Cancel

`0x3C` is `60` in decimal. i.e 1 hour
`0x5F` is `95` in decimal. i.e 1 hour + 35 mins

You clearly need to capture a timer message greater than `0xFF` / `255` minutes (4 hours & 15 mins) to see where the rest of the timer is stored.

You should also work out why those other values change. Either (or both) could mean timer is enabled, or the type of timer etc.
e.g. Check if there are on & off timers, rather than sleep timers.

See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data for ways **_you_** can attack the analysis.

crankyoldgit, 2021-10-03:
@kARTechnology Friendly chase up.

kARTechnology, 2021-10-05:
Sorry, I replaced the unit with a Mitsubishi Heavy unit, now trying to
connect directly with AC via CNS connector in the PCB of the indoor unit to
communicate with esp.
When I get some free time I will capture the sleep and post it here.

crankyoldgit, 2021-12-01:
Closing this due to inactivity.

closed 2021-12-01 by crankyoldgit

jeremyvisser, 2023-05-14:
I ended up having the same problem, and realised the root cause: in sendAc() in src/IRac.cpp, we are not passing the sleep parameter. Specifically:

https://github.com/crankyoldgit/IRremoteESP8266/blob/03885a6853e67c6274a78003223e740fc01274f3/src/IRac.cpp#L3230C1-L3239

Notice how we are calling `fujitsu()` with no `send.sleep` parameter. But the function itself does accept `sleep`:

https://github.com/crankyoldgit/IRremoteESP8266/blob/03885a6853e67c6274a78003223e740fc01274f3/src/IRac.cpp#L1218C15-L1224

So I presume the fix is as simple as passing `send.sleep` to `fujitsu()` in `sendAc()`.

crankyoldgit, 2023-05-14:
Thanks for the additional information.
I'll look into that, however that was subtly different from the original reported problem/scenario, which didn't involve the `IRac` class.

crankyoldgit, 2023-05-14:
@jeremyvisser I believe the issue you have found is different one to the original issue reported here.
It's asking for "Sleep" to work/be reported for a model within a protocol where we don't have the support (yet) for it.

However, you did find a lapse the code. Yes, we should have been passing `sleep` for the models that do support it.
Hopefully that will be fix in PR #1992