crankyoldgit/IRremoteESP8266 #1616: Potential security issue

zidingz, 2021-09-24:
Hey there!

I belong to an open source security research community, and a member (@srikanthprathi) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

NiKiZe, 2021-09-24:
I might be wrong here, but seems that you are posting almost the same thing in many repos?
You can find the repo owners email in the git commits - SECURITY.md sure has it's place, but it can also be a hindrance, and out of date if just created. I will be against it in this repo/project.

JamieSlome, 2021-09-24:
Hello @NiKiZe - we request a SECURITY.md, so we can be sure that the person receiving the report has explicit write permissions to the repository.

We reach out via Github Issue in this way, as we want to follow a responsible disclosure process and ensure we send the security concerns to the right, and authorised individuals.

Let me know if you have any questions or further thoughts! 😊

NiKiZe, 2021-09-24:
That documentation is only valid if keept up to date, which it might not be even if it is there.
Not sure what security has anything to do with write permission.

Now 3 accounts are involved. Seems weird-ish.

crankyoldgit, 2021-09-24:
> I belong to an open source security research community

What is the name of the community? Got a link to something?

> Hello @NiKiZe - we request a SECURITY.md, so we can be sure that the person receiving the report has explicit write permissions to the repository.

@JamieSlome I definitely have explicit write permission (as the owner of the repo), @NiKiZe has write access too.

> Let me know if you have any questions or further thoughts! 😊

This is a _completely_ open project. Feel free to create a new issue (or reuse this one) for any security problems.
I've created a draft `Security.md` in PR #1617 Please have a read. It expresses my views I think.
Unless there is something implicitly confidential/sensitive in nature, I'm fine with it being public knowledge with in the repo and in the open so to speak.
e.g. If you're dealing with an undisclosed confidential vuln in some component we are using etc., or we are the root cause of some RCE in a major project (e.g. Tasmota), then sure then lets keep it under-wraps etc. If it's a vuln in one of our examples etc, then a normal bug is fine etc.

That said, sure, we'd love to know of & fix the vuln before it is widely publicised etc. e.g. Before you go making announcements etc.

FWIW, I've previously worked at a Nation-level CERT, and Multi-national IT/Internet company on their IT Security teams. Not bragging, just saying I'm fully aware of the things, topics, & issues involved and willing to assist etc.

GPG key is available if required, but nothing in this project/library really warrants that level of paranoia. (plus I'd have to dust off the passphrase ;-) 

Oh, and thanks for following "Responsible Disclosure" guidelines. Appreciated. Also happy to give appropriate attribution too.

JamieSlome, 2021-09-25:
@crankyoldgit:

> What is the name of the community? Got a link to something?

https://huntr.dev

> @JamieSlome I definitely have explicit write permission (as the owner of the repo), @NiKiZe has write access too.

https://huntr.dev/bounties/4da00a75-50dc-458b-acc6-cc216e1c854a/ is the actual report. We don't share even this when we create the issue, as it can be seen as breaking GitHub's Guidelines and Community rules (i.e. advertising).

> I've created a draft Security.md in PR #1617 Please have a read. It expresses my views I think.

Dropped a review on it!

crankyoldgit, 2021-09-25:
> https://huntr.dev/bounties/4da00a75-50dc-458b-acc6-cc216e1c854a/ is the actual report. We don't share even this when we create the issue, as it can be seen as breaking GitHub's Guidelines and Community rules (i.e. advertising).

@JamieSlome I looking at the proffered link _without logging in_, it looks like (from the context surrounding the blurred text) it is a case of a poor quality regex (i.e. CWE-1333 etc). I don't think we have any regex in the core of the library, or even in the examples we offer to users.
There are however regex in some of our internal tools we use when doing releases.
i.e.
https://github.com/crankyoldgit/IRremoteESP8266/blob/ca44fa33bbe872c8b646e6c7be4b270968268bd7/tools/scrape_supported_devices.py#L13-L21
And that tool (`scrape_supported_devices.py`) isn't even exposed to untrusted data.

As the "tools" directory is pretty much the only one with regex in/used by the code, it's of very low priority/importance in the scheme of things. Thus, I'm more than happy for your team to drop the details in a new (public) Issue on this repo.

Per [your FAQ](https://huntr.dev/faq/):
> I'm a maintainer, but don't want to sign up. What should I do?
>
> If you have been notified of a vulnerability disclosure against your repository, we will send a magic link to the elected e-mail address in your SECURITY.md. This magic link will allow you to view the private advisory, and validate/invalidate if necessary.
If you do not have a SECURITY.md in your repository, we will ping you the advisory information once you have added this.

Feel free to send the _magic_ link to myself & @NiKiZe.
Per our code review policy, the `Security.md` will only be merged once it's been through our [process](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Library-Maintainers-Guide#code-review). Thanks for your additional review, but I'll wait for @NiKiZe as he seems to have some thing to say on the issue.

crankyoldgit, 2021-09-25:
> And that tool (`scrape_supported_devices.py`) isn't even exposed to untrusted data.

Oh, that's incorrect now that I think about it. Someone could submit a PR with something evil in it. The worst that will happen is they get an RCE in a GitHub Actions container. Thankfully they (GitHub Actions) have no additional privileges above any other GitHub user w.r.t. this repo. So we are still fairly safe.

NiKiZe, 2021-09-25:
> Oh, that's incorrect now that I think about it. Someone could submit a PR with something evil in it. The worst that will happen is they get an RCE in a GitHub Actions container. Thankfully they [...]

I would be more worried about a code execution and running it on my local machine - but since we must review and approved any new contributors PRs this is probably low risk (even if we don't check for non visible chars)

crankyoldgit, 2021-09-25:
@zidingz @JamieSlome @srikanthprathi In the interest of saving time, as it seems GitHub's "privacy" email system doesn't seem to work as advertised, please feel free to email me directly at david@xyzzy.com.au

zidingz, 2021-09-28:
Thank you for your patience and understanding! 

An email should be with you in an hour.

closed 2021-09-28 by zidingz

crankyoldgit, 2021-09-28:
I'll keep any eye out for it.

crankyoldgit, 2021-09-29:
@zidingz I haven't seen any email to the above address from you or your org yet.

crankyoldgit, 2021-09-29:
FYI, the PR has been merged, the repo now as a Security.md that you can read/reference etc.

crankyoldgit, 2021-09-29:
Confirming I got the email with the magic link.

PR #1619 (2021-09-29): Fix potential security issue with `scrape_supported_devices.py`
**Note**: The tool in question is used internally by library developers to automatically generate documentation prior to a release and to warn when files are missing the required documentation.

* Update Regex used to eliminate potential denial of service. Deemed a **_Low_** severity & impact threat.
  - Does **NOT** require or necessitate a new version/release of the library.
  - The tool is only used after review of code/PR etc which would already be very suspicious.
  - The tool is not something a user would typically use or ru [... cut]
commit d2b52085 (2021-09-29): Fix potential security issue with `scrape_supported_devices.py` (#1619)

Note: Tool is used internally by library developers to automatically generate documentation prior to a release and to warn when files are missing the required documentation.

* Update Regex used to eliminate potential denial of service.
  - Deemed a Low severity & impact threat.
  - Does NOT require or necessitate a new version/release of the library.
  - The tool is only used after review of code/PR etc which would already be very suspicious.
  - The tool is not something a user would typically use or run. (i.e. Internal [... cut]