crankyoldgit/IRremoteESP8266 #1639: String manipulations giving grief. As usual.

jimmys01, 2021-10-15:
### Version/revision of the library used
Latest and greatest self compiled build

### Describe the bug
The Compiler is nagging about 
`IRutils.cpp:98:50: error: conversion from 'const __FlashStringHelper* const' to 'const StringSumHelper' is ambiguous
     return kDashStr + uint64ToString(-input, base);`
My fix ( I do not know if is the best solution)
`return String(kDashStr) + uint64ToString(-input, base);`

Also 
```
ir_Coolix.cpp:614:57: error: operands to ?: have different types 'const __FlashStringHelper* const' and 'StringSumHelper'
           ? kOffStr : uint64ToString(getSensorTemp()) + 'C', kSensorTempStr);
```
My fix (also dont know if this is the best solution)
 `? kOffStr : String(uint64ToString(getSensorTemp()) + 'C'), kSensorTempStr);`

#### Expected behaviour
To compile.

Keep up the excelent work
Mutsos Gracias

crankyoldgit, 2021-10-16:
I'm asking because this isn't showing up in our automated build system (which is PlatformIO).

jimmys01, 2021-10-16:
The IDE is Visual Studio with PlatformIO

```
Processing nodemcuv2 (board: nodemcuv2; platform: espressif8266; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.3) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
```

crankyoldgit, 2021-10-16:
Oh, and I'm using the current _master_ branch [head](https://github.com/crankyoldgit/IRremoteESP8266/commit/df27ef5acf607c16438cfeda6caf4175219744fc), not that I think it matters. Just for reference.

jimmys01, 2021-10-16:
> Looking through my build and your environment, it looks like this may have been an issue fixed in the more recent Espressif 8266 platform (i.e. 3.0.2 vs your 2.6.3) and framework-arduinoespressif8266 3.30002.0 (3.0.2 vs your 2.7.4)

@TD-er

TD-er, 2021-10-16:
I don't know why that "custom IR" build env. is still using 2.6.3
As soon as this fix of yours is done, we may need to look into why that one is not set to 2.7.4 at least.

Still working on 3.0.x support as it had a lot more to change than I would have expected it to be.

crankyoldgit, 2021-10-16:
My current leading candidate is: https://github.com/esp8266/Arduino/pull/7781 where they fixed/corrected the _+_ overload and rework etc.
i.e. 3.0.0

TD-er, 2021-10-16:
Yep, looks like a good candidate for what we're seeing here.

Concatenating directly to flashstring items used to be quite problematic in the past.
Glad to know that will be less of an issue in the future :)

TD-er, 2021-10-16:
With the suggested changes made by @jimmys01 the build was successful.
See [GH actions](https://github.com/letscontrolit/ESPEasy/actions/runs/1346114695) of that build.

crankyoldgit, 2021-10-16:
> With the suggested changes made by @jimmys01 the build was successful. See [GH actions](https://github.com/letscontrolit/ESPEasy/actions/runs/1346114695) of that build.

Thanks.

crankyoldgit, 2021-11-19:
FYI, the changes mentioned above have now been included in the new [v2.8.0 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.0) of the library.

TD-er, 2021-11-19:
Thanks for the heads up.
I guess @jimmys01 is already preparing a new PR for ESPEasy then :)

commit 9f79edde (2021-10-16): Fix compilation issue when using old 8266 Arduino Frameworks. (#1640)

* Try to add backward compatibility to older 8266 Arduino frameworks
* Add code comments to document why this has been done.
* Doesn't seem to be required as of 3.0.0 of the ESP8266 Arduino Framework/platform code.
  - The fix doesn't seem to hurt either way. 🤷 

Fixes #1639