crankyoldgit/IRremoteESP8266 #1669: Setting "Beep" doesn't work on Samsung AR12TXEAAWKNEU

xorbital, 2021-11-10:
### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
master (817517cc367c51cd2a092695fdc865505cba45c5)

### Describe the bug
Setting "Beep: On" doesn't toggle the beep state on my unit.

#### To Reproduce
Set "Beep: On" when sending a Samsung IR message. 

#### Example code used
N/A

#### Expected behaviour
The code should spit out a valid IR message that toggles the beep state.

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
I can upload a dump if you need it - however I've figured out what's causing the issue and fixed it in #1668 . Also I don't want to raise your blood pressure by posting VS1838b dumps.

### What brand/model IR demodulator are you using?
I think it's called B14Y, however somehow the results from that are even worse than from my VS1838b

_If VS1838b: That is likely your problem. Please buy a better one & try again. Reporting a capture or decode issue when you use one of these is effectively wasting our & your time. See the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator)_
Yes, I agree, it's a piece of ****, however it's been working fine with 30% tolerance for all captures.
Also, I've put the whole circuit in a long, dark box, gave it a few headpats and the results are good.

### Circuit diagram and hardware used (if applicable)
N/A

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes_

### Has this library/code previously worked as expected for you?
_No, the beep never worked_

### Other useful information
_More information is always welcome. Be verbose._
When Sending "Beep: On", the last nibble will be set to 2 (0010), while it's required to be set to 4 (0100). This is fixed in #1668 , however I have no knowledge of C++, so I'll need you to take a look at my PR to see if I just broke the whole thing / broke compatibility with other models.
Using the fix in the PR makes my unit beep like it's supposed to, exactly like the original remote.

crankyoldgit, 2021-11-11:
@xorbital You've mentioned this is (`beep`) is a toggle.
Just confirming: Does that mean when it is present in a message it will change what ever the current setting is on the A/C?
i.e. If the bit is zero in a message, everything stays the same beep-wise. A single message with the beep-bit set will change on->off, and off->on.

If so, I'll need to make some additional changes other than your PR #1668

crankyoldgit, 2021-11-11:
Assuming it is a toggle, PR #1671 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1669 should handle it better. Let me know etc. If it isn't a toggle, then we can ignore #1671

xorbital, 2021-11-11:
Yes, exactly, it does behave like a toggle. IIRC that's also true for the CLEAN setting.

PR #1671 (2021-11-11): SamsungAc: Change `beep` setting to a toggle.
* Use the `beep` bit as a toggle setting.
* Update `IRSamsungAc` & `IRac` classes.
* Revise unit tests accordingly.

Fixes #1669
commit 291f13ba (2021-11-11): SamsungAc: Change `beep` setting to a toggle. (#1671)

* Use the `beep` bit as a toggle setting.
* Update `IRSamsungAc` & `IRac` classes.
* Revise unit tests accordingly.

Fixes #1669
closed 2021-11-11 by crankyoldgit