crankyoldgit/IRremoteESP8266 #1670: Add support for Airton AC

jonesy2, 2021-11-10:
Hi,
Hoping to contribute to this great project.
I'm adding all that I have, and will add more/ do more test as required.
The AC did not respond to any codes in the library version v2.7.18

The AC is AIRTON (bought in France). There is no model number, but it is the 2500W version. The remote has no part number, but the chip underneath the lcd is SH77P1652 SINO WEALTH. On the inside of the battery cover is the code RD1A1

Following the I have captured the usual modes (hot/cool etc) at all temperatures. I don't use the timers, or change away from auto fan etc.
The spreadsheet is here 
[https://docs.google.com/spreadsheets/d/1l-qo6LPPkBYUGSJ65j3-j3iQJg7lGAo96AzR4s6NDOA/edit?usp=sharing](url)
56 bit.
The  auto_analyse_raw_data.py code produced code that worked with the slight modification of 56 bit, rather than 57 bit that was suggested.

Manual here [https://cdn.shopify.com/s/files/1/2600/0244/files/409730-409731-409732-409733_MANUAL_INSTRUCTION_V01052021.pdf?v=1619861501](url)

[image]

crankyoldgit, 2021-11-12:
Hey, we can probably add this, but you need to follow: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol#obtaining-the-data

Without some of the raw data etc, we can't be sure we can decode the underlying signal. We need it for testing

Also, please update to 2.7.20 (or the `master` branch) as the protocol may have already been added.

jonesy2, 2021-11-15:
I've updated to 2.7.20, and the protocol hasn't been added.
I've updated the spreadsheet to show uint16_t rawData I used for testing (the 'all' tab)
I think that's it, but happy to do more if you point me in the right direction

crankyoldgit, 2021-11-17:
Thanks for the raw data. That allowed me to build basic support for your protocol.
You'll need to recollect, recapture, & reanalyse all of your data because after analysis of the temperature data, the protocol uses LSBF bit ordering. Your existing spreadsheet has MSBF ordering.

To help your analysis:
The first byte of data will be the checksum byte. (0x**5E**1400090C11D3) _You will need to work out how that is calculated_
The temp is stored in this nibble (0x5E14000**9**0C11D3). i.e. That nibble + 16 gives the temp in Celsius. e.g. `0x9` + 16 = `25C`

Once you've mapped out all the bits, what they mean, how they work, what the values mean in terms of modes/swing positions etc etc, and have it all nicely & clearly documented in a spreadsheet or some document, I'll re-look at this and code up the detailed support (i.e. setting modes, power, temp etc).

For now, it should decode to a 56-bit hex number in `IRrecvDumpV2` or `V3` etc if you download & compile using branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/airton_basic / PR #1681

jonesy2, 2021-11-17:
Thank you for the update.
I have compiled PR #1681 and it correctly decodes the remote

`IRrecvDump is now running and waiting for IR input on Pin 14
Timestamp : 000140.459
Library   : v2.7.20

Protocol  : AIRTON
Code      : 0x591400060C11D3 (56 Bits)
uint16_t rawData[115] = {6648, 3310,  408, 1258,  406, 1258,  408, 426,  434, 398,  406, 1258,  406, 426,  404, 1258,  406, 1258,  406, 1258,  408, 426,  434, 398,  406, 426,  406, 1258,  408, 426,  406, 426,  434, 398,  406, 426,  406, 426,  406, 1236,  430, 1256,  408, 424,  406, 426,  406, 426,  432, 398,  406, 426,  406, 1258,  406, 1258,  408, 426,  406, 426,  436, 398,  404, 426,  404, 426,  406, 426,  406, 426,  432, 398,  406, 426,  406, 426,  406, 426,  406, 428,  404, 426,  406, 426,  406, 426,  406, 1256, 
 408, 426,  406, 1258,  408, 426,  404, 426,  406, 426,  406, 1258,  408, 426,  406, 426,  406, 1258,  408, 1256,  408, 426,  406, 1258,  408, 426,  406};  // AIRTON 591400060C11D3
uint64_t data = 0x591400060C11D3;`

I've done a new spreadsheet LSBF, which is here [https://docs.google.com/spreadsheets/d/1Kpq7WCkh85heLnTQGlwUfCR6eeu_vfBHvhii8wtP4LU/edit?usp=sharing](url)
There were a few mistakes which I've fixed.
The old link is still valid.

The checksum escapes me still.  You can see in state[0] that is works for some codes, but not all.

crankyoldgit, 2021-11-17:
Thanks for confirming receiving/decoding works. Have you tried transmission/sending also?

> The checksum escapes me still. You can see in state[0] that is works for some codes, but not all.

No idea also. I tried a few things, but it alludes me too. Until you've worked that out, there is next to no point on me starting the coding for detailed support.

Try also playing with the data in MSBF format too, ya never know.

jonesy2, 2021-11-18:
Transmission is fine both rawData and irsend.sendAirton.
I can't help feeling the checksum is simple, but after spending a couple of days I decided to save my sanity and stuck with sending codes. 
Thank you for your help. Hopefully someone else will want Airton support and will crack it.

crankyoldgit, 2021-11-19:
Thanks for the confirmation. Sorry about the checksum or what ever it is etc.

Marking this issue closed.

PtilopsisLeucotis, 2021-11-19:
I think I have solved the riddle with Airton checksum.
As we know, Airton packet structure is `|CS|D0|D1|D2|D3|D4|D5|`, where `CS` - checksum byte, `D0-D5` - data bytes.
Checksum calculation formula is `CS = ((0x7F - (D0+D1+D2+D3+D4+D5)) % 0x100) ^ 0x2C`. 
For example, test this formula against sample packet `0xDB9800000C11D3`: `CS = ((0x7F - (0x98+0x00+0x00+0x0C+0x11+0xD3)) % 0x100) ^ 0x2C = 0xDB`. As expected, we have same value `0xDB`. I have test this formula against data, listed on the https://docs.google.com/spreadsheets/d/1Kpq7WCkh85heLnTQGlwUfCR6eeu_vfBHvhii8wtP4LU/edit?usp=sharing on the tab `all`, everything works correct, except `0x580400091A11D4` in cell `C101`, where there is likely a typo, probably is `0x580400091A11D3` is correct value, because `D4` at the end of packet never seen anywhere else.

crankyoldgit, 2021-11-19:
FYI, the code changes mentioned above have now been included in the new [v2.8.0 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.0) of the library.

jonesy2, 2021-11-20:
@PtilopsisLeucotis Great work. My sanity is saved!
I've just re-tried the remote a few times for cell `C101`, and `D3` is the correct value.
I've upated the spreadsheet.

PtilopsisLeucotis, 2021-11-20:
I'm like to solve such riddles! And it is very good, that your sanity is saved 😀

crankyoldgit, 2021-11-21:
FYI, We still need data from you for the Eco, Health, & Turbo buttons.

The data in the the byte next to the checksum needs some more explanation too.
e.g. Why is the first nibble `9` (_LED on_) & `1` (_LED off_) for heat when the power is on, where all other modes (inc heat power off) are `8` (_LED on_) & `0` (_LED off_)?

crankyoldgit, 2021-11-21:
@jonesy2 I've created a branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/airton_detailed) with experimental detailed support.
Can you please test it with IRrecvDumpV2+ and with sending to your A/C via the `IRAirtonAc` class to see what works, what doesn't, and what we still need to add?

jonesy2, 2021-11-21:
Added  data to the spreadsheet for the Eco, Health, Sleep & Turbo buttons.
There is an anomaly for the LED lit/not lit as it affects 2 bits in different nibbles.  It can probably be ignored in the lower nibble (notes in the spreadsheet). 
I've added notes how different options affect operation eg can't have sleep in Auto.
I've tested the branch  (https://github.com/crankyoldgit/IRremoteESP8266/tree/airton_detailed)  IRrecvDumpV2 and it works accurately for receive eg

```
Timestamp : 000126.900
Library   : v2.8.0

Protocol  : AIRTON
Code      : 0x268800000911D3 (56 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Fan: 0 (Auto), Temp: 16C, Swing(V): Off, Light: On
uint16_t rawData[115] = {6602, 3364,  384, 1282,  384, 1282,  384, 450,  382, 450,  382, 1282,  384, 450,  438, 1226,  384, 1282,  384, 1282,  384, 450,  382, 450,  436, 398,  382, 1282,  384, 450,  438, 396,  382, 450,  382, 1282,  384, 448,  384, 448,  384, 1282,  384, 450,  432, 400,  382, 450,  382, 450,  382, 450,  382, 450,  434, 398,  384, 450,  382, 448,  384, 450,  382, 450,  384, 450,  382, 450,  382, 450,  382, 450,  382, 450,  382, 450,  438, 392,  382, 450,  382, 450,  382, 450,  382, 450,  436, 396,  382, 1282,  384, 450,  382, 450,  382, 450,  382, 1282,  384, 448,  384, 1282,  382, 1282,  384, 450,  384, 450,  382, 1280,  440, 394,  382, 450,  382};  // AIRTON 268800000911D3
uint64_t data = 0x268800000911D3;
```
Good work. Thank you.
I'll try and test sending tomorrow.

crankyoldgit, 2021-11-22:
@jonesy2 Thanks for the prompt testing and new information & analysis. I've coded up the changes so it meets your descriptions. Please re-download the latest version of that branch and let me know how capture and sending go for you.
Also PR #1688 now exists to get these changes merged into the `master` branch.
It will get merged when you confirm it's all working as expected.

jonesy2, 2021-11-22:
New branch tested for receive and send.  All modes/functions work as expected. However, it's not possible to say whether the AC was in ECO mode, sleep or health as there is no indication from the unit. However, it bleeped which it won't do when the CRC is bad.
```
IRrecvDumpV2.ino
Timestamp : 000465.458
Library   : v2.8.0

Protocol  : AIRTON
Code      : 0xD9CA0102D911D3 (56 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Fan: 5 (Maximum), Temp: 18C, Swing(V): On, Econo: Off, Turbo: On, Light: On, Health: On, Sleep: On
uint16_t rawData[115] = {6620, 3344,  404, 1262,  404, 1264,  402, 432,  400, 430,  404, 1260,  406, 428,  404, 1260,  406, 1262,  428, 1236,  404, 430,  404, 428,  402, 432,  400, 1260,  404, 432,  400, 430,  402, 454,  378, 1260,  404, 430,  402, 430,  400, 1262,  404, 1262,  404, 430,  402, 1262,  404, 1262,  404, 430,  404, 1260,  406, 430,  400, 456,  378, 430,  400, 430,  404, 430,  400, 430,  402, 1262,  404, 454,  378, 430,  402, 430,  402, 430,  398, 434,  400, 430,  402, 430,  400, 432,  400, 1262,  404, 430,  402, 1262,  404, 432,  400, 432,  400, 1262,  404, 1260,  402, 1264,  404, 432,  398, 432,  400, 1262,  404, 1262,  404, 428,  404, 1262,  404, 1260,  404};  // AIRTON D9CA0102D911D3
uint64_t data = 0xD9CA0102D911D3;
```

I'll start modifying my AC control app to use the  `IRAirtonAc` class commands  like ac.off(); ac.setFan(kAirtonFanAuto); etc
A big thank you to the team.

crankyoldgit, 2021-11-22:
Thanks for confirming it works. If there are any problems, please create a new issue.

commit 5f6e3daa (2021-11-19): [AIRTON] Update working status

Has been confirmed working by the user.
Fixes #1670
commit c1cb3b79 (2021-11-22): Airton: Improve detailed support.
* Re-work Light/LED control bits.
* Add support for:
  - Econo (Eco) in Cool only
  - Turbo
  - Health (Filter)
  - Sleep (Not in Fan or Auto, chaning modes removes it)
* Add & update unit tests accordingly.
* Add new features to `IRac` class.

Fixes #1670
PR #1688 (2021-11-22): Airton: Add detailed A/C support.
* Add checksum verification to `decodeAirton()`.
* Add detailed support for Airton AC's via `IRAirtonAc` class.
  - Power
  - Temp
  - Mode
  - Fan Speed
  - SwingV
  - LED/Light/Display
  - Health/Filter
  - Turbo
  - Econo/Eco
  - Sleep
* Add `AIRTON` to `IRac` common API.
* Add unit test coverage for new changes.

H/T to @PtilopsisLeucotis for working out the checksum alg. Kudos!

Fixes #1670
commit c657b27f (2021-11-22): Airton: Add detailed A/C support. (#1688)

* Add checksum verification to `decodeAirton()`.
* Add detailed support for Airton AC's via `IRAirtonAc` class.
  - Power
  - Temp
  - Mode
  - Fan Speed
  - SwingV
  - LED/Light/Display
  - Health/Filter
  - Turbo
  - Econo/Eco
  - Sleep
* Add `AIRTON` to `IRac` common API.
* Add unit test coverage for new changes.

H/T to @PtilopsisLeucotis for working out the checksum alg. Kudos!

Fixes #1670
closed 2021-11-22 by crankyoldgit