crankyoldgit/IRremoteESP8266 #1675: IRSamsungAc::convertFan doesn't work for kMax Fan Speed

xorbital, 2021-11-11:
### Version/revision of the library used
Latest master (d896068)

### Describe the bug
I was testing the new changes to the Samsung_AC protocol with a freshly compiled Tasmota(IRhvac, this is using irAC) when I noticed that setting the fan speed to Turbo/Max doesn't work - the Fan Speed gets set to Auto instead. All other fan speeds work.
Something seems to go wrong when converting the FanSpeed from irAC to irSamsungAC.

#### To Reproduce
1. Send a message with ControlSamsunAC.ino with Turbo Fan Speed
2. Send a message with CommonACControl.ino with Max Fan Speed
3. Compare the results, the Nibble holding the Fan Speed is different

#### Example code used
##### First Step
```  
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRutils.h>

const uint16_t kIrLed = 4;  // The ESP GPIO pin to use that controls the IR LED.
IRac ac(kIrLed);  // Create a A/C object using GPIO to sending messages with.

void setup() {
  Serial.begin(115200);
  delay(200);
  ac.next.protocol = decode_type_t::SAMSUNG_AC;  // Set a protocol to use.
  ac.next.model = 1;  // Some A/Cs have different models. Try just the first.
  ac.next.mode = stdAc::opmode_t::kCool;  // Run in cool mode initially.
  ac.next.celsius = true;  // Use Celsius for temp units. False = Fahrenheit
  ac.next.degrees = 25;  // 25 degrees.
  ac.next.fanspeed = stdAc::fanspeed_t::kMax;
  ac.next.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
  ac.next.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
  ac.next.light = false;  // Turn off any LED/Lights/Display that we can.
  ac.next.beep = false;  // Turn off any beep from the A/C if we can.
  ac.next.econo = false;  // Turn off any economy modes if we can.
  ac.next.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
  ac.next.turbo = false;  // Don't use any turbo/powerful/etc modes.
  ac.next.quiet = false;  // Don't use any quiet/silent/etc modes.
  ac.next.sleep = -1;  // Don't set any sleep time or modes.
  ac.next.clean = false;  // Turn off any Cleaning options if we can.
  ac.next.clock = -1;  // Don't set any current time if we can avoid it.
  ac.next.power = false;  // Initially start with the unit off.
}

void loop() {
      ac.next.fanspeed = stdAc::fanspeed_t::kLow;
      ac.sendAc();  // Have the IRac class create and send a message.
      delay(2000); 
      
      ac.next.fanspeed = stdAc::fanspeed_t::kMedium;
      ac.sendAc();  // Send the message.
      delay(2000);
      
      ac.next.fanspeed = stdAc::fanspeed_t::kHigh;
      ac.sendAc();  // Send the message.
      delay(2000);
      
      ac.next.fanspeed = stdAc::fanspeed_t::kMax;
      ac.sendAc();  // Send the message.
      delay(2000);
  Serial.println("Starting from the begining again ...");
}
```
##### Second Step
```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
[... 155 lines cut ...]
Timestamp : 000716.526
Library   : v2.7.20

Protocol  : SAMSUNG_AC
Code      : 0x02920F000000F001A2FE71901FF0 (112 Bits)
Mesg Desc.: Power: On, Mode: 1 (Cool), Temp: 25C, Fan: 7 (Turbo), Swing(V): Off, Swing(H): Off, Beep: Off, Clean: Off, Quiet: Off, Powerful: Off, Econo: Off, Breeze: Off, Light: On, Ion: Off
uint16_t rawData[233] = {670, 17878,  3070, 8880,  646, 414,  610, 1410,  612, 412,  610, 416,  612, 412,  610, 414,  612, 412,  612, 412,  610, 416,  610, 1412,  612, 416,  612, 442,  584, 1414,  612, 422,  606, 416,  610, 1412,  612, 1410,  612, 1410,  610, 1410,  610, 1412,  610, 414,  612, 414,  610, 414,  612, 438,  584, 432,  596, 442,  584, 422,  606, 416,  612, 416,  610, 416,  610, 416,  612, 416,  612, 414,  612, 436,  586, 416,  612, 414,  612, 412,  612, 414,  610, 440,  586, 414,  612, 414,  610, 442,  588, 414,  612, 416,  612, 416,  610, 442,  584, 416,  610, 416,  612, 414,  612, 412,  610, 438,  588, 414,  610, 1410,  612, 1408,  610, 1410,  610, 1410,  612, 2866,  3042, 8884,  638, 1412,  610, 440,  588, 416,  610, 442,  586, 442,  586, 416,  610, 416,  610, 418,  612, 414,  612, 1408,  612, 414,  610, 430,  596, 414,  610, 1410,  612, 440,  586, 1412,  612, 414,  612, 1410,  612, 1412,  612, 1412,  612, 1438,  584, 1412,  614, 1428,  592, 1414,  612, 1408,  612, 414,  610, 416,  610, 414,  612, 1408,  612, 1408,  614, 1408,  610, 440,  586, 416,  612, 416,  612, 442,  586, 416,  610, 1412,  612, 414,  612, 440,  586, 1412,  612, 1408,  612, 1410,  612, 1410,  612, 1424,  598, 1408,  612, 414,  610, 414,  610, 414,  612, 416,  612, 416,  610, 416,  612, 414,  612, 1410,  610, 1412,  612, 1436,  586, 1412,  612};  // SAMSUNG_AC
uint8_t state[14] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xA2, 0xFE, 0x71, 0x90, 0x1F, 0xF0};
```

### What brand/model IR demodulator are you using?
It says "LFN", I can't find the exact model. I'm using [this device](https://templates.blakadder.com/ytf_ir_bridge.html) to capture the codes. I'm sure it's not a VS1838b.

### Circuit diagram and hardware used (if applicable)
See link above

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes_

### Has this library/code previously worked as expected for you?
_No, this specific thing didn't_

### Other useful information

I've looked at [ir_Samsung.cpp](https://github.com/crankyoldgit/IRremoteESP8266/blob/d8960688ce63772768055e0ea2132ad311d56683/src/ir_Samsung.cpp#L835) and found the spot where the fan speed is converted, so I changed the `switch` to return `kSamsungAcFanTurbo` in all cases - now all of them are set to auto, so there's no issue in this part, it needs to be somewhere else.

crankyoldgit, 2021-11-12:
I tracked this down to `setPowerful()`:
https://github.com/crankyoldgit/IRremoteESP8266/blob/a727486f7c0fdf98746ba0a17396eb63fe6b13d7/src/ir_Samsung.cpp#L648-L649

In short, `setTurbo(false)` deliberately removes that fan speed because it isn't supposed to be available in normal operation afaik.
Defaulting to "auto" is just the best guess at what the previous non-turbo fan speed was in use. i.e. it's a safe default.

Per your code:
>       ac.next.fanspeed = stdAc::fanspeed_t::kMax;
>       ac.sendAc();  // Send the message.

In order to unlock the `kSamsungAcFanTurbo` fan speed, it needs to be in `turbo` mode per se.
e.g.
```c++
  ac.next.turbo = true;
  ac.sendAc();  // Send the message.
```

I can probably change the conversion routines, to make `stdAc::fanspeed_t::kMax` <=> `kSamsungAcFanHigh` if you think that will help with the confusion.

xorbital, 2021-11-12:
Ah, I see.
Well, I could send a picture of the remote, and maybe my unit is a bit different, so I could just change the code on my end and recompile.
For me, pressing the Fan buttion cycles through the speeds, them being Auto, then 1/4 bars, 2/4 bars, 3/4 bars and then (nothing really special about it) 4/4 bars.
"Turbo mode" or "Fast mode" is only available in Heat/Cool mode and sets the temp. to 16°C (Cool Mode) or 30°C (Heat Mode) for 30min and then automatically falls back to the temperature that was transmitted, so it's just for initially cooling/heating the room quickly.

Maybe other units are different with the turbo fan speed only available while in turbo mode, that's not the case for me though, it's just another fan speed setting

crankyoldgit, 2021-11-12:
So, is the highest "normal" (non-turbo) speed for the fan the `kSamsungAcFanTurbo` value?
https://github.com/crankyoldgit/IRremoteESP8266/blob/a727486f7c0fdf98746ba0a17396eb63fe6b13d7/src/ir_Samsung.h#L174-L179
Can you achieve that speed from the real remote? If so, then I can undo that bit of logic.

xorbital, 2021-11-12:
Yes, exactly, I might have been unclear before, Turbo/Fast **mode** is to heat/cool quickly, it sets temp high/low and the fan to 4/4.
The Turbo **fan** speed is nothing special, just the highest fan speed [4/4] you can set at any mode (excl. dry mode).
I've removed the previously mentioned code and I can now control my A/C almost perfectly, all fan speeds are at my disposal now via irAC

crankyoldgit, 2021-11-13:
PR #1678 should fix it as discussed.

PR #1678 (2021-11-13): [SAMSUNG_AC] Highest fan speed is available without Powerful setting.
Per user feedback, it seems the highest fan speed is just a normal fan speed, not locked behind `setPowerful(true)` as previous thought.

Already confirmed working.

Fixes #1675
commit 42ebfd3e (2021-11-13): [SAMSUNG_AC] Highest fan speed is available without Powerful setting. (#1678)

Per user feedback, it seems the highest fan speed is just a normal fan speed, not locked behind `setPowerful(true)` as previous thought.

Fixes #1675
closed 2021-11-13 by crankyoldgit