crankyoldgit/IRremoteESP8266 #1691: Is it possible to use remote control module driver by esp-idf

Nakul93, 2021-11-27:
Has anyone explored remote control driver which decodes nec and rc-5 protocols. Can that be ported to arduino.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#_CPPv410rmt_configPK12rmt_config_t

crankyoldgit, 2021-11-27:
Yes, it can be ported/re-coded to use those features of the ESP32. Someone will need to do it, because it's not high on my list of (personal) priorities.

i.e. You're welcome to do it.

If you want to use those features, you might want to look at these libraries instead, they _might_ use them. I don't know.

- https://github.com/Arduino-IRremote/Arduino-IRremote
- https://github.com/ukw100/IRMP

closed 2021-12-01 by crankyoldgit

NiKiZe, 2021-12-01:
So duplicate of #1575 #1368 #1329 #1328 #1223

Nakul93, 2021-12-01:
Checkout this code for esp-idf, it does the basic ir receive (raw data).
 FYI: I have only tested it using 2 different remotes(38khz). Worked fine

`#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "esp_err.h"
#include "esp_log.h"
#include "driver/rmt.h"
#include "soc/rmt_reg.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include <math.h>
#include "ir_tools.h"

#define INCLUDE_vTaskDelay 1

#define rrmt_item32_tIMEOUT_US  9500   /*!< RMT receiver timeout value */

// Struct for rmt
typedef struct {
	uint32_t* num;
	int raw_len;
	bool display;
} ir_params;

volatile ir_params params; 

// structure used to initialize RMT inputs
// NOTE: tag is used on the monitor outputs to distinguish between channels
struct rmt_rx_inits {
	rmt_channel_t  channel;
	gpio_num_t     gpio_num;
	uint8_t        clk_div;
	uint8_t        mem_block_num;
	bool           config_filter_en;
	uint8_t        config_filter_tics_thresh;
	uint16_t       config_idle_threshold;
	char           tag[10];
}rx_inputs[1] = {
	{ 0, 18, 80, 7, false, 2, 50000, "Ch0" }
};

// initialize RMT receive channels
void rx_channels_init() {
    rmt_config_t rmt_rx;
    size_t i;
    size_t len = sizeof(rx_inputs) / sizeof( rx_inputs[0] );
    
    for ( i=0; i<len; i++ ) {
		rmt_rx.channel       = rx_inputs[i].channel;
		rmt_rx.gpio_num      = rx_inputs[i].gpio_num;
		rmt_rx.clk_div       = rx_inputs[i].clk_div;
		rmt_rx.mem_block_num = rx_inputs[i].mem_block_num;
		rmt_rx.rmt_mode      = RMT_MODE_RX;
		
		rmt_rx.rx_config.filter_en           = rx_inputs[i].config_filter_en;
		rmt_rx.rx_config.filter_ticks_thresh = rx_inputs[i].config_filter_tics_thresh;
		rmt_rx.rx_config.idle_threshold      = rx_inputs[i].config_idle_threshold;
		
		rmt_config(&rmt_rx);
		rmt_driver_install(rx_inputs[i].channel, 2000, 0);
	}
}

/* Converts the RMT level, duration into a positive or negative integer
 * Compatible with the ESP32-RMT-server application
 * Note: most IR receivers have active-low outputs, where the
 *   ESP32-RMT-server application has active-high oututs
 * This function inverts the RMT receive level so the text output is
 *   compatible with ESP32-RMT-server application
 */
uint32_t dur( uint32_t level, uint32_t duration ) {
	if ( level == 0 ) { return duration; }
	else { return 1.0 * duration; }
}

// RMT receiver task
static void rmt_example_nec_rx_task() {
	size_t num_channels = sizeof(rx_inputs) / sizeof( rx_inputs[0] );
	size_t c, i;
    size_t rx_size = 0;
	volatile int j=0;
    rmt_item32_t* items = NULL;

	params.num=(uint32_t*) malloc(1024 * sizeof(uint32_t));

    RingbufHandle_t rb;
	
    // start receiving IR data
    for ( c=0; c<num_channels; c++ ) {
		rmt_rx_start(rx_inputs[c].channel, 1);
	}
    
    // loop forever
    while (1) {
		// check each receive channel
		for ( c=0; c<num_channels; c++ ) {
			// get the ring buffer handle
			rmt_get_ringbuf_handle(rx_inputs[c].channel, &rb);
			
			// get items, if there are any
			items = (rmt_item32_t*) xRingbufferReceive(rb, &rx_size, 1000);
			if(items) {
				
				params.display=true;
				params.raw_len=rx_size/2;
				
				// print the RMT received durations to the monitor
				printf( "  %s received %i items\n", rx_inputs[c].tag, rx_size/4 );
				for ( i=0; i<rx_size/4; i++ ) {

					params.num[j]=items[i].duration0;	
					
					// Stop capturing if we get zero(end of transmission)
					if(items[i].duration1!=0)	{
						params.num[j+1]=items[i].duration1;	

					}else break;
					j=j+2;
					
				}
				
				printf("\n");
				
				
				// free up data space
				vRingbufferReturnItem(rb, (void*) items);
			}

		

		}
		j=0;
		// delay 100 milliseconds. No need to overheat the processor
		vTaskDelay( 200 / portTICK_PERIOD_MS );

	}
	for ( c=0; c<num_channels; c++ ) {
		rmt_driver_uninstall(rx_inputs[c].channel);
	}
	
	vTaskDelete(NULL);
}

static void disp_rmt_raw_data(void* arg){
	while(1){
		if(params.display){
		printf("raw_data[%d]= {",params.raw_len);
		for(int i=0;i<(params.raw_len-1);i++){
			if(i>0) printf(",");
			printf("%d ",params.num[i]);
		
		}
		printf("}");
		printf("\n\n");	
		params.display=false;

		}
		vTaskDelay(pdMS_TO_TICKS(750));
		
	}
	
	
	vTaskDelete(NULL);
}

void app_main() {
	// initialize hardware
    rx_channels_init();
    
    // start receive processing task
    
	
	xTaskCreate(rmt_example_nec_rx_task, "rmt_nec_rx_task", 2048, NULL, 10, NULL);
	xTaskCreate(disp_rmt_raw_data, "disp_rmt_raw_data", 2048, NULL, 5, NULL);

	
	
}`