crankyoldgit/IRremoteESP8266 #1692: Toshiba Heat Pump - Pure mode

janpfo, 2021-11-29:
I tried to add it myself, but I couldn't solve it. 

It looks like it just sets one of the bytes to 1. 
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/janpf/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/janpf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

With pure | Temp |   |  
-- | -- | -- | --
  | 21 | 0xF20D03FC0140030042 | 0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x40, 0x03, 0x00, 0x42
  | 22 | 0xF20D03FC0150030052 | 0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x50, 0x03, 0x00, 0x52
  | 23 | 0xF20D03FC0160030062 | 0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x60, 0x03, 0x00, 0x62
  |   |   |  
Without pure |   |   |  
  | 21 | 0xF20D03FC0140031052 | 0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x40, 0x03, 0x10, 0x52
  | 22 | 0xF20D03FC0150031042 | 0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x50, 0x03, 0x10, 0x42
  | 23 | 0xF20D03FC0160031072 | 0xF2, 0x0D, 0x03, 0xFC, 0x01, 0x60, 0x03, 0x10, 0x72

</body>

</html>

NiKiZe, 2021-11-29:
Where is
#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable) ?

You seem to have ignored the issue template?

crankyoldgit, 2021-11-29:
Also, what is Pure Mode?

janpfo, 2021-11-29:
Sorry, I don't have the raw data from IRrecvDump. 

I tried with the included Climate IR component in EspHome, where I set byte[7] to 0x10, and it turns on Pure mode. 

@crankyoldgit Pure Mode is an "air cleaner" mode, where it uses an ionic filter.

crankyoldgit, 2021-11-29:
@janpfo Can you please download and test the branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1692 / PR #1693  as it should add `setFilter()`/`getFilter()` aka. Pure support for Toshiba. Please let us know the results so we can merge this.

Also, is your remote & A/C model numbers listed in:
https://github.com/crankyoldgit/IRremoteESP8266/blob/94080aa9adb899ddd311fe3e9050545b3f34591f/src/ir_Toshiba.h#L12-L24

If not, please tell us what they are so we can document that the library works with them.

janpfo, 2021-11-29:
@crankyoldgit thank you, that was fast!

I can confirm that filter/pure mode works with the updated code!

` //   Brand: Toshiba,  Model: WH-TA01JE remote `

crankyoldgit, 2021-11-29:
> I can confirm that filter/pure mode works with the updated code!

Thanks for that.

> `// Brand: Toshiba, Model: WH-TA01JE remote`

Added. What about the A/C unit itself?

janpfo, 2021-11-29:
> 

RAS-25SKVP2-ND

crankyoldgit, 2021-11-29:
Thankyou

PR #1693 (2021-11-29): TOSHIBA_AC: Add Filter setting support. aka. Pure.
* Add `setFilter()` & `getFilter()`.
* Add filter support for Toshiba in `IRAc` class.
* Minor code style/formating cleanup.
* Update & add unit tests for code coverage and verification.

Fixes #1692
commit 187c533a (2021-11-29): TOSHIBA_AC: Add Filter setting support. aka. Pure. (#1693)

* Add `setFilter()` & `getFilter()`.
* Add filter support for Toshiba in `IRAc` class.
* Minor code style/formating cleanup.
* Update & add unit tests for code coverage and verification.

Fixes #1692
closed 2021-11-29 by crankyoldgit