crankyoldgit/IRremoteESP8266 #1694: Bosch AC Timer Captures

jimmys01, 2021-12-01:
### Version/revision of the library used
v2.7.20

I had problems with powering off the AC but  the problems where mainly between the chair and the monitor. Of course the library is it is working flawlessly :-) ..

I thought to post anyway the captures I have done for my project (I send them using RAW encoding with ESPEasy), it would be nice If can be added to the library. If not this post can be deleted. 

Bosch Remote RG36B4/BGE (Uses COOLIX protocol)
Turn Off IR Capture from original Remote (irrecvdump V3)

Turn Off Timer captures
```
1.0 Hour
uint16_t rawData[199] = {4342, 4454,  486, 1724,  436, 658,  438, 1748,  464, 1718,  462, 634,  440, 656,  462, 1696,  488, 634,  462, 634,  436, 1722,  516, 608,  462, 660,  436, 1694,  488, 1720,  440, 630,  488, 1700,  488, 1704,  458, 660,  462, 1698,  490, 632,  462, 634,  436, 684,  436, 1700,  464, 1748,  462, 634,  462, 1720,  436, 658,  462, 1700,  488, 1692,  512, 1696,  438, 684,  410, 686,  434, 688,  408, 1696,  488, 1694,  464, 682,  414, 1748,  436, 1722,  488, 632,  438, 686,  408, 662,  462, 1696,  488, 1722,  462, 1696,  462, 1746,  436, 1798,  386, 1694,  490, 1720,  516, 5234,  4370, 4446,  490, 1690,  492, 658,  434, 1726,  436, 1746,  464, 604,  488, 658,  412, 1718,  490, 636,  460, 660,  438, 1698,  460, 662,  458, 632,  436, 1718,  490, 1720,  488, 608,  436, 1754,  462, 1726,  438, 682,  414, 1748,  464, 632,  460, 660,  410, 658,  438, 1748,  464, 1694,  464, 660,  436, 1720,  488, 634,  460, 1726,  462, 1724,  462, 1692,  490, 606,  462, 714,  384, 660,  460, 1722,  460, 1722,  490, 606,  464, 1718,  490, 1670,  486, 634,  462, 662,  410, 660,  460, 1722,  464, 1718,  460, 1696,  464, 1720,  462, 1720,  462, 1722,  486, 1700,  462};  // UNKNOWN 1F691B97

1.5 Hour
uint16_t rawData[199] = {4242, 4550,  490, 1720,  438, 686,  408, 1724,  486, 1696,  490, 632,  436, 684,  436, 1720,  462, 634,  438, 660,  460, 1698,  514, 610,  434, 658,  436, 1746,  464, 1718,  464, 634,  460, 1728,  460, 1702,  490, 630,  438, 1748,  438, 658,  436, 632,  488, 1720,  438, 658,  438, 1724,  464, 686,  432, 1724,  460, 636,  434, 1750,  488, 1698,  462, 686,  384, 1746,  468, 632,  436, 662,  436, 1748,  464, 1720,  438, 654,  440, 1722,  488, 1722,  460, 660,  436, 632,  438, 700,  398, 1822,  386, 1722,  438, 1744,  436, 1746,  462, 1720,  490, 1694,  462, 1696,  488, 5306,  4322, 4474,  542, 1642,  462, 658,  462, 1700,  486, 1720,  460, 662,  436, 658,  410, 1744,  464, 634,  488, 610,  460, 1700,  488, 630,  436, 682,  438, 1696,  486, 1700,  460, 634,  486, 1726,  460, 1728,  464, 630,  438, 1746,  464, 630,  464, 660,  408, 1774,  440, 628,  462, 1750,  414, 660,  460, 1720,  438, 658,  462, 1700,  488, 1694,  490, 634,  434, 1722,  492, 658,  410, 658,  438, 1776,  410, 1720,  464, 658,  434, 1750,  438, 1746,  464, 630,  438, 682,  412, 660,  436, 1748,  460, 1720,  438, 1772,  434, 1722,  438, 1744,  440, 1742,  436, 1750,  438};  // UNKNOWN F53BBAC8

2 Hour
uint16_t rawData[199] = {4348, 4452,  518, 1696,  488, 630,  464, 1720,  468, 1692,  486, 656,  438, 632,  464, 1718,  464, 684,  386, 686,  412, 1774,  408, 660,  462, 658,  434, 1722,  438, 1744,  468, 654,  438, 1724,  466, 1698,  490, 658,  434, 1726,  440, 658,  460, 682,  386, 1750,  436, 1716,  464, 1748,  460, 634,  464, 1696,  486, 632,  438, 1700,  512, 1692,  490, 656,  438, 630,  464, 632,  464, 634,  438, 1722,  514, 1668,  464, 686,  434, 1726,  488, 1694,  464, 634,  462, 660,  410, 660,  486, 1696,  464, 1718,  466, 1718,  464, 1720,  438, 1742,  442, 1720,  462, 1718,  492, 5280,  4350, 4474,  490, 1666,  490, 606,  488, 1724,  438, 1720,  484, 634,  440, 658,  460, 1692,  488, 662,  438, 608,  462, 1724,  488, 634,  462, 632,  462, 1720,  436, 1722,  464, 656,  436, 1728,  488, 1726,  436, 658,  462, 1700,  460, 658,  464, 656,  436, 1720,  464, 1698,  460, 1746,  442, 710,  408, 1698,  486, 660,  436, 1724,  458, 1722,  488, 632,  436, 684,  412, 686,  410, 662,  434, 1726,  464, 1720,  436, 658,  436, 1722,  488, 1696,  490, 660,  432, 636,  462, 634,  438, 1720,  490, 1692,  462, 1720,  462, 1720,  464, 1718,  464, 1744,  464, 1694,  492};  // UNKNOWN FB167210

2.5 Hour
uint16_t rawData[199] = {4344, 4434,  522, 1688,  494, 602,  494, 1690,  496, 1686,  496, 600,  494, 602,  494, 1686,  496, 604,  492, 602,  496, 1690,  496, 600,  494, 602,  494, 1688,  494, 1688,  496, 602,  494, 1694,  496, 1666,  522, 600,  494, 1690,  496, 600,  494, 1692,  496, 600,  494, 600,  494, 1690,  496, 602,  494, 1690,  494, 600,  494, 1690,  496, 602,  492, 1666,  520, 1688,  494, 602,  494, 602,  494, 1690,  496, 1688,  494, 600,  494, 1666,  522, 1686,  494, 602,  494, 600,  496, 602,  496, 1688,  494, 1688,  494, 1688,  494, 1688,  496, 1688,  494, 1688,  496, 1688,  470, 5326,  4356, 4438,  496, 1688,  496, 598,  496, 1692,  496, 1688,  494, 600,  494, 600,  494, 1688,  496, 600,  496, 602,  494, 1668,  520, 600,  496, 600,  494, 1688,  496, 1662,  522, 600,  494, 1694,  496, 1692,  496, 602,  492, 1690,  496, 600,  496, 1666,  520, 600,  494, 600,  494, 1690,  496, 602,  496, 1662,  518, 600,  496, 1690,  496, 600,  494, 1690,  496, 1688,  496, 600,  494, 602,  494, 1692,  496, 1688,  494, 600,  496, 1690,  496, 1686,  496, 600,  494, 604,  494, 602,  494, 1688,  496, 1688,  496, 1686,  496, 1662,  520, 1688,  494, 1662,  520, 1690,  470};  // UNKNOWN FCBBB588

3 Hour
uint16_t rawData[199] = {4352, 4454,  522, 1688,  496, 600,  494, 1666,  520, 1688,  496, 600,  494, 600,  494, 1688,  494, 604,  492, 602,  496, 1690,  496, 600,  494, 602,  494, 1688,  494, 1688,  494, 602,  494, 1694,  496, 1694,  494, 602,  492, 1692,  494, 600,  494, 1692,  494, 600,  494, 1692,  494, 1690,  496, 604,  494, 1688,  494, 600,  494, 1692,  494, 600,  496, 1666,  520, 600,  494, 602,  496, 602,  494, 1688,  494, 1688,  494, 602,  494, 1692,  494, 1690,  494, 602,  494, 602,  494, 600,  496, 1688,  496, 1688,  494, 1688,  494, 1666,  518, 1664,  520, 1688,  494, 1690,  470, 5302,  4380, 4416,  518, 1688,  494, 602,  492, 1694,  494, 1688,  494, 600,  494, 602,  494, 1688,  494, 602,  494, 604,  494, 1692,  494, 604,  490, 602,  494, 1688,  494, 1688,  496, 600,  494, 1694,  496, 1694,  494, 602,  494, 1666,  520, 600,  494, 1692,  494, 602,  494, 1692,  494, 1690,  496, 602,  494, 1688,  494, 602,  494, 1666,  520, 600,  494, 1692,  494, 602,  494, 602,  494, 602,  494, 1688,  494, 1688,  494, 602,  494, 1692,  494, 1688,  496, 600,  496, 602,  494, 602,  494, 1662,  520, 1662,  522, 1688,  494, 1688,  494, 1688,  494, 1664,  518, 1692,  470};  // UNKNOWN 1FBF6450

3.5 Hour
uint16_t rawData[199] = {4402, 4384,  548, 1638,  544, 576,  520, 1666,  520, 1662,  522, 574,  520, 576,  520, 1636,  546, 576,  520, 578,  520, 1666,  520, 574,  522, 574,  520, 1662,  520, 1640,  544, 574,  520, 1668,  522, 1644,  546, 576,  518, 1666,  522, 576,  520, 1666,  520, 1640,  544, 576,  518, 1646,  544, 576,  520, 1638,  544, 576,  518, 1666,  520, 578,  518, 574,  520, 1664,  520, 576,  520, 578,  520, 1664,  522, 1662,  520, 576,  518, 1666,  520, 1664,  518, 574,  522, 578,  518, 578,  520, 1662,  520, 1636,  548, 1664,  520, 1662,  520, 1662,  520, 1662,  520, 1664,  470, 5302,  4406, 4416,  520, 1664,  520, 576,  520, 1666,  522, 1662,  520, 576,  518, 576,  520, 1662,  520, 578,  520, 576,  520, 1666,  520, 576,  520, 576,  518, 1664,  518, 1664,  520, 576,  520, 1668,  522, 1668,  520, 576,  518, 1666,  522, 574,  520, 1666,  520, 1662,  520, 574,  522, 1668,  520, 578,  520, 1662,  520, 576,  518, 1668,  520, 576,  520, 576,  518, 1662,  522, 578,  518, 578,  520, 1664,  520, 1662,  520, 576,  518, 1668,  520, 1636,  548, 574,  520, 576,  520, 578,  518, 1664,  520, 1662,  522, 1662,  522, 1638,  544, 1662,  522, 1636,  546, 1664,  470};  // UNKNOWN 7DEF7380

4 Hour
uint16_t rawData[199] = {4326, 4454,  546, 1664,  518, 576,  518, 1668,  518, 1664,  518, 578,  518, 578,  518, 1664,  520, 578,  518, 578,  518, 1668,  520, 576,  520, 576,  518, 1666,  518, 1664,  520, 576,  518, 1670,  518, 1646,  544, 576,  518, 1668,  520, 574,  520, 1668,  520, 1666,  518, 1664,  520, 1666,  520, 578,  518, 1664,  520, 576,  518, 1668,  520, 576,  518, 576,  520, 576,  520, 578,  518, 580,  516, 1666,  518, 1664,  520, 576,  518, 1668,  518, 1664,  520, 576,  518, 580,  518, 580,  518, 1664,  520, 1664,  520, 1664,  518, 1640,  544, 1664,  520, 1664,  520, 1642,  518, 5254,  4404, 4392,  546, 1664,  518, 576,  520, 1668,  520, 1640,  542, 578,  516, 576,  518, 16 [... cut ...] 78,  518, 578,  518, 1666,  520, 576,  520, 576,  520, 1662,  522, 1662,  520, 576,  520, 1670,  518, 1668,  520, 576,  518, 1666,  520, 1664,  520, 1662,  520, 1664,  518, 1662,  520, 1640,  544, 578,  520, 1662,  520, 574,  520, 576,  518, 576,  520, 574,  520, 576,  520, 576,  520, 578,  518, 1636,  546, 1664,  520, 576,  518, 1666,  520, 1664,  520, 1664,  520, 1664,  520, 580,  518, 1662,  520, 576,  518, 1642,  546, 1662,  520, 1662,  520, 1662,  522, 1640,  494};  // UNKNOWN A2C83678

```

crankyoldgit, 2021-12-03:
Hey @jimmys01
So, it seems those messages are not as we expect for normal COOLIX messages.
e.g.
> Off timer
> 1.0 Hour
uint16_t rawData[199] = {4342, 4454,  486, 1724,  436, 658,  438, 1748,  464, 1718,  462, 634,  440, 656,  462, 1696,  488, 634,  462, 634,  436, 1722,  516, 608,  462, 660,  436, 1694,  488, 1720,  440, 630,  488, 1700,  488, 1704,  458, 660,  462, 1698,  490, 632,  462, 634,  436, 684,  436, 1700,  464, 1748,  462, 634,  462, 1720,  436, 658,  462, 1700,  488, 1692,  512, 1696,  438, 684,  410, 686,  434, 688,  408, 1696,  488, 1694,  464, 682,  414, 1748,  436, 1722,  488, 632,  438, 686,  408, 662,  462, 1696,  488, 1722,  462, 1696,  462, 1746,  436, 1798,  386, 1694,  490, 1720,  516, 5234,  4370, 4446,  490, 1690,  492, 658,  434, 1726,  436, 1746,  464, 604,  488, 658,  412, 1718,  490, 636,  460, 660,  438, 1698,  460, 662,  458, 632,  436, 1718,  490, 1720,  488, 608,  436, 1754,  462, 1726,  438, 682,  414, 1748,  464, 632,  460, 660,  410, 658,  438, 1748,  464, 1694,  464, 660,  436, 1720,  488, 634,  460, 1726,  462, 1724,  462, 1692,  490, 606,  462, 714,  384, 660,  460, 1722,  460, 1722,  490, 606,  464, 1718,  490, 1670,  486, 634,  462, 662,  410, 660,  460, 1722,  464, 1718,  460, 1696,  464, 1720,  462, 1720,  462, 1722,  486, 1700,  462};  // UNKNOWN 1F691B97

works out to be 2 x 48 bit messages (`0xB24DA35C6C7F` (MSB first))
It differs from the Coolix protocol in that it fails the integrity test. That test is every second byte is an inverted copy of the previous byte.
e.g.
First byte: `0xB2 ^ 0xFF == 0x4D` (second byte)
Third byte: `0xA3 ^ 0xFF == 0x5C` (fourth byte)
Fifth byte: `0x6C ^ 0xFF == 0x93` **!= 0x7F** (sixth byte)

So that is why it isn't being detected as a normal 24-bit `COOLIX` message.

I can't see anyway I can retrofit this into the existing `sendCOOLIX()`/`decodeCOOLIX()` as they are.
What I can do is add a `COOLIX48` protocol that will capture these messages and allow you to send it via a 48bit code.

That work for you?

NiKiZe, 2021-12-03:
If it mostly matches otherwise, what if we ignore som checks for larger messages?

jimmys01, 2021-12-03:
It feels weird that the same remote to use 2 different ways of communicating with the AC. 
I think that it will be a major source of confusion for people trying to use the library.
For me this would make my life a little easier but I can work around it. I dont know if it is worth your time to implement it. It may be this one remote that uses this protocol?

crankyoldgit, 2021-12-03:
> It feels weird that the same remote to use 2 different ways of communicating with the AC.

It's the same way from it's perspective.

> I think that it will be a major source of confusion for people trying to use the library. For me this would make my life a little easier but I can work around it. 

I see your point, but I think I can make it work.

> I dont know if it is worth your time to implement it. It may be this one remote that uses this protocol?

Adding basic support will be trivial so don't worry about that aspect.

crankyoldgit, 2021-12-03:
> Bosch Remote RG36B4/BGE (Uses COOLIX protocol)

Can we also get the details of the AC itself?

jimmys01, 2021-12-05:
Sure its the Bosch B1ZAI2441W/B1ZAO2441W

crankyoldgit, 2021-12-06:
Hey @jimmys01 Can you please download/test PR #1697 and confirm if it captures the codes for you and that `sendCoolix48()` works for you.

jimmys01, 2021-12-07:
Both encoding and decoding works! 
Thank you!!

crankyoldgit, 2021-12-07:
Excellent. Thanks for that.

PR #1697 (2021-12-07): Add basic support for COOLIX48 protocol.
* As far as we can tell COOLIX48 is a subtle version of `COOLIX` but does not obey the "every second byte is inverted" verification pattern. Thus, COOLIX48 is twice the bit size of COOLIX (24 bits).
* This seems to be used by a Bosch A/C to transmite On & Off timer values.
* Update & Extend Coolix Unit tests to cover this.
* Other minor code style changes.

Fixes #1694
commit 81da75d8 (2021-12-07): Add basic support for COOLIX48 protocol. (#1697)

* As far as we can tell COOLIX48 is a subtle version of `COOLIX` but does not obey the "every second byte is inverted" verification pattern. Thus, COOLIX48 is twice the bit size of COOLIX (24 bits).
* This seems to be used by a Bosch A/C to transmite On & Off timer values.
* Update & Extend Coolix Unit tests to cover this.
* Other minor code style changes.

Fixes #1694
closed 2021-12-07 by crankyoldgit