crankyoldgit/IRremoteESP8266 #1695: ESP32-C3 Support

sh-user, 2021-12-05:
```
ESP32-C3
RISC-V processor (RV32IMC)
Arduino 1.8.13
IRremoteESP8266 library 2.8.0
```
Is it possible to support ESP32-C3?
Example `IRrecvDemo.ino` does not compile

```
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'uint32_t IRrecv::ticksLow(uint32_t, uint8_t, uint16_t)':
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1082:8: error: no matching function for call to 'max(int32_t, int)'
       0));
        ^
In file included from c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\algorithm:61,
                 from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:155,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:219:5: note: candidate: 'template<class _Tp> const _Tp& std::max(const _Tp&, const _Tp&)'
     max(const _Tp& __a, const _Tp& __b)
     ^~~
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:219:5: note:   template argument deduction/substitution failed:
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1082:8: note:   deduced conflicting types for parameter 'const _Tp' ('long int' and 'int')
       0));
        ^
In file included from c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\algorithm:61,
                 from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:155,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:265:5: note: candidate: 'template<class _Tp, class _Compare> const _Tp& std::max(const _Tp&, const _Tp&, _Compare)'
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^~~
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:265:5: note:   template argument deduction/substitution failed:
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1082:8: note:   deduced conflicting types for parameter 'const _Tp' ('long int' and 'int')
       0));
        ^
In file included from c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:155,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3462:5: note: candidate: 'template<class _Tp> _Tp std::max(std::initializer_list<_Tp>)'
     max(initializer_list<_Tp> __l)
     ^~~
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3462:5: note:   template argument deduction/substitution failed:
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1082:8: note:   mismatched types 'std::initializer_list<_Tp>' and 'long int'
       0));
        ^
In file included from c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:155,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3468:5: note: candidate: 'template<class _Tp, class _Compare> _Tp std::max(std::initializer_list<_Tp>, _Compare)'
     max(initializer_list<_Tp> __l, _Compare __comp)
     ^~~
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3468:5: note:   template argument deduction/substitution failed:
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1082:8: note:   mismatched types 'std::initializer_list<_Tp>' and 'long int'
       0));
        ^
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'bool IRrecv::matchAtLeast(uint32_t, uint32_t, uint8_t, uint16_t)':
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1165:75: error: no matching function for call to 'min(uint32_t&, unsigned int)'
   return measured >= ticksLow(std::min(desired, MS_TO_USEC(params.timeout)),
                                                                           ^
In file included from c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\algorithm:61,
                 from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:155,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:195:5: note: candidate: 'template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)'
     min(const _Tp& __a, const _Tp& __b)
     ^~~
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:195:5: note:   template argument deduction/substitution failed:
[... 14 lines cut ...]
In file included from c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:155,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note: candidate: 'template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)'
     min(initializer_list<_Tp> __l)
     ^~~
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note:   template argument deduction/substitution failed:
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1165:75: note:   mismatched types 'std::initializer_list<_Tp>' and 'long unsigned int'
   return measured >= ticksLow(std::min(desired, MS_TO_USEC(params.timeout)),
                                                                           ^
In file included from c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32/Arduino.h:155,
                 from C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
[... cut ...]
c:\users\alex1\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note:   template argument deduction/substitution failed:
C:\Users\Alex1\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:1165:75: note:   mismatched types 'std::initializer_list<_Tp>' and 'long unsigned int'
   return measured >= ticksLow(std::min(desired, MS_TO_USEC(params.timeout)),
                                                                           ^
exit status 1
Error compiling for board ESP32C3 Dev Module.
```

crankyoldgit, 2021-12-06:
Looks like the compiler is slightly different for the C3. Should be fixable.
Are you using PlatformIO? If so, can you please include the `platformio.ini` file please so we/I can try to replicate this build environment.

crankyoldgit, 2021-12-06:
PR #1696 should fix the compilation errors.

Note: This isn't a problem with the library per se, well, as far as I can tell. This seems to be an issue with the experimental Arduino ESP32-C3 support. I made up a mock ESP32-C3 environment (I have no C3 boards to test with) and I was able to get it to compile successfully. i.e. The IRrecvDumpV2 & IRMQTTServer examples.

Anyway, the workarounds I've added shouldn't cause an issue with any other platforms.

As I have no way of testing the resulting code, that's as far as I can go at the moment.

In short, it builds a binary as expected now, at least if you use a `platformio.ini` entry like: https://github.com/platformio/platform-espressif32/issues/497#issuecomment-984089226
That cause similar errors to the ones your reported prior to the fix.

Please download & test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1695

Warning: As far as I can tell ESP32-C3 Arduino support is in it's infancy. I wouldn't describe it as well tested/finished yet.

PR #1696 (2021-12-06): ESP32-C3: Work around for some C3 specific compiler issues.
Seems the experimental Arduino support for the ESP32-C3 is a bit more finicky than the other environments.
Work around those issues by forcing the compiler to chose the correct `std::min()` & `std::max()` routines by casting the arguments.

Fixes #1695
commit 3f871e4b (2021-12-06): ESP32-C3: Work around for some C3 specific compiler issues. (#1696)

Seems the experimental Arduino support for the ESP32-C3 is a bit more finicky than the other environments.
Work around those issues by forcing the compiler to chose the correct `std::min()` & `std::max()` routines by casting the arguments.

Fixes #1695
PR #1732 (2022-01-09): ESP32-C3: Work around for some C3 specific compiler issues again.
This is a sibling of 3f871e4b7989789fa0093548b4d7b7eb240f7908.

Fixes #1695
commit d3c8abc5 (2022-01-09): ESP32-C3: Work around for some C3 specific compiler issues again. (#1732)

This is a sibling of 3f871e4b7989789fa0093548b4d7b7eb240f7908.

Fixes #1695
closed 2021-12-06 by crankyoldgit