crankyoldgit/IRremoteESP8266 #1703: Generate raw data from hex code

Nakul93, 2021-12-15:
I am trying to use rmt module from esp-idf and was able to receive raw data(raw timings) from it, in order for matching the protocols I have the hex codes for each control(which is unique).
 Is there a way where we can generate the raw data (raw timings) from hex (hash) code so that I can send data.

eg: for GREE

convert:
code: 0x09072050002000C0

to:
uint16_t rawData[139] = {9048, 4422,  706, 1616,  674, 474,  726, 524,  650, 1642,  676, 514,  684, 496,  702, 524,  652, 550,  672, 1616,  674, 1620,  650, 1644,  652, 522,  676, 520,  702, 516,  682, 524,  676, 522,  652, 518,  702, 498,  678, 548,  652, 546,  676, 524,  676, 1620,  674, 476,  700, 520,  704, 524,  674, 524,  652, 548,  678, 522,  652, 1616,  676, 548,  676, 1620,  676, 522,  676, 524,  652, 1642,  676, 498,  700, 19896,  706, 526,  676, 522,  678, 520,  678, 520,  680, 522,  676, 494,  706, 520,  678, 520,  678, 520,  680, 518,  680, 520,  682, 516,  684, 516,  682, 1582,  712, 512,  684, 516,  686, 512,  684, 514,  684, 514,  684, 514,  684, 514,  686, 514,  684, 514,  684, 514,  684, 516,  682, 516,  684, 516,  682, 518,  682, 516,  680, 494,  704, 1586,  708, 1588,  706};  // GREE

crankyoldgit, 2021-12-16:
See `resultToSourceCode()`
 https://github.com/crankyoldgit/IRremoteESP8266/blob/31432b4ef442f0c0548a5a4068feac7e1d5b32c9/test/IRutils_test.cpp#L165-L206

Nakul93, 2021-12-16:
From the resultToSourceCode() function, I understood that the rawData is not the rawbuf in decode_results structure. 
How can I get rawbuf data and how does it vary from rawData? since in the lib I can see rawbuf is used for decoding

Nakul93, 2021-12-16:
Thanks for the detailed explanation, couple of more questions:
1. If I make some changes and use one of the above solutions will I be able to get the raw ticks, because the approach which I use using esp-idf I get the raw timings(rawData) directly so I need to go back a step and get the  raw ticks(rawbuf) if I need to decode
eg:
` void dumpRawResult() { 
   std::cout << std::dec; 
   if (capture.rawlen == 0) return; 
   std::cout << "uint16_t rawbuf[" << capture.rawlen - 1 << "] = {"; 
   for (uint16_t i = 1; i < capture.rawlen; i++) { 
     if (i % 8 == 1) std::cout << std::endl << "    "; 
     std::cout << (capture.rawbuf[i] /kRawTick); 
     // std::cout << "(" << capture.rawbuf[i] << ")"; 
     if (i < capture.rawlen - 1) std::cout << ", "; 
   } 
   std::cout << "};" << std::endl; 
 } `
2. Why do we use krawtick as 2, is there a specific reason

crankyoldgit, 2021-12-18:
> 1. If I make some changes and use one of the above solutions will I be able to get the raw ticks, because the approach which I use using esp-idf I get the raw timings(rawData) directly so I need to go back a step and get the  raw ticks(rawbuf) if I need to decode

If you're getting `usecs` from `rmt` natively, then you probably want to set `kRawTick` to `1` to use those routines as is, or just remove the `kRawTick` multiplications.

> 2. Why do we use krawtick as 2, is there a specific reason

Yes. For some protocols we need to be able to capture/record marks/spaces that are longer than 66ms (66000us).
`rawbuf` is a `uint16_t` array, so the max value is 65535. So to get a value long enough to cover 66+ms and still fit into a `uint16_t`, we need to lose resolution, hence the `2`. Which allows us to record up to 131,072 usecs (with a resolution of 2usecs) i.e. 17 bits. This large a value covers pretty much all known marks/spaces in all protocols. (One of the protocols requires detecting a gap of 90ms IIRC). So `2` is the smallest tick to usec ratio/resolution loss that fits us into 16 bits.

We don't use `uint32_t` arrays for `rawbuf` to store the usec timings because it literally cost twice as much memory for the capture array. The hack of storing it as ticks, rather than raw `uSeconds` saves memory at the very small price of a tiny bit of integer math. Plus multiplying/dividing by `2` is a simple single bit shift, so it is _very_ fast integer math.

crankyoldgit, 2021-12-30:
@Nakul93 How are you going with this? Can we mark this issue resolved now?

crankyoldgit, 2022-01-14:
Marking closed due to no response in approx 1 month.

closed 2022-01-14 by crankyoldgit