crankyoldgit/IRremoteESP8266 #1706: Arton.h file fails to compile

Jeebus678, 2021-12-19:
### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
v2.8.0
### Describe the bug
A clear and concise description of what the bug is.
When I compile my program and include Arton.h it throws a massive exit code that shows up like this:
```c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x42): undefined reference to `_ZN10IRAirtonAc6setFanEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x4e): undefined reference to `_ZN10IRAirtonAc7setModeEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x5a): undefined reference to `_ZN10IRAirtonAc7setTempEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x64): undefined reference to `_ZN10IRAirtonAc8setPowerEb'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text._Z18setStateByProtocolv+0x20): undefined reference to `_ZN10IRAirtonAc4sendEt'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o: in function `_Z18setStateByProtocolv':```

#### To Reproduce
_What steps did you do, and what did or didn't actually happen? How can we reproduce the issue?_

I copy-pasted the raw Arton.h file and included it in my main.cpp file then compiled.

_e.g._
_1. Initialise the IRsend class._
_2. `IRsend.sendFoobar(0xdeadbeef);`_
_3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead._

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable in the report](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._
``` #include <ir_Airton.h>
IRAirtonAc airtonAC(12); 
void setAirtonState(bool status, int temp, int mode, int fan)
{
    airtonAC.begin();
    airtonAC.setFan(fan);
    airtonAC.setMode(mode);
    airtonAC.setTemp(temp);
    airtonAC.setPower(status);
}
```

#### Expected behaviour
_What steps did you do and what should it have done? A clear and concise description of what you expected to happen._

_e.g._
_1. Initialise the IRsend class._
_2. `IRsend.sendFoobar(0xdeadbeef);`_
_3. Foobar branded BBQ turns on and cooks me some ribs._

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
_Include some serial text of the raw dumps of what the device saw._
```
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text._Z14setAirtonStatebiii+0x4): undefined reference to `_ZN10IRAirtonAc5beginEv'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text._Z14setAirtonStatebiii+0x8): undefined reference to `_ZN10IRAirtonAc6setFanEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text._Z14setAirtonStatebiii+0xc): undefined reference to `_ZN10IRAirtonAc7setModeEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text._Z14setAirtonStatebiii+0x10): undefined reference to `_ZN10IRAirtonAc7setTempEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text._Z14setAirtonStatebiii+0x14): undefined reference to `_ZN10IRAirtonAc8setPowerEb'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o: in function `_Z14setAirtonStatebiii':
main.ino.cpp:(.text._Z14setAirtonStatebiii+0x33): undefined reference to `_ZN10IRAirtonAc5beginEv'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x42): undefined reference to `_ZN10IRAirtonAc6setFanEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x4e): undefined reference to `_ZN10IRAirtonAc7setModeEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x5a): undefined reference to `_ZN10IRAirtonAc7setTempEh'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: main.ino.cpp:(.text._Z14setAirtonStatebiii+0x64): undefined reference to `_ZN10IRAirtonAc8setPowerEb'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text._Z18setStateByProtocolv+0x20): undefined reference to `_ZN10IRAirtonAc4sendEt'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o: in function `_Z18setStateByProtocolv':
main.ino.cpp:(.text._Z18setStateByProtocolv+0xba): undefined reference to `_ZN10IRAirtonAc4sendEt'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o:(.text.startup._GLOBAL__sub_I_Blynk+0xc): undefined reference to `_ZN10IRAirtonAcC1Etbb'
c:/users/root/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\src\main.ino.cpp.o: in function `_GLOBAL__sub_I_Blynk':
main.ino.cpp:(.text.startup._GLOBAL__sub_I_Blynk+0x70): undefined reference to `_ZN10IRAirtonAcC1Etbb'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp12e\firmware.elf] Error 1
```
_**Note**: Output from Tasmota is not acceptable. We can't easily use their raw format._

### What brand/model IR demodulator are you using?
_Brand/Model code_ or _None_

_If VS1838b: That is likely your problem. Please buy a better one & try again. Reporting a capture or decode issue when you use one of these is effectively wasting our & your time. See the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Help_Im_getting_very_inconsistent_results_when_capturing_an_IR_message_using_a_VS1838b_IR_demodulator)_

### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_
Yes, i've been using it for over a week and it's worked flawlessly. 
### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2021-12-20:
> I copy-pasted the raw Arton.h file and included it in my main.cpp file then compiled.

&
 
> ```
> IRAirtonAc airtonAC(12); 
> void setAirtonState(bool status, int temp, int mode, int fan)
> {
>     airtonAC.begin();
>     airtonAC.setFan(fan);
>     airtonAC.setMode(mode);
>     airtonAC.setTemp(temp);
>     airtonAC.setPower(status);
> }
> ```

That code / approach will not work. You seem to be missing a whole bunch of header includes.
Pasting in a header file into your main file is not a good approach.

In short, you'll need to include this in your Arduino Sketch:
```c++
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Airton.h>
```

e.g. Start with something similar to [`TurnOnKelvinatorAC.ino`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/TurnOnKelvinatorAC/TurnOnKelvinatorAC.ino)

The error messages are from the `linker` trying to link/include the parts of the compiled library into your binary, and it not being able to find the compiled library. i.e. Your either including the library poorly, don't have it installed in the correct place, and/or haven't told the compiler to include or where to locate the library when compiling/linking.

This does NOT look like a problem with the library based on the information provided thus far.

PS. we are very confident the code compiles & works because every change to the library gets tested with https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/ir_Airton_test.cpp and we've had user reports that it works fine.

Jeebus678, 2021-12-20:
I have many more include statements for a variety of other remote header
files that work flawlessly, so the error doesn’t seem to be in my include
statements. I have all the files you’ve commented included and more.

crankyoldgit, 2021-12-20:
Can you please provide the smallest code example that we can compile that produces the errors we are seeing?

Also, this is probably your problem:

> Typically located in the library.json & src/IRremoteESP8266.h files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc.
**v2.8.0**

Have you checked that your build environment is actually using the revision of the library you think you're using? It needs to be _later_ than `2.8.0`.

**Note**: _Detailed_ support for Airton was added _after_ version `2.8.0` (7b13d24a895837d2f5ab7f1faea43b062a7cf90d) of the library. It is **NOT** included in any _currently (as of Dec 20th 2021)_ released version. You need to be using the [`master` branch](https://github.com/crankyoldgit/IRremoteESP8266/tree/master) after commit c657b27f3b79dea99d5801cdf4ce196b0351e566 for it to work/compile.

crankyoldgit, 2021-12-23:
Ping @Jeebus678

crankyoldgit, 2021-12-30:
Another ping for @Jeebus678 . If we don't hear from you soon, we will assume our suggestion was correct, and this is a "user" issue.

crankyoldgit, 2022-01-01:
FYI, if you update to [v2.8.1 of library](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.1) (which includes full Airton support) you'll probably find your issue is resolved.

Marking this resolved as `ir_Airton.h` actually exists in 2.8.1, it didn't in 2.8.0

closed 2022-01-01 by crankyoldgit