crankyoldgit/IRremoteESP8266 #1707: Raw information

maiiikell1995, 2021-12-21:
Hi there, 

Is there away to the raw code of the midea ac's? 
Im looking everywhere but cant find them anywhere in the repository. 

The readon im asking is I'm looking for a way go get my Comfee(midea) AC to working with homekit (which i know this repo doesnt support)

NiKiZe, 2021-12-21:
You want to control an AC with RAW codes? Please read the FAQ why that is impractical at best.

maiiikell1995, 2021-12-21:
Not really i want to get the raw codes so i can convert them into the IR command that HAA can deal with. Since this repo doesn't support native homekit implementation im using HAA found here ( https://github.com/RavenSystem/esp-homekit-devices/wiki/HeaterCooler ) but since i figured some people did the work here already with the raw code it would be a waste of time to do that whole process again.

NiKiZe, 2021-12-21:
This lib is free standing, you might want to poke homekit to use it or support it. Modes has several different protocols, if you follow the source, or search the issues history you can probably find the source for how they where reverse engineered.

crankyoldgit, 2021-12-21:
If you look in the `test` directory you can find examples of how protocols and specific settings can be converted into "raw" codes. I'm not even sure what you mean by raw codes in the context of your question/issue.

maiiikell1995, 2021-12-21:
Sorry for not being clear, i am a complete noob with these IR commands since its the first time working with them and trying to figure out how to get stuff done like i want them too. Im a little im over my head, but i hope with the help of you people i can get it working.

Im looking for IR commands in this format: 

+  420 -  486 +  590 - 1593 +  559 - 1557 +  593 -  482 +  593 -  482 +  591 -  517 +  559 -  481 +  591 -  484 
+  588 - 1561 +  593 -  482 +  594 -  480 +  589 - 1595 +  556 -  485 +  591 -  519 +  552 -  487 +  590 -  485 
+  588 -  518 +  558 -  520 +  556 - 1557 +  588 - 1561 +  593 - 1557 +  593 - 1557 +  593 - 1556 +  593 - 1557 
+  592 - 1558 +  589 - 1596 +  553 - 1561 +  589 - 1561 +  592 - 1557 +  593 - 1557 +  593 - 1556 +  589 - 1561 
+  591 - 1559 +  590 - 1559J+  594 -  481 +  602 - 1547 +  591 -  522 +  526 - 1585 +  591 - 1595 +  553 - 1561 
+  591 -  517 +  558 -  482 +  592 - 5151 + 4443 - 4337 +  591 -  484 +  589 - 1559 +  593 -  484 +  585 - 1564 
+  592 - 1556 +  587 - 1562 +  589 - 1599 +  529 -  508 +  591 -  483 +  590 - 1592 +  557 - 1559 +  588 - 1563 
+  561 - 1586 +  592 - 1592 +  558 -  483 +  591 - 1558 +  589 - 1595 +  554 -  486 +  587 - 1596 +  553 - 1563 
+  591 - 1557 +  588 - 1597 +  554 - 1560 +  588 - 1562 +  588 -  486 +  591 -  515 +  557 -  485 +  589 -  485 
+  592 -  516 +  557 -  519 +  552 -  487 +  591 -  484 +  591 -  52J +  552 -  483 +  591 -  521 +  526 -  510 
+  592 -  482 +  590 -  520 +  552 -  486 +  589 -  485 +  591 - 1559 +  589 -  486 +  588 - 1593 +  558 -  514 
+  560 -  484 +  590 -  517 +  557 - 1559 +  590 - 1592 +  559 

As an example or any other way to get my Midea (Comfee) Mobile AC working in HomeKit, I choose HAA as mentioned earlier because im familiar with this firmware, but not with IR commands.

NiKiZe, 2021-12-21:
That means that you will have one raw code for every combination.

So lets say you want to have 18-30 degrees C, that is 12 different codes, you also want to have 3 different fan modes, now you are up to 12*3=36 different codes, now you want with swing on, and of so that is *2=72 different codes, and that is only the basic basic.

You probably don't really want this.

crankyoldgit, 2021-12-22:
@maiiikell1995 Please try out the tool `code_to_raw` in tools/ in PR #1708 (branch https://github.com/crankyoldgit/IRremoteESP8266/tree/code_to_raw)

I've made it about as easy for someone as practically possible.
You will still have to write code to generate the state/hex code for the Midea settings you want, but that tool should convert it to raw for you.

dudanov, 2021-12-23:
Hi
@maiiikell1995 if you need to control your Midea air conditioner from HomeAssistant, then the best option is to use [ESPHome](https://esphome.io/). And you don't need RAW codes. I wrote the implementation of the Midea IR protocol and the IR climate device. By the way, thanks to David @crankyoldgit , I took the protocol description from this repository. Usually, Midea air conditioners, in addition to their own protocol, also support the Coolix protocol. [Coolix support](https://esphome.io/components/climate/ir_climate.html) is already working in the official release, but my [PR](https://github.com/esphome/esphome/pull/2847) with Midea IR climate support I think will be included only in the next January 2022.01 release. You can also build ESPHome from [my repository](https://github.com/dudanov/esphome/tree/midea_ir) without waiting for the official release.

crankyoldgit, 2021-12-23:
> if you need to control your Midea air conditioner from HomeAssistant

I think @maiiikell1995 is after Homekit, not Home Assistant. I think that is Apple's system.

maiiikell1995, 2021-12-27:
Thanks everybody for commenting and pointing me in the right direction! 
Will continue to figure it out in 2022. Issue may be closed now thanks a lot again!

crankyoldgit, 2021-12-27:
Good luck. Thanks for getting back to us.

closed 2021-12-27 by crankyoldgit

crankyoldgit, 2022-01-01:
FYI, the changes/tool mentioned above have now been included in the new [v2.8.1 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.1) of the library.