crankyoldgit/IRremoteESP8266 #1718: Some CI tests/checks not marked as "Required"

crankyoldgit, 2021-12-31:
Hey @NiKiZe 
I just noticed with new PRs that some of the checks/tests are no longer marked as "(Required)" 
e.g.
Build / Gen_Matrix (pull_request)
Build / Build_Example (./examples/IRGCTCPServer) (pull_request)
etc.

e.g. https://github.com/crankyoldgit/IRremoteESP8266/pull/1717

My guess this was introduced/changed in/after #1709 

They should be marked as "(Required)" IMHO.

Or does the final "Build_Examples Expected — Waiting for status to be reported" one (which is marked as "(Required)") enforce they are required too?

Probably some very minor tweak is required, but as you touched it last. ;-P

NiKiZe, 2021-12-31:
Yes, That is why I added Build_Examples test that depends on the others. (It don't show up until the others is complete, so even if it don't show up as failing, it should still block merge)

I did not find where this required comes from, but guess that is from project settings that I don't have access to?
The other build_tools test (don't have the name) is missing this, so might need some more checks for that.

crankyoldgit, 2021-12-31:
Seems the "(Required)" comes from https://github.com/crankyoldgit/IRremoteESP8266/settings/branch_protection_rules/17453042
"Settings" -> "Branches" -> "Branch protection rules" -> "master" (Edit) -> "Protect matching branches" -> ...

So yeah, it's in "Settings" ... stupid system in a way. :-/
I've tagged everything I can think of.

Fixed.

closed 2021-12-31 by crankyoldgit

NiKiZe, 2021-12-31:
I don't think I have access to that.
Adding the dynamic named tests is probably not a good idea, so the last test that depends on the others is probably a good idea.

But we might want to create a failing test and see how it behaves.