crankyoldgit/IRremoteESP8266 #1729: Can help join HITACHI RAR-2P2 remote (264 bit)

step3528, 2022-01-06:
### Version/revision of the library used
Version of the library : 2.8.1

### Describe the bug
My air conditioner is Hitachi RAK-25NH5 with RAR-2P2 remote, which is unrecognizable, can help to join?

the protocol appears to be just a shorter version of HITACHI.
view https://github.com/crankyoldgit/IRremoteESP8266/issues/1134

same value for Temperature

I created this file.

https://docs.google.com/spreadsheets/d/1qDnKM3ZslWrR-skgQjaConq3U2b7RuzZK1tA64WYRBM/edit?usp=sharing

NiKiZe, 2022-01-06:
Do you have at least one Raw dump using `IRrecvDumpV3`? (as asked for in the issue template)

step3528, 2022-01-06:
Hello,
Yes, i used IRrecvDumpV3.

NiKiZe, 2022-01-06:
Please post the raw dump in the issue as text from at least on of the commands.

step3528, 2022-01-06:
attached the raw dump

*hot/power off/ fan 1 /27°C***
PROTOCOL : Unknow
Code      : 0xC2B825B6 (266 Bits)
uint16_t rawData[531] = {3392, 1752,  372, 1356,  344, 504,  344, 504,  344, 480,  370, 480,  370, 478,  370, 478,  372, 480,  370, 478,  370, 480,  370, 478,  370, 480,  370, 1330,  370, 478,  370, 480,  370, 478,  370, 478,  370, 480,  370, 478,  370, 480,  370, 504,  346, 504,  344, 480,  370, 504,  346, 478,  370, 478,  370, 478,  372, 480,  370, 480,  370, 504,  346, 1330,  370, 480,  370, 1356,  346, 1330,  370, 1354,  346, 1328,  370, 1330,  370, 1328,  370, 478,  372, 1330,  370, 1328,  370, 1356,  346, 1354,  344, 1328,  372, 1354,  344, 1356,  344, 1330,  370, 1328,  372, 480,  370, 478,  370, 504,  346, 478,  372, 476,  372, 504,  346, 480,  370, 504,  344, 478,  372, 504,  344, 1328,  372, 1328,  372, 478,  370, 504,  346, 1330,  370, 1328,  372, 1328,  370, 1356,  344, 478,  370, 478,  370, 1356,  344, 1328,  372, 480,  370, 478,  372, 504,  346, 1330,  370, 478,  370, 480,  370, 1330,  370, 478,  370, 480,  370, 1328,  370, 1330,  370, 480,  372, 1352,  346, 1328,  372, 504,  346, 1328,  370, 1328,  370, 480,  370, 1328,  372, 1328,  370, 504,  344, 478,  372, 1330,  370, 478,  370, 480,  370, 480,  370, 478,  370, 480,  370, 1356,  346, 1328,  370, 480,  370, 1330,  370, 1330,  370, 1330,  370, 478,  370, 506,  344, 1328,  372, 1328,  372, 478,  372, 1330,  370, 1328,  372, 478,  370, 1330,  370, 1328,  372, 504,  344, 480,  370, 1330,  370, 504,  344, 504,  346, 1354,  346, 504,  346, 478,  372, 478,  370, 480,  370, 478,  370, 492,  358, 478,  370, 478,  370, 1328,  372, 1330,  370, 1328,  372, 1354,  346, 1328,  372, 1328,  370, 1330,  370, 1328,  372, 476,  372, 504,  346, 478,  372, 480,  370, 480,  370, 504,  344, 478,  370, 480,  370, 1328,  370, 1330,  372, 1328,  370, 1330,  370, 1330,  370, 1330,  368, 1330,  370, 1330,  370, 480,  370, 480,  370, 478,  370, 504,  344, 480,  370, 478,  370, 504,  344, 478,  372, 1328,  370, 1330,  370, 1330,  370, 1330,  370, 1328,  372, 1356,  344, 1330,  370, 1330,  370, 480,  370, 478,  370, 482,  368, 480,  370, 480,  370, 480,  370, 480,  370, 480,  370, 1330,  370, 1354,  346, 1330,  370, 1354,  346, 1330,  370, 1330,  370, 1330,  370, 1330,  370, 504,  346, 480,  370, 478,  372, 478,  372, 504,  344, 480,  370, 480,  370, 504,  344, 1330,  370, 1328,  370, 1330,  372, 1328,  370, 1354,  346, 1328,  370, 1330,  370, 1330,  370, 478,  370, 1330,  370, 1328,  372, 480,  370, 1330,  370, 480,  370, 480,  370, 480,  370, 1356,  344, 478,  370, 506,  344, 1330,  370, 478,  370, 1330,  372, 1354,  346, 1354,  346, 1328,  370, 478,  372, 478,  370, 480,  370, 504,  346, 480,  370, 1328,  370, 1330,  370, 478,  370, 1330,  370, 1328,  370, 1330,  372, 1354,  344, 1328,  372, 504,  346, 478,  370, 504,  346, 504,  346, 478,  370, 478,  370, 482,  368, 480,  370, 478,  370, 480,  370, 1328,  372, 1328,  370, 1330,  370, 1328,  370, 1328,  370, 1330,  370, 1354,  346, 1328,  372, 478,  370, 478,  370, 478,  372, 478,  372, 478,  372, 478,  372, 478,  370, 478,  372, 1330,  370, 1328,  370, 1328,  372, 1328,  372, 1330,  444, 1256,  370, 1330,  370, 1330,  442}

this is what I identified
state 11 et 12 : press button
on/off           : 13 EC
change FAN : 42 BD
decrease T°  : 43 BC 
increase T°   : 44 BD
shutter swing : 81 7E

 State 25 et 26 :mode et FAN
cool     : 3C
HOT     : 69
FAN      : 1E
DRY      : 5A
AUTO   : 78

FAN Auto : 5A
FAN 1       : 1E
FAN 2       : 3C
FAN 3       : 4B

state 27 et 28 : ON ou OFF
ON : D1 2E
OFF : C1 3E

crankyoldgit, 2022-01-07:
I've added "basic" support for the protocol in PR #1730 / Branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/rar-2p2

Please download that, test it works (both sending & decoding please), and base your analysis on those results in your spreadsheet/documentation.

It should be 33 bytes in LSB order now.

step3528, 2022-01-08:
here are the results with IRrecvDumpV3

Timestamp : 002824.971
13:07:58.607 -> Library   : v2.8.1
13:07:58.607 -> 
13:07:58.607 -> Protocol  : HITACHI_AC264
13:07:58.607 -> Code      : 0x01100040BFFF00CC33926D13EC50AF00FF00FF00FF00FF00FF16E9C13E00FF00FF (264 Bits)

https://docs.google.com/spreadsheets/d/1qDnKM3ZslWrR-skgQjaConq3U2b7RuzZK1tA64WYRBM/edit?usp=sharing

NiKiZe, 2022-01-08:
Looks good so far.
Did you have a chance to also test sending?

step3528, 2022-01-08:
Yes i am tested with the program CommonAcControl, the console bip when sending hitachi344 message but she does not start.

crankyoldgit, 2022-01-08:
You can't use the "common" IRac stuff yet for this protocol. You'll need to use the more basic IR send example code.

step3528, 2022-01-09:
I am tested with IRsenddemo to modify and the console start.
the message sent is detected by the console. 
I tested on and OFF message

crankyoldgit, 2022-01-09:
Excellent. Thanks for confirming. I'll merge that PR

crankyoldgit, 2022-01-09:
Ok, you're now at this stage: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#working-out-the-details

We have already established the bit ordering and the checksum (i.e. every second byte after `state[3]`, is the inverse of the previous byte.)

So, you need to analyse which bit(s) is critical to power etc. eg. `state[27]` has `0xD1`/`0b11010001` for `on` and `0xC1`/`0b11000001` for `off`, which means `0b00010000` is the bit that controls power, `1` for on, `0` for off.

You need to do that sort of documentation for all the controls you want supported. When you have that all done, let me know and I'll write the code for it.

Bonus points if you can document it like:
https://github.com/crankyoldgit/IRremoteESP8266/blob/93af543dcb849918b0c342ed1b6af68953a61d06/src/ir_Hitachi.h#L177-L213

step3528, 2022-01-10:
`// / Représentation native d'un message A/C Hitachi 264 bits.
union Hitachi1Protocole{
  uint8_t raw[ kHitachiAc264StateLength ] ;  // /< L'état dans le code natif.
  structure {
   struct {
    // Byte 0~10
    uint8_t pad0[11];
    // Byte 11
    uint8_t Button  :8;
    // Byte 12
    uint8_t       :8;
    // Byte 13
    uint8_t       :2;
    uint8_t Temp  :6;
    // Byte 14
    uint8_t       :8;
    // Byte 14~24
    uint8_t pad1[10];
    // Byte 25
    uint8_t Mode  :4;
    uint8_t Fan   :4;
    // Byte 26
    uint8_t       :8;
    // Byte 27
    uint8_t Power :8;
    // Byte 28
    uint8_t        :8;
    // Byte 29~32
    uint8_t pad2[4];
 
   } ;
} ;
// HitachiAc264
const uint8_t kHitachiAc264ButtonPowerMode = 0x13;
const uint8_t kHitachiAc264ButtonFan = 0x42;
const uint8_t kHitachiAc264ButtonTempDown = 0x43;
const uint8_t kHitachiAc264ButtonTempUp = 0x44;
const uint8_t kHitachiAc264ButtonSwing = 0x81;

//Temp min max for Cool Hot Dry (Fan no T°) 
const uint8_t kHitachiAc264MinTemp = 16;   // 16C
const uint8_t kHitachiAc264MaxTemp = 32;   // 32C

const uint8_t kHitachiAc264Fan = 1;
const uint8_t kHitachiAc264Cool = 3;
const uint8_t kHitachiAc264Dry = 5;
const uint8_t kHitachiAc264Heat = 6;

const uint8_t kHitachiAc264FanLow = 1;
const uint8_t kHitachiAc264FanMedium = 3;
const uint8_t kHitachiAc264FanHigh = 4;
const uint8_t kHitachiAc264FanAuto = 5;
const uint8_t kHitachiAc264FanDry = 1; //mode dry only speed low
const uint8_t kHitachiAc264FanLowfan = 1;// mode Fan no speed fan auto
const uint8_t kHitachiAc264FanMediumfan = 3;
const uint8_t kHitachiAc264FanHighfan = 4;

const uint8_t kHitachiAc264PowerOn = 0xD1;
const uint8_t kHitachiAc264PowerOff = 0xC1;
`

crankyoldgit, 2022-01-10:
> const uint8_t kHitachiAc264ButtonPowerMode = 0x13;
> const uint8_t kHitachiAc264ButtonFan = 0x42;
> const uint8_t kHitachiAc264ButtonTempDown = 0x43;
> const uint8_t kHitachiAc264ButtonTempUp = 0x44;
> const uint8_t kHitachiAc264ButtonSwing = 0x81;

Which location do these values go into?

crankyoldgit, 2022-01-10:
> const uint8_t kHitachiAc264FanDry = 1; //mode dry only speed low
> const uint8_t kHitachiAc264FanLowfan = 1;// mode Fan no speed fan auto

Why are these fan speeds the same?

step3528, 2022-01-11:
> > const uint8_t kHitachiAc264ButtonPowerMode = 0x13;
> > const uint8_t kHitachiAc264ButtonFan = 0x42;
> > const uint8_t kHitachiAc264ButtonTempDown = 0x43;
> > const uint8_t kHitachiAc264ButtonTempUp = 0x44;
> > const uint8_t kHitachiAc264ButtonSwing = 0x81;
> 
> Which location do these values go into?

these values ​​relate to byte 11 (button)

step3528, 2022-01-11:
> > const uint8_t kHitachiAc264FanDry = 1; //mode dry only speed low
> > const uint8_t kHitachiAc264FanLowfan = 1;// mode Fan no speed fan auto
> 
> Why are these fan speeds the same?

const uint8_t kHitachiAc264FanLowfan = 1;// mode Fan no speed fan auto
const uint8_t kHitachiAc264FanMediumfan = 3;
const uint8_t kHitachiAc264FanHighfan = 4;

these 3 lines can be deleted and add that in FAN mode auto speed is not possible.

crankyoldgit, 2022-01-11:
> I can't insert as code

When using multiple lines of code you need to use "```", not "`".

crankyoldgit, 2022-01-13:
Please download & test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/rar-2p2-detailed / PR #1735 and let us know how it goes.

step3528, 2022-01-15:
Hello,

Which sample program I can use to easily test
thank you

crankyoldgit, 2022-01-15:
[IRrecvDumpV3.ino](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3) should tell you if the library is correctly detecting and decoding the messages. i.e. power, mode, temp, fan speed.
[TurnOnArgoAC.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/TurnOnArgoAC/TurnOnArgoAC.ino) should be an easy send test, once you've modified the code to use the new `IRHitachiAc264` class and constants etc.

A more involved/complex send test would be to use [IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) which should support `HITACHI_AC264` now in the aircon section. You won't need to modify the code for that one.

Note: The only controls that work are power, mode, temp, & fan speed. (Per https://github.com/crankyoldgit/IRremoteESP8266/issues/1729#issuecomment-1008889121)

step3528, 2022-01-16:
I tested with IRrecvDumpV3.ino and it detects all data from the remote control.

With IRMQTTServer the console start but i can't stop it.

crankyoldgit, 2022-01-16:
> With IRMQTTServer the console start but i can't stop it. the code for off sends On, for hitachi344 code too

Can you please expand on this? I'm not sure exactly what you're saying.

Instructions for how to setup & use IRMQTTServer can be found in the source code here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9228143207ab7e2968a42d9a92e246237af7bb83/examples/IRMQTTServer/IRMQTTServer.ino#L22-L62

step3528, 2022-01-16:
when I choose power OFF in the AirCon menu it sends the code power ON.
[image]

crankyoldgit, 2022-01-17:
That's interesting.

I've written a unit test that checks that it (library) thinks it is sending what it thinks is an "Off" message, when you transition from "On" to "Off".
That is checking at the `IRHitachi_Ac264` class & the `IRac` class level. (The `IRac` class uses the `IRHitachi_Ac264` class, and `IRMQTTServer` uses the `IRac` class).

The "Off" message code it is sending should be:
```c++
uint8_t state[33] = {
     0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92,
     0x6D, 0x13, 0xEC, 0x64, 0x9B, 0x00, 0xFF, 0x00, 0xFF, 0x00,
     0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x56, 0xA9, 0xC1, 0x3E, 0x00,
     0xFF, 0x00, 0xFF};
```
Or in human form:
```
Power: Off, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
```

> I tested with IRrecvDumpV3.ino and it detects all data from the remote control.

My tests indicate the library is working. It thinks it's turning it off. However, we might have a bug in generation/position of the bits for an "off" message. That's unusual/unexpected as you indicated it was all working working earlier (i.e. the quoted line above.)

Can you please confirm that is the correct `state` code produced (via IRrecvDumpV3) when your remote has the settings of "On, Heat, 25C, Fan Auto", and _then_ you press the "Power" button. i.e. "Off"? e.g. Please include the complete capture from IRrecvDumpV3.

step3528, 2022-01-17:
Test with remote control and IRrecvDumpV3.in
ON

Protocol  : HITACHI_AC264
Mesg Desc.: Power: On, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
uint8_t state[33] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x64, 0x9B, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x56, 0xA9, 0xD1, 0x2E, 0x00, 0xFF, 0x00, 0xFF};

Off 
Mesg Desc.: Power: Off, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
 uint8_t state[33] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x64, 0x9B, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x56, 0xA9, 0xC1, 0x3E, 0x00, 0xFF, 0x00, 0xFF};

test with IRMQTTServer and IRrecvDumpV3.ino
On
 Mesg Desc.: Power: On, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
 uint8_t state[33] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x64, 0x9B, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x56, 0xA9, 0xD1, 0x2E, 0x00, 0xFF, 0x00, 0xFF};
 
OFF
 Code      : 0x01100040BFFF00CC33926D13EC649B00FF00FF00FF00FF00FF56A9D12E00FF00FF (264 Bits)
 Mesg Desc.: Power: On, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
uint8_t state[33] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x64, 0x9B, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x56, 0xA9, 0xD1, 0x2E, 0x00, 0xFF, 0x00, 0xFF};

crankyoldgit, 2022-01-17:
Thanks for that. I'll take another look at it to see what I can work out that's not working as expected.

crankyoldgit, 2022-01-17:
I've been staring at the code for over an hour or and I can't see anything that is jumping out at me where I've made a mistake. I'm stumped at the moment.

@NiKiZe Calling in a second/fresh set of eyes to take a look to see if I'm missing something.

Notes:
- `handleToggles()` isn't/shouldn't be changing anything for this protocol. (but I'll add a test just in case)
- We've heard of not problems with HITACHI_AC424 which this class and code is literally mostly a copy of (via class inheritance)
- The unit tests I added in https://github.com/crankyoldgit/IRremoteESP8266/commit/99f2210e5f17cc588a2e4ecc29dd6d5520194c06 seem to indicate that everything is working fine/as expected in the `IRHitachiAc264` & `IRac` classes.
- That kind of points to a problem in:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9228143207ab7e2968a42d9a92e246237af7bb83/examples/IRMQTTServer/IRMQTTServer.ino#L3030-L3032
which is very unlikely as every protocol uses it, and we would have had other reports (and it hasn't changed in ages)
or
https://github.com/crankyoldgit/IRremoteESP8266/blob/99f2210e5f17cc588a2e4ecc29dd6d5520194c06/src/IRac.cpp#L2664-L2670
which should be unlikely as it (`hitachi264`) doesn't use any `prev`ious state and is pretty much an exact copy of `hitachi424`s code. Which I've also gone over.
- The data provided by @step3528 in https://github.com/crankyoldgit/IRremoteESP8266/issues/1729#issuecomment-1014412318 really points to a failing somewhere in the example code/library.

@step3528 Did you modify **any** of the IRMQTTServer header or source files at all? If so, what setting/defines/flags etc did you change?

crankyoldgit, 2022-01-17:
@NiKiZe Just extended the test to verify `handleToggles()` isn't the problem/is behaving as expected. :-/

crankyoldgit, 2022-01-17:
@step3528 Can you please include a copy of the "info" page on IRMQTTServer too please, after you've tried to turn it off via the Web Interface? It might give some useful info. :-/

step3528, 2022-01-17:
> @step3528 Did you modify **any** of the IRMQTTServer header or source files at all? If so, what setting/defines/flags etc did you change?

I did not modify the file

> @step3528 Can you please include a copy of the "info" page on IRMQTTServer too please, after you've tried to turn it off via the Web Interface? It might give some useful info. :-/

General
Hostname: ir_server
IP address: 192.168.0.70
MAC address: CC:50:E3:63:6D:93
Booted: 1 Minute 45 Seconds ago
Version: v1.6.0
Built: Jan 17 2022 11:41:11
Period Offset: -6us
IR Lib Version: 2.8.1
ESP8266 Core Version: 3.0.2
Free Sketch Space: 1444k
Cpu Freq: 80MHz
Sanity Check: Ok
IR Send GPIO(s): 4 (default)
Inverting GPIO output: Off
Total send requests: 1
Last message sent: FAILED (Never)
IR Recv GPIO: 14
Total IR Received: 0
Last IR Received: None (Never)
Duplicate WiFi networks: Show
Min WiFi signal required: 8%
Serial debugging: Off

MQTT Information
Server: 10.0.0.4:1883 (Disconnected Never)
Disconnections: 0
Buffer Size: 768 bytes
Client id: ir_server636d93
Command topic(s): ir_server/send, ir_server/send_0
Acknowledgements topic: ir_server/sent
IR Received topic: ir_server/received
Log topic: ir_server/log
LWT topic: ir_server/status
QoS: 1
Last MQTT command seen: (topic) 'None' (payload) 'None' (Never)
Total published: 3
Total received: 0

Climate Information
IR Send GPIO: 4
Last update source: HTTP
Total sent: 1
Last send: Ok (1 Minute 6 Seconds ago)
State listen period: 5 Seconds
State broadcast period: 10 Minutes
Last state broadcast: Never
Last discovery sent: Locked
Discovery topic: homeassistant/climate/ir_server/config
Command topics: ir_server/ac/cmnd/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend)
State topics: ir_server/ac/stat/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend)

I must have a problem at home because I tried with LG and  IRrecvDumpV3.ino  detected ON

crankyoldgit, 2022-01-19:
@step3528 I'm really stumped.

I've tested everything I can think of in the library itself. It all appears to be working as expected, and producing the correct hex/state codes.
The fact you've tested it with `LG` and it appears to be working indicates you are probably using `IRMQTTServer` correctly.
`IRMQTTServer` uses **exactly** the same code for parsing the web interface and turning power on & off for all AC protocols.
So it's unlikely that's the problem too.

I'm going to try hair-brained hail-mary suggestions/solutions now. Can you delete the library code you have on your computer. Download the latest copy from that branch, and build `IRMQTTServer` again from scratch, and wipe & re-flash the device.

@NiKiZe Basically the only things left to verify are:

- Is @step3528 using the IRMQTTServer program correctly? (Probably is)
- Perhaps @step3528 's build is corrupted in some funky way. (Unlikely)
- A problem in `sendClimate()` in IRMQTTServer as described above. (Extremely unlikely as it's also working for `LG` for him).

step3528, 2022-01-19:
Sorry @crankyoldgit I have trouble expressing myself. I have the same problem with LG.

I will reinstall IRMQTTServer.

crankyoldgit, 2022-01-19:
> I have trouble expressing myself. I have the same problem with LG.

Ahhhh. That's helpful info. i.e. LG being broken too. That gives me something to work on.
But yes, please re-install.

NiKiZe, 2022-01-19:
Add a test for the IRMQTTServer path that creates the packet. maybe there is on/off issues that is not propagating from generic to specific AC classes? (sorry short on time ATM)

crankyoldgit, 2022-01-19:
@NiKiZe Yeah. I'm going to have to instrument IRMQTTServer more I think to debug what might be going on there.

I temporarily added debug stuff to print `ac->toString()` in `IRac::hitachi264()` just before the `send()` at:
https://github.com/crankyoldgit/IRremoteESP8266/blob/56e48e2b0eafce0d820de7f3d0e57d08db4abe66/src/IRac.cpp#L1326-L1345
In the following (new) test:
https://github.com/crankyoldgit/IRremoteESP8266/blob/56e48e2b0eafce0d820de7f3d0e57d08db4abe66/test/ir_Hitachi_test.cpp#L2213-L2232
it was working as expected.
e.g.
```
[ RUN      ] TestIRHitachiAc264Class.Issue1729_PowerOntoOff
/* Manually calling hitachi264() */
DEBUG: Power: On, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
DEBUG: Power: Off, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
/* Then calling it via IRac::sendAc()` */
DEBUG: Power: On, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
DEBUG: Power: Off, Mode: 6 (Heat), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)
[       OK ] TestIRHitachiAc264Class.Issue1729_PowerOntoOff (1 ms)
```
So that leads me to think I'm doing something wrong in the `IRMQTTServer` code compared to the unit test code.

step3528, 2022-01-19:
sorry but I think it was me who did not use the interface correctly.

By carrying out a new test, I found that it was the "mode" parameter that allowed to turn off and not the "power" parameter.

[image]

I'm really sorry for the lost time.

NiKiZe, 2022-01-19:
So mode off is power off?
And power off makes no difference?

step3528, 2022-01-19:
> So mode off is power off?

yes, it sends the message off with cool mode

Mesg Desc.: Power: Off, Mode: 3 (Cool), Temp: 25C, Fan: 5 (Auto), Button: 19 (Power/Mode)

> And power off makes no difference?

yes power off has no action

crankyoldgit, 2022-01-20:
> By carrying out a new test, I found that it was the "mode" parameter that allowed to turn off and not the "power" parameter.

Ahhhh. That makes all the difference in pinpointing the issue.

You need to change this parameter to `false`:
https://github.com/crankyoldgit/IRremoteESP8266/blob/030b4033d27cf22ac507a75780bb70e1082e8c7f/examples/IRMQTTServer/IRMQTTServer.h#L138-L152

I'll look into fixing it (power doing what is expected via the web interface) in your scenario. That's sort of a bug. But now I know what is causing it at least. Phew!!

closed 2022-01-20 by crankyoldgit

crankyoldgit, 2022-03-15:
FYI, the changes mentioned above have now been included in the new [v2.8.2 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.2) of the library.

anklav24, 2022-09-05:
Hi guys.
I have a HITACHI RAC-09CH7 with a RAR-2P2 remote and tuya-ir.
I've tried to integrate it in a Home Assistant, but works only raw commands as a couple buttons.
How to make it work as a climate device?

`tuya-ir.yaml`
```yaml
esphome:
  name: tuya-ir
  platform: ESP8266
  board: esp01_1m
  includes:
    - include/hitachi_ir.h
  libraries:
    - crankyoldgit/IRremoteESP8266@^2.8.2
  platformio_options:
    build_flags: 
      - -DSEND_HITACHI_AC264
      - -DDEBUG_AC

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:

api:
  password: !secret esphome_api_password

ota:
  password: !secret esphome_api_password

# Use the blue LED as a status light.
#   Blink slowly = Temporary issue (WiFi/MQTT disconnected, sensor unavailable, ...)
#   Blink quickly = Error encountered
status_led:
  pin: GPIO4

# Configure the IR receiver. Handy to pickup confirmation messages
# from your AC (or capture commands from the actual remote)
remote_receiver:
  pin:
    number: GPIO5
    inverted: true
    mode:
      input: true
      pullup: true
  # high 55% tolerance is recommended for some remote control units
  tolerance: 55%
  dump: raw  # all or raw for my case

# Configure the IR LED: this one sends commands to other devices
remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 50%

# Configure the AC unit you have. Mine is a Hitachi.

switch:
  - platform: template
    name: Turn On A/C
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          # Hitachi AC 24C FanLow SwingOff
          code: [3442, -1657, 440, -1260, 440, -410, 439, -410, 464, -386, 440, -409, 440, -412, 438, -410, 440, -409, 467, -383, 444, -406, 466, -383, 441, -409, 465, -1235, 439, -410, 464, -386, 440, -410, 463, -387, 463, -385, 465, -385, 465, -385, 443, -407, 440, -410, 439, -410, 465, -384, 465, -385, 440, -410, 440, -410, 439, -411, 439, -410, 440, -410, 440, -1259, 441, -409, 440, -1259, 441, -1258, 443, -1257, 439, -1261, 443, -1255, 441, -1259, 466, -383, 440, -1260, 440, -1259, 440, -1259, 445, -1255, 440, -1260, 438, -1260, 440, -1259, 440, -1260, 465, -1235, 444, -405, 465, -385, 440, -409, 465, -385, 439, -411, 439, -410, 440, -410, 465, -384, 465, -385, 441, -412, 437, -1262, 438, -1259, 440, -410, 439, -410, 464, -1235, 440, -1259, 440, -1260, 441, -1258, 441, -409, 463, -386, 466, -1234, 463, -1237, 440, -409, 441, -408, 440, -414, 460, -1236, 439, -410, 441, -410, 439, -1259, 442, -408, 466, -384, 466, -1233, 441, -1259, 440, -411, 463, -1235, 441, -1257, 442, -408, 465, -1235, 441, -1258, 464, -385, 441, -1259, 440, -1260, 442, -407, 441, -409, 440, -1259, 440, -410, 464, -385, 466, -384, 468, -382, 466, -383, 465, -1235, 466, -1232, 466, -385, 442, -1257, 466, -1233, 466, -1234, 439, -411, 439, -410, 464, -386, 467, -385, 463, -384, 465, -1235, 440, -1259, 439, -411, 464, -1235, 439, -1260, 440, -1260, 466, -1236, 440, -1257, 440, -409, 464, -385, 441, -1259, 441, -409, 440, -410, 439, -412, 439, -409, 440, -409, 465, -385, 466, -388, 436, -410, 439, -1259, 441, -1259, 465, -1235, 439, -1260, 441, -1258, 465, -1235, 439, -1260, 440, -1259, 442, -408, 468, -381, 441, -409, 465, -385, 439, -411, 440, -409, 440, -410, 444, -406, 441, -1262, 436, -1261, 440, -1258, 439, -1260, 441, -1258, 442, -1258, 440, -1260, 439, -1259, 440, -410, 440, -410, 440, -410, 439, -410, 466, -383, 440, -411, 441, -408, 440, -411, 439, -1259, 466, -1233, 465, -1235, 440, -1262, 438, -1258, 487, -1212, 467, -1233, 440, -1260, 440, -409, 440, -410, 440, -409, 441, -409, 465, -385, 464, -385, 442, -410, 464, -383, 442, -1258, 465, -1234, 442, -1257, 465, -1235, 440, -1259, 441, -1258, 466, -1234, 444, -1255, 465, -386, 438, -411, 464, -387, 463, -385, 464, -386, 465, -385, 440, -410, 465, -385, 439, -1259, 441, -1259, 440, -1259, 464, -1235, 440, -1260, 438, -1261, 466, -1233, 440, -1260, 439, -1260, 441, -1258, 440, -410, 464, -386, 464, -1235, 465, -384, 465, -385, 442, -408, 441, -409, 439, -411, 439, -1260, 464, -1235, 465, -384, 441, -1259, 440, -1260, 440, -1258, 466, -1238, 437, -408, 467, -383, 465, -385, 443, -1256, 465, -384, 465, -1235, 443, -1256, 465, -385, 464, -1236, 442, -1258, 438, -1261, 438, -411, 465, -1234, 466, -388, 462, -383, 466, -1234, 443, -407, 445, -1254, 465, -384, 442, -409, 438, -410, 466, -385, 441, -409, 440, -409, 464, -1235, 440, -410, 440, -1259, 440, -1260, 441, -1258, 440, -1260, 439, -1260, 469, -1230, 440, -410, 465, -385, 440, -1258, 442, -409, 440, -411, 438, -410, 440, -409, 441, -410, 440, -1258, 465, -1235, 465, -385, 440, -1258, 441, -1259, 466, -1237, 437, -1259, 465]

  - platform: template
    name: Turn On A/C Swing Toggle
    turn_on_action:
[... 153 lines cut ...]
      this->swing_mode = swing_mode;
      this->publish_state();
    }

    //Code for what to do when the temperature is changed on the dashboard
    if (call.get_target_temperature().has_value()) {
      // User requested target temperature change
      float temp = *call.get_target_temperature();
      // Send target temp to climate
      ac.setTemp(temp);

      this->target_temperature = temp;
      this->publish_state();
    }

  //Send the IR code you've built basis all the above data
  ac.send();
  #if DEBUG_AC
  ESP_LOGD("DEBUG", "Home A/C remote is in the following state:");
  ESP_LOGD("DEBUG", "%s\n", ac.toString().c_str());
  #endif // DEBUG_AC
  }
};

```

ppkstat, 2023-08-19:
@anklav24 Have you managed to find a solution to this?

anklav24, 2023-08-19:
@ppkstat No, I keep using raw commands.

ppkstat, 2023-08-19:
That's unfortunate. I'll let you know if I find a solution, but that;s not
very likely since I am not a developer.

Have you tried posting this in the home assistant forums? In any case,
thank you for your work.

Best,
Tom

anklav24, 2023-08-19:
@ppkstat I tried. There's no answer there like in irremote library repo too.

PR #1735 (2022-01-20): HITACHI_AC264: Add minimal detailed support.
* Supports Power, Temp, Mode, & Fanspeed. That's it.
  - Effectively a parent class of `IRHitachiAc424`.
* Add `IRac` integration.
* Update & add unit tests.

Fixes #1729
commit 91a528ec (2022-01-20): HITACHI_AC264: Add minimal detailed support. (#1735)

* Supports Power, Temp, Mode, & Fan speed. That's it.
  - Effectively a parent class of `IRHitachiAc424`.
* Add `IRac` integration.
* Update & add unit tests.

Fixes #1729