crankyoldgit/IRremoteESP8266 #1733: Additional Midea toggle codes

normanr, 2022-01-10:
[#1318](https://github.com/crankyoldgit/IRremoteESP8266/issues/1318#issuecomment-723171666) mentioned that self-clean doesn't produce an ir signal, but it only works in some modes. The `Silence/FP` button is also mentioned, but it was not fully documented. The full codes are:
```
const uint64_t kMideaACToggleSelfClean = 0xA20DFFFFFF70;         // Mode must be Auto, Cool, or Dry
const uint64_t kMideaACToggleFreezeProtection = 0xA20FFFFFFF73;  // Mode must be Heat
const uint64_t kMideaACToggleSilenceOn = 0xA212FFFFFF6E;
const uint64_t kMideaACToggleSilenceOff = 0xA213FFFFFF6F;
```
The Lennox mini-split wireless controller user manual can be found here: https://coolairsolutions.com/wp-content/uploads/Lennox-Minisplit-Wireless-Remote-M0STAT60Q-1-User-Manual.pdf. The Mr COOL mini-split systems look identical to the Lennox systems, their remote manual can be found here: https://mrcool.com/wp-content/dox_repo/mc-rmt-um-en-01.pdf. I suspect they're the same base-unit just with slightly different control electronics, eg Mr COOL offers a Smart Wifi Kit, but I can't find one for Lennox.

Freeze Protection is also called "Frost Prevention" in some places. Silence is a toggle and actives a lower-than-low fan speed, replacing the fan speed indicator with a picture of "a person holding their index finger up to their mouth" ("hush").

crankyoldgit, 2022-01-10:
> Lennox systems, their remote manual can be found here: https://mrcool.com/wp-content/dox_repo/mc-rmt-um-en-01.pdf

FYI, the "direct" link above does not work.

> Freeze Protection is also called "Frost Prevention" in some places.

There are other brands/models that have a similar feature. I may use different text to describe it when decoding, but I'll see what I can do/add.

crankyoldgit, 2022-01-11:
Hi @normanr Can you please download/test PR #1734 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1733 and confirm if that adds the functionality you've documented?

Fingers crossed I got it all correct.

normanr, 2022-01-11:
> > Lennox systems, their remote manual can be found here: https://mrcool.com/wp-content/dox_repo/mc-rmt-um-en-01.pdf
> FYI, the "direct" link above does not work.

blegh, it works fine if the referrer is missing (copy the link and paste it manually, you may need to refresh) or one of the well known search engines - github.com is not on the "allow list".

> > Freeze Protection is also called "Frost Prevention" in some places.
> 
> There are other brands/models that have a similar feature. I may use different text to describe it when decoding, but I'll see what I can do/add.

sgtm.

> Hi @normanr Can you please download/test PR #1734 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1733 and confirm if that adds the functionality you've documented?
> 
> Fingers crossed I got it all correct.

I don't have an ESP8266 that I can test with. I have not yet tried to integrate the library with the code I wrote for the BL602 device I finally got NuttX running on, but it has been extremely useful to understand how to decode/encode the midea signals.

I reviewed the code: `M0STAT60Q-1` doesn't appear anywhere on the physical remote (only the manual). I assume it's a Lennox specific code. The model number of the back of the remote is `RG57A6/BGEFU1` (like the existing Mr Cool brand remote listed in the code).

I also noticed that quiet/silent mode is (currently) a separate toggle, where-as on the remote it behaves more like an extra fan speed. If you enable silent (toggle on code is sent), then fan speed "disappears" on the remote (silent shows instead). If you disable silent then fan speed re-appears however it was before (only the toggle silence off code is sent by the remote). If silent is on and you change fan speed, then silent is forced off on the remote's display (and only the regular mode/fan/setpoint code is sent by the remote).

I'm not 100% sure that my indoor unit understands what to do with the silent on/off codes because it doesn't beep, or seem to change behavior (I have not confirmed that the outdoor unit behavior is unchanged). Interestingly the Lennox manual annotates the figure on the remote screen as "Not used" (like ECO and the leaf), but the text provides details about silent/quite mode, so it's not consistent. The Mr Cool manual  does not include the "Not used" graphics (including the silence icon), but also describes the function in the text. (I noticed that the "%" and "H" units are not described at all, and also never seem to be used).

crankyoldgit, 2022-01-12:
> I don't have an ESP8266 that I can test with. I have not yet tried to integrate the library with the code I wrote for the BL602 device I finally got NuttX running on, but it has been extremely useful to understand how to decode/encode the midea signals.

Bummer, then I guess you can't verify it then.

> 
> I reviewed the code: `M0STAT60Q-1` doesn't appear anywhere on the physical remote (only the manual). I assume it's a Lennox specific code. The model number of the back of the remote is `RG57A6/BGEFU1` (like the existing Mr Cool brand remote listed in the code).

Okay, I'll adjust it.
Just confirming. The remote is Lennox branded?
And what is the brand/model etc of the actual AC unit?

> I also noticed that quiet/silent mode is (currently) a separate toggle, where-as on the remote it behaves more like an extra fan speed. If you enable silent (toggle on code is sent), then fan speed [...]

Alas, that is the best way to implement it given the rest of the library.

> I'm not 100% sure that my indoor unit understands what to do with the silent on/off codes because it doesn't beep, or seem to change behavior (I have not confirmed that the outdoor unit behavior is [...]

Bummer. Oh well, hopefully someone will find it useful for their A/C.

normanr, 2022-01-12:
> Bummer, then I guess you can't verify it then.

Yea, I was trying to figure out if I could hook IRremoteESP8266 into my existing code. Maybe if I define `UNIT_TEST` so that the ESP code is ~mostly disabled, and then populate rawbuf and rawlen from my capture routine it would work. I currently wait for a POSIX signal (triggered from an interrupt) that the gpio pin has changed, and then disable task scheduling and tight loop reading the gpio pin. I get very reasonable results that way (I have not tried to go 100% with interrupts/signals, because I feared it wouldn't be good enough. Also I currently only have a 1ms resolution clock, so raw loop counts are more accurate). Annoyingly the BL602 has a hardware support for receiving IR code, but the receive buffer only has 64-bits of data (so fine for "regular" remotes, but useless for HVAC).

> Okay, I'll adjust it. Just confirming. The remote is Lennox branded? And what is the brand/model etc of the actual AC unit?

Yes, It has Lennox branding (pictures: [front](https://darkskies.za.net/~norman/images/lennox/Remote%20Front.jpg) [back](https://darkskies.za.net/~norman/images/lennox/Remote%20Back.jpg)). The indoor units are model `MWMA009S4-3P` (and `MWMA012S4-3P` for the large room). Based on the convert page of their [product specifications pdf](https://tech.lennoxintl.com/C03e7o14l/VIu12Ch2uV/ehb_mpb_multizone_2012.pdf) (11MB) I'd guess the remote works with all of their mini-split non-ducted indoor units (`MWMA/MFMA/M22A/M33A/MCFA/MCFB`)

> Alas, that is the best way to implement it given the rest of the library.

acknowledged. It wasn't clear from #1528 whether the logic was the same or not (I suspect it might be). Would need to get @thanxx to check if TCL works in the same way or not.

crankyoldgit, 2022-01-13:
> Yea, I was trying to figure out if I could hook IRremoteESP8266 into my existing code. Maybe if I define `UNIT_TEST` so that the ESP code is ~mostly disabled, and then populate rawbuf and rawlen [...]

Correct, enabling `UNIT_TEST` will make it safe to compile on a "normal" system. e.g. See the programs under `test/` e.g. `test/ir_Midea_test.cpp`. There are plenty of examples in that directory of how to simulate/extract/reproduce signals generated by the library.

> Yes, It has Lennox branding (pictures: [front](https://darkskies.za.net/~norman/images/lennox/Remote%20Front.jpg) [back](https://darkskies.za.net/~norman/images/lennox/Remote%20Back.jpg)). The [...]

Thanks for the info.

PR #1734 (2022-01-13): MIDEA: Add support for Quiet, Clean & Freeze Protect controls.
* Add support for Quiet setting controls.
* Add support for Clean and Freeze Protect toggle controls/settings/logic.
* Add & update unit test coverage.
* Minor code and output cleanup.
* Update supported model info.

Fixes #1733
commit 9da665c7 (2022-01-13): MIDEA: Add support for Quiet, Clean & Freeze Protect controls. (#1734)

* Add support for Quiet setting controls.
* Add support for Clean and Freeze Protect toggle controls/settings/logic.
* Add & update unit test coverage.
* Minor code and output cleanup.
* Update supported model info.

Fixes #1733
closed 2022-01-13 by crankyoldgit