crankyoldgit/IRremoteESP8266 #1737: Adjust for AKB74955603 (LG2 Model 3)

kzagatti, 2022-01-15:
Hi @crankyoldgit,

It's been a while since we worked on AKB74955603 and AKB73757604, rock solid with IRMQTTServer and Home Assistant.

I have a bedroom where along with HA, I started using the original remote (you know, wife). The device properly decodes the IR sent by the remote and things such as modes, on/off, temp, etc, are all OK. But when I press Light Toggle, it shows as "off" on Home Assistant (although it's still on).

This is the output from IRrecvDumpV2:

```
Protocol  : LG2
Code      : 0x88C00A6 (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Light Toggle: On
uint16_t rawData[59] = {3424, 9810,  588, 1494,  588, 492,  538, 502,  538, 494,  538, 1524,  544, 502,  538, 494,  538, 496,  536, 1538,  540, 1526,  540, 474,  564, 494,  538, 494,  538, 496,  538, 494,  538, 504,  538, 476,  564, 494,  538, 496,  538, 484,  564, 1516,  566, 474,  564, 1534,  542, 494,  538, 496,  538, 1518,  566, 1532,  564, 476,  564};  // LG2 88C00A6
uint32_t address = 0x88;
uint32_t command = 0xC00A;
uint64_t data = 0x88C00A6;
```

And these are the MQTT messages (latest first):

1-) When I turn the A/C on:

```
Message 11 received on ir_server/ac/stat/fanspeed at 4:20 PM:
Min
QoS: 0 - Retain: false
Message 10 received on ir_server/ac/stat/temp at 4:20 PM:
18
QoS: 0 - Retain: false
Message 9 received on ir_server/ac/stat/mode at 4:20 PM:
fan_only
QoS: 0 - Retain: false
Message 8 received on ir_server/ac/stat/power at 4:20 PM:
on
QoS: 0 - Retain: false
```

2-) When I press the Light button:

```
Message 16 received on ir_server/ac/stat/light at 4:20 PM:
off
QoS: 0 - Retain: false
Message 15 received on ir_server/ac/stat/fanspeed at 4:20 PM:
High
QoS: 0 - Retain: false
Message 14 received on ir_server/ac/stat/temp at 4:20 PM:
15
QoS: 0 - Retain: false
Message 13 received on ir_server/ac/stat/mode at 4:20 PM:
off
QoS: 0 - Retain: false
Message 12 received on ir_server/ac/stat/power at 4:20 PM:
off
QoS: 0 - Retain: false
````

Please, is it possible to adjust the code to not update the power mode when the Light Toggle is pressed?

Thank you

crankyoldgit, 2022-01-16:
I'll see what I can do.

crankyoldgit, 2022-01-16:
@kzagatti Can you please download and try out branch https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1737
Hopefully it will do what you want. Please let us know asap. If it works, I'll make it into a PR and get it merged.

kzagatti, 2022-01-16:
@crankyoldgit compiled IRMQTT from the branch but the final result is the same, when I press the Light button on the remote, it shows as "off" on HA. Not sure if it helps, but below you'll also find the prints from IRMQTTServer Air Conditioner Control:

1-) When I press the power on button:

[image]

2-) Right after pressing Light button:

[image]

crankyoldgit, 2022-01-17:
Hmmm.
My tests simulating the issue indicate it should be fixed. i.e. Power: On should remain after it receives the "Light Toggle" button press from a real remote.
Can you please confirm/check you're building the example code against the downloaded version from _**that**_ branch by modifying the `src/IRremoteESP8266.h` file where you downloaded it. Modify these lines:
https://github.com/crankyoldgit/IRremoteESP8266/blob/9228143207ab7e2968a42d9a92e246237af7bb83/src/IRremoteESP8266.h#L55-L61
to something unique. e.g. `#define _IRREMOTEESP8266_VERSION_MAJOR 4` or something.
You can check on the "Info" page of your ESP running IRMQTTServer what version it's running.

Also, can you provide the annotated MQTT log your provided earlier for your tests? That was quite helpful in track it down.

crankyoldgit, 2022-01-17:
and while I think about it, Please also include the MQTT messages sent to the "IR Received topic" (which is listed on the "Info" page.

That should be `ir_server/received` if I've read the data you've supplied correctly.

crankyoldgit, 2022-01-17:
> I deleted the original library and pasted from the branch and it is working!

Thanks for doing that. Is that it is _all_ working as expected now, or is there something else/more I need to address?

kzagatti, 2022-01-17:
> Thanks for doing that. Is that it is _all_ working as expected now, or is there something else/more I need to address?

 It is _all_ working as expected!

PR #1738 (2022-01-18): LG2: Improve Light toggle msg handling.
Don't override previous settings if the current message being processed is a Light Toggle message.

Confirmed working.

Fixes #1737
commit 030b4033 (2022-01-18): LG2: Improve Light toggle msg handling. (#1738)

Don't override previous settings if the current message being processed is a Light Toggle message.

Confirmed working.

Fixes #1737
closed 2022-01-18 by crankyoldgit