crankyoldgit/IRremoteESP8266 #1739: IRMQTTServer: Web AC power control not behaving as expected

crankyoldgit, 2022-01-20:
> By carrying out a new test, I found that it was the "mode" parameter that allowed to turn off and not the "power" parameter.

Ahhhh. That makes all the difference in pinpointing the issue.

You need to change this parameter to `false`:
https://github.com/crankyoldgit/IRremoteESP8266/blob/030b4033d27cf22ac507a75780bb70e1082e8c7f/examples/IRMQTTServer/IRMQTTServer.h#L138-L152
i.e.
```c++
#define MQTT_CLIMATE_HA_MODE false
```

I'll look into fixing it (power doing what is expected via the web interface) in your scenario. That's sort of a bug. But now I know what is causing it at least. Phew!!

_Originally posted by @crankyoldgit in https://github.com/crankyoldgit/IRremoteESP8266/issues/1729#issuecomment-1017010479_

crankyoldgit, 2022-01-20:
@step3528 Can you please download and try out IRMQTTServer from branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/irmqttserver_161

I think that will fix "power" control via the web interface. I haven't tested it, as it easier for you to test it in your environment. I _think_ that should fix it.

Please let me know.

step3528, 2022-01-20:
Yes this solved the problem.

thank you

crankyoldgit, 2022-01-20:
Huzzah!

PR #1740 (2022-01-20): IRMQTTServer V1.6.1
- Fix the case when `MQTT_CLIMATE_HA_MODE` is enabled **and** Home Assistant is **_NOT_** being used, **and** `power` is turned _Off_ via the http interface, that it also sets the `mode` to _Off_, and vice-versa; i.e. Turning `mode` _"Off"_ also sets `power` _"Off"_.
- Move some literal strings to constants for consistency.
  - Helps with potential internationalisation.
- Bump version patch number.

Ref #1729
Fixes #1739
commit ccb8542d (2022-01-20): IRMQTTServer V1.6.1 (#1740)

- Fix the case when `MQTT_CLIMATE_HA_MODE` is enabled & Home Assistant is NOT being used, and `power` is turned off via the http interface, that it also sets the `mode` to Off, and viceversa. Mode "Off" also sets Power "Off".
- Move some literal strings to constants for consistency.
  - Helps with potential internationalisation.
- Bump version patch number.

Ref #1729
Fixes #1739
closed 2022-01-20 by crankyoldgit