crankyoldgit/IRremoteESP8266 #1745: Add basic support for 168 bit KELON protocol.

crankyoldgit, 2022-01-28:
@crankyoldgit,
Thanks for your all replies. Codes I get from IRrecvDumpV2:

```
13:23:57.265 -> Timestamp : 000153.655
13:23:57.265 -> Library   : v2.8.0
13:23:57.265 -> 
13:23:57.265 -> Protocol  : KELON
13:23:57.265 -> Code      : 0x178D000070030683 (48 Bits)
13:23:57.265 -> Mesg Desc.: Temp: 25C, Mode: 0 (Heat), Fan: 3 (High), Sleep: Off, Dry: 0, Timer: Off, Turbo: Off
13:23:57.265 -> uint16_t rawData[343] = {8922, 4494,  548, 1714,  524, 1718,  524, 572,  524, 550,  550, 556,  550, 560,  548, 586,  524, 1724,  526, 564,  524, 1692,  550, 1696,  550, 552,  550, 554,  550, 560,  548, 586,  524, 550,  548, 1688,  550, 1694,  548, 548,  550, 552,  548, 556,  548, 582,  524, 586,  524, 576,  524, 540,  548, 568,  526, 572,  526, 548,  550, 1702,  550, 1706,  550, 1734,  526, 574,  526, 564,  526, 544,  550, 548,  548, 550,  550, 554,  550, 582,  526, 586,  524, 550,  548, 540,  550, 544,  550, 570,  526, 550,  550, 558,  546, 582,  524, 586,  524, 562,  524, 7978,  548, 1690,  548, 568,  524, 1696,  548, 1700,  548, 554,  550, 560,  548, 560,  550, 1722,  526, 1686,  550, 1692,  548, 1696,  550, 576,  524, 1704,  548, 558,  550, 562,  548, 550,  550, 564,  524, 570,  524, 548,  548, 576,  524, 578,  526, 558,  550, 586,  524, 574,  524, 566,  524, 568,  526, 548,  548, 576,  526, 556,  548, 558,  550, 560,  550, 550,  550, 564,  526, 544,  548, 548,  548, 552,  548, 556,  548, 582,  526, 564,  548, 550,  550, 540,  548, 568,  526, 546,  550, 550,  550, 580,  524, 558,  550, 560,  550, 1698,  550, 542,  548, 542,  550, 546,  550, 1722,  524, 1706,  548, 582,  526, 586,  524, 574,  526, 1690,  548, 544,  550, 546,  550, 552,  548, 1726,  526, 1730,  524, 1734,  526, 562,  524, 7976,  550, 566,  524, 544,  550, 570,  526, 550,  548, 554,  550, 1706,  550, 562,  548, 574,  526, 542,  548, 544,  550, 572,  526, 548,  552, 554,  550, 582,  526, 584,  526, 574,  524, 542,  548, 544,  548, 572,  524, 552,  548, 578,  524, 560,  548, 562,  548, 550,  550, 540,  550, 546,  548, 572,  524, 576,  526, 556,  548, 560,  546, 564,  548, 574,  526, 540,  550, 546,  546, 546,  550, 1698,  550, 1728,  524, 1706,  548, 564,  548, 574,  524, 544,  548, 568,  526, 548,  548, 574,  524, 554,  550, 558,  550, 562,  548, 550,  550, 566,  524, 544,  548, 546,  552, 1698,  548, 1702,  550, 560,  546, 586,  524, 538,  546};  // KELON 178D000070030683
13:23:57.466 -> uint32_t address = 0x80000000;

13:23:57.466 -> uint32_t command = 0x207118;

13:23:57.466 -> uint64_t data = 0x178D000070030683;

13:23:57.466 -> 

13:23:57.466 ->
```

Tomorrow maybe I try with "rawData" that should be easier

_Originally posted by @mp3-10 in https://github.com/crankyoldgit/IRremoteESP8266/discussions/1744#discussioncomment-2061968_

mp3-10, 2022-01-28:
Hello,
More info:

- Device: Hisense/Kelon AC interior unit AST-09UW4RVETG00A
- Remote: DG11R2-01.

PowerOn/Power Off etc is the same (probably) - I spend for now only half day on it - as on Whirpool AC 

I add soon more dumped codes (in spreadsheet), and I will try decode.

I'm focused on "iFeel" function, because Hisense thermometer (and logic on mainboard of AC) is stupid, so when you put, on heating mode, 23 degrees, you have  up to 28 at room  (+5 degree) and starts from 19 (this i'm not pretty sure yet).

crankyoldgit, 2022-01-28:
Thanks for the model info.

I've now created PR #1747 / Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/kelon168_basic which should support basic (sending of `state[]`s) support for the device.

You're now at this step in the process for adding detailed (i.e. changing settings) support: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#a-note-on-collecting-data

Please download that branch, recompile IRrecvDumpV2 etc and try to re-capture. etc etc. Please let us know how it goes.

crankyoldgit, 2022-02-01:
@mp3-10 ping

mp3-10, 2022-02-01:
@crankyoldgit  ack :) 
This weekend I don't have opporunity to play with my Wemos 😭 I will try this weekend spend some time

mp3-10, 2022-02-05:
@crankyoldgit, 

Today I started working with that remote again :) As I see normal (without iFeel) control is using: WHIRLPOOL_AC  Model:  2 (DG11J191)

But during iFeel Mode, ( now i'm catching data):

13:38, 24 degrees on remote, 23 set on AC unit, heating:
13:39:08.429 -> Code      : 0x8306007000008D2600000080184300000000380038 (168 Bits)

13:38, 25 degrees on remote, 23 set on AC unit, heating:
13:39:47.256 -> Code      : 0x8306007000000D270000008019C300000000380038 (168 Bits)

I'm now sniffing, so I will be back soon with more data and sheet where i mark differences :)

mp3-10, 2022-02-05:
After some hours, I decoded some of the code:

Results: https://1drv.ms/x/s!AqpZPk4W5RXIg-8136-zyOU_diZHjA?e=2oY5CX.

closed 2022-02-20 by crankyoldgit

crankyoldgit, 2022-03-15:
FYI, the changes mentioned above have now been included in the new [v2.8.2 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.2) of the library.

gianricod, 2022-11-15:
> After some hours, I decoded some code:
> 
> Results: https://1drv.ms/x/s!AqpZPk4W5RXIg-8136-zyOU_diZHjA?e=2oY5CX.
> 
> I found also what way is checksum generated. Works for 58 from 59 tests.
> 
> Next week, I try to check if my dump is working (for now I only based on "excel sheet" :)

@mp3-10  were you able to come up with something that can be used to send temperature readings from an ESPx to the split without using the remote (that sucks)?

mp3-10, 2022-11-16:
@gianricod I found that, WiFi module have pinout "A" and "B" so that looks like RS485, but after connection any usable signal :(

I can't provide more, because my split is mounted on wall, so I can't disassemble to check mainboard ;)

PR #1747 (2022-02-20): Experimental basic support for Kelon 168 bit / 21 byte protocol.
* Add `sendKelon168()` & `decodeKelon168()`
* Add & extend unit test coverage.
* Code style cleanup.

Fixes #1745
Ref #1744