crankyoldgit/IRremoteESP8266 #1748: Need help with COOLIX vs COOLIX48

Pigi-102, 2022-01-29:
I've followed the web-ac-controller example to create a sketch to control my AC unit.
The unit use the 24 bit protocol ( derived by testing the remote control with the irdump example ) but whatever I do in my sketch I always send out a 48-bit message vs a 24-bit.
Cannot find anywhere a way to force the use of the 24-bit protocol.

Can someone share a bit of light on this ?

Here to sketch:

```
#include <WebOTA.h>
#include <EEPROM.h>

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Coolix.h>
const uint16_t kIrLed = D7;  
#define AUTO_MODE kCoolixAuto
#define COOL_MODE kCoolixCool
#define DRY_MODE kCoolixDry
#define HEAT_MODE kCoolixHeat
#define FAN_MODE kCoolixFan

#define FAN_AUTO kCoolixFanAuto
#define FAN_MIN kCoolixFanMin
#define FAN_MED kCoolixFanMed
#define FAN_HI kCoolixFanMax
#define SEND_COOLIX true
#undef SEND_COOLIX48 

#include <PubSubClient.h>

#include <DHT_U.h>
#include "DHT.h"
#include "PubSubClient.h" // Connect and publish to the MQTT broker
#include "ESP8266WiFi.h"  // Enables the ESP8266 to connect to the local network (via WiFi)

IRCoolixAC ac(kIrLed);  // Set the GPIO to be used to sending the message.

int HeatingPin = D7;
String switch1 = "ON" ;
String strTopic;
String strPayload;

#define DHTPIN D5         // Pin connected to the DHT sensor
#define DHTTYPE DHT11     // DHT11 or DHT22
DHT dht(DHTPIN, DHTTYPE);
float variaz=0.01;

const char* vers = "v 0.1";                   // Project version
                                              // WiFi
const char* ssid = "SSID";                 // Your personal network SSID
const char* wifi_password = "PASSWORD";  // Your personal network password

String location = "AC_REMOTE";

String AllHum   = ("homeassistant/"+location+"/humidity");
String AllTem   = ("homeassistant/"+location+"/temperature");
String AllSwi1  = ("homeassistant/"+location+"/switch1");
String AllTtemp = ("homeassistant/"+location+"/target_temp");
String AllMode  = ("homeassistant/"+location+"/mode");
String AllModeSt  = ("homeassistant/"+location+"/mode_st");
String AllFan   = ("homeassistant/"+location+"/fan");
String AllSwing = ("homeassistant/"+location+"/swing");
String ModeSt_Rec = "";
String ModeSt_Rec1 = "";
String FanSt_Rec = "";
String TempSt_Rec = "";
String ModeSt_temp ="";
uint8  Epr_mod=0;
uint8  firstboot=7;
[... 222 lines cut ...]
     // If the message failed to send, we will try again, as the connection may have broken.
     else {
//       Serial.println("Temperature failed to send. Reconnecting to MQTT Broker and trying again");
       client.connect(clientID, mqtt_username, mqtt_password);
       delay(10); // This delay ensures that client.publish doesn't clash with the client.connect call
       client.publish(temperature_topic, String(t).c_str());
     }

     // PUBLISH to the MQTT Broker (topic = Humidity, defined at the beginning)
     if (client.publish(humidity_topic, String(h).c_str())) {
       delay(1); // Solo per riempire l' if
//        Serial.println("Humidity sent!");
     }
     // Again, client.publish will return a boolean value depending on whether it succeded or not.
     // If the message failed to send, we will try again, as the connection may have broken.
     else {
          Serial.println("Humidity failed to send. Reconnecting to MQTT Broker and trying again");
          client.connect(clientID, mqtt_username, mqtt_password);
          delay(10); // This delay ensures that client.publish doesn't clash with the client.connect call
          client.publish(humidity_topic, String(h).c_str());
     }
  }

}
```

crankyoldgit, 2022-01-29:
> but whatever I do in my sketch I always send out a 48-bit message vs a 24-bit.

Can you supply a capture of what it is sending? And what kind of settings you set the `IRCoolixAC` class to before sending it. 

At first glance, I can't see anything obviously wrong in your code.
Also, what version/revision of the library are you using?

Have you tried simplifying your code down to the bare minimum that produces the fault/issue? e.g. Like one of the many "TurnOn" examples.

`IRCoolixAC::send()` should only ever produce a "COOLIX" message, not a "COOLIX48" message.
Basically, a COOLIX message is a subset of COOLIX48 messages. COOLIX messages physically contain 48-bits too, but there is a pattern to them. i.e. every second byte is inverted. That means we can shrink it to 24 bits. A COOLIX48 message is just one that does not follow that pattern.

crankyoldgit, 2022-01-29:
> #define SEND_COOLIX true
> #undef SEND_COOLIX48

FYI, that will not have the effect you think it does.

Pigi-102, 2022-01-29:
Hello, 
 that's what get captured:
```
Timestamp : 000013.780
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : COOLIX48
Code      : 0xB24D7B84E01F (48 Bits)
uint16_t rawData[199] = {4642, 4502,  514, 1706,  516, 624,  488, 1704,  514, 1702,  516, 624,  488, 624,  488, 1702,  514, 626,  488, 620,  488, 1702,  4
90, 620,  512, 620,  488, 1728,  488, 1704,  514, 624,  488, 1704,  514, 620,  488, 1702,  490, 1722,  488, 1724,  514, 1728,  488, 600,  512, 1728,  490, 1706,  512, 1698,  488, 646,  486, 622,  462, 646,  488
, 624,  488, 1704,  514, 626,  488, 628,  460, 1724,  514, 1702,  514, 1724,  462, 646,  488, 624,  488, 624,  488, 626,  486, 602,  488, 646,  460, 648,  486, 626,  488, 1704,  486, 1724,  488, 1748,  488, 170
4,  514, 1708,  488, 5312,  4648, 4494,  488, 1704,  486, 646,  486, 1698,  512, 1700,  488, 646,  462, 646,  486, 1728,  462, 648,  484, 622,  462, 1724,  510, 622,  488, 626,  488, 1702,  514, 1728,  490, 626
,  488, 1730,  462, 646,  488, 1704,  512, 1724,  486, 1698,  514, 1728,  488, 626,  488, 1728,  488, 1704,  514, 1700,  512, 620,  486, 620,  488, 620,  486, 626,  490, 1728,  488, 626,  488, 628,  460, 1750, 
 488, 1728,  488, 1704,  488, 646,  488, 620,  488, 624,  488, 626,  488, 626,  462, 646,  462, 644,  488, 626,  488, 1728,  490, 1704,  486, 1724,  514, 1724,  488, 1728,  488};  // COOLIX48 B24D7B84E01F
uint64_t data = 0xB24D7B84E01F;
```
This is ( or at least should be )  an "off"  messages.
Using the same decoder, with the original IR Remote the message for an of is the following:
```
Timestamp : 000023.275
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : COOLIX
Code      : 0xB27BE0 (24 Bits)
Mesg Desc.: Power: Off
uint16_t rawData[199] = {4386, 4384,  528, 1618,  528, 546,  526, 1618,  528, 1620,  528, 546,  526, 546,  526, 1618,  528, 546,  526, 546,  528, 1618,  528, 546,  528, 546,  526, 1618,  528, 1620,  528, 546,  528, 1618,  526, 546,  526, 1620,  528, 1618,  504, 1642,  528, 1618,  504, 572,  502, 1644,  528, 1618,  504, 1644,  528, 544,  502, 570,  502, 570,  502, 570,  524, 1622,  502, 572,  502, 570,  502, 1644,  502, 1644,  504, 1642,  504, 570,  502, 570,  502, 570,  502, 570,  502, 570,  502, 570,  502, 572,  502, 570,  526, 1620,  502, 1644,  502, 1644,  502, 1644,  504, 1644,  502, 5232,  4412, 4360,  526, 1620,  502, 570,  526, 1620,  502, 1644,  502, 572,  526, 546,  526, 1618,  502, 572,  502, 572,  526, 1620,  522, 550,  502, 572,  526, 1618,  528, 1620,  526, 546,  526, 1620,  526, 546,  526, 1618,  528, 1618,  528, 1618,  528, 1618,  528, 546,  526, 1618,  526, 1620,  528, 1618,  528, 546,  526, 546,  528, 546,  526, 546,  526, 1618,  528, 546,  528, 544,  528, 1618,  528, 1618,  528, 1618,  528, 544,  528, 546,  526, 546,  528, 544,  528, 544,  528, 544,  526, 546,  528, 544,  528, 1616,  528, 1618,  528, 1618,  528, 1618,  530, 1616,  528};  // COOLIX B27BE0
uint64_t data = 0xB27BE0;
```

Library version should be the lastone as I have d/loaded with the Arduino ide a couple of days ago
From the Ide library manager I have the 2.8.1 version.

Pigi

crankyoldgit, 2022-01-29:
`0xB27BE0` is the correct code for "Power Off" and it turns out `0xB24D7B84E01F` is the 48 bit version of the 24-bit `0xB27BE0`. I hand checked it.
So that leaves a slight timing tolerance issue in the COOLIX detection/decoding. i.e. This is a false negative. I'll try to see if I can diagnose it deeper to see which part of the message if failing.

While I do that, I suggest you tweak this value/line in the dump code:
https://github.com/crankyoldgit/IRremoteESP8266/blob/f8e5662776102e57cf4c03af5e9fc54d14a07b19/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L112
Try something larger than `25`. i.e. `30` or `35`. It _may_ give you the correct results with that.

crankyoldgit, 2022-01-29:
Yep. a Tolerance value of `27` would have matched that "Off" message as a 24-bit COOLIX message.

crankyoldgit, 2022-01-29:
In short, the real device might accept it as is.

Pigi-102, 2022-01-29:
I'm actually using the IRrecvDumpV3 as receiver and already has the line you suggest.
Could I modify the line 
```
const uint8_t kTolerancePercentage = kTolerance;  // kTolerance is normally 25%
```

with a bit more tollerance so I can see the message ?
```
const uint8_t kTolerancePercentage = 30 
```

I'm not sure about my code and would like to see all messages I can send  before trying to the real device.

Thnks for your time  !

Pigi-102, 2022-01-29:
Yes ! I do confim that modifying with 30 will make dump works

Thanks a million for you time !

PR #1750 (2022-01-29): Refactor `decodeCOOLIX()` code & add another test case.
* Remove the old auto scaling version of the code. It made it harder to diagnose problems.
* Use `decodeGeneric()` as much as we can, and decode in whole 8-bit chuncks at a time.
* Try to make the code more understandable and smaller.
* Standardise on the same extra tolerance for Coolix protocols.
* Add a unit test to confirm the library now matches the provided example that was previously slight out of tolerance.

Fixes #1748
commit c5b36c8d (2022-01-29): Refactor `decodeCOOLIX()` code & add another test case. (#1750)

* Remove the old auto scaling version of the code. It made it harder to diagnose problems.
* Use `decodeGeneric()` as much as we can, and decode in whole 8-bit chuncks at a time.
* Try to make the code more understandable and smaller.
* Standardise on the same extra tolerance for Coolix protocols.
* Add a unit test to confirm the library now matches the provided example that was previously slight out of tolerance.

Fixes #1748
closed 2022-01-29 by Pigi-102