crankyoldgit/IRremoteESP8266 #1749: IRsend::encodePioneer documentation might be misleading

shin10, 2022-01-29:
Hi David,

thanks for this awesome library. I just tried it for the first time following the codes listed [here](https://www.pioneerelectronics.com/pio/pe/images/portal/cit_3424/319012430PDP4270HD_IR.pdf) and I'm super happy that my remote is working great.
Anyway, I had some issues on the way and thought I might leave a note.

A single code run as command works as expected:
```
irsend.sendPioneer(irsend.encodePioneer(0, command), 32, 1);
```
The docs in ir_Pioneer.cpp state

>"A556+AF20" is an Address of 0xA556 & a Command of 0xAF20.

So I assumed the following call would be correct, but didn't work out:
```
irsend.sendPioneer(irsend.encodePioneer(address, command), 32, 1);
```
Running both codes as command in sequence does the trick:
```
        if (address) {
            irsend.sendPioneer(irsend.encodePioneer(0, address), 32, 1);
        }
        irsend.sendPioneer(irsend.encodePioneer(0, command), 32, 1);
```
Since I don't know anything about NEC I might have gotten something else completely wrong.

Keep up your great work. :)

Kind regards

Michael

crankyoldgit, 2022-01-29:
See:
https://github.com/crankyoldgit/IRremoteESP8266/blob/f8e5662776102e57cf4c03af5e9fc54d14a07b19/src/IRremoteESP8266.h#L1192

That's probably because Pioneer is a `64` bit protocol, not `32` like you're calling it.

Try:
```c++
irsend.sendPioneer(irsend.encodePioneer(address, command), 64, 1);
// or
irsend.sendPioneer(irsend.encodePioneer(address, command), kPioneerBits, 1);
```

crankyoldgit, 2022-01-29:
And yes, it is essentially two NEC 32-bit messages back to back, making a 64 bit message, hence why your alternative code/way of constructing it works.
i.e. Sending a 32-bit pioneer message will effectively send a just the `command` portion. Glue two together, and you've kind of made full 64-bit pioneer message.

crankyoldgit, 2022-01-29:
Also, another clue:
https://github.com/crankyoldgit/IRremoteESP8266/blob/f8e5662776102e57cf4c03af5e9fc54d14a07b19/src/ir_Pioneer.cpp#L69
&

https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#:~:text=A%20raw-,64%2Dbit,-Pioneer%20message%20code

shin10, 2022-01-29:
Ah, right. Didn't have much sleep. So I'd either didn't see it or it's because sending 64 bit with an `address = 0` seems to have no effect on my TV. I really can't remember 😅 🤷 

So now I ended up with
```
irsend.sendPioneer(irsend.encodePioneer(address, command), kPioneerBits >> (address ? 0 : 1), 1);
```
Or should I rather double the command?

crankyoldgit, 2022-01-29:
I feel like I'm missing something here. What is with the `address` of `0` and sending 32 bits?
(Note: I don't have a Pioneer device, so I'm unfamiliar with it. Everything I based the code on is from other people's comments or code)

shin10, 2022-02-02:
From my experience the TV only responds to codes, that are repeated at least once.

For example `AA1C` to toggle the `on/off` state has to be sent either as 32 bit with repeat=1; or just once without repetition but with an abused address parameter and 64 bit (to repeat the same command). Sending only the command with 64 bit will not work, since the address parameter will interfere and the TV does nothing.
```
// working
irsend.sendPioneer(irsend.encodePioneer(0, 0xAA1C), 32, 1);
// or
irsend.sendPioneer(irsend.encodePioneer(0xAA1C, 0xAA1C), 64, 0);

// will not work
irsend.sendPioneer(irsend.encodePioneer(0, 0xAA1C), 64, 1);
```

On the other hand `AA5A+AF7C` to select `Input 3` has to be sent like documented, with repeat=1:
```
irsend.sendPioneer(irsend.encodePioneer(0xAA5A, 0xAF7C), 64, 1);
```

crankyoldgit, 2022-02-15:
Good to know. I'll add a link to this conversation in the code/documentation for future Pioneer users.

commit 61e4edfa (2022-02-15): Improve documentation for `encodePioneer()`
Add notes about how to use it if the Address is `0`.

Fixes #1749
PR #1761 (2022-02-15): Improve documentation for `encodePioneer()`
Add notes about how to use it if the Address is `0`.

Fixes #1749
commit a3e2c81f (2022-02-15): Improve documentation for `encodePioneer()` (#1761)

Add notes about how to use it if the Address is `0`.

Fixes #1749
closed 2022-02-15 by crankyoldgit