crankyoldgit/IRremoteESP8266 #1751: esp32c3

hydrafi, 2022-01-29:
with esp32c3 device 

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420193de  RA      : 0x42000d4c  SP      : 0x3fc948e0  GP      : 0x3fc8b000  
TP      : 0x3fc8b424  T0      : 0x4005890e  T1      : 0x403848e6  T2      : 0x00000000  
S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x0000c350  
A2      : 0x0000c350  A3      : 0x00000000  A4      : 0x0000009f  A5      : 0x00000032  
A6      : 0x3fe00000  A7      : 0x00040000  S2      : 0x3fc8d1bc  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0xffffffff  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000001  
MHARTID : 0x00000000

NiKiZe, 2022-01-29:
You will probably need to provide more details, exactly which code is running and when does it happen?

hydrafi, 2022-01-30:
sorry, 
I tried IRrecvDemo example and with esp32 run correctly, with esp32c3 reboot continuosly after irrecv.enableIRIn(); istruction ...
thanks
rgds
Max

crankyoldgit, 2022-01-31:
My **guess** is that there is something _not quite right_ with the ESP core library for the ESP32-C3.
I don't have a C3, so I'll look at ordering one.

As you said it is working on the normal ESP32, I'm going to have to ask you for all the exact versions of this library you're using, and the relevant core library versions for **BOTH** your ESP32 _and_ the ES_32-C3 so a) it can be replicated, and b) determined if it is our library's code that is the problem, or c) If it's a bug in the C3 version.

Also, can you please add the following to the "IRremoteESP8266.h" file:
```c++
#define _ESP32_IRRECV_TIMER_HACK false
```
Put it after:
https://github.com/crankyoldgit/IRremoteESP8266/blob/6bc095af80e5aec47d66f8c6263f3a943ea3b4d5/src/IRremoteESP8266.h#L46

Also can you please confirm/check what the values of the following `#define`s are, or if they are defined at all:
`ESP32`, `ESP_ARDUINO_VERSION_MAJOR`, `ESP_ARDUINO_VERSION_MINOR`, `ESP_ARDUINO_VERSION_PATCH`

crankyoldgit, 2022-01-31:
Oh, and what ESP32-C3 dev board are you using, so I can try to get the same?

hydrafi, 2022-02-01:
> Oh, and what ESP32-C3 dev board are you using, so I can try to get the same?

ESP32-C3-DevKitM-1

crankyoldgit, 2022-02-01:
Oh, and can you please try installing/running the same example code without anything plugged in to the board, other than the USB cable of course. Just to eliminate any hardware/wiring issues.

crankyoldgit, 2022-02-01:
I've ordered a board, but that will take weeks & weeks to arrive.

crankyoldgit, 2022-02-01:
> > Oh, and what ESP32-C3 dev board are you using, so I can try to get the same?
> 
> ESP32-C3-DevKitM-1
> 
> library -> Arduino by Espressif System 2.0.2
> 
> I will try your proposal and update you tks M

Also, what is the version of the IRremoteESP8266 library?

crankyoldgit, 2022-02-12:
> Oh, and can you please try installing/running the same example code without anything plugged in to the board, other than the USB cable of course. Just to eliminate any hardware/wiring issues.

@hydrafi Have you tried the above?

crankyoldgit, 2022-03-01:
The ESP-C3 board I ordered arrived today.
I'll try to test it out soon.
Did you try the earlier suggestions?

crankyoldgit, 2022-03-01:
I just tried building this and got an error message of:
```
Error: This board doesn't support arduino framework!
```

@hydrafi Can you please post the [`platformio.ini`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/platformio.ini) file you are using to build with so I can replicate it?

hydrafi, 2022-03-01:
Hi,
sorry for my inaction,
I use Arduino, Visualmicro or Visualcode
tomorrow I will dedicate a slot, I will do the tests and I will send you 
everything,

thanks a lot
Massimo

-- 
Hydra srl

(Le informazioni contenute in questo messaggio di posta elettronica o nei
suoi allegati sono riservate e confidenziali e ne è vietata la diffusione in
qualunque modo eseguita senza autorizzazione scritta. Qualora Lei non fosse
la persona a cui il presente messaggio è destinato, La invitiamo ad
eliminarlo e a non utilizzare in alcun caso il suo contenuto, dandone
gentilmente comunicazione al mittente. Grazie.
This e-mail message, the information it contains and any files attached are
subject to attorney-client privilege and D.Lgs. 196/2003 and contain
confidential information intended only for the person(s) to whom it is
addressed.  If you are not the intended recipient, you are hereby notified
that any use or distribution of this e-mail is strictly prohibited:  please
notify the sender and delete the original message.  Thank you.)

hydrafi, 2022-03-02:
Hi,
I tried again with #define _ESP32_IRRECV_TIMER_HACK false, no change ....

In Arduino IDE 1.8.19 I have installed library -> Arduino by Espressif System 2.0.2 and no problem with ESP32C3 dev module
this is the ref to install that
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
Visualmicro & Visual code works too
I do not know if platformio support C3

example used is IRrecvDEMO and the target has only 2 components, ESP-kit + IRreceiver. Only my change is the pin used to IRreceived ..

crankyoldgit, 2022-03-04:
Good news: I can kind-of replicate the crash. Bad news: I have no idea what is causing it yet.

crankyoldgit, 2022-03-04:
I've seemed to have narrowed it down to the `pinMode()` call in `IRrecv::enableIRIn()`, not sure why that is causing the device to go crazy.

crankyoldgit, 2022-03-04:
I think I've found a/the problem. It is the use GPIO 14, that seems to be used by the USB connection on the dev board for C3 devices.
It causes the board to crash by making that pin an INPUT. Probably causing a short circuit/current loop etc.

Using:
```c++
const uint16_t kRecvPin = 10;
```

Seemed to get to the main loop.

@hydrafi Can you please test that out?

crankyoldgit, 2022-03-04:
FYI: GPIO `10` is Pin 11 on the board.
see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html#j1

crankyoldgit, 2022-03-04:
@hydrafi Can you please try the code in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/espc3 / PR #1768 ?

Please let me know how it goes.
You will need to use the new versions of the examples. The core of the library has not been changed.

I've partly tested this on my ESP32-C3 board. I've not tried to capture an IR signal on it yet, but it doesn't crash/boot-loop like it did, thus I'm considering it fixed.

crankyoldgit, 2022-03-04:
As a note for those trying to use PlatformIO to build the examples on an `ESP32-C3`, you need something like the following in the `platformio.ini` file:
```ini
[env:esp32c3]
platform = espressif32
platform_packages =
    toolchain-riscv-esp
    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#cbcba53
    platformio/tool-esptoolpy @ ~1.30100
framework = arduino
board = esp32dev
board_build.mcu = esp32c3
;board_build.partitions = huge_app.csv
;board_build.partitions = custom_part.csv
board_build.variant = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.flash_mode = dio
board_build.arduino.ldscript = esp32c3_out.ld
build_unflags =
    -DARDUINO_ESP32_DEV
    -DARDUINO_VARIANT="esp32"
build_flags =
    -DARDUINO_ESP32C3_DEV
    -DARDUINO_VARIANT="esp32c3"
; build_flags = -D_IR_LOCALE_=en-AU
```

hydrafi, 2022-03-04:
I had used gpio3 however I tried your new example sketch with gpio10 but nothing has changed. I don't know platformio, can you post the bin file that I try it?

crankyoldgit, 2022-03-05:
> I had used gpio3 however I tried your new example sketch with gpio10 but nothing has changed. I don't know platformio, can you post the bin file that I try it?

@hydrafi Try this:
[IRrecvDumpV3-espc3-20220305.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/8190092/IRrecvDumpV3-espc3-20220305.zip)

hydrafi, 2022-03-05:
Hi, good news.
Your bin run.
Not clear why?
I debugged / traced verifying that the statement "timer = timer Begin (_timer_num, 80, true);" in line 455 of IRrecv.cpp which defaults to 3 in _timer_num, but ESP32C3 has only 2 timers (https://docs.espressif.com/projects/arduino-esp32/en/latest/api/timer.html). 
If you set 0 or 1  my compilation works too.

crankyoldgit, 2022-03-05:
That's great to know. I'll look at the timers used soon as well.
It could be that I'm using a slightly different version of the ESP32 Arduino core. Not sure why

crankyoldgit, 2022-03-05:
@hydrafi I've updated that branch  & PR #1768. This time there is a library change. Can you please build against that version of the library.
It should ensure that only a timer of 0 or 1 is used for an ESP32-C3, thus hopefully fix the issue you're having.

crankyoldgit, 2022-03-08:
@hydrafi Chasing up on this. Can you confirm if the recent changes now construct a binary for the ESP32-C3 that doesn't crash from the examples without _any_ modifications in your build environment/hardware?
i.e. Can we safely merge this into the `master` branch?

hydrafi, 2022-03-08:
ok

closed 2022-03-08 by hydrafi

crankyoldgit, 2022-03-15:
FYI, the changes mentioned above have now been included in the new [v2.8.2 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.2) of the library.

Jason2866, 2022-03-15:
@crankyoldgit  I made a custom setup for Tasmota for Platformio to use the C3, S2 and S3. Feel free to use :-)

Thx for your awesome work with IRremoteESP8266

````
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/733/framework-arduinoespressif32-v4.4_dev-2509ea4227.tar.gz
```

commit 03e1f067 (2022-03-05): [BUG] Ensure ESP32 system timer is valid before use.
* Add debug messages & a runtime assert to ensure we correctly allocated a system timer in `IRrecv::enableIRIn()`.
* Limit the max number of the system timer to the appropriate size for the ESP32. ESP32-C3's only have two system timers (0-1), the rest have 4 (0-3).
* Done in such a way it should be correct for future ESP32 models.

Ref: https://docs.espressif.com/projects/arduino-esp32/en/latest/api/timer.html

Fixes #1751
PR #1768 (2022-03-08): [BUG] ESP32-C3: Fix reboot/crashes on ESP32-C3s when receiving.
- Seems an irrecv gpio of `14` was a poor default for the ESP-C3. Using it causes a reboot.
  - Using `10` for the input GPIO avoids the reboot.
- Ensure when we allocate/use a system timer, that it actually succeeds.
  - ESP32-C3 only has two system timers (0 & 1). Try to automatically cater for that scenario.
  - Add extra debug messages
  - Add a run-time assert to check it succeeded.

Fixes #1751
commit d2fa2067 (2022-03-15): [BUG] ESP32-C3: Fix reboot/crashes on ESP32-C3s when receiving. (#1768)

- Seems an irrecv gpio of `14` was a poor default for the ESP-C3. Using it causes a reboot.
  - Using `10` for the input GPIO avoids the reboot.
- Ensure when we allocate/use a system timer, that it actually succeeds.
  - ESP32-C3 only has two system timers (0 & 1). Try to automatically cater for that scenario.
  - Add extra debug messages
  - Add a run-time assert to check it succeeded.

Fixes #1751