crankyoldgit/IRremoteESP8266 #1754: Support For Daikin Series 17 FTXB24

qcarver, 2022-02-03:
[IRremoteESP8266 Daikin128 protocol](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.cpp#L2511) has worked for other repo-cloners with Series 17 Daikin FTXB09 and 12. However, it doesn't work for my FTXB24 (Specifically FTXB24AXVJU).

The protocol is wicked close though. I get the acknowledgement beep from the wall unit and can issue some commands that are recieved incorrectly. The only feedback I have from the unit is the blue/red/violet LED. So far what I can observe visually... Where Daikin128 -> FTXB24

Mode Heat -> Mode Cool,
Mode Cool -> OFF
VerticalSwing* -> Cool
can't seem to mash any button which puts it in Heat mode.

I downloaded, ginned and ran the de-rigueur arduino project as described in the ["Preparation" paragraph](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#preparation). The [output from the program](https://docs.google.com/document/d/1bhPlwsUE4ppL1dlX6_aYmqCSxeE6HlOumYsR35S0CKA/edit?usp=sharing) also seems to think the codes are Daikin128.

[THE SPREADSHEET DATA IS HERE](https://docs.google.com/spreadsheets/d/1p1Xo03JRA7SGMDiSvs_jigtsV2Ye7T7GNmghGxOLfHk/edit?usp=sharing)

**Of note** (1) the remote does not have a useful ARCxxxxxxx identifier on it. I see a dime sized QR code that equates to "3P605270-1". If i look under the batteries well I see the black printed numbers "TL701" under the batteries. (2) *There is no H swing

gladson1990, 2022-04-28:
In the attached spreadsheet I see that the ON and OFF codes are both same. Is that valid ? just thought of conveying

crankyoldgit, 2022-05-06:
@qcarver I was able to recreate the exact same `state[]` code from your [doc](https://docs.google.com/document/d/1bhPlwsUE4ppL1dlX6_aYmqCSxeE6HlOumYsR35S0CKA/edit#heading=h.luu34vdlzof1) i.e. "**HEAT MODE**"
You can see in PR#1795 that it creates the same state (and thus message), if what you captured is correct, then it does seem to be able to set heat mode correct. I'll need more info or explanation before I proceed further.

@gladson1990 
> In the attached spreadsheet I see that the ON and OFF codes are both same. Is that valid ? just thought of conveying

Yes, that is expected. The protocol does not have a discrete on or off message/setting sadly. Only a "toggle the power" setting.

qcarver, 2022-05-06:
> 

@crankyoldgit hello, I'm unsure what you'd like me to explain. How can I help?

crankyoldgit, 2022-05-08:
@qcarver You said:
> can't seem to mash any button which puts it in Heat mode.

I took the data from your document (which appears to be a capture from your remote) where you put it in heat mode.
I was able to reproduce that message/state code via the library. See below. Therefore I'm not sure what you mean by "you can't put it in Heat mode".

Testing code that makes the same state/message as your heat mode capture:
https://github.com/crankyoldgit/IRremoteESP8266/blob/21804f5bdf83c97f77b00f66d38fd3ea8ba85559/test/ir_Daikin_test.cpp#L3883-L3921

qcarver, 2022-05-08:
@crankyoldgit Ah yes. Okay. 

Yes the document was captured from the remote that came with the unit.

My comment was to say that Running v2.8.1 from HomeBridge. I couldn't find a function that would make the heat mode led go on. It is red. It turns out, that is typical of the manufacturers remote too. So my comment was infact irrelevant.

I updated to v2.8.2 today, and I am able to use the automode through the homebridge-"daikin-esp8266-platform". The LED goes on when automode has determined that it is time to heat.

I didn't see anything about v2.8.2 in the RELEASE notes but as far as I can tell everything works as long as the power state is in synch with homebridge. Did you make any changes? Are my codes the same as another unit (and so no changes were needed?)

Regardless, Thank you.

crankyoldgit, 2022-05-08:
> Did you make any changes? Are my codes the same as another unit (and so no changes were needed?)

No. I've made no changes. It appears your remote & AC is the same as another already supported model.

So, of in understanding it correctly, everything is working as expected, therefore there is nothing to fix? Correct?

qcarver, 2022-05-09:
In SW it has been my experience that as soon as I say that, I find out I am wrong.

However, here goes;  "Yes, it appears everything is working as expected. There is nothing to fix."

I really appreciate your repo. Thank you.

QC

closed 2022-05-10 by crankyoldgit