crankyoldgit/IRremoteESP8266 #1755: Sending IR Raw works on Arduino UNO but not on Wemos D1 mini

winner093, 2022-02-05:
### Version/revision of the library used
V2.8.1

### Describe the bug
I am trying to control my vacuum cleaner robot via ESP8266. 
It is controlled via an IR remote.
I scanned the IR signals for one button on the remote with the given example code with Arduino UNO and esp8266 and they are roughly the same.

After that, I tried to send the raw values with the esp8266 (PIN D2) and my vacuum didn't respond. I used an IR led controlled with a NPN transistor. The frequency is 38 kHz. After that, I tried to increase the CPU frequency, but it still didn't work.

Later I changed the Wemos D1 mini for an Arduino UNO and used the library yours is based on. There everything worked with the example.

I compared what I am sending with the Wemos with what it should send.
The upper four rows show what I am receiving, and the lowest row shows what it should receive.

[image]

What am I doing wrong or is there a setting for the ESP8266 I need to change?

Many thanks in advance!!

crankyoldgit, 2022-02-05:
FYI, you shouldn't need to increase the CPU frequency of the ESP8266. It's already way faster than the UNO.

In theory, our library should be able to send the same "raw" message with your captured data. Try putting that data in our [`IRsendDemo.ino`](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) and use that code.

Now, looking at your data above. The first two lines seem "okay-ish" compared to your reference line at the bottom. Honestly, they are not great and I suspect something odd is going on. But the next two lines, they are terrible. Something is very very wrong with those captures. They are nothing like what you're sending. i.e. Values of 50 etc, they are way way too small. I'd be suspect of any signal with sub 250us timings to be honest.

Can I suggest you put the IR demodulator on the ESP8266 and use our `IRrecvDumpV3` program to re-capture the original remote?

Another possibility is the a different duty cycle. I our library defaults to 50% on, 50% off when transmitting with raw. I _think_ the library we forked from ages ago now uses 33% on, 66% off.

My guess is you have a hardware issue on the "sending" side. Without more details (circuits, components, & code etc) it's hard to tell what you should do next.

winner093, 2022-02-11:
Sorry for the late reply.
I found the problem today after not looking at the project for a while.

The reason it didnt worked is that I didnt know I had to change the XXX in the following line depending on the rawData size.

irsend.sendRaw(rawData, XXX, 38);  // Send a raw data capture at 38kHz.

Thank you for your help. Maybe it will help somebody else ;)

closed 2022-02-11 by winner093