crankyoldgit/IRremoteESP8266 #1757: Support for Hitachi RAS-70YHA3 (remote RAR-3U3)

jeef3, 2022-02-12:
I am trying to add support for my **Hitachi RAS-70YHA3** heatpump with remote **RAR-3U3**. [I forked your repo and started an attempt](https://github.com/jeef3/IRremoteESP8266/tree/hitachi-296) (you'll have to forgive me, I'm not a c++ developer) a few months back, but I struggled a bit. I called it `HITACHI_AC296`.

My attempt is based on [someone else's research](https://perhof.wordpress.com/2015/03/29/reverse-engineering-hitachi-air-conditioner-infrared-remote-commands/) on the same protocol. Using what I learnt there I was able to send working signals, I just wasn't able to get it into IRremoteESP8266 properly.

At the time I did not know about [auto_analyse_raw_data.py](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/auto_analyse_raw_data.py), so I will give that a shot. I just wanted to open this ticket to track progress and have a place for guidance 🙏.

---

Raw output from **IRrecvDumpV2** (I am using a VS1838b, but I don't seem to have had too many issues with it, I don't think):

```
Timestamp : 000016.193
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Protocol  : UNKNOWN
Code      : 0x4EB022A8 (298 Bits)
uint16_t rawData[595] = {3280, 1714,  360, 1278,  358, 518,  358, 516,  362, 514,  362, 512,  362, 514,  358, 516,  358, 520,  360, 516,  358, 514,  362, 516,  356, 518,  360, 1278,  360, 514,  360, 516,  360, 518,  360, 516,  360, 518,  358, 516,  358, 516,  358, 516,  360, 514,  362, 512,  364, 518,  360, 514,  362, 514,  360, 516,  358, 514,  360, 514,  362, 514,  358, 1280,  360, 518,  360, 1276,  362, 1278,  360, 1278,  360, 1278,  358, 1280,  360, 1280,  358, 518,  356, 1282,  360, 1278,  358, 1282,  358, 1278,  360, 1278,  360, 1278,  360, 1276,  360, 1278,  358, 1282,  360, 516,  360, 516,  360, 514,  360, 516,  358, 516,  362, 516,  358, 518,  356, 520,  362, 514,  360, 514,  360, 1276,  362, 1280,  358, 516,  360, 516,  358, 1278,  360, 1284,  358, 1276,  360, 1278,  360, 516,  360, 518,  360, 1274,  362, 1280,  358, 516,  360, 518,  360, 516,  358, 1280,  360, 514,  360, 514,  360, 1280,  360, 516,  358, 514,  360, 1282,  360, 1280,  358, 514,  360, 1278,  360, 1280,  358, 516,  358, 1278,  360, 1280,  358, 518,  360, 516,  360, 514,  360, 1278,  360, 516,  358, 518,  356, 518,  358, 1278,  360, 518,  362, 1278,  358, 1278,  360, 516,  358, 1276,  362, 1278,  358, 1280,  358, 516,  356, 1284,  362, 514,  360, 514,  360, 516,  360, 514,  360, 514,  362, 1278,  360, 1276,  362, 518,  358, 1278,  362, 1276,  360, 1278,  360, 1280,  358, 1280,  358, 516,  360, 516,  360, 1282,  362, 512,  362, 516,  360, 516,  360, 514,  360, 514,  360, 514,  362, 516,  358, 520,  358, 1278,  362, 1278,  358, 1280,  360, 1276,  362, 1278,  360, 1278,  360, 1278,  360, 1282,  362, 518,  356, 518,  358, 516,  358, 516,  360, 516,  358, 514,  360, 516,  358, 520,  358, 1280,  358, 1280,  358, 1280,  358, 1278,  362, 1278,  358, 1278,  360, 1276,  362, 1284,  358, 518,  358, 516,  358, 516,  360, 516,  358, 516,  358, 516,  360, 514,  360, 520,  360, 1278,  360, 1278,  362, 1278,  360, 1278,  362, 1278,  358, 1278,  362, 1278,  360, 1282,  360, 516,  360, 514,  360, 516,  360, 514,  360, 514,  360, 516,  360, 516,  360, 518,  360, 1278,  360, 1276,  362, 1278,  360, 1278,  362, 1278,  358, 1278,  362, 1280,  358, 1282,  358, 516,  360, 514,  360, 516,  360, 514,  360, 518,  358, 516,  358, 516,  360, 520,  360, 1278,  360, 1278,  360, 1278,  360, 1278,  362, 1278,  360, 1276,  360, 1278,  360, 1282,  360, 514,  360, 1276,  360, 1280,  358, 516,  360, 1278,  362, 512,  360, 1278,  360, 518,  358, 1278,  362, 516,  356, 516,  360, 1278,  360, 518,  356, 1280,  360, 518,  358, 1284,  358, 1280,  358, 516,  358, 516,  360, 516,  358, 1282,  356, 1278,  360, 1278,  360, 1282,  358, 518,  360, 1276,  360, 1278,  360, 1276,  362, 514,  358, 516,  360, 516,  358, 518,  362, 514,  360, 516,  358, 516,  358, 516,  362, 512,  360, 516,  360, 512,  360, 518,  360, 1278,  360, 1276,  362, 1278,  360, 1278,  360, 1278,  360, 1278,  360, 1278,  358, 1284,  360, 514,  358, 516,  360, 516,  360, 514,  358, 514,  362, 514,  362, 514,  358, 520,  360, 1278,  360, 1278,  362, 1278,  360, 1276,  362, 1278,  360, 1278,  360, 1278,  358, 1284,  360, 514,  362, 514,  362, 514,  360, 514,  360, 514,  362, 514,  358, 516,  360, 520,  358, 1278,  360, 1280,  358, 1276,  360, 1278,  360, 1278,  360, 1278,  360, 1278,  360, 1282,  358, 1280,  360, 1278,  362, 512,  362, 514,  360, 514,  360, 514,  360, 516,  360, 518,  360, 514,  362, 514,  360, 1278,  360, 1278,  358, 1278,  360, 1276,  362, 1276,  360, 1280,  362};  // UNKNOWN 4EB022A8
```

Output from **auto_analyse_raw_data.py**.

```
Found 595 timing entries.
Potential Mark Candidates:
[3280, 364]
Potential Space Candidates:
[1714, 1284, 520]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3280
kHdrSpace  = 1714
kBitMark   = 359
kOneSpace  = 1278
kZeroSpace = 515

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+10000000000010000000000000000010111111011111111100000000001100111100110001001001101101100010001011011101000001101111100100000000111111110000000011111111000000001111111100000000111111110000000011111111011010101001010110001111011100000000000011111111000000001111111100000000111111111100000000111111
  Bits: 296
  Hex:  0x80080002FDFF0033CC49B622DD06F900FF00FF00FF00FF00FF6A958F7000FF00FF00FFC03F (MSB first)
        0xFC03FF00FF00FF000EF1A956FF00FF00FF00FF00FF009F60BB446D9233CC00FFBF40001001 (LSB first)
  Dec:  63672915700286784428888561416304607301127352178493058802278520882964001991547211996971071 (MSB first)
        125333218691812209784605858600489852403040852452316227886439093576240819778267537704226817 (LSB first)
  Bin:  0b10000000000010000000000000000010111111011111111100000000001100111100110001001001101101100010001011011101000001101111100100000000111111110000000011111111000000001111111100000000111111110000000011111111011010101001010110001111011100000000000011111111000000001111111100000000111111111100000000111111 (MSB first)
        0b11111100000000111111111100000000111111110000000011111111000000000000111011110001101010010101011011111111000000001111111100000000111111110000000011111111000000001111111100000000100111110110000010111011010001000110110110010010001100111100110000000000111111111011111101000000000000000001000000000001 (LSB first)

Total Nr. of suspected bits: 296
```

NiKiZe, 2022-02-12:
You seem to have been doing quite good.
Start with https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol
Get the receiver and sending logic in place.
For an AC it is best to record the temperature range and post the raw data.
That is from 16-32 Degrees C or whatever the remote does without changing any other setting.
This gives us the best way to arrange the receive and sending functions for easy conversation.
(it also serves as documentation if anyone ever needs to go back and check the protocol)

The values you have seem to match the generic IRrecv::decodeHitachiAC only a "new number of bits"

closed 2022-03-13 by crankyoldgit

crankyoldgit, 2022-03-15:
FYI, the changes mentioned above have now been included in the new [v2.8.2 release](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.8.2) of the library.

Onepamopa, 2022-12-08:
I'm having some troubles with HITACHI_AC296 detection.
Remote is RAR-3E1, Hitachi RAS-10EX2 -- It gets detected as HITACHI_AC296 but not all the time. I've tried adjusting kTolerancePercentage & kTimeout without any improvement:

11:47:33.449 -> Tolerance : 24%
11:47:33.449 -> Protocol  : HITACHI_AC296
11:47:33.449 -> Code      : 0x01100040BFFF00CC33926D13EC708F00FF00FF00FF00FF00FF53ACF10E00FF00FF00FF03FC (296 Bits)
11:47:33.449 -> Mesg Desc.: Power: On, Mode: 3 (Cool), Temp: 28C, Fan: 5 (Auto)
11:47:33.449 -> uint16_t rawData[595] = {3434, 1642,  464, 1202,  484, 436,  484, 436,  484, 436,  484, 436,  484, 440,  480, 440,  484, 458,  458, 462,  464, 456,  464, 458,  462, 462,  460, 1222,  462, 458,  464, 456,  464, 436,  484, 462,  458, 462,  460, 462,  458, 462,  438, 482,  464, 458,  462, 458,  464, 436,  484, 440,  484, 436,  484, 436,  484, 436,  484, 436,  486, 436,  484, 1202,  484, 458,  462, 1224,  462, 1198,  484, 1202,  480, 1228,  458, 1222,  464, 1202,  484, 436,  484, 1222,  460, 1200,  484, 1202,  484, 1222,  462, 1218,  468, 1196,  484, 1202,  458, 1250,  458, 1202,  484, 436,  484, 436,  488, 432,  488, 436,  484, 436,  484, 436,  484, 436,  484, 456,  464, 456,  464, 458,  464, 1222,  462, 1202,  484, 436,  486, 436,  484, 1196,  490, 1196,  484, 1202,  484, 1202,  484, 436,  486, 434,  486, 1222,  464, 1218,  464, 456,  464, 458,  462, 436,  484, 1222,  464, 458,  462, 458,  462, 1222,  464, 458,  462, 458,  464, 1222,  464, 1196,  490, 436,  480, 1222,  464, 1222,  464, 458,  464, 1202,  484, 1196,  486, 456,  468, 1218,  464, 1202,  484, 436,  484, 436,  484, 1196,  490, 432,  490, 434,  486, 456,  464, 458,  462, 458,  464, 1222,  464, 1196,  490, 434,  486, 1196,  486, 1222,  464, 1196,  490, 432,  488, 436,  484, 436,  484, 436,  484, 1198,  484, 1222,  464, 1222,  464, 436,  484, 1202,  480, 1226,  464, 1218,  464, 1200,  484, 436,  484, 436,  484, 436,  484, 1224,  462, 458,  464, 456,  464, 458,  462, 458,  464, 458,  462, 462,  458, 462,  464, 436,  484, 1198,  484, 1200,  486, 1222,  464, 1218,  468, 1196,  484, 1202,  484, 1222,  464, 1196,  490, 430,  488, 432,  490, 436,  484, 436,  484, 436,  486, 436,  484, 436,  484, 458,  462, 1224,  462, 1224,  462, 1196,  486, 1200,  486, 1222,  464, 1218,  468, 1196,  486, 1222,  436, 484,  464, 456,  464, 458,  464, 456,  464, 456,  464, 462,  458, 462,  464, 436,  484, 1196,  486, 1202,  484, 1222,  464, 1218,  442, 1224,  484, 1196,  486, 1222,  462, 1202,  484, 436,  464, 456,  464, 458,  488, 432,  490, 432,  488, 432,  490, 436,  458, 484,  462, 1222,  464, 1218,  442, 1222,  484, 1198,  484, 1222,  442, 1244,  464, 1200,  484, 1224,  458, 462,  458, 462,  438, 482,  464, 458,  464, 458,  462, 458,  462, 458,  442, 458,  464, 1222,  486, 1196,  484, 1224,  436, 1250,  464, 1196,  464, 1222,  484, 1224,  464, 1196,  488, 1198,  484, 1200,  484, 436,  484, 436,  484, 1222,  464, 456,  464, 1222,  462, 436,  486, 456,  464, 456,  442, 1222,  486, 1196,  488, 436,  486, 1222,  432, 488,  464, 1196,  488, 1198,  484, 436,  484, 436,  464, 458,  484, 1224,  462, 1224,  464, 1196,  464, 1244,  464, 456,  464, 1224,  462, 1198,  484, 1202,  484, 436,  484, 436,  486, 436,  488, 452,  464, 458,  462, 462,  436, 484,  436, 484,  464, 456,  464, 458,  462, 458,  468, 432,  488, 1198,  484, 1198,  484, 1224,  464, 1222,  464, 1196,  464, 1222,  486, 1222,  464, 1196,  490, 430,  488, 432,  490, 436,  484, 436,  484, 436,  458, 462,  462, 458,  484, 458,  462, 1222,  442, 1244,  464, 1196,  488, 1198,  484, 1222,  464, 1222,  464, 1196,  488, 1220,  462, 458,  462, 458,  464, 458,  436, 488,  438, 482,  438, 484,  462, 458,  464, 434,  464, 1222,  484, 1198,  458, 1248,  442, 1244,  464, 1196,  490, 1196,  484, 1224,  462, 1198,  490, 1196,  484, 1224,  462, 458,  464, 458,  462, 458,  464, 458,  462, 458,  442, 458,  462, 478,  468, 458,  464, 1196,  458, 1228,  484, 1224,  464, 1218,  436, 1228,  486, 1222,  464};  // HITACHI_AC296
11:47:33.768 -> uint8_t state[37] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x70, 0x8F, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x53, 0xAC, 0xF1, 0x0E, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x03, 0xFC};
11:47:33.768 -> 
11:47:33.768 -> 
11:47:37.620 -> Timestamp : 000036.258
11:47:37.620 -> Library   : v2.8.4
11:47:37.620 -> 
11:47:37.620 -> Tolerance : 24%
11:47:37.620 -> Protocol  : UNKNOWN
11:47:37.665 -> Code      : 0x9926467C (299 Bits)
11:47:37.665 -> No description
11:47:37.665 -> uint16_t rawData[597] = {3414, 1620,  484, 1202,  484, 436,  484, 436,  484, 436,  484, 436,  486, 436,  488, 432,  484, 462,  464, 436,  484, 436,  486, 436,  484, 436,  486, 1222,  458, 462,  464, 456,  464, 436,  484, 436,  484, 436,  486, 434,  486, 436,  488, 436,  484, 436,  486, 436,  484, 458,  462, 458,  462, 458,  464, 462,  462, 458,  462, 458,  464, 456,  464, 1198,  484, 462,  458, 1224,  464, 1222,  464, 1196,  484, 1202,  484, 1224,  462, 1218,  468, 458,  460, 1222,  464, 1222,  464, 1196,  488, 1196,  486, 1222,  462, 1224,  462, 1198,  484, 1202,  484, 1202,  484, 458,  464, 456,  464, 458,  464, 456,  464, 456,  464, 458,  462, 458,  464, 436,  488, 432,  488, 436,  484, 1222,  462, 1218,  464, 458,  468, 452,  468, 1198,  484, 1224,  462, 1218,  468, 1196,  486, 436,  484, 436,  484, 1202,  484, 1222,  464, 456,  464, 458,  464, 436,  484, 1196,  490, 436,  484, 436,  484, 1222,  460, 462,  462, 458,  462, 1202,  484, 1196,  488, 436,  484, 119 [... cut ...] , 1228,  428, 1280,  434, 1226,  460, 1226,  428, 1280,  432, 488,  432, 488,  406, 514,  406, 514,  438, 484,  410, 488,  432, 514,  410, 510,  438, 1222,  458, 1228,  428, 1280,  406, 1276,  410, 1254,  428, 1280,  438};  // HITACHI_AC296
10:54:26.370 -> uint8_t state[37] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x70, 0x8F, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x21, 0xDE, 0xE1, 0x1E, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x03, 0xFC};

Onepamopa, 2022-12-08:
Setting kTimeout to ~15 or less does reduce the amount of unknown messages to short messages (6-8-9 bits):
13:37:42.256 -> Protocol  : HITACHI_AC296
13:37:42.256 -> Code      : 0x01100040BFFF00CC33926D13EC5CA300FF00FF00FF00FF00FF46B9E11E00FF00FF00FF03FC (296 Bits)
13:37:42.256 -> Mesg Desc.: Power: Off, Mode: 6 (Heat), Temp: 23C, Fan: 4 (High)
13:37:42.302 -> uint16_t rawData[595] = {3436, 1646,  458, 1202,  484, 436,  486, 436,  484, 436,  484, 436,  486, 436,  484, 440,  484, 458,  458, 462,  458, 462,  462, 458,  464, 456,  464, 1222,  464, 456,  464, 456,  464, 436,  484, 462,  458, 462,  460, 460,  460, 460,  464, 458,  462, 458,  464, 456,  464, 436,  484, 440,  486, 436,  484, 436,  484, 436,  484, 436,  484, 436,  484, 1202,  484, 458,  462, 1222,  464, 1200,  480, 1202,  484, 1222,  464, 1222,  464, 1202,  484, 436,  484, 1218,  464, 1200,  484, 1200,  480, 1228,  462, 1218,  464, 1200,  486, 1196,  484, 1224,  462, 1202,  484, 436,  484, 436,  484, 436,  490, 432,  488, 436,  484, 436,  486, 436,  458, 484,  462, 458,  464, 456,  464, 1222,  464, 1196,  490, 436,  484, 436,  484, 1198,  488, 1196,  486, 1200,  484, 1198,  484, 436,  488, 436,  480, 1228,  462, 1218,  464, 456,  464, 462,  458, 436,  486, 1222,  462, 462,  458, 462,  464, 1218,  464, 462,  464, 456,  458, 1222,  464, 1200,  486, 434,  486, 1222,  462, 1224,  462, 458,  464, 1196,  484, 1200,  484, 458,  462, 1224,  462, 1200,  484, 436,  484, 436,  486, 1200,  484, 436,  484, 436,  484, 458,  462, 458,  464, 462,  464, 1218,  464, 1202,  484, 436,  484, 1198,  484, 1222,  464, 1202,  484, 436,  484, 436,  486, 1196,  484, 1224,  462, 1222,  464, 458,  462, 1202,  484, 458,  462, 1224,  462, 1218,  464, 458,  442, 482,  464, 458,  462, 1196,  490, 432,  488, 1218,  464, 458,  462, 458,  464, 456,  464, 462,  460, 460,  460, 460,  460, 462,  462, 436,  486, 1200,  484, 1196,  484, 1222,  464, 1222,  462, 1202,  484, 1196,  484, 1222,  464, 1200,  484, 436,  486, 436,  462, 458,  488, 432,  490, 430,  490, 436,  484, 436,  484, 456,  464, 1222,  464, 1222,  462, 1198,  484, 1202,  484, 1224,  462, 1218,  468, 1198,  484, 1222,  460, 462,  438, 482,  464, 458,  462, 458,  462, 458,  464, 458,  462, 458,  442, 458,  488, 1196,  486, 1200,  486, 1218,  468, 1218,  464, 1202,  486, 1196,  486, 1222,  462, 1202,  484, 436,  484, 436,  484, 436,  462, 458,  490, 430,  490, 432,  488, 432,  484, 462,  464, 1222,  462, 1218,  442, 1222,  486, 1196,  484, 1224,  462, 1224,  464, 1196,  490, 1218,  462, 458,  464, 458,  462, 458,  462, 462,  458, 462,  458, 462,  438, 482,  464, 436,  484, 1198,  488, 1198,  484, 1222,  464, 1222,  462, 1198,  458, 1226,  486, 1222,  462, 1196,  488, 458,  464, 1196,  488, 1196,  486, 436,  462, 458,  464, 458,  484, 1222,  442, 458,  488, 1218,  462, 458,  442, 478,  468, 1196,  484, 1196,  490, 1222,  458, 462,  464, 1196,  464, 1222,  484, 436,  484, 436,  460, 462,  488, 432,  462, 1244,  442, 1244,  462, 1222,  464, 436,  484, 1222,  464, 1218,  442, 1224,  484, 1196,  490, 432,  488, 432,  490, 456,  464, 458,  462, 458,  462, 458,  464, 456,  464, 456,  464, 462,  436, 484,  464, 436,  484, 1198,  488, 1198,  484, 1222,  464, 1218,  442, 1222,  486, 1196,  464, 1244,  442, 1224,  458, 462,  464, 456,  464, 458,  488, 432,  488, 432,  490, 432,  488, 436,  460, 482,  464, 1218,  468, 1218,  442, 1222,  486, 1196,  484, 1224,  462, 1224,  462, 1196,  490, 1218,  464, 458,  462, 462,  438, 482,  438, 484,  436, 484,  464, 456,  464, 458,  436, 462,  464, 1222,  484, 1198,  462, 1244,  442, 1244,  436, 1224,  488, 1198,  484, 1222,  464, 1196,  488, 1198,  458, 1248,  462, 458,  462, 458,  438, 488,  436, 484,  436, 484,  438, 462,  458, 484,  436, 484,  442, 1222,  484, 1198,  462, 1244,  442, 1244,  438, 1222,  464, 1244,  442};  // HITACHI_AC296
13:37:42.578 -> uint8_t state[37] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x5C, 0xA3, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x46, 0xB9, 0xE1, 0x1E, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x03, 0xFC};
13:37:42.623 -> 
13:37:42.623 -> 
13:37:43.541 -> Timestamp : 000033.236
13:37:43.541 -> Library   : v2.8.4
13:37:43.541 -> 
13:37:43.541 -> Protocol  : HITACHI_AC296
13:37:43.588 -> Code      : 0x01100040BFFF00CC33926D13EC5CA300FF00FF00FF00FF00FF46B9F10E00FF00FF00FF03FC (296 Bits)
13:37:43.588 -> Mesg Desc.: Power: On, Mode: 6 (Heat), Temp: 23C, Fan: 4 (High)
13:37:43.588 -> uint16_t rawData[601] = {3438, 1642,  462, 1224,  458, 462,  464, 456,  464, 458,  462, 458,  464, 456,  464, 462,  464, 430,  490, 458,  464, 456,  464, 458,  462, 458,  464, 1196,  490, 432,  488, 436,  480, 460,  464, 458,  462, 458,  462, 458,  464, 458,  462, 458,  462, 462,  460, 462,  458, 436,  488, 436,  484, 436,  484, 436,  486, 436,  484, 436,  484, 436,  484, 1224,  462, 436,  486, 1200,  484, 1196,  486, 1222,  462, 1224,  462, 1198,  484, 1200,  486, 436,  484, 1202,  484, 1196,  486, 1222,  462, 1222,  464, 1200,  484, 1198,  484, 1222,  464, 1222,  464, 1222,  464, 436,  486, 434,  484, 436,  484, 436,  486, 436,  484, 436,  484, 436,  490, 456,  464, 456,  464, 456,  464, 1200,  486, 1196,  484, 436,  490, 432,  484, 1224,  462, 1202,  484, 1198,  484, 1224,  462, 458,  464, 456,  442, 1244,  464, 1202,  484, 436,  484, 436,  484, 458,  464, 1222,  462, 458,  464, 456,  464, 1198,  488, 436,  484, 436,  484, 1222,  460, 1222,  464, 458,  462, 1222,  464, 1202,  484, 436,  484, 1196,  486, 1222,  464, 436,  484 [... cut ...]
13:37:45.057 -> uint8_t state[37] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x5C, 0xA3, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x46, 0xB9, 0xE1, 0x1E, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x03, 0xFC};
13:37:45.102 -> 
13:37:45.102 -> 
13:37:45.102 -> Timestamp : 000034.765
13:37:45.102 -> Library   : v2.8.4
13:37:45.102 -> 
13:37:45.102 -> Protocol  : UNKNOWN
13:37:45.102 -> Code      : 0x7F3F35A9 (6 Bits)
13:37:45.102 -> No description
13:37:45.102 -> uint16_t rawData[11] = {628, 864,  328, 21846,  272, 2926,  194, 328,  356, 6724,  324};  // UNKNOWN 7F3F35A9
13:37:45.102 -> 
13:37:45.102 -> 

How to get rid of those ?

crankyoldgit, 2022-12-08:
> Also, looks like 2 of the modes aren't recognized (took me a looooot of tries to be able to extract a "working" match:

Here are the mode we have had reported to us:
https://github.com/crankyoldgit/IRremoteESP8266/blob/00b27cc7ea2e7ac1e48e91740723c805a38728e0/src/ir_Hitachi.h#L345-L353

Unfortunately the make/model you are dealing with here has more mode options than our library can easily report.

Onepamopa, 2022-12-08:
Min temp: 16, Max: 32
[image]
[image]
[image]

Onepamopa, 2022-12-08:
For the modes -- const uint8_t kHitachiAc296Cool                = 0b0011;  -- how is "0b0011" derived? I.e. How can I get it from my remote to add to supported.

crankyoldgit, 2022-12-08:
> How to get rid of those?

You can disable "UNKNOWN"s using:
https://github.com/crankyoldgit/IRremoteESP8266/blob/00b27cc7ea2e7ac1e48e91740723c805a38728e0/src/IRremoteESP8266.h#L113-L116

i.e. Set `DECODE_HASH` to `false`
or
use [`setUnknownThreshold()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#a02693553aad1decd67bdae60402e48bf)

crankyoldgit, 2022-12-08:
Also, a good "issue" ettiquite is to NOT add information about a new problem to an existing "closed"/resolved issue.
You're dealing with a different remote and A/C, it's not "the same thing", nor is your issue related to adding support for a complete new protocol.

PR #1758 (2022-03-13): Support for Hitachi RAS-70YHA3 (remote RAR-3U3)
Add support for Hitachi RAS-70YHA3 (remote RAR-3U3). Fixes #1757

This PR provides some initial support for the RAR-3U3 remote. It works of initial research [found here](https://perhof.wordpress.com/2015/03/29/reverse-engineering-hitachi-air-conditioner-infrared-remote-commands/).
commit 434f65e9 (2022-03-13): Support for Hitachi RAS-70YHA3 (remote RAR-3U3) (#1758)

Add support for Hitachi RAS-70YHA3 (remote RAR-3U3). Fixes #1757

This PR provides some initial support for the RAR-3U3 remote. It works of initial research [found here](https://perhof.wordpress.com/2015/03/29/reverse-engineering-hitachi-air-conditioner-infrared-remote-commands/).
PR #1776 (2022-03-13): HITACHI_AC296: Add `IRac` class support & tests.
* Fix max temp issue.
* Set special temp for auto operation mode.
* Merge into the `IRac` class so it is supported fully.
* Add `.toString()` output.
* Add real & synthetic decoding examples.
* General code style cleanup(s).
* Add supporting Unit Tests.

Ref: #1758
Fixes #1757
commit f63b9be6 (2022-03-13): HITACHI_AC296: Add `IRac` class support & tests. (#1776)

* Fix max temp issue.
* Set special temp for auto operation mode.
* Merge into the `IRac` class so it is supported fully.
* Add `.toString()` output.
* Add real & synthetic decoding examples.
* General code style cleanup(s).
* Add supporting Unit Tests.

Ref: #1758
Fixes #1757