crankyoldgit/IRremoteESP8266 #1759: Mitsubishi: Unable to determine remote code for MLZ-RX5017AS

nao-pon, 2022-02-13:
### Version/revision of the library used
v2.7.20 or later

### Describe the bug
Due to Commit (2d4659cd174ee5b555b2ef70e674e54053f282c5), the remote code of the MLZ-RX5017AS, which had no problem before, cannot be determined.

#### To Reproduce
Receive MLZ-RX5017AS remote code using v2.7.20 or later.

#### Example code used
IRrecvDumpV3.ino

#### Expected behaviour
omit

#### Output of raw data from [IRrecvDumpV3.ino]
Turn On Heat 20℃
```
Timestamp : 000074.412Library   : v2.8.0
Protocol  : UNKNOWNCode      : 0x76CCC723 (292 Bits)uint16_t rawData[583] = {3392, 1638,  456, 1220,  456, 1220,  456, 382,  454, 382,  456, 380,  456, 1222,  454, 380,  456, 382,  456, 1220,  456, 1220,  456, 382,  454, 1222,  454, 382,  456, 384,  454, 1222,  454, 1220,  454, 382,  456, 1220,  456, 1220,  456, 382,  456, 382,  456, 1220,  456, 380,  456, 382,  456, 1220,  456, 380,  456, 382,  456, 382,  456, 380,  456, 382,  456, 380,  458, 382,  456, 380,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 380,  458, 382,  456, 380,  456, 1220,  456, 380,  456, 382,  456, 382,  456, 382,  456, 382,  456, 1220,  456, 382,  456, 382,  456, 382,  456, 380,  456, 382,  456, 382,  456, 1220,  456, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  454, 384,  454, 382,  456, 380,  456, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 1222,  454, 1220,  456, 382,  456, 382,  456, 382,  454, 1220,  456, 1220,  454, 1222,  454, 1220,  454, 382,  456, 382,  456, 382,  454, 382,  454, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 380,  456, 382,  454, 382,  454, 384,  454, 382,  456, 382,  454, 384,  454, 382,  456, 382,  456, 382,  456, 384,  454, 384,  454, 382,  456, 382,  454, 382,  454, 384,  454, 382,  456, 1220,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  454, 1220,  456, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  454, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  454, 1220,  456, 1220,  456, 380,  456, 1220,  456, 1220,  456, 1222,  454, 1220,  456, 1220,  456, 11280,  3392, 1638,  454, 1220,  456, 1220,  456, 382,  456, 382,  456, 382,  456, 1220,  456, 380,  456, 382,  456, 1220,  456, 1220,  454, 382,  456, 1220,  456, 382,  456, 382,  456, 1220,  456, 1222,  454, 382,  456, 1220,  456, 1220,  454, 382,  456, 382,  456, 1220,  454, 382,  456, 382,  456, 1220,  456, 382,  456, 382,  454, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 1220,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 1220,  456, 380,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 1220,  456, 382,  454, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 380,  456, 382,  456, 380,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  454, 382,  456, 382,  456, 1220,  454, 1220,  456, 382,  456, 382,  456, 382,  454, 1220,  456, 1220,  454, 1222,  454, 1220,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 380,  456, 382,  456, 382,  456, 382,  456, 1220,  456, 382,  456, 382,  456, 382,  454, 382,  456, 382,  456, 1222,  454, 380,  456, 382,  456, 382,  456, 382,  456, 380,  456, 382,  456, 382,  454, 382,  454, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 382,  456, 1220,  456, 1220,  454, 382,  456, 1222,  454, 1220,  454, 1220,  454, 1222,  454, 1220,  456};  // UNKNOWN 76CCC723
```
Turn Off
```
Timestamp : 000077.222Library   : v2.8.0
Protocol  : UNKNOWNCode      : 0xEC67773B (292 Bits)uint16_t rawData[583] = {3388, 1642,  450, 1224,  452, 1224,  452, 388,  450, 388,  450, 388,  450, 1224,  452, 388,  450, 388,  450, 1224,  452, 1224,  452, 386,  450, 1224,  450, 388,  450, 386,  450, 1224,  450, 1226,  450, 388,  450, 1224,  452, 1224,  452, 388,  450, 386,  450, 1224,  452, 386,  450, 386,  450, 1224,  450, 388,  450, 388,  450, 386,  450, 388,  450, 388,  450, 388,  452, 386,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 386,  450, 386,  450, 386,  450, 388,  450, 388,  450, 386,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 1224,  452, 388,  450, 386,  450, 388,  450, 388,  450, 388,  450, 386,  450, 1224,  452, 386,  450, 388,  450, 388,  450, 388,  450, 386,  450, 386,  452, 386,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 386,  450, 388,  450, 388,  450, 386,  450, 388,  450, 388,  450, 388,  450, 1224,  452, 1224,  452, 388,  450, 386,  450, 388,  450, 1224,  452, 1224,  452, 1222,  452, 1224,  452, 386,  450, 388,  452, 386,  450, 388,  450, 388,  450, 386,  450, 388,  450, 388,  450, 386,  450, 386,  450, 388,  450, 388,  450, 388,  450, 386,  450, 388,  450, 388,  450, 386,  450, 386,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 386,  450, 386,  450, 386,  450, 1224,  452, 386,  450, 388,  450, 388,  450, 388,  450, 386,  452, 1222,  452, 386,  450, 386,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 386,  450, 388,  450, 388,  450, 1224,  452, 1224,  452, 386,  450, 1224,  452, 1222,  452, 386,  450, 1224,  450, 1224,  452, 11294,  3386, 1642,  452, 1224,  452, 1224,  452, 388,  450, 386,  450, 386,  450, 1224,  452, 386,  450, 386,  450, 1224,  452, 1224,  452, 388,  450, 1224,  478, 360,  452, 386,  450, 1224,  452, 1222,  480, 360,  478, 1196,  480, 1196,  452, 388,  450, 386,  450, 1224,  452, 386,  450, 386,  450, 1224,  452, 386,  478, 360,  450, 388,  450, 388,  450, 386,  478, 360,  450, 388,  450, 388,  450, 388,  450, 388,  450, 388,  450, 386,  450, 388,  450, 386,  450, 388,  450, 386,  450, 386,  478, 360,  450, 388,  450, 386,  450, 386,  450, 388,  450, 388,  450, 386,  450, 386,  450, 386,  450, 1224,  452, 386,  450, 386,  452, 386,  450, 386,  452, 386,  450, 388,  450, 1224,  452, 386,  450, 388,  450, 386,  450, 386,  450, 386,  450, 388,  450, 388,  450, 386,  452, 386,  450, 388,  450, 388,  450, 386,  450, 386,  450, 388,  450, 388,  450, 388,  450, 386,  450, 388,  450, 386,  450, 1224,  452, 1224,  452, 386,  450, 386,  450, 388,  450, 1224,  452, 1224,  452, 1224,  452, 1224,  452, 386,  450, 388,  450, 386,  450, 386,  452, 386,  450, 386,  452, 386,  450, 386,  478, 360,  450, 386,  452, 386,  478, 360,  452, 386,  450, 386,  450, 386,  450, 388,  450, 386,  452, 386,  478, 360,  450, 386,  478, 360,  452, 386,  450, 386,  450, 386,  478, 360,  452, 386,  452, 1224,  478, 358,  478, 358,  478, 358,  480, 358,  478, 360,  478, 1198,  478, 358,  480, 358,  478, 358,  478, 358,  478, 358,  480, 358,  478, 358,  478, 358,  478, 360,  478, 360,  478, 360,  478, 360,  476, 360,  478, 360,  478, 360,  478, 360,  478, 1198,  476, 1198,  478, 360,  478, 1198,  478, 1198,  476, 360,  478, 1200,  476, 1198,  476};  // UNKNOWN EC67773B
```

### What brand/model IR demodulator are you using?
VS1838B (No problem with v2.7.19)

### Circuit diagram and hardware used (if applicable)
omit

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
Yes, v2.7.19

### Other useful information
Commenting out line 285 of ir_Mitsubishi.cpp works fine.
```c++
// if (std::memcmp(save, results->state, nbits / 8) != 0) return false;
```
```c++
bool IRrecv::decodeMitsubishiAC(decode_results *results, uint16_t offset,
                                const uint16_t nbits,
                                const bool strict) {
  // Compliance
  if (strict && nbits != kMitsubishiACBits) return false;  // Out of spec.
  // Do we need to look for a repeat?
  const uint16_t expected_repeats = strict ? kMitsubishiACMinRepeat : kNoRepeat;
  // Enough data?
  if (results->rawlen <= (nbits * 2 + kHeader + kFooter) *
                         (expected_repeats + 1) + offset - 1) return false;
[... cut ...]
        // Data signature check.
        static const uint8_t signature[5] = {0x23, 0xCB, 0x26, 0x01, 0x00};
        if (std::memcmp(results->state, signature, 5) != 0) return false;
        // Checksum verification.
        if (!IRMitsubishiAC::validChecksum(results->state)) return false;
      }
      // Save a copy of the state to compare with.
      std::memcpy(save, results->state, nbits / 8);
    }
  }

  // Success.
  results->decode_type = MITSUBISHI_AC;
  results->bits = nbits;
  return true;
}
```

crankyoldgit, 2022-02-15:
@nao-pon The reason the messages were not being decoded correctly was the "gap" between the messages (there are technically two messages in your capture) was not the same as expected.
I've tweaked that timing a little to make it match successfully. Please download & confirm if https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue1759 (PR #1760) fixes the issue for you.

closed 2022-02-15 by crankyoldgit

nao-pon, 2022-02-15:
@crankyoldgit Thank you for your prompt response. We have confirmed that it works without problems with the correction.
Thank you so much!

PR #1760 (2022-02-15): MitsubishiAC: Tweak repeat gap timing.
* Update supported devices
* Decrease gap between repeat messages by 1.6ms to improve capture of messages.
* Add/update unit tests accordingly.

Fixes #1759
commit 789b926b (2022-02-15): MitsubishiAC: Tweak repeat gap timing. (#1760)

* Update supported devices
* Decrease gap between repeat messages by 1.6ms to improve capture of messages.
* Add/update unit tests accordingly.

Fixes #1759