crankyoldgit/IRremoteESP8266 #1767: Compile error when using arduino as espidf component

ItaiCuker, 2022-02-23:
### Version/revision of the library used
^2.8.1

### Describe the bug
compile error

#### To Reproduce

add library in platformio.ini with the following enviorment in platformio:

#### Example code used
```
[env:esp32doit-devkit-v1]
extends = esp32
platform = https://github.com/tasmota/platform-espressif32.git#Arduino2022/IDF44
board = esp32doit-devkit-v1
framework = arduino, espidf
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
monitor_speed = 115200
board_build.partitions = partitions.csv
board_build.embed_txtfiles = 
	src\certs\private_key.pem
lib_deps =
	crankyoldgit/IRremoteESP8266 @ ^2.8.1
```

#### Expected behaviour
compile with no error when code doesnt include library yet

### Circuit diagram and hardware used (if applicable)
ESP32

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No

ItaiCuker, 2022-02-23:
potentially fixed it by adding ```add_definitions(-D ESP32 -D ARDUINO)``` to CMakeLists.txt.

I will test the functionality of the library tomorrow.

crankyoldgit, 2022-02-24:
> framework = arduino, espidf

Well, that's something I didn't know, that you can have multiple simultaneous frameworks.

Currently, this library works only under the Arduino framework.

> compiled successfully by adding `add_definitions(-D ESP32 -D ARDUINO)`

Yes, those are two of the required definitions that immediately jump to mind. They are typically included as part of the Arduino Framework.

ItaiCuker, 2022-02-24:
> Well, that's something I didn't know, that you can have multiple simultaneous frameworks.
under Platformios hood this is whats happening [Arduino as an ESP-IDF component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html)
so its not actually 2 frameworks but one above the other.

crankyoldgit, 2022-03-08:
Did you have any success? Any tips we can document for others trying to do the same?

Jason2866, 2022-03-27:
Compiles fine as IDF component. The setup is different. Not done with `lib_deps`
IRremoteESP8266 has to be placed in a folder (root of the project) called `components`.
In sdkconfig check if the entry `CONFIG_AUTOSTART_ARDUINO=y` is there. If not add.

Tested with my platformio framework for esp32. Just this is needed
```
[env:esp32dev]
platform = https://github.com/Jason2866/platform-espressif32
framework = arduino, espidf
```
The main.cpp should look like this
```
//file: main.cpp
#include "Arduino.h"

void setup(){
  Serial.begin(115200);
  while(!Serial){
    ; // wait for serial port to connect
  }
}

void loop(){
    Serial.println("loop");
    delay(1000);
}
```

ItaiCuker, 2022-03-30:
> Any tips we can document for others trying to do the same?

when using Arduino as an ESP-IDF u can use it in 2 ways:

(ESP-IDF docs)
>1. For usage of app_main() function - Turn off Autostart Arduino setup and loop on boot
>2. For usage of setup() and loop() functions - Turn on Autostart Arduino setup and loop on boot
>`CONFIG_AUTOSTART_ARDUINO=y/n`

option 1 requires adding 2 build flags to the `CmakeLists.txt` inside `src` folder:
i added them this way - `add_definitions(-D ESP32 -D ARDUINO)`

I hope I remember correctly but that's everything needed to run the library on Arduino as an ESP-IDF component.

You should try it yourself I think:
[this explains Arduino as ESP-IDF component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html)
This is a `platformio.ini` file you can use to try:
```
[env:esp32doit-devkit-v1]
platform = https://github.com/tasmota/platform-espressif32.git#Arduino2022/IDF44
framework = arduino, espidf
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
lib_deps =
	crankyoldgit/IRremoteESP8266 @ ^2.8.1
```

try first with `CONFIG_AUTOSTART_ARDUINO=y` than `CONFIG_AUTOSTART_ARDUINO=n` with build flags I mentioned above.

crankyoldgit, 2022-04-04:
This kind-a looks resolved now. Marking this closed.

closed 2022-04-04 by crankyoldgit

Jason2866, 2022-04-04:
My latest build does simplify the setup
```
[env:esp32doit-devkit-v1]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip
framework = arduino, espidf
lib_deps =
	crankyoldgit/IRremoteESP8266 @ ^2.8.1
```
using latest core 2.0.3-rc1

ItaiCuker, 2022-04-05:
> My latest build does simplify the setup
> ```
> [env:esp32doit-devkit-v1]
> platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3.zip
> framework = arduino, espidf
> lib_deps =
> 	crankyoldgit/IRremoteESP8266 @ ^2.8.1
> ```
> using latest core 2.0.3-rc1

I had a lot of version syncing problems that's why mine is weird

suffiyanshaikh, 2024-01-10:
I am facing the same issue while using IRremoteESP8266 as a component in ESP32 IDF, please can share what changes need to be make CMake file and which Cmake file

[image]

Jason2866, 2024-01-10:
@suffiyanshaikh To post on a closed issue is not an good idea to get help.
Open a new one and provide as much as possible Infos. Screenshots are no Info at all.
What version used? Setup?

NiKiZe, 2024-01-10:
Opening #2055 will not change the fact of https://github.com/crankyoldgit/IRremoteESP8266/issues/1767#issuecomment-1049561614